merbanan/rtl_433 #1446: Support for Kerui GD-13 Gas Detector Alarm

heavyinfo, 2020-07-26:
First of all, my appreciation for everyone involved with this project.

I've just start rtl_433 to integrate couple of sensors. I found that **Kerui GD-13** gas detector alarm is not being decoded properly.

Here is the pulse analysis,

```
Detected OOK package	2020-07-26 19:18:00
Analyzing pulses...
Total count:   17,  width: 63.66 ms		(15915 S)
Pulse width distribution:
 [ 0] count:    7,  width:  672 us [632;776]	( 168 S)
 [ 1] count:    1,  width: 4168 us [4168;4168]	(1042 S)
 [ 2] count:    1,  width: 6544 us [6544;6544]	(1636 S)
 [ 3] count:    4,  width: 1784 us [1568;1868]	( 446 S)
 [ 4] count:    1,  width: 2996 us [2996;2996]	( 749 S)
 [ 5] count:    1,  width: 1304 us [1304;1304]	( 326 S)
 [ 6] count:    1,  width:  312 us [312;312]	(  78 S)
 [ 7] count:    1,  width:   80 us [80;80]	(  20 S)
Gap width distribution:
 [ 0] count:   11,  width:  516 us [480;544]	( 129 S)
 [ 1] count:    1,  width: 8720 us [8720;8720]	(2180 S)
 [ 2] count:    1,  width:  404 us [404;404]	( 101 S)
 [ 3] count:    1,  width:  132 us [132;132]	(  33 S)
 [ 4] count:    1,  width:   48 us [48;48]	(  12 S)
 [ 5] count:    1,  width: 21400 us [21400;21400]	(5350 S)
Pulse period distribution:
 [ 0] count:    8,  width: 1208 us [1168;1436]	( 302 S)
 [ 1] count:    1,  width: 4708 us [4708;4708]	(1177 S)
 [ 2] count:    1,  width: 7060 us [7060;7060]	(1765 S)
 [ 3] count:    2,  width: 2352 us [2352;2356]	( 588 S)
 [ 4] count:    1,  width: 3528 us [3528;3528]	( 882 S)
 [ 5] count:    1,  width: 10576 us [10576;10576]	(2644 S)
 [ 6] count:    1,  width: 1616 us [1616;1616]	( 404 S)
 [ 7] count:    1,  width: 21712 us [21712;21712]	(5428 S)
Level estimates [high, low]:  14047,   5616
RSSI: -0.7 dB SNR: 4.0 dB Noise: -4.6 dB
Frequency offsets [F1, F2]:   -2146,      0	(-8.2 kHz, +0.0 kHz)
Guessing modulation: No clue...
```
Here is the recorded sample - 
[g008_433.92M_250k.ook.zip](https://github.com/merbanan/rtl_433/files/4977747/g008_433.92M_250k.ook.zip)

Would be happy to provide more data if needed, thank you for your time.

zuckschwerdt, 2020-07-27:
The pulse width and gaps don't look like valid PCM, there is no preamble and the last gaps are too short. I guess the recording has noise or the demod didn't work well. Perhaps add a .cu8.zip here

heavyinfo, 2020-07-27:
Although there is shift keying noise every ~ 20 -30 seconds at my RTL_SDR, I'm confident that I provided the correct sample recording from the GD-13 by noticing the difference in size between it and the noise after triggering its alarm.

Here is the .cu8 of the same recording, I hope its better - 
[g008_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/4980717/g008_433.92M_250k.cu8.zip).

zuckschwerdt, 2020-07-27:
That signal is clipping a bit (gain too high, visible in https://triq.org/iqs/ by too much white color) but the problem might be that the frequency is nearly at the center, using `-f 433.87k` should be better.

heavyinfo, 2020-07-27:
I'm not sure whether this is what you meant, when using `rtl_433 -f 433.87k -A`, No pulse is detected.

```
[FC0012] no valid PLL combination found for 433870 Hz!
WARNING: Failed to set center freq.
```

`rtl_433 -f 433.87k -S all`

Records only usual noise at regular intervals and the pulse from the device is not recorded.

zuckschwerdt, 2020-07-27:
Oops: `-f 433.87M` of course (`M` not `k`), sorry!

heavyinfo, 2020-07-27:
No problem, I figured `433.87M` is what you meant, but that didn't detect any pulse at that time and the README mentions ```A sample rate is detected as (fractional) number suffixed with 'k', 'sps', 'ksps', 'Msps', or 'Gsps'.```, So I thought may be` k` is valid for recording sample.

Anyways, tried again and got results!

`rtl_433 -f 433.87M -A`

```
time      : 2020-07-27 19:31:01
model     : Smoke-GS558  id        : 9712
unit      : 28           learn     : 0             Raw Code  : b4be1c
Analyzing pulses...
Total count:  480,  width: 719.08 ms		(179770 S)
Pulse width distribution:
 [ 0] count:  230,  width:  328 us [316;364]	(  82 S)
 [ 1] count:  249,  width:  916 us [856;960]	( 229 S)
 [ 2] count:    1,  width: 1308 us [1308;1308]	( 327 S)
Gap width distribution:
 [ 0] count:  211,  width:  840 us [800;856]	( 210 S)
 [ 1] count:  248,  width:  252 us [208;276]	(  63 S)
 [ 2] count:   19,  width: 9048 us [9032;9056]	(2262 S)
 [ 3] count:    1,  width:  644 us [644;644]	( 161 S)
Pulse period distribution:
 [ 0] count:  459,  width: 1172 us [1160;1204]	( 293 S)
 [ 1] count:   19,  width: 9376 us [9368;9388]	(2344 S)
 [ 2] count:    1,  width: 1952 us [1952;1952]	( 488 S)
Level estimates [high, low]:  16109,   5057
RSSI: -0.1 dB SNR: 5.0 dB Noise: -5.1 dB
Frequency offsets [F1, F2]:    8875,      0	(+33.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 328, long_width: 916, reset_limit: 9060, sync_width: 1308
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=328,l=916,r=9060,g=0,t=0,y=1308'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {478} c7 82 d2 e3 c1 69 71 e0 b4 b8 f0 5a 5c 78 2d 2e 3c 16 97 1e 0b 4b 8f 05 a5 c7 82 d2 e3 c1 69 71 e0 b4 b8 f0 5a 5c 78 2d 2e 3c 16 97 1e 0b 4b 8f 05 a5 c7 82 d2 e3 c1 69 71 e0 b4 b8
[01] { 1} 00                                                                                                                                                                                  : 0
```

`rtl_433 -f 433.87M -S all`
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-27 19:23:00
model     : Smoke-GS558  id        : 9712
unit      : 28           learn     : 0             Raw Code  : b4be1c
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-27 19:23:02
model     : Smoke-GS558  id        : 9712
unit      : 28           learn     : 0             Raw Code  : b4be1c
*** Saving signal to file g007_433.87M_250k.cu8 (538884 samples, 1179648 bytes)
```

Attaching .cu8 file - 
[g007_433.87M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/4982423/g007_433.87M_250k.cu8.zip)

The issue is that the device is being detected as Smoke-GS558, which conflicts with my actual smoke detector. Seems to be the case with this issue as well #1406. 

How can I change the model name?

zuckschwerdt, 2020-07-27:
Looks good. Use `-R -86` to disable just the GS558 or `-R 0` to disable all, then
`rtl_433 -R -86 -X 'n=Kerui-GD13,m=OOK_PWM,s=330,l=920,g=1100,r=9000' g007_433.87M_250k.cu8` works to get the data.

heavyinfo, 2020-07-27:
Nice, I would need bit more information.

Currently I'm using,
` rtl_433 -f 433.87M -F json -M time | mosquitto_pub -h [IP address] -t home/rtl_433 -l`

To get json data from both my existing smoke detector & Kerui-GD13 gas detector as `"model":"Smoke-GS558"`.

Are you suggesting to use,

` rtl_433 -f 433.87M  -R -86 -X 'n=Gas Kerui-GD13,m=OOK_PWM,s=330,l=920,g=1100,r=9000' -F json -M time | mosquitto_pub -h [IP address] -t home/rtl_433 -l`

to get `"model":"Smoke-GS558"` & `"model":"Gas Kerui-GD13"` when the devices trigger respectively?

Sorry, couldn't test now as it's late here. But will give you the results tomorrow.

zuckschwerdt, 2020-07-27:
In that case, no. Keep the GS558 (don't use -R -86). But you need to filter out the wrong "GS558" messages from the GD13. No way around it.
Note that rtl_433 has MQTT build in, see `rtl_433 -F` and https://triq.org/rtl_433/OPERATION.html#outputs

heavyinfo, 2020-07-27:
Do you mean I have to distinguish the devices from the JSON output and there's no way to set specific device name at source?

The `"code":"xxxxxx"` in the JSON seems to be different for the devices, so I guess I'll have to use that as a filter.

> Note that rtl_433 has MQTT build in, see rtl_433 -F and https://triq.org/rtl_433/OPERATION.html#outputs

Thanks for the heads up, as of now piping does the job and I may require this for complex operations.

gdt, 2023-09-29:
I think this is resolved/inactive.   Please feel free to open a new issue or a discussion against up-to-date git master.

closed 2023-09-29 by gdt