merbanan/rtl_433 #1448: decoder enabling/disabling by frequency

evilpete, 2020-07-27:
Any interest in adding a feature to enable/disable protocol decoders based on current frequency while frequency hopping .
I think this can be efficiently implemented in a couple different ways

I am not sure if this is better implemented  as:

1. a list of min/max values
2. a list of frequencies  and a  +/- offsets
3. a callback function returning a Boolean 

I do not like **3,**   and an no preference between about **1** or **2**.
Decoders without a frequency list will have the filter applied.

any interest ?

zuckschwerdt, 2020-07-28:
I like the idea to add more structured meta-data to the decoders. Ideally the Markdown text would have some common layout to describe the protocol. There is already some work to include the parsed markdown with rtl_433, but it seemed a better idea to put the text "on the web", possibly with commenting and just link there.
I think this comes with too much surprises. Sometimes I'll transmit protocols on different bands just to capture and analyze. The preferred way to reduce processing power is a cut down conf file -- e.g. 1-3 used protocols would fit for the normal user ;)

zuckschwerdt, 2020-07-28:
Currently I'm thinking about adding an Issues-only repo with one issue per protocol (issue numer = protocol number) to dump the markdown info regularly and allow everybody to comment on the decoders. Or perhaps something custom like the https://triq.org/explorer/ -- I'm playing with a small server that uses GitHub OAuth2 and stores data and comments.

evilpete, 2020-08-12:
@zuckschwerdt 

my thought is to add a special case 'F' option to -R  `rtl_443 -f 915M -R F`  that will invoke a later filter to build out `cfg->demod->r_devs` based on the value of `cfg->center_frequency`

in the case of multiple frequencies,  multiple device lists will be build and stored into an new array 'cfg-> rdev_list[MAX_FREQS]'

As the frequency 'hops'  we set `cfg->demod->r_devs = cfg-> rdev_list[frequency_index]` and call `update_protocols()`

Does this sound reasonable?

i've added a `freq_range` array element to r_device,  if the  is freq_range is unset (Null)  

I'm sure i'm missing something but i'll sure i'll figure it out

----

Also what is `r_dev->create_fn()` exactly ?

zuckschwerdt, 2020-08-12:
`create_fn` is a callback for devices to declare that manual setup of the decoder is needed/wanted. I.e. to instanciate the decoder with parameters (say crypto key).

If "decoder per frequency" is really needed then I would suggest placing that in the conf. That way users can create hopping groups each with frequency and protocols.
There actually are some protocols that pop up with different rebranded devices on different bands. And with TPMS you'll likely observer the same protocol on different bands per country.

evilpete, 2020-08-12:
I figured frequency  filtering would be an optimization for reducing CPU,  I also agree it should not be a default..
  (I guess I should profile and check what percentage of time is spent on superfluous device decoding. )

yes, config file grouping may be better solution,   I'll give is some thought

(I am very aware of protocols on different bands my  915MHz lacrosse works because of it)

zuckschwerdt, 2020-08-12:
I don't have the impression that significant time is spend in the decoders but rather in the demod, esp. the FSK demod. But that really needs an evaluation ;)

evilpete, 2020-08-12:
I'll run some tests and ditch the idea if it is less then 10%

evilpete, 2020-08-31:
after looking at executions profiling  this optimization will not help

closed 2020-08-31 by evilpete