merbanan/rtl_433 #1452: Support for CAME remote control (garage door in AM/ASK)

psa-jforestier, 2020-07-31:
Hello All.

I failed to create a decoder for my CAME remote control. The protocol seems easy, but I cant make a flex decoder work.

Pulse analysis :
```
rtl_433 -A -r Button1_433.83M_250k.cu8
(... skip frames)
Detected OOK package    @0.436348s
Analyzing pulses...
Total count:   13,  width: 12.40 ms             ( 3100 S)
Pulse width distribution:
 [ 0] count:    7,  width:  332 us [316;356]    (  83 S)
 [ 1] count:    6,  width:  660 us [652;664]    ( 165 S)
Gap width distribution:
 [ 0] count:    6,  width:  344 us [332;352]    (  86 S)
 [ 1] count:    6,  width:  672 us [668;692]    ( 168 S)
Pulse period distribution:
 [ 0] count:    3,  width:  684 us [680;688]    ( 171 S)
 [ 1] count:    3,  width: 1336 us [1332;1340]  ( 334 S)
 [ 2] count:    6,  width: 1000 us [988;1012]   ( 250 S)
Level estimates [high, low]:  15949,     31
RSSI: -0.1 dB SNR: 27.1 dB Noise: -27.2 dB
Frequency offsets [F1, F2]:    9701,      0     (+37.0 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 332, long_width: 0, reset_limit: 696, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=332,l=0,r=696'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {16} 27 0c     : 00100111 00001100

```

Seems simple : 2 kinds of pulse (332 and 660) and 2 kinds of gap (344 and 672). Cause the pulse and gap modulation are close, and a length of 2 is 2 times the length of 1, the coding seems simple. But how to create a Flex decoder with this ?
I upload a sample, and wrote all my guess  :  https://github.com/psa-jforestier/rtl_433_tests/blob/master/tests/Came/TOP432/README.md

zuckschwerdt, 2020-07-31:
That's not MC, just plain PWM:
[image]

gvanem, 2020-08-01:
How / who created that nice diagram?

merbanan, 2020-08-01:
@gvanem

merbanan, 2020-08-01:
https://triq.org/pdv/

psa-jforestier, 2020-08-06:
Thank you for this analysis. In fact, the 12 bits here are visible on your graphic : ignore the 1st pulse (start/sync signal), then group by 3 gap/pulse.

With the help of a Russian website (http://phreakerclub.com/447) and Google Translate, I finally suceeded to set a flex decoder :
- modulation is OOK_PWM
- short is 320us
- long is 640us
- a bit is 3 pulses length (960us)
- "0" is short gap + long pulse
- "1" is long gap + short pulse
- no crc, no parity, no preamble
- the first pulse is a start bit
- the message should be repeated by multiple of 4 times.

Here is the Flex decoder, I confirmed the value by hand.
```
$> ./rtl_433 -r Button2_433.83M_250k.cu8 -X "n=CameRX,m=OOK_PPM,s=320,l=640,r=10000,invert,unique"
(...)
data      : 78c
$> ./rtl_433 -r Button1_433.83M_250k.cu8 -X "n=CameRX,m=OOK_PPM,s=320,l=640,r=10000,invert,unique"
(...)
data      : b8c

```

I did not succeed to make the "repeats" option work, so I use the "unique" option.

I'm now starting to code the decoder in rtl_433 on my own fork.

psa-jforestier, 2020-08-07:
Hello.

Code is ready on my fork : https://github.com/psa-jforestier/rtl_433/tree/feat-CameTOP432
But I have a formating problem when I run the program with "-F json" or "-F csv" : 

Without "-F" parameter : 
```
$> ./rtl_433 -r Button2_433.83M_250k.cu8 -R 164
(...)
time      : @0.436356s   brand     : CAME
model     : TOP432       type      : remote control code      : 2163         code_hex  : 0x873
```

With the "-F json" parameter : 
```
./rtl_433 -r Button2_433.83M_250k.cu8 -R 164 -F json
(...)
{"time" : "@0.436356s", "brand" : "CAME", "model" : "TOP432", "type" : "remote control", "code" : 2163, "code_hex" : 2163}
```
In the json output, the "code_hex" is not hex formatted. I dont understand why, and the output is correct without the  formatting. In the decode C code, the data_make call should be valid to output a hex formatted number : 
https://github.com/psa-jforestier/rtl_433/blob/bdec5422fe6e9ed0592dbf319522b9298eb34326/src/devices/came_top432.c#L110
```
 "code_hex", "", DATA_FORMAT,   "0x%03x", DATA_INT, code,
```

zuckschwerdt, 2020-08-08:
You are, of course, right. I totally missed that the gaps (logically) preceed the pulses. Some protocols do that, it's really confusing for any decoder. So there is *no* data in the gaps, each gap+pulse are fixed length.
Our current PWM slicer won't check if we have a) fixed gaps, b) fixed pulse+gap (period), or c) fixed gap+pulse. We could perhaps add that someday. Also adding "gap+pulse" to the `-A` heuristic might be nice.

zuckschwerdt, 2020-08-08:
The `DATA_FORMAT` is only for human readable output, esp. the JSON and network outputs never format data.

psa-jforestier, 2020-08-10:
Thank you for the explanation. The decoder is waiting to me merged from here : https://github.com/merbanan/rtl_433/pull/1460
For the `DATA_FORMAT` ok I understand.I will probably add thoses kind of information as comments in the `new_template.c`.

Mindavi, 2020-08-10:
I think a flex decoder like this would be sufficient:

```
# CAME-TOP432.conf

# Decoder for CAME TOP432 remote control

# Add protocol information etc...

decoder {
  name=CAME-TOP432,
  modulation=OOK_PPM,
  short=320,
  long=640,
  reset=10000,
  bits=12,
  invert,
  get=@1:{12}:button_code
}
```

psa-jforestier, 2020-08-20:
Fixed with https://github.com/merbanan/rtl_433/pull/1474

closed 2020-08-20 by psa-jforestier