merbanan/rtl_433 #1459: error when linking: libusb  error adding symbols: DSO missing from command line

rtega, 2020-08-07:
When linking, the linking fails with the following message on Raspbian 10 (buster):
`  CCLD     rtl_433
/usr/bin/ld: sdr.o: undefined reference to symbol 'libusb_strerror'
/usr/bin/ld: //lib/arm-linux-gnueabihf/libusb-1.0.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:1046: rtl_433] Error 1
make[2]: Leaving directory '/home/pi/rtl_433/src'
make[1]: *** [Makefile:497: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/rtl_433'
make: *** [Makefile:383: all] Error 2
`

Anybody any ideas?

rtega, 2020-08-07:
This build was with autoconf. I tried with cmake and it built like a charm.

zuckschwerdt, 2020-08-07:
Ah, thanks for the feedback. We don't invest too much in testing or caring for autotools build these days.

hbarnard, 2021-12-23:
I'm currently trying to build this on a Pi Zero. cmake step appears to work. However make does this:

```
/projects/rtl_433/build $ make
[  0%] Building C object src/CMakeFiles/r_433.dir/abuf.c.o
cc: error: unrecognized command line option ‘-Wimplicit-fallthrough’; did you mean ‘-Wno-fallthrough’?
src/CMakeFiles/r_433.dir/build.make:75: recipe for target 'src/CMakeFiles/r_433.dir/abuf.c.o' failed
make[2]: *** [src/CMakeFiles/r_433.dir/abuf.c.o] Error 1
CMakeFiles/Makefile2:950: recipe for target 'src/CMakeFiles/r_433.dir/all' failed
make[1]: *** [src/CMakeFiles/r_433.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```
I suspect this may be a gcc thing, but not sure where/what to fix?

zuckschwerdt, 2021-12-23:
That warning flag was added with GCC 7 (https://gcc.gnu.org/gcc-7/changes.html) -- you might be using GCC 6 (`gcc --version`)? (note that GCC 7 is from 2017, current GCC is 11).

zuckschwerdt, 2021-12-23:
In any case, just delete that line from [CMakeLists.txt](https://github.com/merbanan/rtl_433/blob/master/CMakeLists.txt#L77)

hbarnard, 2021-12-24:
> In any case, just delete that line from [CMakeLists.txt](https://github.com/merbanan/rtl_433/blob/master/CMakeLists.txt#L77)

Thanks and thanks for quick reply, it's built. It's an old Pi Zero and I probably should have reinstalled the OS.

zuckschwerdt, 2021-12-24:
We do want do support older environments, it's fixed with 2010e8f now.

gdt, 2023-09-25:
autconf is no more in rtl_433

closed 2023-09-25 by gdt