merbanan/rtl_433 #1461: 20.02-94-gf03710a Windows Build Returns Different Results To Recorded File Or Previous Build

OneOfTheInfiniteMonkeys, 2020-08-07:
**Software**
rtl_433.exe from [Official binary builds](https://bintray.com/chzu/dist/rtl_433#) for Windows (32 and 64 bit) at Bintray.

**Software Build**
Windows 0x64 20.02-94-gf03710a

**Hardware**
- SDR
RTL-SDR Compatible DVB hardware based on  [Realtek RTL2832U](https://en.wikipedia.org/wiki/List_of_software-defined_radios), 
VID 0x0bda PID 0x2832 aka Generic RTL2832U (e.g. hama nano).
 - Antenna 
Generic 12 cm (120 mm) whip with magnetic base
- Computer
Windows 10 64 bit, Intel(R) Core(TM) i5-8600K CPU @ 3.60GHz, 8 GB RAM, Build 1909, USB interface (no hub)

**Target Device**
[Geo-Minim energy meter](https://www.geotogether.com/wp-content/uploads/2016/09/160901-Minim-EU-1v2.0.pdf) believed to be using 868.265 Megahertz, FSK PCM encoding based around an [si4022](https://www.silabs.com/documents/public/data-sheets/Si4022.pdf) device.

**Issue Observed**
Live data results displayed from custom decode using -X feature display different data to match.  Though match function identifies transmission correctly.

**Example Command (all builds tested)**
rtl_433 -s 1000000 -f 868.265M -R 0 -M bits -X "n=si4022,m=FSK_PCM,s=25,l=25,r=8000,match={32}0xaaaa2dd4"

**Results displayed:**

**Latest Available Build**
D:\rtl_433>rtl_433 -s 1000000 -f 868.265M -R 0 -M bits -X "n=si4022,m=FSK_PCM,s=25,l=25,r=8000,match={32}0xaaaa2dd4"
rtl_433 version 20.02-94-gf03710a branch  at 202007201253 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\xxxxx\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 158 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.265MHz.

time      : 2020-08-07 17:02:09
model     : si4022       count     : 1             num_rows  : 1             rows      :
len       : 83           data      : aaaaa8b753d3e0c29b086
codes     : {83}aaaaa8b753d3e0c29b086
**Observation:**
Data returned (0xaaaaa8b75) as if matched, though data shown does not reflect match data string (0xaaaa2dd4), other transmissions are ignored. It is believed the remainder of the returned data is similarly miss reported.

**Previous build (18.12-125-gfc8734e):**
rtl_433 -s 1000000 -f 868.265M -R 0 -M bits -X "n=si4022,m=FSK_PCM,s=25,l=25,r=8000,match={32}0xaaaa2dd4"

D:\rtl_433_b>rtl_433 -s 1000000 -f 868.265M -R 0 -M bits -X "n=si4022,m=FSK_PCM,s=25,l=25,r=8000,match={32}0xaaaa2dd4"
rtl_433 version 18.12-125-gfc8734e branch  at 201903021639 inputs file rtl_tcp RTL-SDR
Trying conf file at "D:\rtl_433_b\rtl_433.conf"...
Trying conf file at "C:\Users\xxxxx\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 121 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.265MHz.

time      : 2020-08-07 17:02:23
model     : si4022       count     : 1             num_rows  : 1             rows      :
len       : 87           data      : 7aaaaa2dd4f4f830a7c30e
codes     : {87}7aaaaa2dd4f4f830a7c30e
**Observation:**
Data returned (0xaaaa2dd4, preceding 7 ignored as noise) are same as detailed in the match statement (0xaaaa2dd4), other transmissions are ignored as anticipated.

**Additional Information**
Decoding of TPS using rtl_433.exe defaults appears to operate as expected. i.e. Vehicles correctly identified, type, pressure and temperature data within expected parameters when cross checked with independant measurement. Weather station(s) data as expected and verified against independent measurement to be within expectations. 

Decoding of the recorded file created by build 20.02-94-gf03710a is correctly performed in that it reports the codes anticipated.  A copy of a recorded file is attached that might be used to simulate the issue with suitable hardware.
[g051_868.265M_1000k.zip](https://github.com/merbanan/rtl_433/files/5042343/g051_868.265M_1000k.zip)

It is noted:
  Both builds tested also occasionally return other data as matched e.g (18.12-125-gfc8734e).
     f555545ba9e9f06149801c
  Timining in the examples is not finalised for decoing these signals.

**Proposed containment**
Results returned from build 20.02-94-gf03710a using the -X option should be treated with caution, though off line analysis of captured files did return the data expected. For recording of data build 20.02-94-gf03710a operated as anticipated in the usage case considered. For display of data using the -X feature in build 20.02-94-gf03710a, care should be exercised.

**Proposed next step(s)**
As the available official build is quite old i.e. approx. 18 months, the initial request is to issue a more recent build than the Windows x64 20.02-94-gf03710a build currently available, for further evaluation of the identified issue. When a build is issued, a follow up entry against this issue will be made to confirm the observed performance.

zuckschwerdt, 2020-08-07:
From your description the `-Y classic` option should  get you the old behaviour back.
Note that ` {83}aaaaa8b753d3e0c29b086` is `{81}aaaaa2dd4f4f830a6c211` and not that far off. Use `preamble` instead of `match` if you want to align on the sync word.
The 20.02-94 build is from 2020-06-20 :) A new build (20.02-124-g9c11752) is now available.

OneOfTheInfiniteMonkeys, 2020-08-07:
Hi, thanks for making a new build available.

I have verified the **-Y classic** option you proposed, which worked perfectly on build 20.02-94-gf03710a and the build you made available 20.02-124-g9c11752, aligning behaviour with the older build 18.12-125-gfc8734e.

The symbols **2dd4** are as you point out are sync symbols that I was looking for consistency in. I suspect the following 20 bits to be a device identifier, for my device consistent of course. With data following that and then possibly some form state indication. I am hoping for some form of checksum as well.

I will read further on **preamble**, **match** and **-Y classic** options. 

Thanks for the prompt response and solutions.

Cheers, have a great weekend.

closed 2020-08-07 by OneOfTheInfiniteMonkeys