merbanan/rtl_433 #1462: Optional preamble compression

ehagan, 2020-08-08:
It would be useful for some devices to be able to optionally compress or discard the preamble. 

Implementing a decoder for the Inkbird ITH-20R temperature and humidity sensor has required increasing BITBUF_COLS and PD_MAX_PULSES substantially which increases the memory footprint. The preamble from this device is 14400 on-off bits which pushes the actual payload out of the available data window for decoding.

Please see this conversation in the group for the full context
https://groups.google.com/g/rtl_433/c/oeExmwoBI0w

Dmitriy Kozyrev wrote a decoder and for the device and posted a link to a zip of his work. I've merged the changes into my working copy and have pushed to my fork for convenience.

This is the message format the decoder reads as copied from Dmitriy's post:
```
Total packet length 14563 bits:
Preamble: aa aa aa ... aa aa (14400 on-off sync bits)
Sync Word (16 bits): 2DD4
Data (147 bits):
Byte    Sample      Comment
0-2     D3910F      Always the same across devices, a device type?
3       00          00 - normal work , 40 - unlink sensor (button pressed 5s), 80 - battery replaced
4       01          Changes from 1 to 2 if external sensor present
5-6     0301        Unknown (also seen 0201), sw version?
7       58          Battery % 0-100
8-9     A221        Device id, always the same for a sensor but each sensor is different
10-11   D600        Temperature in °C * 10, little endian, so 0xD200 is 210, 21.0°C or 69.8°F
12-13   F400        Temperature °C * 10 for the external sensor,  0x1405 if not connected
14-15   D301        Relative humidity %  * 10, little endian, so 0xC501 is 453 or 45.3%
16-17   38FB        CRC16
18      0           Unknown 3 bits (seen 0 and 2)

CRC16 (bytes 0-15), without sync word):
poly=0x8005  init=0x2f61  refin=true  refout=true  xorout=0x0000  check=0x3583  residue=0x0000
```
And the full set of changes to get this thing to work are here:
https://github.com/ehagan/rtl_433/tree/feat-inkbird_ith20r

I ran ` rtl_433 -F json -M utc -R 40 ` under memusage with and without the changes. 
feat-inkbird_ith20r branch:
```
 Memory usage summary: heap total: 40818814, heap peak: 38142376, stack peak: 92400
         total calls   total memory   failed calls
 malloc|       9285        2055942              0
realloc|       1110         461800              0  (nomove:483, dec:0, free:0)
 calloc|       3232       38301072              0
   free|      19512       40809762
```

Baseline:
```
Memory usage summary: heap total: 40429260, heap peak: 37873576, stack peak: 47440
         total calls   total memory   failed calls
 malloc|       6867        2035428              0
realloc|       1110         461800              0  (nomove:480, dec:0, free:0)
 calloc|       1442       37932032              0
   free|      14680       40420208
```

Then I ran `rtl_433 -F json -M utc`
feat-inkbird_ith20r branch:
```
Memory usage summary: heap total: 40756790, heap peak: 38167432, stack peak: 92400
         total calls   total memory   failed calls
 malloc|       7538        2064590              0
realloc|       1117         470480              0  (nomove:512, dec:0, free:0)
 calloc|       1815       38221720              0
   free|      15862       40747738
```

Baseline:
```
Memory usage summary: heap total: 40465143, heap peak: 37898448, stack peak: 47440
         total calls   total memory   failed calls
 malloc|       7093        2060671              0
realloc|       1117         470480              0  (nomove:509, dec:0, free:0)
 calloc|       1477       37933992              0
   free|      14965       40456091
```
So about double the peak stack usage and about 1% increase in heap usage.

buxtronix, 2021-01-22:
Reporting here that the changes also successfully decode the Inkbird IBS-P01R pool thermometer.

merbanan, 2021-01-22:
Hi, can you send a few signal samples? And @zuckschwerdt maybe we should allocate the rows and cols dynamically ? Then we can add this protocol disabled by default and if you enable it you get bigger buffers.

zuckschwerdt, 2021-01-22:
A fixed-sized bitbuffer means it can go on the stack and is very easy to handle.
The compression idea is good, but we still need to figure out where/how the slicers can detect this condition and then truncate the preamble. Perhaps a bitbuffer function to look for a very long 0xaa pattern and truncate the row before it fills up.

buxtronix, 2021-01-28:
I've raised  a PR with some samples from this pool thermometer: https://github.com/merbanan/rtl_433_tests/pull/384

hkclark, 2021-03-12:
I have an IBS-P01R and @ehagan's `feat-inkbird_ith20r` branch is working well for me based on initial tests.  Let me know if I can help in some way (more data samples, testing, etc.).

@ehagan Thanks for your work on that!

dfiore1230, 2021-05-26:
I see this was not merged. Is there a straightforward way to add this to the latest build? I have the IBS-P01R and would love to be able to capture the pool temp for my home automation.

zuckschwerdt, 2021-05-26:
The fork (https://github.com/ehagan/rtl_433/tree/feat-inkbird_ith20r) with the changes is still there. We need to implement some kind of compression before we can merge this.

dfiore1230, 2021-05-26:
Forgive my ignorance, I’m at best a hack, but I’m using https://github.com/bademux/rtl_433toMQTT to run this in docker. With it far ahead I suspect using this fork (which is 250+ commits behind) is going to cause me some issues.

Is there a way to drop in the compiled changes ?

zuckschwerdt, 2021-05-26:
I could pull the changes into a local branch, then you'd need to fork the rtl_433toMQTT repo and change the [zip address](https://github.com/bademux/rtl_433toMQTT/blob/master/Dockerfile#L2).
How are you using the docker build? Do just use https://github.com/bademux/rtl_433toMQTT somewhere and could you easily change that to say https://github.com/dfiore1230/rtl_433toMQTT?

merbanan, 2021-05-26:
@zuckschwerdt how about we change it to a circular buffer? (I did not look at the code.) And after the signal is received we can frame the start index based on BITBUF_COLS and some heuristics.

Something like if the circular buffer wraps we enter long preamble mode and just set the start index at the signal end - BITBUF_COLS.

merbanan, 2021-05-26:
And yeah the code looks quite good. Please do a local branch.

dfiore1230, 2021-05-26:
@zuckschwerdt Appreciate that. yes i can just point to a local branch and update my docker run script for sure.

zuckschwerdt, 2021-05-26:
A circular buffer sounds interesting. But we'd then need to have something like a bitbuffer_finish() to warp the the rows into order (i.e. align the actual end to the buffer end and copy the trailing portion (older bits) to the front.

I was pondering the idea to detect repeated bytes (0x55 / 0xaa / 0xcc / ..) at some trigger point (~50 bytes in) and remove the last few (20 or so). That way we effectivly remove a middle portion of the over-long preamble. Should be a "simple" change to bitbuffer_add_bit().

ehagan, 2021-05-26:
If you'd like, I can bring my pull request up to current with the mainline.

dfiore1230, 2021-05-26:
@ehagan that would be awesome.  If your PR is up to mainline, and i can host a local branch i would happy to. 

The responsiveness here is awesome. My background is far from dev, but if i can help in any way please let me know. 

(i might need a little nudge to get this right though)

Dave

zuckschwerdt, 2021-05-26:
Already done ;) https://github.com/merbanan/rtl_433/tree/feat-inkbird-ith20r -- attribution is still there (my changes are amended in)

The zip url for a docker build is https://github.com/merbanan/rtl_433/archive/refs/heads/feat-inkbird-ith20r.zip

zuckschwerdt, 2021-05-26:
I'll try this idea (numbers are random):
- When we hit byte 50
- check if the last 40 bytes are identical
- if so remove the last 30 bytes

I'd guess there is no performance impact and we won't remove important data.

zuckschwerdt, 2021-05-26:
It works (triggering at byte 60 to account for the leading 00's). And we get `{493} 00 00 00 00 00 00 00 00 00 00 00 2a aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa 8b 75 34 e4 43 c0 00 80 c0 4c 05 8f c6 40 4b ff d4 00 51 19 00`
This leaves PD_MAX_PULSES (needs a change from 1200 to 18000) with similar problems.

zuckschwerdt, 2021-05-26:
One more idea, since memory consumption came up on small embedded devices:
Instead of the fixed 25 rows of 450 bytes we could use a single large buffer and put rows at random offsets. That way we could get many more rows if the rows are small and also very long rows if there are few. We would need to change every `bitbuffer->bb[num_row]` to `bitbuffer->bb[bitbuffer->row_offset[num_row]]` or `bitbuffer_get_row(bitbuffer, num_row)` in the decoders though...

dfiore1230, 2021-05-26:
I forked and created a new repo and the docker build is failing with:

```
CMake Error at CMakeLists.txt:28 (include):
  include could not find load file:

    GetGitRevisionDescription

CMake Error at CMakeLists.txt:29 (get_git_head_revision):
  Unknown CMake command "get_git_head_revision".

```

This is a bit above my pay grade. Any pointers?

dfiore1230, 2021-05-26:
CMakeOutput.log
`The system is: Linux - 5.4.0-73-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/build/out/CMakeFiles/3.18.4/CompilerIdC/a.out"

Detecting C compiler ABI info compiled with the following output:
Change Dir: /build/out/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_73a62/fast && /usr/bin/make  -f CMakeFiles/cmTC_73a62.dir/build.make CMakeFiles/cmTC_73a62.dir/build
make[1]: Entering directory '/build/out/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o
/usr/bin/cc   -v -o CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o -c /usr/share/cmake/Modules/CMakeCCompilerABI.c
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
Target: x86_64-alpine-linux-musl
Configured with: /home/buildozer/aports/main/gcc/src/gcc-10.2.1_pre1/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --build=x86_64-alpine-linux-musl --host=x86_64-alpine-linux-musl --target=x86_64-alpine-linux-musl --with-pkgversion='Alpine 10.2.1_pre1' --enable-checking=release --disable-fixed-point --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-werror --disable-symvers --enable-__cxa_atexit --enable-default-pie --enable-default-ssp --enable-cloog-backend --enable-languages=c,c++,d,objc,go,fortran,ada --disable-libssp --disable-libmpx --disable-libmudflap --disable-libsanitizer --enable-shared --enable-threads --enable-tls --with-system-zlib --with-linker-hash-style=gnu
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.1 20201203 (Alpine 10.2.1_pre1) 
COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o' '-c' '-mtune=generic' '-march=x86-64'
 /usr/libexec/gcc/x86_64-alpine-linux-musl/10.2.1/cc1 -quiet -v /usr/share/cmake/Modules/CMakeCCompilerABI.c -quiet -dumpbase CMakeCCompilerABI.c -mtune=generic -march=x86-64 -auxbase-strip CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o -version -o /tmp/cceFegGN.s
GNU C17 (Alpine 10.2.1_pre1) version 10.2.1 20201203 (x86_64-alpine-linux-musl)
        compiled by GNU C version 10.2.1 20201203, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.0, isl version isl-0.22-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/include"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../x86_64-alpine-linux-musl/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/include/fortify
 /usr/include
 /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/include
End of search list.
GNU C17 (Alpine 10.2.1_pre1) version 10.2.1 20201203 (x86_64-alpine-linux-musl)
        compiled by GNU C version 10.2.1 20201203, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.0, isl version isl-0.22-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 0024134b69111392f17b472e854971ed
COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o' '-c' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../x86_64-alpine-linux-musl/bin/as -v --64 -o CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o /tmp/cceFegGN.s
GNU assembler version 2.35.2 (x86_64-alpine-linux-musl) using BFD version (GNU Binutils) 2.35.2
COMPILER_PATH=/usr/libexec/gcc/x86_64-alpine-linux-musl/10.2.1/:/usr/libexec/gcc/x86_64-alpine-linux-musl/10.2.1/:/usr/libexec/gcc/x86_64-alpine-linux-musl/:/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/:/usr/lib/gcc/x86_64-alpine-linux-musl/:/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../x86_64-alpine-linux-musl/bin/
LIBRARY_PATH=/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/:/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../x86_64-alpine-linux-musl/lib/../lib/:/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../lib/:/lib/../lib/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../x86_64-alpine-linux-musl/lib/:/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o' '-c' '-mtune=generic' '-march=x86-64'
Linking C executable cmTC_73a62
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_73a62.dir/link.txt --verbose=1
/usr/bin/cc  -v -rdynamic CMakeFiles/cmTC_73a62.dir/CMakeCCompilerABI.c.o -o cmTC_73a62 
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-alpine-linux-musl/10.2.1/lto-wrapper
Target: x86_64-alpine-linux-musl
Configured with: /home/buildozer/aports/main/gcc/src/gcc-10.2.1_pre1/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --build=x86_64-alpine-linux-musl --host=x86_64-alpine-linux-musl --target=x86_64-alpine-linux-musl --with-pkgversion='Alpine 10.2.1_pre1' --enable-checking=release --disable-fixed-point --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-werror --disable-symvers --enable-__cxa_atexit --enable-default-pie --enable-default-ssp --enable-cloog-backend --enable-languages=c,c++,d,objc,go,fortran,ada --disable-libssp --disable-libmpx --disable-libmudflap --disable-libsanitizer --enable-shared --enable-threads --enable-tls --with-system-zlib --with- [... cut ...]
  collapse library dir [/lib/../lib] ==> [/lib]
  collapse library dir [/usr/lib/../lib] ==> [/usr/lib]
  collapse library dir [/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../../../x86_64-alpine-linux-musl/lib] ==> [/usr/x86_64-alpine-linux-musl/lib]
  collapse library dir [/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1/../../..] ==> [/usr/lib]
  implicit libs: [ssp_nonshared;gcc;gcc_s;c;gcc;gcc_s]
  implicit dirs: [/usr/lib/gcc/x86_64-alpine-linux-musl/10.2.1;/usr/x86_64-alpine-linux-musl/lib;/usr/lib;/lib]
  implicit fwks: []
`

dfiore1230, 2021-05-26:
Scratch the above. I repackaged the zip into a gzip and set the expansion depth and got it to compile

merbanan, 2021-05-27:
> One more idea, since memory consumption came up on small embedded devices:
> Instead of the fixed 25 rows of 450 bytes we could use a single large buffer and put rows at random offsets. That way we [...]

Does it matter on the platforms we run?

zuckschwerdt, 2021-05-27:
> Does it matter on the platforms we run?

We can't keep growing the bitbuffer, but we do want bigger rows. And if we can shrink bitbuffer while getting longer rows that's even better!

Refining the idea above so we don't need api changes:
- spill bits into the next row
- keep increasing the bit length in the original row
- mark that next row as 0 bits length

Since row memory is contiguous that "spill" works cleanly -- just treat the row now as double length.
Now we can make bitbuffer much smaller and still get very long rows or more rows (e.g. EV1527 is 25 bits but very many rows).

zuckschwerdt, 2021-05-27:
Yes, that works transparently without changes. But the ASAN checker is unhelpful, to be technically correct we would need to change `bitbuffer->bb[row][col]` to `*(bitbuffer->bb[row] + col)`. But if the pointer is copied out, like in most instances, to `uint8_t *b = bitbuffer->bb[row]` the access `b[col]` just works.

zuckschwerdt, 2021-05-27:
Sketch is on https://github.com/merbanan/rtl_433/tree/feat-rowspill, i.e. 875d7df
Usage of `BITBUF_COLS` in other code needs to be checked. It's likely broken in the print helpers, flex and oregon decoder.

zuckschwerdt, 2021-10-03:
The "row spill" feature is now implemented.
- Rows can have up to 51200 bits (previously 3600)
- There are now 50 rows available (previously 25)
- Memory usage per bitbuffer is now 6408 bytes (previously 11256)
- For constrainted platforms this could easily be lowered without adverse effects (e.g. back to 25 rows is only 3208 bytes per bitbuffer, and then up to 25600 bits per row won't be a restriction)

We can now represent the needed total packet length of 14563 bits for this Inkbird ITH-20R easily.
Still pulse data buffers can only hold 1200 pulses -- this is the next problem to address.

hkclark, 2021-10-03:
@zuckschwerdt Very nice!  Thank you!  Looking forward to getting rtl_433 version updated and still be able to work with my Inkbird devices. :-)  (It sounds like there could still be a little more work to merge in the full Inkbird stuff, but from what I understand you just took care of a huge obstacle, so thanks.)

zuckschwerdt, 2021-10-05:
The https://github.com/merbanan/rtl_433/tree/feat-inkbird-ith20r branch has been updated. The pulse_data hack needs some more thought and work, but it's looking good already.

niklas-probst, 2023-06-16:
> Reporting here that the changes also successfully decode the Inkbird IBS-P01R pool thermometer.
Has somebody an Idea how to Decode Inkbird IBS-P01R with a microcontroller? Like an ESP32?
I have an FSK receiver connected to an ESP32. I know this message comes a bit late but maybe someone can help me. 
I'm starting to get depressed.
LG Niklas

zuckschwerdt, 2023-06-16:
> Has somebody an Idea how to Decode Inkbird IBS-P01R with a microcontroller? Like an ESP32?

Look at the source to get some infos about timing and bits https://github.com/merbanan/rtl_433/blob/master/src/devices/inkbird_ith20r.c
But you need to tailor this to e.g. a CC1101 connected to the ESP32.

niklas-probst, 2023-06-19:
Thanks, I‘ll try!