merbanan/rtl_433 #1467: Efergy-e2CT False Trigger

evilpete, 2020-08-12:
More False Triggers

Not sure how to fix this one.

Here's some test data:
[efergy_e2_false.tar.gz](https://github.com/merbanan/rtl_433/files/5060723/efergy_e2_false.tar.gz)

```
{"model" : "Efergy-e2CT", "id" : 512, "battery_ok" : 0, "current" : 0.000, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{64} 00 00 02 00 00 00 00 02 : 00000000 00000000 00000010 00000000 00000000 00000000 00000000 00000010

{"model" : "Efergy-e2CT", "id" : 1, "battery_ok" : 0, "current" : 0.000, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{64} 00 01 00 00 00 00 00 01 : 00000000 00000001 00000000 00000000 00000000 00000000 00000000 00000001

{ "model" : "Efergy-e2CT", "id" : 2048, "battery_ok" : 0, "current" : 0.000, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{64} 00 00 08 00 00 00 00 08 : 00000000 00000000 00001000 00000000 00000000 00000000 00000000 00001000

{"model" : "Efergy-e2CT", "id" : 0, "battery_ok" : 0, "current" : 0.004, "interval" : 6, "learn" : "YES", "mic" : "CHECKSUM"}
{64} 00 00 00 80 00 80 00 00 : 00000000 00000000 00000000 10000000 00000000 10000000 00000000 00000000

{"model" : "Efergy-e2CT", "id" : 4096, "battery_ok" : 0, "current" : 0.000, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{64} 00 00 10 00 00 00 00 10 : 00000000 00000000 00010000 00000000 00000000 00000000 00000000 00010000

{"model" : "Efergy-e2CT", "id" : 8192, "battery_ok" : 0, "current" : 0.000, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{64} 00 00 20 00 00 00 00 20 : 00000000 00000000 00100000 00000000 00000000 00000000 00000000 00100000
```

Live Data Report
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-08-11 19:33:12
model     : Efergy-e2CT  Transmitter ID: 4096
Battery   : 0            Current   : 0.00 A        Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
Modulation: FSK          Freq1     : 914.7 MHz     Freq2     : 914.6 MHz
RSSI      : -0.1 dB      SNR       : 30.3 dB       Noise     : -30.4 dB
*** Saving signal to file g016_915M_2000k.cu8 (63407 samples, 131072 bytes)
```

zuckschwerdt, 2020-08-12:
Can we assume that all zero /all one IDs are likely false positives?

merbanan, 2020-08-12:
Low popcount?

evilpete, 2020-08-12:
count null bytes?

all the examples have 6 of the 8 byte equal to 0

```
cnt = 0
foreach byte in packet:
   if byte == 0:
     cnt++

if cnt >= 3:
   fail packet 
```
       
IMOO:  i  strongly suspect this is a symptom of "Picket fencing" from some some other signal  at a different data rate

evilpete, 2020-08-14:
@merbanan  looking at the sample data I am suspecting the data pattern might be a artifact of the FSK classic demodulator 
I used -Y classic 

`-M stats` reports 
`Modulation: FSK          Freq1     : -0.336 MHz    Freq2     : -0.428 MHz`

(yes I added a couple extra printf decimal points in my branch)

visually looking at the signal I to not see a the frequency shifts

Sample data is attached to the first post.

[image]

evilpete, 2020-08-17:
> looking at the sample data I am suspecting the data pattern might be a artifact of the FSK classic demodulator

@zuckschwerdt  Any thoughts on this ? or am I missing something obvious ? 

Samples : [efergy_e2_false.tar.gz](https://github.com/merbanan/rtl_433/files/5060723/efergy_e2_false.tar.gz)

gdt, 2023-09-25:
@evilpete Status with current master?  Should this decoded just be disabled by default?   Or is there really a bug?

evilpete, 2023-10-02:
> @evilpete Status with current master? Should this decoded just be disabled by default? Or is there really a bug?

this issue was fixed,  it can be tested with [test_false_positives](https://github.com/merbanan/rtl_433_tests/tree/master/test_false_positives) code i submitted in https://github.com/merbanan/rtl_433_tests/pull/397

gdt, 2023-10-02:
If it is fixed the issue should be closed.

closed 2023-10-02 by gdt