merbanan/rtl_433 #1471: Bad influxdb timestamps on Windows

thecabinet, 2020-08-13:
I'm running _rtl_433 version 20.02-124-g9c11752 branch  at 202008052201_ on Windows with _-M "time:unix:usec:utc"_ to an influxdb instance in a FreeNAS jail, but I'm seeing a mix of good and "bad" (at least as I understand it) timestamps.  Here's what in my database at some point:

    > select * from "Acurite-Tower";
    name: Acurite-Tower
    time                battery_ok channel humidity id    mic      temperature_C
    ----                ---------- ------- -------- --    ---      -------------
    330490000           1          B       41       13388 CHECKSUM 26.200001
    330962000           1          B       41       13388 CHECKSUM 26.200001
    332502000           1          B       41       13388 CHECKSUM 26.200001
    333936000           1          B       41       13388 CHECKSUM 26.200001
    335331000           1          B       41       13388 CHECKSUM 26.200001
    336077000           1          B       41       13388 CHECKSUM 26.200001
    336305000           1          B       41       13388 CHECKSUM 26.200001
    337136000           1          B       41       13388 CHECKSUM 26.200001
    338378000           1          B       41       13388 CHECKSUM 26.200001
    340276000           1          B       41       13388 CHECKSUM 26.200001
    1597351206567657071 1          B       41       13388 CHECKSUM 26.200001
    1597351254277335352 1          B       41       13388 CHECKSUM 26.200001

Here's what's in it later:

    > select * from "Acurite-Tower";
    name: Acurite-Tower
    time                battery_ok channel humidity id    mic      temperature_C
    ----                ---------- ------- -------- --    ---      -------------
    315216000           1          B       42       13388 CHECKSUM 26.200001
    317267000           1          B       42       13388 CHECKSUM 26.200001
    317289000           1          B       42       13388 CHECKSUM 26.200001
    317453000           1          B       42       13388 CHECKSUM 26.200001
    317488000           1          B       42       13388 CHECKSUM 26.200001
    319239000           1          B       41       13388 CHECKSUM 26.200001
    320336000           1          B       41       13388 CHECKSUM 26.200001
    320571000           1          B       41       13388 CHECKSUM 26.200001
    330490000           1          B       41       13388 CHECKSUM 26.200001
    330962000           1          B       41       13388 CHECKSUM 26.200001
    332502000           1          B       41       13388 CHECKSUM 26.200001
    333936000           1          B       41       13388 CHECKSUM 26.200001
    335331000           1          B       41       13388 CHECKSUM 26.200001
    336077000           1          B       41       13388 CHECKSUM 26.200001
    336305000           1          B       41       13388 CHECKSUM 26.200001
    337136000           1          B       41       13388 CHECKSUM 26.200001
    338378000           1          B       41       13388 CHECKSUM 26.200001
    340276000           1          B       41       13388 CHECKSUM 26.200001
    1597351206567657071 1          B       41       13388 CHECKSUM 26.200001
    1597351254277335352 1          B       41       13388 CHECKSUM 26.200001

I was definitely running it with _-M "time:unix:usec:utc"_ while all those new rows (that have **earlier** timestamps) were inserted.

gdt, 2023-09-25:
Please update to the latest git and retest.

gdt, 2023-09-25:
Also, please debug log the influx data.

zuckschwerdt, 2023-10-21:
Numbers that short could be from non-live mode, i.e. reading test data. We don't output timestamps but something like file positions there.

gdt, 2023-10-21:
It's been more than 30 days with no repsonse and this seems reasonably likely not a bug, so I'm hitting close but happy to reopen if we get fingerpointing data from an up-to-date build.

closed 2023-10-21 by gdt