merbanan/rtl_433 #1473: Works on Windows but not on RPi - help

pmknowles, 2020-08-14:
I'm probably missing something obvious
This is the output from rtl_sdr -test
```
pi@raspberrypi:~ $ rtl_sdr test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
Tuned to 100000000 Hz.
Tuner gain set to automatic.
Reading samples in async mode.
```.
Which, I think, means that rtl-sdr is working correctly
If I then enter rtl-433 -f 838.3M -R 119 -F json
```
pi@raspberrypi:~ $ rtl_433 -f 868.3M -R 119 -F json
rtl_433 version 20.02-125-g57c924d branch master at 202008102239 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 163 device decoding protocols [ 119 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.

```
And nothing else happens. I've reinstalled several times and get the same results. Any ideas?

zuckschwerdt, 2020-08-14:
Looks good. You are just not receiving any "Bresser Weather Center 5-in-1" (protocol 119) transmissions.
Try running without `-R 119` to see what else is there, or add `-A` to see all received data.

pmknowles, 2020-08-14:
Tried that. It worked on my Windows laptopn with the Bresser.
`pi@raspberrypi:~ $ rtl_433 -A
rtl_433 version 20.02-125-g57c924d branch master at 202008102239 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 134 out of 163 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
`

pmknowles, 2020-08-14:
This is the rtl-sdr output
pi@raspberrypi:~ $ rtl_sdr -f 838300000 -n 100 -
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
Tuned to 838300000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...

User cancel, exiting...
~|�y�}y{}{�~�{|~~��~��~{�x}����y�|}{|�w~{|����}}~|���~}����~��}�����~�~~}��}~~||�y�x����|������||}|~|t}y~��|~�~y}�{�|z�}~~~~~����~������~{��~�x|{�}���~��||{�~���}�~���~

zuckschwerdt, 2020-08-14:
Did you try `rtl_433 -A` and `rtl_433 -f 868.3M -A` and both don't show any transmissions? Looks like the Windows output (note different frequency) doesn't show any output either?
If there were no data from the RTL-SDR then rtl_433 would abort with a "stalled" warning.
(note `RTL-SDR` is a type of receiver, `rtl_sdr` is a program, `rtl_433` is this software)

You can run the same "some rf data incoming" test as `rtl_sdr -n 100 - |hexdump -C` with `rtl_433 -n 100 -w - |hexdump -C` to check if the SDR receives anything. Data around 0x80 (say 0x70 to 0x90) is "silence".

pmknowles, 2020-08-14:
With Windows I could output MQTT data to openHAB and it worked fine - the Bresser was indoors at the time. The Pi is in my shed and the Bresser is about 5m away.
```
```
pi@raspberrypi:~ $ rtl_433 -A
rtl_433 version 20.02-125-g57c924d branch master at 202008102239 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 134 out of 163 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
```
pi@raspberrypi:~ $ rtl_433 -f 868.3M -A
rtl_433 version 20.02-125-g57c924d branch master at 202008102239 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 134 out of 163 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
```
It doesn't seem to stall - the last one sayed the same for 5 minutes.
The data with the dump seems to be in the 'silence' range.
Am going to go outside with the laptop when the rain stops to check that it's still working.

zuckschwerdt, 2020-08-14:
Does not look like a problem at the rtl_433 end. Perhaps best to grab a few minutes of RF and then view the spectrogram manually. E.g. `rtl_433 -T 120 -w FILENAME.cu8` then get that file and drop on https://triq.org/iqs/

pmknowles, 2020-08-14:
I think there may be an issue with the dongle. Although the LED is on it’s stone cold and it was very hot yesterday.

zuckschwerdt, 2020-08-14:
Heat already kill one of my RTL-SDR and maimed another: it only works with a low sample rate now, like `-s 250k` -- could be worth a try.

pmknowles, 2020-08-14:
I think it's a faulty RTL-SDR so will close it

closed 2020-08-14 by pmknowles

pmknowles, 2020-08-15:
Update. In trying to test the device I installed SoapySDR. I don't know whether that fixed the issue but it's now working. I'm just going to leave it as it is!