merbanan/rtl_433 #1476: BIOWIN / 2measure No. 270208 - weather station

pinolec, 2020-08-17:
Hi,

I try to crack this device:
https://www.mediaexpert.pl/dom-i-ogrod/do-domu/stacje-pogodowe-i-termometry/stacja-pogody-bioterm-270208#reviews
also, this seems to be the same weather station:
https://it.aliexpress.com/item/32830480338.html
This is probably available under many different brands and model numbers.

By default it is recognised as:

```
time      : 2020-08-17 15:45:48
model     : Smoke detector GS 558                  id        : 19754
unit      : 21           learn     : 0             Raw Code  : 19a555
pulse_demod_pwm(): Wireless Smoke and Heat Detector GS 558 
bitbuffer:: Number of rows: 7 
[00] { 7} 00                                  : 0000000
[01] {96} 55 a5 19 5c 01 27 1f 09 f0 c6 8d 00 
[02] {96} aa a5 98 5c 01 27 1f 09 f0 c6 8d 00 
[03] {96} aa a5 98 5c 01 27 1f 09 f0 c6 8d 00 
[04] {96} aa a5 98 5c 01 27 1f 09 f0 c6 8d 00 
[05] {96} aa a5 98 5c 01 27 1f 09 f0 c6 8d 00 
[06] {89} aa a5 98 5c 01 27 1f 09 f0 c6 8d 00 
```

[cu8.zip](https://github.com/merbanan/rtl_433/files/5084589/cu8.zip)

I used 
`rtl_433 -R -86 -X 'n=Smoke-GS558,m=OOK_PWM,s=450,l=980,g=1499,r=9000'`

to collected some data available here: [bitbeanch](http://triq.net/bitbench?c=555a67a3ff0fdbf7bf996eff%2024C%2036%25&c=555a67a3ff0cdbf7bf996dff%2024%2C3C%2036%25&c=555a67a3ff13dbf7bf9972ff%2023%2C6C%2036%25&c=555a67a3ff1bdaf7bfa97bff%2022%2C8C%2037%25%201006hPa&c=555a67a3ff1adaf7bfa675ff%2022%2C9C%2037%25%201006hPa%205km%2Fh&c=555a67a3ff19daf7bfb565ff%2023C%2037%25%2010006hPa%204km%2Fh&c=555a67a3ff18daf7bfa677ff%2023%2C1C%2037%25%2010006hPa%205km%2Fh%20&c=555a67a3ff17daf7bfa47aff%2023%2C2C%2037%25%2010006hPa%205km%2Fh%20&c=555a67a3ff15daf7bfa874ff%2023%2C4C%2037%25%2010006hPa%205km%2Fh%20&c=555a67a3fef7ddf7bfa9317f%2026%2C4%2034%25%2010005hPa%205km%2Fh&c=555a67a3feecdef7bf26f6ff%2027.5%2033%25%2010005hPa%2013km%2Fh%20&c=555a67a3fedfdcf7bfa928ff%2028.8%2035%25%201005hPa%205km%2Fh%20SE&c=555a67a3fee2dcf7bfb8047f%2028.5%2035%25%201005hPa%204km%2Fh%20SSE&c=555a67a3fee7dcf7bfb70eff%2028%2C0%2035%25%201005hPa%204km%2Fh%20S&c=555a67a3fedcddf7bf79ca7f%2029.1%2034%25%20%20%20%20%20%20%20%20%208km%2Fh%20SE&c=555a67a3feddddf7bf78ca7f&c=555a67a3fee6ddf78f9b227f%20--rain&c=555a67a3fededdf78f78f9ff&c=555a67a3fedfddf78fc545ff&c=555a67a3fedcdcf74fb8ca7f%20-more%20rain&c=555a67a3fee4ddf74f69127f%20-RAIN%2024h%205.5mm%201004hPA&c=555a67a3fee1ddf62fa7b8ff%20-Rain%2024h%2019.7mm%20total%20111.3mm&c=555a67a3fedbe1f61f7d147f%20-last%20hour%200.7mm%20more%20rain%20&c=555a67a3fedae1f60f97ef7f%20-last%20hour%201.5mm%20last%2024h%2021.3%20total%20112.9mm&c=555a67a3fed8e2f60f8ba2ff%20wind%20E&c=555a67a3fed9e1f60f58237f%201004hPa%20SSE&f=ID%3F%3Ahhhhhh%20hhh%20%20TEMP_C%3A~12d%20b%20HUMID%3A~7d%20RAIN%3A~12d%203d%20WIND_KM%3A~5d%2020b%20&a=Preamble&m=
)

This is what I determined: 
ID?:hhhhhh hhh  TEMP_C:~12d b HUMID:~7d RAIN_TOTAL:~12d 3d WIND_KM:~5d 20b 

There probably should be somewhere sign bit for temperature but I had no chance to test it yet. I couldn't find the wind direction and Pressure.

Please advise what to do next.

Kind regards,
Piotr

zuckschwerdt, 2020-09-13:
Your BitBench looks good. I've updated the link to invert all bits and use `PRE:hhh ID?:hh hh hh TEMP_C:12d HUMID:8d RAIN:12d WIND_KM:8d 4d 4d 4d 1b 7b`
All fields align nicely to bytes. For wind direction I would expect 4 bit for 16 cardinal directions or 9 (12?) bits for a full degrees value. But also 8 bits for a checksum at the end. We have 13 bits left, so maybe 5+8? But I can't spot the wind dir.

Can you get readings for at least the 16 cardinal directions? Or even better slowly turn the direction indicator to see if there is a field at the end that changes slowly?
Once we figure that out I'll add the device support.

zuckschwerdt, 2020-09-13:
Or perhaps keep the direction fixed (due north should give all 0's) and let other values change so we can see which bits at the end stay the same.

gdt, 2023-09-29:
@pinolec Are you still working on this?