merbanan/rtl_433 #1481: Inconsistent signals for 345 mhz devices

roo-punch, 2020-08-29:
I followed this guide https://community.home-assistant.io/t/honeywell-5800-series-345mhz-door-and-window-sensor-integration-with-ha/18970 in order to read in info from my sensors. When I first set it up it was working perfectly. It was almost as good as plug and play. Since then, without changing anything, I am having trouble seeing any signals come in. Most of the time they do not. It seems very sporadic. I've tried adjusting the sample rate, as well as tweaking other things. I've gone through the issues and can't seem to find anything. Tried re-inserting USB, and changing the antenna.  I'm a bit out of my depth here, although good at following directions. I'm wondering if it is the dongle. I'm using the Nooelec NESDR Smart v4 Bundle - Premium RTL-SDR. Not really sure what to do now. I appreciate the project and any help troubleshooting this is also appreciated.

roo-punch, 2020-08-29:
Told myself I was on to other things with my day but still futzing with this. I'm suspecting it is a calibration issues. Trying to nail that down. Will post results here if I get a solution.

merbanan, 2020-08-29:
Run rtl_433 -f 344975000 and see if the signals arrive as often as you need. Otherwise you might need to change the frequency.

roo-punch, 2020-08-31:
Yeah the issue seems to be the dongle or interference. This doesn't appear to be an rtl_433. Adjusting the ppm yields better results but still inconsistent. I'm playing around with it and may try another dongle.

roo-punch, 2020-09-01:
Any troubleshooting tips? Still struggling with consistency. Seems to not work sometimes, and then I come back later to find what appears to be latent signals. Only two sensors located in the same room as the usb dongle appear to have signals appear promptly.

zuckschwerdt, 2020-09-01:
Generally if you don't know exactly where your signals are and if there is interference it's best to take a look at the spectrum with CubicSDR, Gqrx, or SDRangel. As a poor-man replacement you can also record a short time (e.g. `-T 120`) and then drop the .cu8 on https://triq.org/iqs/ to view the spectrum.

roo-punch, 2020-09-01:
Great. I'll check these out. Thanks!

roo-punch, 2020-09-07:
So, feeling a bit silly but I've made a bit of headway here. It turns out, some of my doors have wireless sensors and some have wired sensors. Coincidentally, the one I was testing the most and is closest has a wireless sensor. I figured it was a distance issue. After trying another dongle, I realized this. I have a 2gig GC2 panel, and it appears that they have a "takeover module", which transmits the signals wirelessly from the hardwired sensors (previously ADT) to the GC2 panel. As such, I'm guessing this protocol is likely different than the "honeywell sensors" and that is why I'm not picking up readings from these sensors?

zuckschwerdt, 2020-09-07:
We did add some 2Gig support a while ago (#747). They have a different CRC and thus would silently fail. We currently guess that "channel" 2 and 10 are 2Gig, which might not apply to your bridge. Perhaps add `bitbuffer_print(bitbuffer)` after `bitbuffer_invert(bitbuffer)` in [honeywell.c L67](https://github.com/merbanan/rtl_433/blob/master/src/devices/honeywell.c#L67) for a quick check. Otherwise we need samples files :)

roo-punch, 2020-09-07:
So, I added that as mentioned and no changes. Hope I'm not being too bothersome, I've looked through the issues and can't seem to find instructions. What is the proper command to run to record the type of sample files you would need? Thanks.

zuckschwerdt, 2020-09-08:
It's simply `rtl_433 -S unknown` or `rtl_433 -S all` to save sample files.
You should also be able to read the raw signal data with `rtl_433 -X "n=Honeywell-Test,m=OOK_MC_ZEROBIT,s=156,r=292"` which is easier to gather and transfer than binary sample files.

roo-punch, 2020-09-08:
[samples.zip](https://github.com/merbanan/rtl_433/files/5191643/samples.zip)

Thank you very much for the help. I did run the last command and got this: 
[image]

So I've attached the sample files from running 'rtl_433 -S all' for about 1 minute. While it was running I went and opened / closed the door with the sensor 3 times.

zuckschwerdt, 2020-09-09:
The files are empty (drop them on https://triq.org/iqs/ ) to check. I guess you need `-f 345M` / `-f 344.97M`?

roo-punch, 2020-09-09:
[samples34497.zip](https://github.com/merbanan/rtl_433/files/5198264/samples34497.zip)

Ok. I uploaded these ones. Looks like that last one has something. I did the same thing (ran for 1 minute and opened/closed 3 times)

zuckschwerdt, 2020-09-10:
There are one or two files in there that vaguely might have a signal, but it's fat too weak. Look at these samples: https://github.com/merbanan/rtl_433_tests/tree/master/tests/honeywell/2Gig-DW10 -- download one of the .cu8 files and drop on https://triq.org/iqs/ (or click here [to see 2Gig-DW10/g001_344.975M_250k.cu8](https://triq.org/iqs/#/honeywell/2Gig-DW10/g001_344.975M_250k.cu8)) then compare to your files.
Perhaps the frequency selected is wrong and the shadows in your captures are just harmonics or the gain is too low? You need to find a stronger signal.

gdt, 2023-09-29:
This has gone quiet.  What's status and next steps, or should we close it?

carefulcomputer, 2024-01-29:
i am looking for same sensors. here is my capture with 
`rtl_433 -f 345M -X "n=Honeywell-Test,m=OOK_MC_ZEROBIT,s=156,r=292"`
pressed the switch on sensor first, then triggered reed switch few times.

[output.txt](https://github.com/merbanan/rtl_433/files/14080468/output.txt)

closed 2025-07-04 by gdt