merbanan/rtl_433 #1488: Question: Compatibility of Renkforce RF4406812 weather station

vistalba, 2020-09-04:
I'm looking for a weather station that I can connect to Homeassistant with rtl_433 as there are no really nice solutions out there without using clouds.

So I'm not sure about the compatibility of weatherstations. There is the following in the list:

```
    [32]  Fine Offset Electronics WH1080/WH3080 Weather Station
```

As I understand is, that Fine Offset doesn't sell this stations. They were sold as other brands.

Does anyone know if this station is compatible and which values can be received/decoded by rtl_433?

https://www.conrad.ch/de/p/renkforce-rf-4406812-funk-wetterstation-vorhersage-fuer-12-bis-24-stunden-2203406.html (In the manuel they call them WH2315)

On the picture it looks similar like the Wh1080/WH3080 on the website: http://www.foshk.com/Weather_Professional/

If there are any other nice weatherstation that works great... let me know.

I wand at least the following values:
- temperature
- humidity
- rain (mm)
- uv
- windspeed

closed 2020-09-09 by vistalba

zuckschwerdt, 2020-09-09:
Sorry, I don't think any of the main developers own a weather station. I would be really happy to see a wiki page with contributed information on user experience with weather stations.
If you are up to some hacking then we should be able to add support for another weather station. Mainly you'd need to capture data packets and annotate with the values shown on some display. We'll help with the actual decoding.

vistalba, 2020-09-10:
I've now bought this one:
https://de.elv.com/elv-funk-wetterstation-ws3080-inkl-kombi-aussensensor-868-mhz-und-pc-software-102690?fs=3350976843

This one was working out of the box as it is similar to Fine Offset WH3080... but they sell them with another brand "ELV WS3080" ;)

But a list with compatible & tested one would be nice.

merbanan, 2020-09-10:
@vistalba can you show exactly what rtl_433 identified is as?