merbanan/rtl_433 #1490: Add custom fields to flex spec?

mzealey, 2020-09-10:
https://github.com/merbanan/rtl_433/blob/b091ab68b58b5ddce463f06c6b6d87a492775ce5/src/devices/flex.c#L321

It would be nice to be able to specify custom fields (for example `id`, `command`) from within a flex spec so that we can use it to spoof other devices. For example if I do it for a doorbell, for domoticz to register it at present it needs to have a `command` field, and `id` would be useful to be able to add in. Perhaps passing a key of `field_id`, `field_command` (ie `field_<xxx>` and the value being the static value that will be sent in the csv would be a good way to start?

zuckschwerdt, 2020-09-10:
If I understand that correctly you want "dummy" fields with fixed values?
There is a more extensive spec for flex planned which will make that available.
For now you should be able to just use extractors for a single bit and with a mapping of 0 and 1 to the desired value.
E.g. `get=@0:{1}:id:[0:foo 1:foo], get=@0:{1}:command:[0:bar 1:bar]`

mzealey, 2020-09-10:
Thanks, this seems to work well (I didn't see it documented anywhere?). One suggestion would be to allow a certain bitrange to be passed as-is rather than via a lookup table, for example the first 16 bits as the id field rather than having to hardcode for a couple of devices.

zuckschwerdt, 2020-09-11:
Even easier :) `get=@0:{16}:id`

closed 2020-09-22 by zuckschwerdt