merbanan/rtl_433 #1492: Could you add support for Explore Scientific 7 in 1 Weather Center at 868 MHz

Tuxyso, 2020-09-12:
Hi!

I own:

https://www.bresser.de/en/EXPLORE-SCIENTIFIC-professional-7-in-1-Wi-Fi-Weather-Centre-with-Light-Intensity-and-UV-Measurement-Function.html

The sensor transmits data at 868 MHz. 
With 

`rtl_433 -f 868.3M`

I do not receive any data.

Can I provide some raw data to you so that you can add support?

Kind regards,
Tuxyso

zuckschwerdt, 2020-09-13:
Sure, get sample data with `-S unknown`. Note down the expected measurement values from a read-out or head unit.
Try analyzing each sample with `rtl_433 -A gfile.cu8` to see if there is some real data.
Also check the spectrogram by dropping samples on https://triq.org/iqs/ (it should [look "busy" like this](https://triq.org/iqs/#/honeywell/2Gig-DW10/g001_344.975M_250k.cu8))
Then upload the zipped samples here and post a description and tabled values per sample file. Happy hacking!

Tuxyso, 2020-09-13:
I gathered data with 
`rtl_433 -f 868.3M -S unknown`

For the received files g001_868.3M_1000k.cu8 ... g007_868.3M_1000k.cu8

I called `rtl_433 -A g001_868.3M_1000k.cu8` 

Output is as follows:

`
Reading samples from file: g001_868.3M_1000k.cu8
Detected FSK package    @0.292000s
Analyzing pulses...
Total count:   99,  width: 130.69 ms            (32673 S)
Pulse width distribution:
 [ 0] count:    4,  width: 1544 us [1464;1788]  ( 386 S)
 [ 1] count:   82,  width:  488 us [484;492]    ( 122 S)
 [ 2] count:    8,  width:  976 us [972;980]    ( 244 S)
 [ 3] count:    3,  width: 2600 us [2440;2928]  ( 650 S)
 [ 4] count:    1,  width: 1952 us [1952;1952]  ( 488 S)
 [ 5] count:    1,  width: 4396 us [4396;4396]  (1099 S)
Gap width distribution:
 [ 0] count:   82,  width:  488 us [484;492]    ( 122 S)
 [ 1] count:    5,  width: 1464 us [1460;1468]  ( 366 S)
 [ 2] count:    8,  width:  972 us [972;980]    ( 243 S)
 [ 3] count:    2,  width: 2684 us [2440;2928]  ( 671 S)
 [ 4] count:    1,  width: 1952 us [1952;1952]  ( 488 S)
Pulse period distribution:
 [ 0] count:   12,  width: 2060 us [1948;2444]  ( 515 S)
 [ 1] count:   72,  width:  976 us [972;980]    ( 244 S)
 [ 2] count:    7,  width: 1464 us [1460;1468]  ( 366 S)
 [ 3] count:    4,  width: 2928 us [2928;2932]  ( 732 S)
 [ 4] count:    3,  width: 4392 us [3904;4884]  (1098 S)
Level estimates [high, low]:  15887,     42
RSSI: -0.1 dB SNR: 25.8 dB Noise: -25.9 dB
Frequency offsets [F1, F2]:    5542,  -6781     (+21.1 kHz, -25.9 kHz)
Guessing modulation: No clue...
`

`
Reading samples from file: g002_868.3M_1000k.cu8
Detected FSK package    @0.105900s
Analyzing pulses...
Total count:   54,  width: 89.93 ms             (22483 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:   26,  width:  400 us [376;412]    ( 100 S)
 [ 2] count:   10,  width: 1196 us [1192;1208]  ( 299 S)
 [ 3] count:   12,  width:  796 us [792;808]    ( 199 S)
 [ 4] count:    3,  width: 1596 us [1596;1600]  ( 399 S)
 [ 5] count:    2,  width: 2196 us [2004;2392]  ( 549 S)
Gap width distribution:
 [ 0] count:    1,  width: 9956 us [9956;9956]  (2489 S)
 [ 1] count:   27,  width:  396 us [388;408]    (  99 S)
 [ 2] count:   15,  width:  800 us [788;812]    ( 200 S)
 [ 3] count:    5,  width: 1200 us [1196;1204]  ( 300 S)
 [ 4] count:    5,  width: 2000 us [1996;2008]  ( 500 S)
Pulse period distribution:
 [ 0] count:    1,  width: 9956 us [9956;9956]  (2489 S)
 [ 1] count:   14,  width:  796 us [784;808]    ( 199 S)
 [ 2] count:    8,  width: 1596 us [1592;1604]  ( 399 S)
 [ 3] count:   12,  width: 1200 us [1196;1212]  ( 300 S)
 [ 4] count:   15,  width: 2184 us [1996;2408]  ( 546 S)
 [ 5] count:    3,  width: 2796 us [2792;2804]  ( 699 S)
Level estimates [high, low]:   1370,     45
RSSI: -10.8 dB SNR: 14.8 dB Noise: -25.6 dB
Frequency offsets [F1, F2]:    2974,  -2090     (+11.3 kHz, -8.0 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_width: 400, long_width: 400, reset_limit: 409600, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=400,l=400,r=409600'
[... 171 lines cut ...]
 [ 3] count:    3,  width: 3092 us [2928;3416]  ( 773 S)
 [ 4] count:    1,  width:   64 us [64;64]      (  16 S)
Gap width distribution:
 [ 0] count:   78,  width:  488 us [484;492]    ( 122 S)
 [ 1] count:    5,  width: 1464 us [1460;1468]  ( 366 S)
 [ 2] count:    2,  width: 1952 us [1952;1952]  ( 488 S)
 [ 3] count:    8,  width:  972 us [972;976]    ( 243 S)
 [ 4] count:    2,  width: 2684 us [2444;2928]  ( 671 S)
 [ 5] count:    1,  width: 35992 us [35992;35992]       (8998 S)
Pulse period distribution:
 [ 0] count:   16,  width: 2212 us [1948;2444]  ( 553 S)
 [ 1] count:   67,  width:  976 us [972;980]    ( 244 S)
 [ 2] count:    7,  width: 1464 us [1460;1472]  ( 366 S)
 [ 3] count:    4,  width: 4024 us [3904;4392]  (1006 S)
 [ 4] count:    1,  width: 2932 us [2932;2932]  ( 733 S)
 [ 5] count:    1,  width: 36484 us [36484;36484]       (9121 S)
Level estimates [high, low]:  15927,     34
RSSI: -0.1 dB SNR: 26.7 dB Noise: -26.8 dB
Frequency offsets [F1, F2]:    6551,  -5607     (+25.0 kHz, -21.4 kHz)
Guessing modulation: No clue...
`

Does it help? Cu8 are following...

`

Tuxyso, 2020-09-13:
Additional info: According to datasheet the outdoor sensor transmits data every 12 seconds. 
The files g004 and g007have an exact difference of 12 seconds, so I guess these files are from the outdoor sensor.

g004_868.3M_1000k.cu8: 2020-09-13T10:46:32.000Z
g007_868.3M_1000k.cu8: 2020-09-13T10:46:44.000Z

Signal is as follows:

[image]

zuckschwerdt, 2020-09-13:
You captured 3 different devices:
- g001, g004, g005, g007: strongest, maybe 125µs PCM
- g002, g003: weak, maybe 100µs PCM
- g006: probably 30µs MC, likely a TPMS

Your assumption fits with the first group. I'll take a closer look now.

Tuxyso, 2020-09-13:
Further analysis: The **indoor sensor** transmits every minute. I received 

g031_868.3M_1000k.cu8:2020-09-13T11:33:49.000Z
g037_868.3M_1000k.cu8:2020-09-13T11:34:49.000Z

The occurence at the console corresponds to the tansmit led from the indoor sensor. The signal characteristic looks very similiar to the one from above.

Test mode active. Reading samples from file: **g031_868.3M_1000k.cu8**
Detected FSK package    @0.292064s
Analyzing pulses...
Total count:   96,  width: 130.63 ms            (32658 S)
Pulse width distribution:
 [ 0] count:   11,  width: 1488 us [1460;1728]  ( 372 S)
 [ 1] count:   77,  width:  488 us [484;492]    ( 122 S)
 [ 2] count:    6,  width:  976 us [972;980]    ( 244 S)
 [ 3] count:    2,  width: 2196 us [1956;2440]  ( 549 S)
Gap width distribution:
 [ 0] count:   78,  width:  488 us [484;492]    ( 122 S)
 [ 1] count:    7,  width: 1464 us [1460;1468]  ( 366 S)
 [ 2] count:    5,  width:  972 us [972;976]    ( 243 S)
 [ 3] count:    1,  width: 3908 us [3908;3908]  ( 977 S)
 [ 4] count:    2,  width: 2684 us [2440;2932]  ( 671 S)
 [ 5] count:    2,  width: 1952 us [1952;1952]  ( 488 S)
Pulse period distribution:
 [ 0] count:   17,  width: 2024 us [1948;2444]  ( 506 S)
 [ 1] count:   68,  width:  976 us [972;980]    ( 244 S)
 [ 2] count:    3,  width: 1464 us [1464;1464]  ( 366 S)
 [ 3] count:    2,  width: 4884 us [4396;5372]  (1221 S)
 [ 4] count:    5,  width: 3024 us [2928;3416]  ( 756 S)
Level estimates [high, low]:  16003,     52
RSSI: -0.1 dB SNR: 24.9 dB Noise: -25.0 dB
Frequency offsets [F1, F2]:    3838,  -7093     (+14.6 kHz, -27.1 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_width: 488, long_width: 488, reset_limit: 499712, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=488,l=488,r=499712'

Test mode active. Reading samples from file: **g037_868.3M_1000k.cu8**
Detected FSK package    @0.291976s
Analyzing pulses...
Total count:   97,  width: 166.78 ms            (41694 S)
Pulse width distribution:
 [ 0] count:    9,  width: 1504 us [1464;1812]  ( 376 S)
 [ 1] count:   77,  width:  488 us [484;492]    ( 122 S)
 [ 2] count:    8,  width:  976 us [972;980]    ( 244 S)
 [ 3] count:    1,  width: 2928 us [2928;2928]  ( 732 S)
 [ 4] count:    1,  width: 4396 us [4396;4396]  (1099 S)
 [ 5] count:    1,  width:   32 us [32;32]      (   8 S)
Gap width distribution:
 [ 0] count:   80,  width:  488 us [480;492]    ( 122 S)
 [ 1] count:    5,  width: 1464 us [1460;1468]  ( 366 S)
 [ 2] count:    2,  width: 1948 us [1948;1952]  ( 487 S)
 [ 3] count:    6,  width:  972 us [972;976]    ( 243 S)
 [ 4] count:    1,  width: 5372 us [5372;5372]  (1343 S)
 [ 5] count:    1,  width: 2928 us [2928;2928]  ( 732 S)
 [ 6] count:    1,  width: 36028 us [36028;36028]       (9007 S)
Pulse period distribution:
 [ 0] count:   16,  width: 2124 us [1948;2448]  ( 531 S)
 [ 1] count:   68,  width:  976 us [972;980]    ( 244 S)
 [ 2] count:    7,  width: 1464 us [1460;1468]  ( 366 S)
 [ 3] count:    1,  width: 6840 us [6840;6840]  (1710 S)
 [ 4] count:    2,  width: 4640 us [4396;4888]  (1160 S)
 [ 5] count:    1,  width: 3416 us [3416;3416]  ( 854 S)
 [ 6] count:    1,  width: 36516 us [36516;36516]       (9129 S)
Level estimates [high, low]:  16003,     58
RSSI: -0.1 dB SNR: 24.4 dB Noise: -24.5 dB
Frequency offsets [F1, F2]:    6007,  -6489     (+22.9 kHz, -24.8 kHz)
Guessing modulation: No clue...

[image]

[image]

[g037_868.3M_1000k.zip](https://github.com/merbanan/rtl_433/files/5214015/g037_868.3M_1000k.zip)
[g031_868.3M_1000k.zip](https://github.com/merbanan/rtl_433/files/5214016/g031_868.3M_1000k.zip)

zuckschwerdt, 2020-09-13:
Note that with 868M the default sample rate will be 1000k, so you need to add that when running `-A` on samples: `-s 1000k`.

zuckschwerdt, 2020-09-13:
You can use `rtl_433 -s 1000k -X 'n=Bresser-7in1,m=FSK_PCM,s=125,l=125,r=9000,preamble=2dd4' gfile.cu8` to read samples and also the -X option for live readings.

Tuxyso, 2020-09-13:
I called

`rtl_433 -f 868.3M -s 1000k -X 'n=Bresser-7in1,m=FSK_PCM,s=125,l=125,r=9000,preamble=2dd4'`

and got:

<code>
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:33:13
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : 1b1905c0a9ba18abaabaaaaaaaaa899afcac8cbdae8aaaaaaa000000000000000000
codes     : {271}1b1905c0a9ba18abaabaaaaaaaaa899afcac8cbdae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:33:25
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : d16905c0bffa18aa9aa9aaaaaaaa899afeac8db3ae8aaaaaaa000000000000000000
codes     : {271}d16905c0bffa18aa9aa9aaaaaaaa899afeac8db3ae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:33:37
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : eab705c0bbea18abbabbaaaaaaaa899afeac8f3eae8aaaaaaa000000000000000000
codes     : {271}eab705c0bbea18abbabbaaaaaaaa899afeac8f3eae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:33:49
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : c63005c08e3a18aa9aa9aaaaaaaa899af9ac8c99ae8aaaaaaa000000000000000000
codes     : {271}c63005c08e3a18aa9aa9aaaaaaaa899af9ac8c99ae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:34:01
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : f21a05c08cba18aa9aa9aaaaaaaa899af9ac8c9fae8aaaaaaa000000000000000000
codes     : {271}f21a05c08cba18aa9aa9aaaaaaaa899af9ac8c9fae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:34:13
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : 02b105c0beda18aaeaaeaaaaaaaa899af9ac8fc9ae8aaaaaaa000000000000000000
codes     : {271}02b105c0beda18aaeaaeaaaaaaaa899af9ac8fc9ae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:34:25
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : 480605c0b2ba18aa8aa8aaaaaaaa899afeac8dbfae8aaaaaaa000000000000000000
codes     : {271}480605c0b2ba18aa8aa8aaaaaaaa899afeac8dbfae8aaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:34:37
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : 812705c08aca18abaabaaaaaaaaa899afeac822cae8aaaaaaa000000000000000000
codes     : {271}812705c08aca18abaabaaaaaaaaa899afeac822cae8aaaaaaa000000000000000000

</code>

zuckschwerdt, 2020-09-13:
This seems to be good data:
```
g001_868.3M_1000k.cu8  {271}9f5d05c08b2a18abcabeaaaaaaaa8adac8acffa9afcaaaaaaa000000000000000000
g004_868.3M_1000k.cu8  {271}631d05c09e9a18abaabaaaaaaaaa8adacbacff9cafcaaaaaaa000000000000000000
g007_868.3M_1000k.cu8  {271}3fb905c08dca18ab9ab9aaaaaaaa8adaceacfcd8afcaaaaaaa000000000000000000
```
Not sure what's encoded though, it doesn't look like the 5-in-1. Can you table more values with expected measurement values in a [BitBench](https://triq.net/bitbench?c=9f5d05c08b2a18abcabeaaaaaaaa8adac8acffa9afcaaaaaaa%5Bolder%5D&c=631d05c09e9a18abaabaaaaaaaaa8adacbacff9cafcaaaaaaa%5Bolder%5D&c=3fb905c08dca18ab9ab9aaaaaaaa8adaceacfcd8afcaaaaaaa%5Bolder%5D&c=1b1905c0a9ba18abaabaaaaaaaaa899afcac8cbdae8aaaaaaa&c=d16905c0bffa18aa9aa9aaaaaaaa899afeac8db3ae8aaaaaaa&c=eab705c0bbea18abbabbaaaaaaaa899afeac8f3eae8aaaaaaa&c=c63005c08e3a18aa9aa9aaaaaaaa899af9ac8c99ae8aaaaaaa&c=f21a05c08cba18aa9aa9aaaaaaaa899af9ac8c9fae8aaaaaaa&c=02b105c0beda18aaeaaeaaaaaaaa899af9ac8fc9ae8aaaaaaa&c=480605c0b2ba18aa8aa8aaaaaaaa899afeac8dbfae8aaaaaaa&c=812705c08aca18abaabaaaaaaaaa899afeac822cae8aaaaaaa&f=8h%20)?

Tuxyso, 2020-09-13:
Additionaly (with len 240, with 1 minute period probably indoor sensor):

<code>
time      : 2020-09-13 14:41:57
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 240          data      : ec76197005fd29000000000023865600004c000007000000000000000000
codes     : {240}ec76197005fd29000000000023865600004c000007000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-13 14:42:57
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 240          data      : 5b3b197005fd29000000000023765600005c000007000000000000000000
codes     : {240}5b3b197005fd29000000000023765600005c000007000000000000000000
</code>

zuckschwerdt, 2020-09-13:
Yes, that matches g005_868.3M_1000k.cu8
`{240}6ab0197005fd29000000000022965700003c000007000000000000000000`
something much different but likely the same protocol, so indoor sensor sounds good.

Tuxyso, 2020-09-13:
Current values:
Outdoor: 23.3° C / 54 % / 1026 hPa
Light: 61.5 kLux
Indoor: 23.7°C / 56 %
no rain currently

zuckschwerdt, 2020-09-13:
There is so much changing data in the codes that it's hard to spot where the value fields are. You need to play with the BitBench to hopefully pin down some values. Humidity might be easiest to spot, with `8d` it should show up somewhere.

Tuxyso, 2020-09-13:
Sorry, but I do not really understand what to do with BitBench. Later on this day I will provide a more comprehensive und systematic measurements - I will exactly describe with values change from file to file. And I will put the indoor sender into the fridge and will additionaly change the channel.

Tuxyso, 2020-09-13:
I made a rather strange observation while analysing the indoor sensor:

56e4197005fd290000000000**242**6**57**0000aa000007000000000000000000  (24,2 / 57 / CH 1)
2b7e197005fd210000000000**255**6**56**000082000007000000000000000000  (25,5 / 56 / CH 1)
118f197005fd220000000000**250**6**57**0000d0000007000000000000000000  (25,0 / 57 / CH 2)
8358197005fd270000000000**253**6**56**00009c000007000000000000000000  (25,3 / 56 / CH 7)

Temperature and humidity is directly hex encoded, hex values = decimal values.

zuckschwerdt, 2020-09-13:
Great! The format is "BCD".
Marking the values (like bold in your post) is exactly what BitBench does :)
This [BitBench](https://triq.net/bitbench?c=56e4197005fd2900000000002426570000aa000007000000000000000000%20%5B24%2C2%20%2F%2057%20%2F%20CH%201%5D&c=2b7e197005fd210000000000255656000082000007000000000000000000%20%5B25%2C5%20%2F%2056%20%2F%20CH%201%5D&c=118f197005fd2200000000002506570000d0000007000000000000000000%20%5B25%2C0%20%2F%2057%20%2F%20CH%202%5D&c=8358197005fd27000000000025365600009c000007000000000000000000%20%5B25%2C3%20%2F%2056%20%2F%20CH%207%5D&f=8h8h%208h8h%208h8h%208h8h%208h8h%208h8h%20TEMP%3A12h%204h%20HUM%3A8h%208h8h%20CHK%3F8h%208h%208h8h%20TRAILER%3F8h8h%208h8h%208h8h%208h8h%208h) should better show how to mark and extract values.

Tuxyso, 2020-09-13:
Further analysis for outdoor sensor:

<code>
2b1f05c0beda18aaaaaaaaaaaaaa8faae**3a**b892faa2aaaaaaa000000000000000000 (25,0 / **49**)
f85505c0bfba18aabaabaaaaaaaa8faae**3a**b88acaa3aaaaaaa000000000000000000 (25,0 / **49**)
8c2a05c0a8fa18aabaabaaaaaaaa8efae**3a**ed3b9a92aaaaaaa000000000000000000 (24,5 / **49**)

f4a105c0beaa18aa9aa9aaaaaaaa8f_aae_**2a**b882baa3aaaaaaa000000000000000000 (_25,0_ / **48**)
26c305c0bc9a18aa8aa8aaaaaaaa8f_aae_**2a**b8aadaa3aaaaaaa000000000000000000 (_25,0_ / **48**)
439a05c0bcba18aaaaaaaaaaaaaa8f_bae_**2a**b83fdaa3aaaaaaa000000000000000000 (_25,1_ / **48**)
1d4005c0888a18aabaabaaaaaaaa8ecae**2a**efef2a9daaaaaaa000000000000000000   (24,6 / **48**)
c7dd05c08bda18aa8aa8aaaaaaaa8eeae**2a**e2d83aeaaaaaaaa000000000000000000   (24,4 / **48** )
967005c09b3a18aa8aa8aaaaaaaa8efae**2a**e28cea93aaaaaaa000000000000000000   (24,5 / **48** )
</code>

zuckschwerdt, 2020-09-13:
Here is the [BitBench with this data](https://triq.net/bitbench?c=2b1f05c0beda18aaaaaaaaaaaaaa8f%20aae%203a%20b892faa2aaaaaaa000000000000000000%20%5B25%2C0%2049%25%5D&c=f85505c0bfba18aabaabaaaaaaaa8f%20aae%203a%20b88acaa3aaaaaaa000000000000000000%20%5B25%2C0%2049%25%5D&c=8c2a05c0a8fa18aabaabaaaaaaaa8e%20fae%203a%20ed3b9a92aaaaaaa000000000000000000%20%5B24%2C5%2049%25%5D&c=f4a105c0beaa18aa9aa9aaaaaaaa8f%20aae%202a%20b882baa3aaaaaaa000000000000000000%20%5B25%2C0%2048%25%5D&c=26c305c0bc9a18aa8aa8aaaaaaaa8f%20aae%202a%20b8aadaa3aaaaaaa000000000000000000%20%5B25%2C0%2048%25%5D&c=439a05c0bcba18aaaaaaaaaaaaaa8f%20bae%202a%20b83fdaa3aaaaaaa000000000000000000%20%5B25%2C1%2048%25%5D&c=1d4005c0888a18aabaabaaaaaaaa8e%20cae%202a%20efef2a9daaaaaaa000000000000000000%20%5B24%2C6%2048%5D&c=c7dd05c08bda18aa8aa8aaaaaaaa8e%20eae%202a%20e2d83aeaaaaaaaa000000000000000000%20%5B24%2C4%2048%5D&c=967005c09b3a18aa8aa8aaaaaaaa8e%20fae%202a%20e28cea93aaaaaaa000000000000000000%20%5B24%2C5%2048%5D&f=8h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h%20TEMP%3A12d%20HUM%3A8d%208h8h%20CHK%3F8h%208h%208h8h8h4h%20TRAILER%3F8h8h%208h8h%208h8h%208h8h%208h). The fields seem right but are not easy to decode?

Tuxyso, 2020-09-13:
The temperature / humidity field behaves strange. I summarize my current insights;

1) **same hex value means same humidity** (but some hex values stand for different humidities). 

hex | humidity (%)
-- | --
2a | 48
3a | 49
aa | 50
fa | 75
  | 76
  | 77
2a | 78
3a | 79
aa | 80
ba | 81
8a | 82
9a | 83
ea | 84

Strange is that _2a_ stands for 48% and for 78 %:

<code>
f4a105c0beaa18aa9aa9aaaaaaaa8f aae 2a b882baa3aaaaaaa000000000000000000 [25,0 48]
106205c09aca18aaaaaaaaaaaaaabd aad 2a aaaaaaaaaaaaaaa000000000000000000 [17,0 78]
</code>

2) **hex values for humidity increment, but there are jumps**:
2a = 48
3a = 49
aa = 50
8a =  82
9a =  83
ea =  84

3) **same temperature / different values** 
temperature increments, but there seem to be a strange interdependency from humidy when changing the tenner (e.g. 79 -> 80). 

See the following annomaly:

<code>
c3c805c09afa18aaaaaaaaaaaaaabc dad 3a aaaaaaaaaaaaaaa000000000000000000 [16,7 79]
d74c05c09afa18aaaaaaaaaaaaaabc da2 aa aaaaaaaaaaaaaaa000000000000000000 [16,7 80]
</code>

Here you get for 16,7°C different values (probably dependent from humidity). Another example:

<code>
f85505c0bfba18aabaabaaaaaaaa8f aae 3a b88acaa3aaaaaaa000000000000000000 [25,0 49] 
9c7205c0bcba18aaaaaaaaaaaaaa8f baf aa b8d92aa3aaaaaaa000000000000000000 [25,1 50] 
</code>
Normally 0.1 change increments in the first number. Thus after aae (for 25,0) it should be bae (for 25.1) but it is rather different.

Any hints / ideas, experts?

Complete sample:

<code>
2b1f05c0beda18aaaaaaaaaaaaaa8f aae 3a b892faa2aaaaaaa000000000000000000 [25,0 49]
f85505c0bfba18aabaabaaaaaaaa8f aae 3a b88acaa3aaaaaaa000000000000000000 [25,0 49] A
9c7205c0bcba18aaaaaaaaaaaaaa8f baf aa b8d92aa3aaaaaaa000000000000000000 [25,1 50] A 
8c2a05c0a8fa18aabaabaaaaaaaa8e fae 3a ed3b9a92aaaaaaa000000000000000000 [24,5 49]
f4a105c0beaa18aa9aa9aaaaaaaa8f aae 2a b882baa3aaaaaaa000000000000000000 [25,0 48]
26c305c0bc9a18aa8aa8aaaaaaaa8f aae 2a b8aadaa3aaaaaaa000000000000000000 [25,0 48]
439a05c0bcba18aaaaaaaaaaaaaa8f bae 2a b83fdaa3aaaaaaa000000000000000000 [25,1 48]
1d4005c0888a18aabaabaaaaaaaa8e cae 2a efef2a9daaaaaaa000000000000000000 [24,6 48]
c7dd05c08bda18aa8aa8aaaaaaaa8e eae 2a e2d83aeaaaaaaaa000000000000000000 [24,4 48]
967005c09b3a18aa8aa8aaaaaaaa8e fae 2a e28cea93aaaaaaa000000000000000000 [24,5 48]
93ad05c09aca18aaaaaaaaaaaaaabd 2ad fa aab9faaaaaaaaaa000000000000000000 [17,8 75]
106205c09aca18aaaaaaaaaaaaaabd aad 2a aaaaaaaaaaaaaaa000000000000000000 [17,0 78]
57b105c09aca18aaaaaaaaaaaaaabd aad 3a aaaaaaaaaaaaaaa000000000000000000 [17,0 79]
87ef05c09aca18aaaaaaaaaaaaaabc 3a2 ba aaaaaaaaaaaaaaa000000000000000000 [16,9 81]
4f9a05c09aca18aaaaaaaaaaaaaabc 3a2 8a aaaaaaaaaaaaaaa000000000000000000 [16,9 82]
5bb705c09aca18aaaaaaaaaaaaaabc 2a2 9a aaaaaaaaaaaaaaa000000000000000000 [16,8 83]
4fde05c09aca18aaaaaaaaaaaaaabc da2 9a aaaaaaaaaaaaaaa000000000000000000 [16,7 83]
080d05c09aca18aaaaaaaaaaaaaabc da2 8a aaaaaaaaaaaaaaa000000000000000000 [16,7 82]
a3db05c09ada18aaaaaaaaaaaaaabc ca2 ba aaaaaaaaaaaaaaa000000000000000000 [16,6 81]
e75205c09a2a18aaaaaaaaaaaaaabc ca2 aa aaaaaaaaaaaaaaa000000000000000000 [16,6 80]
54ac05c09baa18aaaaaaaaaaaaaabc cad 3a aaaaaaaaaaaaaaa000000000000000000 [16,6 79]
64f105c09bba18aaaaaaaaaaaaaabc cad 3a aaaaaaaaaaaaaaa000000000000000000 [16,6 79] A
c3c805c09afa18aaaaaaaaaaaaaabc dad 3a aaaaaaaaaaaaaaa000000000000000000 [16,7 79]
d74c05c09afa18aaaaaaaaaaaaaabc da2 aa aaaaaaaaaaaaaaa000000000000000000 [16,7 80]
909f05c09afa18aaaaaaaaaaaaaabc da2 ba aaaaaaaaaaaaaaa000000000000000000 [16,7 81]
58ea05c09afa18aaaaaaaaaaaaaabc da2 8a aaaaaaaaaaaaaaa000000000000000000 [16,7 82]
4fde05c09aca18aaaaaaaaaaaaaabc da2 9a aaaaaaaaaaaaaaa000000000000000000 [16,7 83]
1c2005c09aca18aaaaaaaaaaaaaabc ca2 9a aaaaaaaaaaaaaaa000000000000000000 [16,6 83] A
ff1605c09afa18aaaaaaaaaaaaaabc fa2 8a aaaaaaaaaaaaaaa000000000000000000 [16,5 82]
ff1605c09afa18aaaaaaaaaaaaaabc fa2 8a aaaaaaaaaaaaaaa000000000000000000 [16,5 82]
ace805c09afa18aaaaaaaaaaaaaabc ea2 8a aaaaaaaaaaaaaaa000000000000000000 [16,4 82] 
205705c09aca18aaaaaaaaaaaaaabc fa2 aa aaaaaaaaaaaaaaa000000000000000000 [16,5 80]
ff1605c09afa18aaaaaaaaaaaaaabc fa2 8a aaaaaaaaaaaaaaa000000000000000000 [16,5 82]
7cc405c09aca18aaaaaaaaaaaaaabc ea2 ea aaaaaaaaaaaaaaa000000000000000000 [16,4 84]
a00705c09baa18aaaaaaaaaaaaaabc ea2 ba aaaaaaaaaaaaaaa000000000000000000 [16,4 81]
</code>

zuckschwerdt, 2020-09-13:
Looks like humidity is not the two nibbles you selected but one nibble to the left. Also it dies look BCD but with XOR 0xaa (something like whitening maybe?) -- which fits all the 0xaa we see, they would be 0x00 then):

```
2b1f05c0beda18aaaaaaaaaaaaaa 8faae3 ab892faa2aaaaaaa [25,0 49]  : 8faae3^aaaaaa = 250 0 49
f85505c0bfba18aabaabaaaaaaaa 8faae3 ab88acaa3aaaaaaa [25,0 49]  : 8faae3^aaaaaa = 250 0 49
9c7205c0bcba18aaaaaaaaaaaaaa 8fbafa ab8d92aa3aaaaaaa [25,1 50]  : 8fbafa^aaaaaa = 251 0 50
8c2a05c0a8fa18aabaabaaaaaaaa 8efae3 aed3b9a92aaaaaaa [24,5 49]  : 8efae3^aaaaaa = 245 0 49
f4a105c0beaa18aa9aa9aaaaaaaa 8faae2 ab882baa3aaaaaaa [25,0 48]  : 8faae2^aaaaaa = 250 0 48
26c305c0bc9a18aa8aa8aaaaaaaa 8faae2 ab8aadaa3aaaaaaa [25,0 48]  : 8faae2^aaaaaa = 250 0 48
439a05c0bcba18aaaaaaaaaaaaaa 8fbae2 ab83fdaa3aaaaaaa [25,1 48]  : 8fbae2^aaaaaa = 251 0 48
1d4005c0888a18aabaabaaaaaaaa 8ecae2 aefef2a9daaaaaaa [24,6 48]  : 8ecae2^aaaaaa = 246 0 48
c7dd05c08bda18aa8aa8aaaaaaaa 8eeae2 ae2d83aeaaaaaaaa [24,4 48]  : 8eeae2^aaaaaa = 244 0 48
967005c09b3a18aa8aa8aaaaaaaa 8efae2 ae28cea93aaaaaaa [24,5 48]  : 8efae2^aaaaaa = 245 0 48
93ad05c09aca18aaaaaaaaaaaaaa bd2adf aaab9faaaaaaaaaa [17,8 75]  : bd2adf^aaaaaa = 178 0 75
106205c09aca18aaaaaaaaaaaaaa bdaad2 aaaaaaaaaaaaaaaa [17,0 78]  : bdaad2^aaaaaa = 170 0 78
57b105c09aca18aaaaaaaaaaaaaa bdaad3 aaaaaaaaaaaaaaaa [17,0 79]  : bdaad3^aaaaaa = 170 0 79
87ef05c09aca18aaaaaaaaaaaaaa bc3a2b aaaaaaaaaaaaaaaa [16,9 81]  : bc3a2b^aaaaaa = 169 0 81
4f9a05c09aca18aaaaaaaaaaaaaa bc3a28 aaaaaaaaaaaaaaaa [16,9 82]  : bc3a28^aaaaaa = 169 0 82
5bb705c09aca18aaaaaaaaaaaaaa bc2a29 aaaaaaaaaaaaaaaa [16,8 83]  : bc2a29^aaaaaa = 168 0 83
4fde05c09aca18aaaaaaaaaaaaaa bcda29 aaaaaaaaaaaaaaaa [16,7 83]  : bcda29^aaaaaa = 167 0 83
080d05c09aca18aaaaaaaaaaaaaa bcda28 aaaaaaaaaaaaaaaa [16,7 82]  : bcda28^aaaaaa = 167 0 82
a3db05c09ada18aaaaaaaaaaaaaa bcca2b aaaaaaaaaaaaaaaa [16,6 81]  : bcca2b^aaaaaa = 166 0 81
e75205c09a2a18aaaaaaaaaaaaaa bcca2a aaaaaaaaaaaaaaaa [16,6 80]  : bcca2a^aaaaaa = 166 0 80
54ac05c09baa18aaaaaaaaaaaaaa bccad3 aaaaaaaaaaaaaaaa [16,6 79]  : bccad3^aaaaaa = 166 0 79
64f105c09bba18aaaaaaaaaaaaaa bccad3 aaaaaaaaaaaaaaaa [16,6 79]  : bccad3^aaaaaa = 166 0 79
c3c805c09afa18aaaaaaaaaaaaaa bcdad3 aaaaaaaaaaaaaaaa [16,7 79]  : bcdad3^aaaaaa = 167 0 79
d74c05c09afa18aaaaaaaaaaaaaa bcda2a aaaaaaaaaaaaaaaa [16,7 80]  : bcda2a^aaaaaa = 167 0 80
909f05c09afa18aaaaaaaaaaaaaa bcda2b aaaaaaaaaaaaaaaa [16,7 81]  : bcda2b^aaaaaa = 167 0 81
58ea05c09afa18aaaaaaaaaaaaaa bcda28 aaaaaaaaaaaaaaaa [16,7 82]  : bcda28^aaaaaa = 167 0 82
4fde05c09aca18aaaaaaaaaaaaaa bcda29 aaaaaaaaaaaaaaaa [16,7 83]  : bcda29^aaaaaa = 167 0 83
1c2005c09aca18aaaaaaaaaaaaaa bcca29 aaaaaaaaaaaaaaaa [16,6 83]  : bcca29^aaaaaa = 166 0 83
ff1605c09afa18aaaaaaaaaaaaaa bcfa28 aaaaaaaaaaaaaaaa [16,5 82]  : bcfa28^aaaaaa = 165 0 82
ff1605c09afa18aaaaaaaaaaaaaa bcfa28 aaaaaaaaaaaaaaaa [16,5 82]  : bcfa28^aaaaaa = 165 0 82
ace805c09afa18aaaaaaaaaaaaaa bcea28 aaaaaaaaaaaaaaaa [16,4 82]  : bcea28^aaaaaa = 164 0 82
205705c09aca18aaaaaaaaaaaaaa bcfa2a aaaaaaaaaaaaaaaa [16,5 80]  : bcfa2a^aaaaaa = 165 0 80
ff1605c09afa18aaaaaaaaaaaaaa bcfa28 aaaaaaaaaaaaaaaa [16,5 82]  : bcfa28^aaaaaa = 165 0 82
7cc405c09aca18aaaaaaaaaaaaaa bcea2e aaaaaaaaaaaaaaaa [16,4 84]  : bcea2e^aaaaaa = 164 0 84
a00705c09baa18aaaaaaaaaaaaaa bcea2b aaaaaaaaaaaaaaaa [16,4 81]  : bcea2b^aaaaaa = 164 0 81
```

zuckschwerdt, 2020-09-13:
Your big table of values and readings was great help for this satisfying finding, thanks!

I would have said "encryption" at first, but actually it make sense: If you use PCM and expect long runs of 0's you are in trouble with timings. Using a pseudo-whitening of 0xaa is the easiest trick to get around that.

@merbanan @evilpete I have not seen this before but it might be a trick used in other protocols where we suscpect some encryption. Worth remembering and trying out!

I'll add a whitening feature for this to BitBench so we can check easily :)

Tuxyso, 2020-09-13:
**Stunning, great work!**

If you give me a handy way to decode the signals I can also help to identify the other data fields, like (wind speed, wind direction, rain values, light intensity) from the outdoor sensor.

zuckschwerdt, 2020-09-13:
I've now added the xor /whitening to BitBench. Take a look [in this BitBench](https://triq.net/bitbench?c=2b1f05c0beda18aaaaaaaaaaaaaa%208faae3%20ab892faa2aaaaaaa%20%5B25%2C0%2049%5D&c=f85505c0bfba18aabaabaaaaaaaa%208faae3%20ab88acaa3aaaaaaa%20%5B25%2C0%2049%5D&c=9c7205c0bcba18aaaaaaaaaaaaaa%208fbafa%20ab8d92aa3aaaaaaa%20%5B25%2C1%2050%5D&c=8c2a05c0a8fa18aabaabaaaaaaaa%208efae3%20aed3b9a92aaaaaaa%20%5B24%2C5%2049%5D&c=f4a105c0beaa18aa9aa9aaaaaaaa%208faae2%20ab882baa3aaaaaaa%20%5B25%2C0%2048%5D&c=26c305c0bc9a18aa8aa8aaaaaaaa%208faae2%20ab8aadaa3aaaaaaa%20%5B25%2C0%2048%5D&c=439a05c0bcba18aaaaaaaaaaaaaa%208fbae2%20ab83fdaa3aaaaaaa%20%5B25%2C1%2048%5D&c=1d4005c0888a18aabaabaaaaaaaa%208ecae2%20aefef2a9daaaaaaa%20%5B24%2C6%2048%5D&c=c7dd05c08bda18aa8aa8aaaaaaaa%208eeae2%20ae2d83aeaaaaaaaa%20%5B24%2C4%2048%5D&c=967005c09b3a18aa8aa8aaaaaaaa%208efae2%20ae28cea93aaaaaaa%20%5B24%2C5%2048%5D&c=93ad05c09aca18aaaaaaaaaaaaaa%20bd2adf%20aaab9faaaaaaaaaa%20%5B17%2C8%2075%5D&c=106205c09aca18aaaaaaaaaaaaaa%20bdaad2%20aaaaaaaaaaaaaaaa%20%5B17%2C0%2078%5D&c=57b105c09aca18aaaaaaaaaaaaaa%20bdaad3%20aaaaaaaaaaaaaaaa%20%5B17%2C0%2079%5D&c=87ef05c09aca18aaaaaaaaaaaaaa%20bc3a2b%20aaaaaaaaaaaaaaaa%20%5B16%2C9%2081%5D&c=4f9a05c09aca18aaaaaaaaaaaaaa%20bc3a28%20aaaaaaaaaaaaaaaa%20%5B16%2C9%2082%5D&c=5bb705c09aca18aaaaaaaaaaaaaa%20bc2a29%20aaaaaaaaaaaaaaaa%20%5B16%2C8%2083%5D&c=4fde05c09aca18aaaaaaaaaaaaaa%20bcda29%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2083%5D&c=080d05c09aca18aaaaaaaaaaaaaa%20bcda28%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2082%5D&c=a3db05c09ada18aaaaaaaaaaaaaa%20bcca2b%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2081%5D&c=e75205c09a2a18aaaaaaaaaaaaaa%20bcca2a%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2080%5D&c=54ac05c09baa18aaaaaaaaaaaaaa%20bccad3%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2079%5D&c=64f105c09bba18aaaaaaaaaaaaaa%20bccad3%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2079%5D&c=c3c805c09afa18aaaaaaaaaaaaaa%20bcdad3%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2079%5D&c=d74c05c09afa18aaaaaaaaaaaaaa%20bcda2a%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2080%5D&c=909f05c09afa18aaaaaaaaaaaaaa%20bcda2b%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2081%5D&c=58ea05c09afa18aaaaaaaaaaaaaa%20bcda28%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2082%5D&c=4fde05c09aca18aaaaaaaaaaaaaa%20bcda29%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2083%5D&c=1c2005c09aca18aaaaaaaaaaaaaa%20bcca29%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2083%5D&c=ff1605c09afa18aaaaaaaaaaaaaa%20bcfa28%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2082%5D&c=ff1605c09afa18aaaaaaaaaaaaaa%20bcfa28%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2082%5D&c=ace805c09afa18aaaaaaaaaaaaaa%20bcea28%20aaaaaaaaaaaaaaaa%20%5B16%2C4%2082%5D&c=205705c09aca18aaaaaaaaaaaaaa%20bcfa2a%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2080%5D&c=ff1605c09afa18aaaaaaaaaaaaaa%20bcfa28%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2082%5D&c=7cc405c09aca18aaaaaaaaaaaaaa%20bcea2e%20aaaaaaaaaaaaaaaa%20%5B16%2C4%2084%5D&c=a00705c09baa18aaaaaaaaaaaaaa%20bcea2b%20aaaaaaaaaaaaaaaa%20%5B16%2C4%2081%5D&f=8h8h%20ID%3F8h8h%208h8h%208h8h%208h8h%208h8h%208h8h%20TEMP%3A8h.4hC%204h%20HUM%3A8h%25%208h8h%20CHK%3F8h%208h%208h8h8h8h&x=a).

Just add codes in the first text box (green text) and then modify the string in the second text box. All uppercase letters will appear in the output, number and lowercase letters select some decoding.

Tuxyso, 2020-09-13:
From the recorded values I could figure out some other data fields:

<code>
8h8h ID?8h8h WIND-DIR:8h4h.4h° 8h8h WIND-SPEED:8h8h 8h8h 8h8h TEMP:8h.4hC 4h HUM:8h% LIGHT:8h8hW/m² CHK?8h 8h 8h8h8h8h
</code>

WIND-DIR is 100% sure / WIND-SPEED I am not sure about the unit here; LIGHT in W/m² is 95% sure (fits to the data I have recorded during the day), I think the fields after WIND-SPEED have to do with rain (currently I cannot verify it due to missing rain)

With this decoding I get for example:

[image]

evilpete, 2020-09-14:
> @merbanan @evilpete I have not seen this before but it might be a trick used in other protocols where we suscpect some encryption. Worth remembering and trying out!

Insteon devices pad  between packets with 00110011001 till it is done

Tuxyso, 2020-09-16:
I could further improve decoding, see [Bitbench](https://triq.net/bitbench?c=2b1f05c0beda18aaaaaaaaaaaaaa%208faae3%20ab892faa2aaaaaaa%20%5B25%2C0%2049%5D&c=f85505c0bfba18aabaabaaaaaaaa%208faae3%20ab88acaa3aaaaaaa%20%5B25%2C0%2049%5D&c=9c7205c0bcba18aaaaaaaaaaaaaa%208fbafa%20ab8d92aa3aaaaaaa%20%5B25%2C1%2050%5D&c=8c2a05c0a8fa18aabaabaaaaaaaa%208efae3%20aed3b9a92aaaaaaa%20%5B24%2C5%2049%5D&c=f4a105c0beaa18aa9aa9aaaaaaaa%208faae2%20ab882baa3aaaaaaa%20%5B25%2C0%2048%5D&c=26c305c0bc9a18aa8aa8aaaaaaaa%208faae2%20ab8aadaa3aaaaaaa%20%5B25%2C0%2048%5D&c=439a05c0bcba18aaaaaaaaaaaaaa%208fbae2%20ab83fdaa3aaaaaaa%20%5B25%2C1%2048%5D&c=1d4005c0888a18aabaabaaaaaaaa%208ecae2%20aefef2a9daaaaaaa%20%5B24%2C6%2048%5D&c=c7dd05c08bda18aa8aa8aaaaaaaa%208eeae2%20ae2d83aeaaaaaaaa%20%5B24%2C4%2048%5D&c=967005c09b3a18aa8aa8aaaaaaaa%208efae2%20ae28cea93aaaaaaa%20%5B24%2C5%2048%5D&c=93ad05c09aca18aaaaaaaaaaaaaa%20bd2adf%20aaab9faaaaaaaaaa%20%5B17%2C8%2075%5D&c=106205c09aca18aaaaaaaaaaaaaa%20bdaad2%20aaaaaaaaaaaaaaaa%20%5B17%2C0%2078%5D&c=57b105c09aca18aaaaaaaaaaaaaa%20bdaad3%20aaaaaaaaaaaaaaaa%20%5B17%2C0%2079%5D&c=87ef05c09aca18aaaaaaaaaaaaaa%20bc3a2b%20aaaaaaaaaaaaaaaa%20%5B16%2C9%2081%5D&c=4f9a05c09aca18aaaaaaaaaaaaaa%20bc3a28%20aaaaaaaaaaaaaaaa%20%5B16%2C9%2082%5D&c=5bb705c09aca18aaaaaaaaaaaaaa%20bc2a29%20aaaaaaaaaaaaaaaa%20%5B16%2C8%2083%5D&c=4fde05c09aca18aaaaaaaaaaaaaa%20bcda29%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2083%5D&c=080d05c09aca18aaaaaaaaaaaaaa%20bcda28%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2082%5D&c=a3db05c09ada18aaaaaaaaaaaaaa%20bcca2b%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2081%5D&c=e75205c09a2a18aaaaaaaaaaaaaa%20bcca2a%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2080%5D&c=54ac05c09baa18aaaaaaaaaaaaaa%20bccad3%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2079%5D&c=64f105c09bba18aaaaaaaaaaaaaa%20bccad3%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2079%5D&c=c3c805c09afa18aaaaaaaaaaaaaa%20bcdad3%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2079%5D&c=d74c05c09afa18aaaaaaaaaaaaaa%20bcda2a%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2080%5D&c=909f05c09afa18aaaaaaaaaaaaaa%20bcda2b%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2081%5D&c=58ea05c09afa18aaaaaaaaaaaaaa%20bcda28%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2082%5D&c=4fde05c09aca18aaaaaaaaaaaaaa%20bcda29%20aaaaaaaaaaaaaaaa%20%5B16%2C7%2083%5D&c=1c2005c09aca18aaaaaaaaaaaaaa%20bcca29%20aaaaaaaaaaaaaaaa%20%5B16%2C6%2083%5D&c=ff1605c09afa18aaaaaaaaaaaaaa%20bcfa28%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2082%5D&c=ff1605c09afa18aaaaaaaaaaaaaa%20bcfa28%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2082%5D&c=ace805c09afa18aaaaaaaaaaaaaa%20bcea28%20aaaaaaaaaaaaaaaa%20%5B16%2C4%2082%5D&c=205705c09aca18aaaaaaaaaaaaaa%20bcfa2a%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2080%5D&c=ff1605c09afa18aaaaaaaaaaaaaa%20bcfa28%20aaaaaaaaaaaaaaaa%20%5B16%2C5%2082%5D&c=7cc405c09aca18aaaaaaaaaaaaaa%20bcea2e%20aaaaaaaaaaaaaaaa%20%5B16%2C4%2084%5D&c=a00705c09baa18aaaaaaaaaaaaaa%20bcea2b%20aaaaaaaaaaaaaaaa%20%5B16%2C4%2081%5D&c=0f7705c0ad8a18a8ea89aaaaaaaa8ccaf9a92bdfa9baaaaaaa%20%5B2%2C3%20m%2Fs%20ENE%5D&c=dad305c0bbca18aa8aa8aaaaaaaa8ccafea9282ea99aaaaaaa%20%5B0%2C4m%2Fs%5D&c=bd9305c08dca18aa8aa8aaaaaaaa8ccaf9a92b8ca99aaaaaaa%20%5B0%2C2%20W%5D&c=281405c0ba9a18ab8ab8aaaaaaaa8ccaf9a9d2e2a99aaaaaaa%20%5B1%2C2%20ESE%5D&c=914705c0a3da18abdabcaaaaaaaa8cfaf9a9d898a9baaaaaaa%20%5B1%2C6%20E%5D&c=eb2e05c0baaa18a89a88aaaaaaaa8c9afba999aaa8daaaaaaa%20%5B33%2C3%20kLux%5D&c=4c6f05c09aaa18aaeaaeaaaaaaaa8c8afba98fd9a8eaaaaaaa%20%5B32%2C5%20kLux%5D&f=8h8h%20ID%3F8h8h%20WIND-DIR%3A8h4h.4h%C2%B0%208h8h%206h%20WIND-SPEED%3A2h2h2h.4h%20RAIN%3F8h8h%20RAIN%3F8h8h%20TEMP%3A8h.4hC%204h%20HUM%3A8h%25%20LIGHT%3A8h4h%2C4h%20kl%20CHK%3F8h%208h%208h8h8h8h%0A&x=a):
<code>
8h8h ID?8h8h WIND-DIR:8h4h.4h° 8h8h 6h WIND-SPEED:2h2h2h.4h RAIN?8h8h RAIN?8h8h TEMP:8h.4hC 4h HUM:8h% LIGHT:8h4h,4h kl CHK?8h 8h 8h8h8h8h
</code>

Unit of light is kLux (not W/m²). 

Any ideas on pressure? I have absolutely no clue.

zuckschwerdt, 2020-09-16:
I guess I found the checksum: The first two bytes are an LFSR-16 digest, generator 0x8810 with the next two bytes (ID) as init.

Usually barometric pressure is measured indoors in the display unit. There would be no difference to a remote sensor reading (apart from height offset).

I'll write a decoder for this in the next days.

zuckschwerdt, 2020-09-20:
There is some early test code in https://github.com/merbanan/rtl_433/tree/feat-bresser7in1 now. The checksum (likely digest) needs more work -- we really need that to validate the packets. Can you grab a good number of unique codes (say 50-100)? Either the raw bitbuffer codes like before, or the preamble-aligned msg from this branch or the xor'd ones, doesn't matter.

Tuxyso, 2020-09-21:
I've recorded 85 unique checksums (see attachment).  Do you need a similiar sample for the indoor sensor?

Preview: I've just ordered the 

https://www.bresser.de/en/Weather-Time/Accessories/EXPLORE-SCIENTIFIC-Soil-Moisture-and-Soil-Temperature-Sensor.html

Soil-Moisture-and-Soil-Temperature-Sensor. I will provide sample data as soon as possible.

Still waiting for rain in a view days :)

[outdoor_sensor_checksum.txt](https://github.com/merbanan/rtl_433/files/5258156/outdoor_sensor_checksum.txt)

zuckschwerdt, 2020-09-22:
There is a working checksum for the tabled codes: LFSR-16 digest, generator 0x8810 key 0xba95 final xor 0x6df1.
That final xor is strange, and I can't say if the key is fixed for all units or specific to yours and negotiated when pairing.

Can you also get codes for the indoor sensor? Then checksum should be something similar and might give more insight.

The test branch doesn't do much, but works:
```
model     : Bresser-7in1outdoor                    id        : 44906
Temperature: 20.7 C      Humidity  : 61            Wind Gust : 1.0 m/s       Wind Speed: 1.0 m/s
       Direction : 343 °        Light     : 65.5 klx
```

Tuxyso, 2020-09-22:
I've recorded 39 samples (see attached file). During recording I changed the channel so you can check a possible effect on checksum calculation. 
Do not be confused about the changing temperature - I put the sensor into the fridge :)

[indoor-sensor-checksum.txt](https://github.com/merbanan/rtl_433/files/5263997/indoor-sensor-checksum.txt)

Tuxyso, 2020-09-22:
Just another remark. Negative temperature are as follows for the indoor sensor:
`
codes     : {242}ac04197005fd290000000000 996e30000014000001c000000000000000000 [-0,4°]
codes     : {240}736e197005fd290000000000 983e30000045ffff07000000000000000000 [-1,7 30%]
codes     : {240}4f3b197005fd290000000000 977e31000005ffff07000000000000000000 [-2,3 31%]
codes     : {240}16e9197005fd290000000000971e31000065ffff07000000000000000000 [-2,9 31%]
codes     : {240}fecd197005fd290000000000966e31000016000007000000000000000000 [-3,4 31 % ]
codes     : {242}902f197005fd290000000000960e30000077ffffc1c000000000000000000 [-4 30 % ]

`
It seems that negative temperature begin at 99.. There is also an effect on humidity encoding if the temperature is negative

zuckschwerdt, 2020-09-23:
Here is a [BitBench of the Indoor sensor](https://triq.net/bitbench?c=56e4197005fd2900000000002426570000aa000007000000000000000000%20%5B24%2C2%20%2F%2057%20%2F%20CH%201%5D&c=2b7e197005fd210000000000255656000082000007000000000000000000%20%5B25%2C5%20%2F%2056%20%2F%20CH%201%5D&c=118f197005fd2200000000002506570000d0000007000000000000000000%20%5B25%2C0%20%2F%2057%20%2F%20CH%202%5D&c=8358197005fd27000000000025365600009c000007000000000000000000%20%5B25%2C3%20%2F%2056%20%2F%20CH%207%5D&c=ac04197005fd290000000000996e30000014000001c00000000000000000%20%5B-0%2C4%C2%B0%5D&c=736e197005fd290000000000983e30000045ffff07000000000000000000%20%5B-1%2C7%2030%25%5D&c=4f3b197005fd290000000000977e31000005ffff07000000000000000000%20%5B-2%2C3%2031%25%5D&c=16e9197005fd290000000000971e31000065ffff07000000000000000000%20%5B-2%2C9%2031%25%5D&c=fecd197005fd290000000000966e31000016000007000000000000000000%20%5B-3%2C4%2031%20%25%20%5D&c=902f197005fd290000000000960e30000077ffffc1c00000000000000000%20%5B-4%2030%20%25%20%5D&f=CHK%3A8h8h%20ID%3F8h8h8h8h%20FLAGS%3A4h%20BATT%3A1b%20CH%3A3d%208h%208h8h%208h8h%20TEMP%3A12h%204b%20HUM%3A8h%208h8h%20CHK%3F8h%208h%208h8h%20TRAILER%3F8h8h%208h8h%208h8h%208h8h%208h) with all the temperatures. Looks like the bit after the Temp could flag a negative temperature, but we can also assume e.g. `if temp > 70 then temp = temp - 100`. Humidity seems okay?

zuckschwerdt, 2020-09-23:
Just to note it here: the checksum at the end is an 8-bit wide add (with carry). For the indoor message this means skip two bytes (digest), sum 16 bytes to 0xff. This should also signal the end of the message as a checksum is usually the last field.

zuckschwerdt, 2020-09-23:
For the indoor sensor the digest is gen 0x8810 key 0x5412 over the bytes excluding the add-checksum :)

Tuxyso, 2020-09-23:
> Humidity seems okay?

As far as I remeber the these values are the correct humidity values.

I've just purchased  Soil moisture Sensor _SM60020_ for the weather station, see:

https://www.bresser.de/en/Weather-Time/Accessories/EXPLORE-SCIENTIFIC-Soil-Moisture-and-Soil-Temperature-Sensor.html

Bresser Explore Scientific SM60020.

The sensor sends every 5 minutes signals to the basis. I've recorded some data and tried decoding in [Bitbench](https://triq.net/bitbench?c=f16e%20187000e34%207%20ffffff0000%20252%20216%20fff%20004%20000%20%5B25%2C2%2C%2099%25%2C%20CH%207%5D&c=9c75%20187000e34%206%20ffffff0000%20279%20216%20fff%20093%20000%20%5B27%2C9%2099%25%20CH%206%5D&c=&c=c26a%20187000e34%206%20ffffff0000%20329%20215%20fff%20287%20000%20%5B32%2C9%2093%25%5D&c=&c=1af2%20187000e34%206%20ffffff0000%20317%20213%20fff%202aa%20000%20%5B31%2C7%2080%25%20CH%206%5D&c=&c=&c=1ce8%20187000e34%204%20ffffff0000%20270%20201%20fff%20238%20000%20%5B27%2C0%200%25%2C%20CH%204%5D&c=547f%20187000e34%206%20ffffff0000%20267%20201%20fff%202c7%20000%20%5B26%2C7%200%25%2C%20CH%206%5D&c=fa18%20187000e34%207%20ffffff0000%20232%20201%20fff%2001b%20000%20%5B23%2C2%2C%200%25%2C%20CH%207%5D&c=&c=c3bf%20187000e34%206%20ffffff0000%20257%20202%20fff%200c9%20000%20%5B25%2C7%207%20%25%2C%20CH%206%5D&c=e2c5%20187000e34%206%20ffffff0000%20287%20202%20fff%200c6%20000%20%5B28%2C7%207%20%25%2C%20CH%206%5D&f=DIGEST%3A8h8h%20ID%3F8h8h8h8h%20FLAGS%3A4h%20BATT%3A1b%20CH%3A3d%208h%208h8h%208h8h%20TEMP%3A12h%20HUM%3F%3A12h%20X%3A12h%20Y%3A12h%20Z%3A12h). Channel and temp are easy, don't know how humidity in decoded.

zuckschwerdt, 2020-09-23:
The moisture seems to be in (maybe non-linear) increments, change `HUM?:12h` to `4h HUM:8d` and see if you can table more values.

Tuxyso, 2020-09-23:
I think we have the solution. According to the manual there are 16 data points:

[image]

16 - 99 %
15 - 93 %
13 - 80 %
02 - 7 %

Tuxyso, 2020-09-26:
Rain today 👍 
I have analysed the rain encoding of the outdoor sensor, see [Bitbench](https://triq.net/bitbench?c=3e7005c0988a18aaeaaeaaaa3cf8bbba2daaaaaaaaaaaaaaaa%20%5B1%2C52mm%2Fh%202%2C0mm%20daily%209%2C6mm%20week%5D&c=3fc405c0838a18aaaaaaaaaa3cf8bbba23aaaaaaaaaaaaaaaa%20%5B1%2C52mm%5D&c=b29805c09bfa18aabaabaaaa3cf8bbba22aaaaaaaaaaaaaaaa%20%20%5B0mm%5D&c=75d405c09e2a18aabaabaaaa3cf8bbba23aaaaaaaaaaaaaaaa%20%5B0mm%2FH%5D&c=b45005c098ba18aaaaaaaaaa33acbbba23aaaaaaaaaaaaaaaa%20%5B1%2C5mm%5D&c=ec6305c098ba18aa8aa8aaaa33acbbaa23aaaaaaaaaaaaaaaa%20%5B1%2C52%3B%202%2C29%20acc%5D%20&c=9aac05c08dca18aaaaaaaaaa33acbbaa3aaaaaaaaaaaaaaaaa%20%5B1%2C52mm%2Fh%5D&c=101205c083ca18aa9aa9aaababcaba2a3aaaaaaaaaaaaaaaaa%20%5B1%2C52mm%2Fh%202%2C54%20acc%5D%20&c=10a805c09f8a18aaaaaaaaababcaba2a3baaaaaaaaaaaaaaaa%20%5B1%2C52mm%202%2C5mm%20daily%5D&c=7b3f05c0999a18aabaabaaabaebebaca38aaaaaaaaaaaaaaaa&c=1ba205c083aa18aaaaaaaaabaebebaca38aaaaaaaaaaaaaaaa&c=d13d05c09fea18aa8aa8aaabaebebaca38aaaaaaaaaaaaaaaa&c=60a505c0bf9a18aaaaaaaaabacc2bafa39aaaaaaaaaaaaaaaa%20%5B3mm%2Fh%5D&c=ac9005c098da18aaaaaaaaabacc2baea39aaaaaaaaaaaaaaaa%20%5B1%2C5mm%2Fh%5D&c=830f05c082da18aaaaaaaaaba388baea3eaaaaaaaaaaaaaaaa%20%5B3mm%2Fh%5D&c=665e05c09f2a18abbabbaaaba388baea3eaaaaaaaaaaaaaaaa%20%5B3mm%2Fh%5D&c=74cd05c08cca18aaaaaaaaabbbdcba9a3eaaaaaaaaaaaaaaaa%20%5B3mm%2Fh%5D&c=cf2705c0a8ca18aaaaaaaaabbe9aba9a3eaaaaaaaaaaaaaaaa%20%5B4%2C5mm%2Fh%5D&c=22de05c08bfa18aaaaaaaaabbc2eba8a3faaaaaaaaaaaaaaaa%20%5B4%2C5mm%2Fh%5D%20%20&c=7b8505c08f9a18aa8aa8aaab8daabaaa3faaaaaaaaaaaaaaaa%20%5B3mm%2Fh%5D&f=CHK%3A8h8h%20ID%3F8h8h%20WDIR%3A8h4h%C2%B0%204h%208h%20WGUST%3A8h.4h%20WAVG%3A8h.4h%20RAIN-WEEKLY%3A8h8h4h.4h%20RAIN%3F%3A8h%20TEMP%3A8h.4hC%204h%20HUM%3A8h%25%20LIGHT%3A8h4h%2C4hKL%20%3F%3A8h8h4h%20TRAILER%3A8h8h8h4h&x=a)

I was really surprised: The outdoor sensor only transmits the total amount of **weekly measured rain in mm**. From this metric every other metrics are calculated, also the rain rate in mm per hour. In combination with the timestamp you can calculate the rain rate in mm/h - but due to the kind of measurement this value is very slowly and time-delayed. 
I could verify this value by comparision with the RAIN per week value at my basis.

Example:

[image]

The meaning of one field is still unknows (denoted as RAIN? after RAIN-WEEKLY) in the image above. Any ideas?

zuckschwerdt, 2020-09-28:
This is expected. The rain measure from weather station sensors is always an accumulated value. The display unit keeps time and derives all the "per x" values.
Not sure about that extra field, more digits for the accumulated rain count don't seem plausible.

zuckschwerdt, 2020-09-28:
I've merge this with the 6-in-1 and new 5-in-1 changes, the branch is now in #1225. I'll add the moisture support soon.

zuckschwerdt, 2020-09-28:
Actually the indoor sensor is a 6-in-1. I've updated the code to reflect that. We still need to find out how the fields are flagged as valid/invalid as the sensors seem to come with different options.

Tuxyso, 2020-09-28:
I've tested the new branch _rtl_433 version 20.02-173-g46680b7 branch feat-bresser6in1 at 202009281822_ and can give the following feedback:

I got some checksum failures. I am not sure if it works as designed (due to a weak signal) of if it is an error in the code (see complete debug information in the attached file:

[checksum_errors.txt](https://github.com/merbanan/rtl_433/files/5295125/checksum_errors.txt)

Example:

`MSG: {200} ac 51 19 70 05 fd 29 00 00 00 00 00 22 26 56 00 00 ad ff ff c1 c0 00 00 00 
XOR: {200} 06 fb b3 da af 57 83 aa aa aa aa aa 88 8c fc aa aa 07 55 55 6b 6a aa aa aa 
bresser_7in1_decode: Digest check failed 06fb vs 0df9 (0b02)`

Another point: I think the naming for the indoor sensor ("Bresser-6in1") in the output ist confusing. Example:

`time      : 2020-09-28 23:20:27
model     : Bresser-6in1 id        : 426771965
channel   : 1            Battery   : 0             Temperature: 22.3 C       Humidity  : 56            UV        : 0.0           Flags     : 2
Integrity : CRC
`

We are talking about this sensor:

https://www.bresser.de/en/Weather-Time/BRESSER-Thermo-Hygro-Sensor-7-Channel-868-MHz.html

UV fied seems strange.  I prefer a more neutral / generic naming because the sensor is compatible to 3 different weather stations:

Suitable for the following BRESSER weather stations: 7002540000000, 7002560, 7002580

What about "Bresser-Thermo-Hygro-7CH-Sensor" ?

zuckschwerdt, 2020-09-28:
Thanks for debugging. The digest fails in the (outdoor) 7-in-1 decoder because the packet is a 6-in-1. A 7-in-1 packet has lots of 0xaa and should contain 0x00 after xor, not like here:
```
MSG: {200} 80 c6 19 70 05 fd 29 00 00 00 00 00 22 36 57 00 00 9c 00 00 01 c0 00 00 00 
XOR: {200} 2a 6c b3 da af 57 83 aa aa aa aa aa 88 9c fd aa aa 36 aa aa ab 6a aa aa aa 
bresser_7in1_decode: Digest check failed 2a6c vs 6bd9 (41b5)
```

Basically both are very similar 6-in-1 18 bytes and 7-in-1 21 (maybe 25) bytes, but the trailer (a long lead-out FSK "space") will give us as many bytes as we want. We thus need to try the digest to find out.

About the name: the data is just 18 bytes of `DIGEST:8h8h ID?8h8h8h8h FLAGS:4h BATT:1b CH:3d WSPEED:~8h~4h ~4h~8h WDIR:12h ?4h TEMP8h.4h ?4h HUM8h UV?~12h ?4h CHKSUM:8h` I'm currently not sure if/how we can determine if the sender is an indoor sensor or a weather station. There could be some feature or model indication in the id (3rd to 6th byte, 0x197005FD here, 0x188002c3 in the 6-in-1 example I have, 0x18c006c0 for the weather station reported as "new 5-in-1")? I guess I'll split the id out and hopefully a good number of users can report what we have?

zuckschwerdt, 2020-09-29:
It looks like `flags` is 4 for your moisture sensor and 1 on all other sensors? Otherwise it looks like any other 6-in-1 packet.

Tuxyso, 2020-10-01:
Do you need further debug data? Please let me know.

zuckschwerdt, 2020-10-01:
Is flags always `4` for your soil moisture sensor and `1` for everything else? Possibly be the only field with a type indication.

Tuxyso, 2020-10-02:
Here you find sample data for further analysis (including indoor, outdoor and soil moisture sensor).

[sample-data.txt](https://github.com/merbanan/rtl_433/files/5316435/sample-data.txt)

Tuxyso, 2020-10-03:
Rain is not shown in current Git version:

`
MSG: {200} d5 a7 05 c0 88 2a 18 aa aa aa aa af a2 aa bb da 32 aa 8e ff aa aa aa aa aa
XOR: {200} 7f 0d af 6a 22 80 b2 00 00 00 00 05 08 00 11 70 98 00 24 55 00 00 00 00 00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-03 15:51:40
model     : Bresser-7in1 id        : 44906
Temperature: 11.7 C      Humidity  : 98            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 228           Rain      : 0.0 mm
Light     : 2.4 klx      Integrity : CRC
`

Should be 50,8. Is corretly shown in latest [Bitbench](https://triq.net/bitbench?c=d5%20a7%2005%20c0%2088%202a%2018%20aa%20aa%20aa%20aa%20af%20a2%20aa%20bb%20da%2032%20aa%208e%20ff%20aa%20aa%20aa%20aa%20aa&f=CHK%3A8h8h%20ID%3F8h8h%20WDIR%3A8h4h%C2%B0%204h%208h%20WGUST%3A8h.4h%20WAVG%3A8h.4h%20RAIN-WEEKLY%3A8h8h4h.4h%20RAIN%3F%3A8h%20TEMP%3A8h.4hC%204h%20HUM%3A8h%25%20LIGHT%3A8h4h%2C4hKL%20%3F%3A8h8h4h%20TRAILER%3A8h8h8h4h&x=a&cw=4)

zuckschwerdt, 2020-10-03:
Thanks, fixed.

Tuxyso, 2020-10-03:
Just recognized another strange behaviour: Rain (mm) is not in mqtt output:

`rtl_433/Bresser-7in1outdoor/44906 {"time":"2020-10-03 17:22:53","model":"Bresser-7in1outdoor","id":44906,"temperature_C":11.4,"humidity":97,"wind_max_m_s":0,"wind_avg_m_s":0,"wind_dir_deg":287,"light_klx":4.5,"mic":"CRC"}`

zuckschwerdt, 2020-10-03:
"Bresser-7in1outdoor" is not the current code, it should read "Bresser-7in1".

Tuxyso, 2020-10-03:
I have started rtl_433 (your testing branch) with the following options:
`./rtl_433 -f 868.3M -s 1000k -C si -F "mqtt://localhost:1883,events=rtl_433[/model][/id]"`
and got the output from above. 

`rtl_433/Bresser-7in1outdoor/44906` is the MQTT topic. That seems OK for me. The point is: rain field is missing.

zuckschwerdt, 2020-10-03:
Your copy of the branch seems outdated, perhaps `git fetch ; git reset --hard origin/feat-bresser6in1` when you are on the branch. Otherwise remove the local branch and checkout fresh.

konserninjohtaja, 2020-10-04:
Could this work with Bresser 6in1(https://www.bresser.de/en/Weather-Time/Weather-Center/BRESSER-Professional-WIFI-Weather-Centre-6in1.html) too?

zuckschwerdt, 2020-10-04:
@Tuxyso so `flags` is always `2` for this sensor? Is that the indoor unit, there is no UV on that, right? what about the soil probe?

zuckschwerdt, 2020-10-04:
@konserninjohtaja there is no interface to the Weather Center (display unit), but the sensor for that is the 6-in-1 outdoor and should work, yes.

konserninjohtaja, 2020-10-04:
> @konserninjohtaja there is no interface to the Weather Center (display unit), but the sensor for that is the 6-in-1 outdoor and should work, yes.

Which branch should i try?

zuckschwerdt, 2020-10-04:
https://github.com/merbanan/rtl_433/tree/feat-bresser6in1 from #1225

Tuxyso, 2020-10-04:
> @Tuxyso so `flags` is always `2` for this sensor? Is that the indoor unit, there is no UV on that, right? what about the soil probe?

That's right, FLAG ist always 2, as you can see in the [Bitbench of the sample](https://triq.net/bitbench?c=7771197005fd290000000000213663000091ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=3504197005fd290000000000214663000081ffffc1c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=7771197005fd290000000000213663000091ffffc1c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=b094197005fd290000000000214664000080000007000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=7771197005fd290000000000213663000091ffffc1c000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=7771197005fd290000000000213663000091ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=e796197005fd29000000000221266300009fffff07000000000000000000&c=7771197005fd290000000000213663000091ffffc1c000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=7771197005fd290000000000213663000091ffffc1c000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=403f197005fd2900000000002116630000b1ffff07000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=f2e1197005fd290000000000213664000090000001c000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=f2e1197005fd290000000000213664000090000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=c5af197005fd2900000000002116640000b0000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=7771197005fd290000000000213663000091ffffc1c000000000000000000&c=6cd6197005fd2900000000002126630000a1ffffc1c000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000007000000000000000000&c=e946197005fd2900000000002126640000a0000001c000000000000000000&c=c5af197005fd2900000000002116640000b0000007000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=6cd6197005fd2900000000002126630000a1ffff07000000000000000000&c=e9461 [... cut ...] 7000000000000000000&f=DIGEST%3A8h8h%20ID%3F8h8h8h8h%20FLAGS%3A4h%20BATT%3A1b%20CH%3A3d%208h%208h8h%208h8h%20TEMP%3A12h%204b%20HUM%3A8h%208h8h%20CHK%3A8h%20TRAILER%3F8h8h%208h8h%208h8h%208h8h%208h8h%208h8h4h&cw=4).

Length (the value {XXX} is between 240-242 for the indoor sensor. Probably that helps for identification?

Indoor sensor has no UV.

Indoor sample:

[indoor-sample.txt](https://github.com/merbanan/rtl_433/files/5324181/indoor-sample.txt)

In the next days I will check the soil sensor.

[14 more replies]

closed 2021-03-22 by merbanan