merbanan/rtl_433 #1493: Support OWL CM130 ?

mzealey, 2020-09-13:
Did a couple of [captures](https://github.com/merbanan/rtl_433/files/5213842/out.zip) while hitting the reset button on this - sister of the CM160 I guess? Do you need anything more to help debug this thing?

```
Analyzing pulses...
Total count:  114,  width: 121.10 ms            (30276 S)
Pulse width distribution:
 [ 0] count:  106,  width:  492 us [492;512]    ( 123 S)
 [ 1] count:    7,  width:  984 us [980;992]    ( 246 S)
 [ 2] count:    1,  width: 3992 us [3992;3992]  ( 998 S)
Gap width distribution:
 [ 0] count:  106,  width:  476 us [472;488]    ( 119 S)
 [ 1] count:    7,  width:  968 us [964;972]    ( 242 S)
Pulse period distribution:
 [ 0] count:  100,  width:  972 us [968;988]    ( 243 S)
 [ 1] count:   12,  width: 1464 us [1456;1468]  ( 366 S)
 [ 2] count:    1,  width: 1952 us [1952;1952]  ( 488 S)
Level estimates [high, low]:  16003,    151
RSSI: -0.1 dB SNR: 20.2 dB Noise: -20.3 dB
Frequency offsets [F1, F2]:   -9258,      0     (-35.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 492, long_width: 984, reset_limit: 976, sync_width: 3992
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=492,l=984,r=976,g=0,t=0,y=3992'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 2 
[00] {113} ff ff ff 7f 76 ff ff ff ff ff ff ff ff f5 00 
[01] { 0}       

*** signal_start = 64435, signal_end = 114709, signal_len = 50274, pulses_found = 114
Iteration 1. t: 565    min: 132 (113)    max: 999 (1)    delta 81
Iteration 2. t: 565    min: 132 (113)    max: 999 (1)    delta 0
Pulse coding: Short pulse length 132 - Long pulse length 999

Short distance: 118, long distance: 119, packet distance: 241

p_limit: 565
bitbuffer:: Number of rows: 8 
[00] {24} 00 00 00                   : 00000000 00000000 00000000 
[01] { 2} 00                         : 00
[02] { 8} 00                         : 00000000 
[03] { 5} 00                         : 00000
[04] { 5} 00                         : 00000
[05] {66} 00 00 00 00 00 00 00 00 00 
[06] { 1} 00                         : 0
[07] { 3} 20                         : 001
Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta -727379968
Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
Distance coding: Pulse length 0

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 0
bitbuffer:: Number of rows: 0                                        : 
```

```
[00] {113} ff ff ff 7f 6f df ff ff ff ff ff ff ff ea 80 
```

```
[00] {114} ff ff ff 7f 6f 7f ff ff ff ff ff ff ff fb 40 
```

zuckschwerdt, 2020-09-13:
Looks somewhat like the CM180 or CM160. I might take a look at this in a few days.

zuckschwerdt, 2020-10-18:
@mzealey  Can you check out the `feat-cm130` branch, run for some time and note the codes along with the expected readings?
Should be something like e.g.:
```
CM130(orig) {97} 00 c8 bf 00 00 00 00 00 00 00 00 69 10 : id 1234, power 42W, energy 9kWh
CM130(orig) {97} 00 c4 0f 00 00 00 00 00 00 00 00 4c 10 : id 1234, power 41W, energy 11kWh
CM130(orig) {97} 00 cc fd 00 00 00 00 00 00 00 00 9d 10 : id 1234, power 0W, energy 17kWh
```

mzealey, 2021-01-02:
Apologies for the delayed response. Here is some output compared with rflink (r48):

```
oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 d2 00 21 7f 50 00 00 00 85 10
20;03;OWL_CM130;ID=0085;WATT=02d1;KWATT=0000006d;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 a2 00 e0 10 60 00 00 00 b5 10
20;08;OWL_CM130;ID=0085;WATT=02a0;KWATT=0000006d;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 82 00 6b 31 60 00 00 00 be 10
20;10;OWL_CM130;ID=0085;WATT=028b;KWATT=0000006f;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 a2 00 60 72 60 00 00 00 fd 10
20;1B;OWL_CM130;ID=0085;WATT=027f;KWATT=00000070;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 72 00 2a 03 60 00 00 00 c1 10
20;1C;OWL_CM130;ID=0085;WATT=027a;KWATT=00000071;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 a2 00 8f c4 60 00 00 00 46 10
20;2A;OWL_CM130;ID=0085;WATT=02ad;KWATT=00000074;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 a2 00 0d 65 60 00 00 00 b4 10
20;2C;OWL_CM130;ID=0085;WATT=027f;KWATT=00000074;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 72 00 21 06 60 00 00 00 5c 10
20;2F;OWL_CM130;ID=0085;WATT=0271;KWATT=00000074;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 a2 00 89 3a 60 00 00 00 9c 10
20;4D;OWL_CM130;ID=0085;WATT=09d7;KWATT=00000079;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 79 00 c1 9e 60 00 00 00 57 10
20;55;OWL_CM130;ID=0085;WATT=0971;KWATT=0000007e;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 0b 00 ad c0 70 00 00 00 fb 10
20;5A;OWL_CM130;ID=0085;WATT=09e8;KWATT=00000081;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 e9 00 4c 13 70 00 00 00 7d 10
20;5D;OWL_CM130;ID=0085;WATT=09ec;KWATT=00000083;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 d9 00 c9 65 70 00 00 00 a3 10
20;60;OWL_CM130;ID=0085;WATT=09d9;KWATT=00000086;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 59 00 ae a7 70 00 00 00 c8 10
20;65;OWL_CM130;ID=0085;WATT=095e;KWATT=00000088;

oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 a2 00 a3 d9 70 00 00 00 51 10 
20;6D;OWL_CM130;ID=0085;WATT=02a3;KWATT=0000008b;
```

mzealey, 2021-01-02:
(so i make that:

```
                "id",         "House Code", DATA_INT, (msg[2]&0x0F << 4) | msg[2] >> 4,
```

but can't figure out how it obtains the watt/kwatt calcs from the raw data...

peteakalad, 2021-01-02:
Raw data is likely to be current...

merbanan, 2021-03-16:
oregon_scientific_v3_decode: CM130(orig) {97} 00 83 58 [a2] 00 a(3) d9 70 00 00 00 51 10 
20;6D;OWL_CM130;ID=0085;WATT=02a3;KWATT=0000008b;

[a2] (3) -> 2a3 =>
WATT=02a3 

feat-cm130 updated with this info

merbanan, 2021-03-16:
But I think we need more messages with matching readings to create a complete decoder.

synx508, 2021-04-17:
If it's any help, I'm using the CM130 branch code at the moment, the version that gives power1_W. The value being returned is 16 times higher than the measured power in Watts, or 16000 times higher than the kW value shown on the Owl display unit.

merbanan, 2021-04-17:
Can you record transmissions with matching readings? Then we can figure out the encoding of values.

synx508, 2021-04-17:
I may be able to do that but I've a few obstacles to overcome. What is the best format/method of doing this, please? At the moment the readings are very close. When displaying 0.25kW on the OWL I log 0.26kW with rtl_433 (after dividing power1_W by 16,000). At 3kW they are similarly close. I would like to give it a few days to see if any unexpected numbers appear, I have already seen a single reading that was much higher than expected, as there's no checksum currently this could be just corrupted data because of interference.

merbanan, 2021-04-17:
Well the code is written based on 1 sample. We really need more samples and readings to verify the code and to figure out the checksum. After we have done that we can merge the code in the main branch.

merbanan, 2021-04-17:
And we are missing energy. For that we need readings that correspond to values on the meter.

synx508, 2021-04-17:
Here's the output of stdout and stderr mashed up with a csv of "Energy (kW)" and "Total (kWh)" below. Created by taking a webcam snapshot of the display after rtl_433 decoded a CM130 message. There may be a problem that the OWL display doesn't update quickly enough. There was about a second delay after the message was received as a result of my ancient laptop taking a while to start ffmpeg each time. A few of the times when I'd expect the highest powers seem to be reading very low powers so possibly a byte/nibble missing somewhere? 
Hope this is of some use. I've since found ssocr, so may have another go with fully automating this if I can get a better webcam… I will add a few seconds delay before taking the photo if I try again.

[webcammed-logs.txt](https://github.com/merbanan/rtl_433/files/6329518/webcammed-logs.txt)

gdt, 2023-09-29:
Status and plans?

gdt, 2025-07-04:
Does anybody want to work on this, or should we close the issue?