merbanan/rtl_433 #1494: How to know what protocol is being used by the garage door opener?

fluppie, 2020-09-14:
When I run RTL_433 regular, it doesn't give any feedback when tapping the remote.
Later on I ran RTL_433 in analyze mode and got this:

```
FIRST PRESS 

[01] {749} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
*** signal_start = 17867554, signal_end = 18136222, signal_len = 268668, pulses_found = 752
Iteration 1. t: 3328    min: 132 (751)    max: 6525 (1)    delta 15876
Iteration 2. t: 3328    min: 132 (751)    max: 6525 (1)    delta 0
Pulse coding: Short pulse length 132 - Long pulse length 6525

Short distance: 159, long distance: 4124, packet distance: 11909

p_limit: 3328
bitbuffer:: Number of rows: 2
[00] { 1} 80                                                                                                                                                                                                                                                                                        : 1

SECOND PRESS:
[01] {751} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
*** signal_start = 20466475, signal_end = 20735083, signal_len = 268608, pulses_found = 751
Iteration 1. t: 3331    min: 139 (750)    max: 6523 (1)    delta 18769
Iteration 2. t: 3331    min: 139 (750)    max: 6523 (1)    delta 0
Pulse coding: Short pulse length 139 - Long pulse length 6523

Short distance: 151, long distance: 4118, packet distance: 11900

p_limit: 3331
bitbuffer:: Number of rows: 2
[00] { 1} 80                                                                                                                                                                                                                                                                                        : 1

THIRD PRESS:
[01] {750} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Any idea what protocol is being used by the garage door opener?

zuckschwerdt, 2020-09-14:
Use `-A`, not `-a`.

fluppie, 2020-09-14:
Aha. That's more information 👍 
```
Detected OOK package    2020-09-14 17:54:35
Analyzing pulses...
Total count:  748,  width: 994.94 ms            (248735 S)
Pulse width distribution:
 [ 0] count:    1,  width: 26144 us [26144;26144]       (6536 S)
 [ 1] count:  435,  width:  412 us [400;420]    ( 103 S)
 [ 2] count:  312,  width:  812 us [804;828]    ( 203 S)
Gap width distribution:
 [ 0] count:    1,  width: 47596 us [47596;47596]       (11899 S)
 [ 1] count:  407,  width:  392 us [380;408]    (  98 S)
 [ 2] count:    4,  width: 4020 us [4016;4028]  (1005 S)
 [ 3] count:  332,  width:  788 us [780;800]    ( 197 S)
 [ 4] count:    3,  width: 16448 us [16444;16460]       (4112 S)
Pulse period distribution:
 [ 0] count:    1,  width: 73740 us [73740;73740]       (18435 S)
 [ 1] count:   95,  width:  800 us [796;812]    ( 200 S)
 [ 2] count:    4,  width: 4424 us [4420;4428]  (1106 S)
 [ 3] count:  644,  width: 1204 us [1196;1212]  ( 301 S)
 [ 4] count:    3,  width: 16864 us [16864;16872]       (4216 S)
Level estimates [high, low]:  15982,    103
RSSI: -0.1 dB SNR: 21.9 dB Noise: -22.0 dB
Frequency offsets [F1, F2]:    -253,      0     (-1.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 412, long_width: 812, reset_limit: 47600, sync_width: 26144
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=412,l=812,r=47600,g=0,t=0,y=26144'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {747} ff ff ff ff ff ff ff ff 97 c4 cb 8a 77 33 39 46 f5 a9 61 02 db f4 71 33 aa ae 37 09 ff fe 5f 13 2e 29 dc cc e5 1b d6 a5 84 0b 6f d1 c4 ce aa b8 dc 27 ff f9 7c 4c b8 a7 73 33 94 6f 5a 96 10 2d bf 47 13 3a aa e3 70 9f ff e5 f1 32 e2 9d cc ce 51 bd 6a 58 40 b6 fd 1c 4c ea ab 8d c2 60

Detected OOK package    2020-09-14 17:54:38
Analyzing pulses...
Total count:    1,  width: 0.04 ms              (   11 S)
Pulse width distribution:
 [ 0] count:    1,  width:   44 us [44;44]      (  11 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1000,     99
RSSI: -12.1 dB SNR: 10.0 dB Noise: -22.2 dB
Frequency offsets [F1, F2]:     -16,      0     (-0.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package    2020-09-14 17:54:40
Analyzing pulses...
Total count:  748,  width: 994.97 ms            (248743 S)
Pulse width distribution:
 [ 0] count:    1,  width: 26172 us [26172;26172]       (6543 S)
 [ 1] count:  435,  width:  412 us [396;420]    ( 103 S)
 [ 2] count:  312,  width:  812 us [804;828]    ( 203 S)
Gap width distribution:
 [ 0] count:    1,  width: 47596 us [47596;47596]       (11899 S)
 [ 1] count:  407,  width:  392 us [380;412]    (  98 S)
 [ 2] count:    4,  width: 4020 us [4020;4024]  (1005 S)
 [ 3] count:  332,  width:  788 us [780;804]    ( 197 S)
 [ 4] count:    3,  width: 16456 us [16448;16460]       (4114 S)
Pulse period distribution:
 [ 0] count:    1,  width: 73768 us [73768;73768]       (18442 S)
 [ 1] count:   95,  width:  800 us [796;816]    ( 200 S)
 [ 2] count:    4,  width: 4420 us [4416;4428]  (1105 S)
 [ 3] count:  644,  width: 1204 us [1196;1216]  ( 301 S)
 [ 4] count:    3,  width: 16868 us [16864;16876]       (4217 S)
Level estimates [high, low]:  15942,    147
RSSI: -0.1 dB SNR: 20.4 dB Noise: -20.5 dB
Frequency offsets [F1, F2]:   -1095,      0     (-4.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 412, long_width: 812, reset_limit: 47600, sync_width: 26172
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=412,l=812,r=47600,g=0,t=0,y=26172'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {747} ff ff ff ff ff ff ff ff 4b bb 1e 12 d7 f4 67 f0 a9 46 21 4f da 27 8d 23 aa ae 37 09 ff fd 2e ec 78 4b 5f d1 9f c2 a5 18 85 3f 68 9e 34 8e aa b8 dc 27 ff f4 bb b1 e1 2d 7f 46 7f 0a 94 62 14 fd a2 78 d2 3a aa e3 70 9f ff d2 ee c7 84 b5 fd 19 fc 2a 51 88 53 f6 89 e3 48 ea ab 8d c2 60

Detected OOK package    2020-09-14 17:54:45
Analyzing pulses...
Total count:  748,  width: 995.04 ms            (248760 S)
Pulse width distribution:
 [ 0] count:    1,  width: 26140 us [26140;26140]       (6535 S)
 [ 1] count:  375,  width:  412 us [400;424]    ( 103 S)
 [ 2] count:  372,  width:  812 us [800;828]    ( 203 S)
Gap width distribution:
 [ 0] count:    1,  width: 47592 us [47592;47592]       (11898 S)
 [ 1] count:  467,  width:  392 us [380;408]    (  98 S)
 [ 2] count:    4,  width: 4020 us [4020;4024]  (1005 S)
 [ 3] count:  272,  width:  788 us [780;804]    ( 197 S)
 [ 4] count:    3,  width: 16460 us [16460;16464]       (4115 S)
Pulse period distribution:
 [ 0] count:    1,  width: 73732 us [73732;73732]       (18433 S)
 [ 1] count:   95,  width:  800 us [796;812]    ( 200 S)
 [ 2] count:    4,  width: 4424 us [4420;4432]  (1106 S)
 [ 3] count:  644,  width: 1204 us [1196;1216]  ( 301 S)
 [ 4] count:    3,  width: 16872 us [16872;16876]       (4218 S)
Level estimates [high, low]:  15919,    151
RSSI: -0.1 dB SNR: 20.2 dB Noise: -20.4 dB
Frequency offsets [F1, F2]:     187,      0     (+0.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 412, long_width: 812, reset_limit: 47596, sync_width: 26140
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=412,l=812,r=47596,g=0,t=0,y=26140'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {747} ff ff ff ff ff ff ff ff 70 29 04 d4 99 eb 20 e8 d8 4b 1b 22 23 30 51 89 aa ae 37 09 ff fd c0 a4 13 52 67 ac 83 a3 61 2c 6c 88 8c c1 46 26 aa b8 dc 27 ff f7 02 90 4d 49 9e b2 0e 8d 84 b1 b2 22 33 05 18 9a aa e3 70 9f ff dc 0a 41 35 26 7a c8 3a 36 12 c6 c8 88 cc 14 62 6a ab 8d c2 60
```

zuckschwerdt, 2020-09-14:
The guess for a OOK flex decoder with 400/800 µs doesn't look right, the pulse/gap counts don't match.
I guess this is a kind of Manchester coding with 400 µs half-bit width.
And seeing the longer gaps it looks like 4 repeats of 4 distinct packets. (It would be too much data for a keyfob if that was a single packet anyway). Also the long warmup of 48 ms is excessive ;)

Try a flex decoder with `-X 'n=name,m=OOK_MC_ZEROBIT,s=400,l=400,g=1000,r=49000'`
You should be able to receive the proper data with that.

Otherwise capture a sample with `-S unknown` then convert to pulse data with `-w gfile.ook gfile.cu8` check in https://triq.org/pdv/ and upload a zip of that data here.

fluppie, 2020-09-14:
Hmm, I feel like I mis some .conf file(s)

[](C:\ProgramData\rtl_433>rtl_433_64bit_static -X 'n=name,m=OOK_MC_ZEROBIT,s=400,l=400,g=1000,r=49000
rtl_433 version 20.02-124-g9c11752 branch  at 202008052201 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\Philippe\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Bad flex spec, unknown keyword ('n)!
Use -X <spec> to add a general purpose decoder. For usage use -X help

C:\ProgramData\rtl_433>rtl_433 -X 'n=name,m=OOK_MC_ZEROBIT,s=400,l=400,g=1000,r=49000
rtl_433 version 20.02-124-g9c11752 branch  at 202008052201 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Trying conf file at "C:\Users\Philippe\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Bad flex spec, unknown keyword ('n)!
Use -X <spec> to add a general purpose decoder. For usage use -X help)

merbanan, 2020-09-14:
Remove one '.

fluppie, 2020-09-14:
OK, it's generating data. But I think it's too long? The same button is used for all commands (open-stop-close-open-stop-close and so on). So it's probably just sending a trigger signal?

```
C:\Users\Philippe\Downloads\rtl_433-20.02-124-g9c11752-win-x64>rtl_433 -X n=name,m=OOK_MC_ZEROBIT,s=400,l=400,g=1000,r=49000
rtl_433 version 20.02-124-g9c11752 branch  at 202008052201 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\Philippe\Downloads\rtl_433-20.02-124-g9c11752-win-x64\rtl_433.conf"...
Trying conf file at "C:\Users\Philippe\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 135 out of 163 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-14 20:09:41
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 921          data      : 400000000000000033fbbb7b9cc1b0dfb1d8f9c79dbb8ceefb1d9e38e6db6c798f780019fdddbdce60d86fd8ec7ce3ceddc6777d8ecf1c736db63cc7bc000cfeeedee7306c37ec763e71e76ee33bbec7678e39b6db1e63de00067f776f7398361bf63b1f38f3b7719ddf63b3c71cdb6d8f31ef0
codes     : {921}400000000000000033fbbb7b9cc1b0dfb1d8f9c79dbb8ceefb1d9e38e6db6c798f780019fdddbdce60d86fd8ec7ce3ceddc6777d8ecf1c736db63cc7bc000cfeeedee7306c37ec763e71e76ee33bbec7678e39b6db1e63de00067f776f7398361bf63b1f38f3b7719ddf63b3c71cdb6d8f31ef0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-14 20:09:49
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 913          data      : 40000000000000003d8037fbbf776fcf3beee70cf763c18ecddde7671b6db1e63de0007b006ff77eeedf9e77ddce19eec7831d9bbbcece36db63cc7bc000f600dfeefdddbf3cefbb9c33dd8f063b37779d9c6db6c798f78001ec01bfddfbbb7e79df773867bb1e0c766eef3b38db6d8f31ef0
codes     : {913}40000000000000003d8037fbbf776fcf3beee70cf763c18ecddde7671b6db1e63de0007b006ff77eeedf9e77ddce19eec7831d9bbbcece36db63cc7bc000f600dfeefdddbf3cefbb9c33dd8f063b37779d9c6db6c798f78001ec01bfddfbbb7e79df773867bb1e0c766eef3b38db6d8f31ef0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-14 20:09:54
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 921          data      : 40000000000000003c76f8306c6dec76fbf7f8f36ec361fbe6766db0c6db6c798f78001e3b7c183636f63b7dfbfc79b761b0fdf33b36d8636db63cc7bc000f1dbe0c1b1b7b1dbefdfe3cdbb0d87ef99d9b6c31b6db1e63de00078edf060d8dbd8edf7eff1e6dd86c3f7ccecdb618db6d8f31ef0
codes     : {921}40000000000000003c76f8306c6dec76fbf7f8f36ec361fbe6766db0c6db6c798f78001e3b7c183636f63b7dfbfc79b761b0fdf33b36d8636db63cc7bc000f1dbe0c1b1b7b1dbefdfe3cdbb0d87ef99d9b6c31b6db1e63de00078edf060d8dbd8edf7eff1e6dd86c3f7ccecdb618db6d8f31ef0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-14 20:09:59
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 925          data      : 40000000000000003b1bd8e7e6671e6cf0f1bccf9de679b67b7ef30db36db63cc7bc000ec6f639f999c79b3c3c6f33e7799e6d9edfbcc36cdb6d8f31ef0003b1bd8e7e6671e6cf0f1bccf9de679b67b7ef30db36db63cc7bc000ec6f639f999c79b3c3c6f33e7799e6d9edfbcc36cdb6d8f31ef0
codes     : {925}40000000000000003b1bd8e7e6671e6cf0f1bccf9de679b67b7ef30db36db63cc7bc000ec6f639f999c79b3c3c6f33e7799e6d9edfbcc36cdb6d8f31ef0003b1bd8e7e6671e6cf0f1bccf9de679b67b7ef30db36db63cc7bc000ec6f639f999c79b3c3c6f33e7799e6d9edfbcc36cdb6d8f31ef0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-14 20:10:03
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 929          data      : 40000000000000000319b66dcef30cff1b3bbb737b863ec0e371ccdbd9b6db1e63de00006336cdb9de619fe367776e6f70c7d81c6e399b7b36db63cc7bc0000c66d9b73bcc33fc6ceeedcdee18fb038dc7336f66db6c798f7800018cdb36e779867f8d9dddb9bdc31f6071b8e66decdb6d8f31ef0
codes     : {929}40000000000000000319b66dcef30cff1b3bbb737b863ec0e371ccdbd9b6db1e63de00006336cdb9de619fe367776e6f70c7d81c6e399b7b36db63cc7bc0000c66d9b73bcc33fc6ceeedcdee18fb038dc7336f66db6c798f7800018cdb36e779867f8d9dddb9bdc31f6071b8e66decdb6d8f31ef0

fluppie, 2020-09-14:
And here some captured samples:
[GarageKeyfob.zip](https://github.com/merbanan/rtl_433/files/5220228/GarageKeyfob.zip)

zuckschwerdt, 2020-09-14:
Drop the .cu8 on https://triq.org/iqs/ -- very good samples, a warmup then 4x some preamble and data.
The data is really that long and looks good. A very big string of encrypted data…

The OOK in your zip was empty. Use this ook with https://triq.org/pdv/ to see your sample in details ;)

[GarageKeyfob-ook.zip](https://github.com/merbanan/rtl_433/files/5220503/GarageKeyfob-ook.zip)

closed 2020-09-22 by zuckschwerdt