merbanan/rtl_433 #1497: Understanding the internal operation

matthiasseemoo, 2020-09-17:
To get started developing new device decoders, I would find it helpful to get an overview on how rtl_433 is internally processing signals, without having to reverse engineer the C code. Is there something like a block diagram showing, what rtl_433 does internally?

That might help to answer some of my questions:
- Do you tune the SDR directly to the selected frequency or do you tune it next to that frequency and then digitally down convert the interesting signal into a new downsampled baseband signal?
- How do you cope with multiple transmissions in parallel? Can you select and differentiate them? At least as long as they are not overlapping in the frequency band?
- How is the sync_width used during decoding?
- ...

Additionally, for getting started, it would be nice to have an example plot of a signal indicating all widths that can be set.

zuckschwerdt, 2020-09-17:
An intro for contributors would be great. A long list of questions might be a good start.

- No fancy signal processing, we just look at the (summary) amplitude or magnitude and instant phase angle over the whole bandwidth.
- `sync_width` is just a third length to consider with `short` and `long`, it's valid but discarded (and counted).
- the valid parameters for the slicers are given in https://github.com/merbanan/rtl_433/blob/master/include/pulse_demod.h

sp372, 2022-11-30:
Accordingly, it would save us many hours of searching for information and understanding.

gdt, 2023-09-25:
This isn't really a a valid issue as phrased.

gdt, 2023-09-29:
There are a bunch of docs in docs/ in the sources.   Please read all of them and then feel free to open a discussion with further questions; ideally that would result in improving the docs rather than answering in an issue or discussion that will be lost eventually.  I'm not claiming the docs are good enough, but am trying to keep issues focued on things that are actionable for improving the contents of the repository.

closed 2023-09-29 by gdt