merbanan/rtl_433 #1499: Support or help for ws 6 in 1 Ventus w835 station

RebeckaJohansson, 2020-09-22:
I trying to decode my weather station ventus w835 and the station is an 6 in 1 consol. 

My ventus have RF on 868mhz.
Information about the ventus can you see on this link 

[https://www.reichelt.com/de/en/weather-station-ventus-w835-p263486.html?r=1](https://www.reichelt.com/de/en/weather-station-ventus-w835-p263486.html?r=1) 

I trying to decode the station via sdr-rtl v3 usb dongel and its installed on my Raspberry pi 4. 

I have checked some signals via gqrx and got pulses from 868 between frequency 868.7Mhz and 868.33Mhz.

I dont know if i need something more.

merbanan, 2020-09-22:
You can start by running the analyzer rtl_433 -f 868.5M -A, then we can see if any signals are picked up. Post the output here.

RebeckaJohansson, 2020-09-22:
Sitting on my phone right now i got some signals every 12 seconds and i hope this can give a clue from my copy paste outpost

 
Detected FSK package    2020-09-22 10:03:33
Analyzing pulses...
Total count:   51,  width: 25.15 ms             (25146 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:    2,  width:    3 us [3;3]        (   3 S)
 [ 2] count:    2,  width:    4 us [4;4]        (   4 S)
 [ 3] count:    2,  width:    5 us [5;5]        (   5 S)
 [ 4] count:   31,  width:  121 us [114;129]    ( 121 S)
 [ 5] count:    7,  width:  244 us [242;245]    ( 244 S)
 [ 6] count:    2,  width:  362 us [357;367]    ( 362 S)
 [ 7] count:    1,  width: 2926 us [2926;2926]  (2926 S)
 [ 8] count:    1,  width: 3903 us [3903;3903]  (3903 S)
 [ 9] count:    1,  width:  489 us [489;489]    ( 489 S)
 [10] count:    1,  width:   33 us [33;33]      (  33 S)
Gap width distribution:
 [ 0] count:    2,  width:    2 us [2;2]        (   2 S)
 [ 1] count:    5,  width:    6 us [6;8]        (   6 S)
 [ 2] count:   26,  width:  122 us [120;129]    ( 122 S)
 [ 3] count:    6,  width:  365 us [359;374]    ( 365 S)
 [ 4] count:    5,  width:  247 us [244;254]    ( 247 S)
 [ 5] count:    5,  width:  884 us [782;976]    ( 884 S)
 [ 6] count:    1,  width:  488 us [488;488]    ( 488 S)
Pulse period distribution:
 [ 0] count:    1,  width:    2 us [2;2]        (   2 S)
 [ 1] count:    5,  width:   10 us [10;11]      (  10 S)
 [ 2] count:   22,  width:  243 us [237;251]    ( 243 S)
 [ 3] count:    8,  width:  488 us [478;495]    ( 488 S)
 [ 4] count:    4,  width:  368 us [366;375]    ( 368 S)
 [ 5] count:    3,  width: 1072 us [1024;1097]  (1072 S)
 [ 6] count:    4,  width:  642 us [609;733]    ( 642 S)
 [ 7] count:    1,  width: 3771 us [3771;3771]  (3771 S)
 [ 8] count:    1,  width: 4748 us [4748;4748]  (4748 S)
 [ 9] count:    1,  width:   35 us [35;35]      (  35 S)
Level estimates [high, low]:  15897,     10
RSSI: -0.1 dB SNR: 32.0 dB Noise: -32.1 dB
Frequency offsets [F1, F2]:  -17970,  -9638     (-274.2 kHz, -147.1 kHz)
Guessing modulation: No clue...

7Detected FSK package   2020-09-22 10:03:45
Analyzing pulses...
Total count:   58,  width: 25.15 ms             (25152 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:    4,  width:    5 us [5;5]        (   5 S)
 [ 2] count:    2,  width:    4 us [4;4]        (   4 S)
 [ 3] count:   36,  width:  122 us [115;129]    ( 122 S)
 [ 4] count:    7,  width:  243 us [234;250]    ( 243 S)
 [ 5] count:    1,  width:  367 us [367;367]    ( 367 S)
 [ 6] count:    2,  width:  822 us [780;864]    ( 822 S)
 [ 7] count:    1,  width: 2926 us [2926;2926]  (2926 S)
 [ 8] count:    1,  width: 1455 us [1455;1455]  (1455 S)
 [ 9] count:    1,  width:  487 us [487;487]    ( 487 S)
 [10] count:    1,  width:   30 us [30;30]      (  30 S)
 [11] count:    1,  width:    3 us [3;3]        (   3 S)
Gap width distribution:
 [ 0] count:    2,  width:    2 us [2;2]        (   2 S)
 [ 1] count:    1,  width:    8 us [8;8]        (   8 S)
 [ 2] count:    4,  width:    5 us [5;6]        (   5 S)
 [ 3] count:   28,  width:  122 us [115;129]    ( 122 S)
 [ 4] count:    7,  width:  363 us [356;374]    ( 363 S)
 [ 5] count:    7,  width:  244 us [236;254]    ( 244 S)
 [ 6] count:    4,  width:  821 us [730;977]    ( 821 S)
 [ 7] count:    1,  width:  487 us [487;487]    ( 487 S)
 [ 8] count:    1,  width:  609 us [609;609]    ( 609 S)
 [ 9] count:    1,  width:    3 us [3;3]        (   3 S)
 [10] count:    1,  width:    1 us [1;1]        (   1 S)
Pulse period distribution:
 [ 0] count:    1,  width:    2 us [2;2]        (   2 S)
 [ 1] count:    5,  width:   10 us [9;13]       (  10 S)
 [ 2] count:   23,  width:  244 us [237;252]    ( 244 S)
 [ 3] count:    7,  width:  486 us [478;489]    ( 486 S)
 [ 4] count:    8,  width:  367 us [365;375]    ( 367 S)
 [ 5] count:    2,  width: 1159 us [1098;1220]  (1159 S)
 [ 6] count:    6,  width:  659 us [609;783]    ( 659 S)
 [ 7] count:    1,  width: 3771 us [3771;3771]  (3771 S)
 [ 8] count:    1,  width:  854 us [854;854]    ( 854 S)
 [ 9] count:    1,  width: 2187 us [2187;2187]  (2187 S)
 [10] count:    1,  width:   31 us [31;31]      (  31 S)
 [11] count:    1,  width:    5 us [5;5]        (   5 S)
Level estimates [high, low]:  15877,      5
RSSI: -0.1 dB SNR: 35.0 dB Noise: -35.2 dB
Frequency offsets [F1, F2]:  -16920,  -9564     (-258.2 kHz, -145.9 kHz)
Guessing modulation: No clue...

RebeckaJohansson, 2020-09-22:
Got some more here

Detected OOK package    2020-09-22 10:16:37
Analyzing pulses...
Total count:    1,  width: 0.02 ms              (   18 S)
Pulse width distribution:
 [ 0] count:    1,  width:   18 us [18;18]      (  18 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1000,     12
RSSI: -12.1 dB SNR: 19.2 dB Noise: -31.4 dB
Frequency offsets [F1, F2]:    -925,      0     (-14.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

n=name,m=OOK_PWM,s=22,l=30,r=1514,g=0,t=3,y=Detected OOK package        2020-09-22 10:16:40
Analyzing pulses...
Total count:    2,  width: 2.70 ms              ( 2697 S)
Pulse width distribution:
 [ 0] count:    1,  width:   13 us [13;13]      (  13 S)
 [ 1] count:    1,  width:   21 us [21;21]      (  21 S)
Gap width distribution:
 [ 0] count:    1,  width: 2663 us [2663;2663]  (2663 S)
Pulse period distribution:
 [ 0] count:    1,  width: 2676 us [2676;2676]  (2676 S)
Level estimates [high, low]:   1000,     23
RSSI: -12.1 dB SNR: 16.4 dB Noise: -28.5 dB
Frequency offsets [F1, F2]:   -1494,      0     (-22.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_width: 13, long_width: 21, reset_limit: 2664, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=13,l=21,r=2664,g=0,t=3,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] { 2} 80        : 10

Detected OOK package    2020-09-22 10:16:45
Analyzing pulses...
Total count:    1,  width: 0.02 ms              (   21 S)
Pulse width distribution:
 [ 0] count:    1,  width:   21 us [21;21]      (  21 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1010,     10
RSSI: -12.1 dB SNR: 20.0 dB Noise: -32.1 dB
Frequency offsets [F1, F2]:    1189,      0     (+18.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package    2020-09-22 10:16:45
Analyzing pulses...
Total count:    1,  width: 0.01 ms              (   15 S)
Pulse width distribution:
 [ 0] count:    1,  width:   15 us [15;15]      (  15 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1029,      8
RSSI: -12.0 dB SNR: 21.1 dB Noise: -33.1 dB
Frequency offsets [F1, F2]:    1016,      0     (+15.5 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package    2020-09-22 10:16:45
Analyzing pulses...
Total count:    1,  width: 0.01 ms              (   10 S)
Pulse width distribution:
 [ 0] count:    1,  width:   10 us [10;10]      (  10 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1000,     32
RSSI: -12.1 dB SNR: 14.9 dB Noise: -27.1 dB
Frequency offsets [F1, F2]:    1418,      0     (+21.6 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

[... 38 lines cut ...]
 [11] count:    1,  width: 4750 us [4750;4750]  (4750 S)
Level estimates [high, low]:  15965,      8
RSSI: -0.1 dB SNR: 33.0 dB Noise: -33.1 dB
Frequency offsets [F1, F2]:  -17898, -10142     (-273.1 kHz, -154.8 kHz)
Guessing modulation: No clue...

Detected OOK package    2020-09-22 10:16:48
Analyzing pulses...
Total count:    2,  width: 0.09 ms              (   93 S)
Pulse width distribution:
 [ 0] count:    1,  width:   29 us [29;29]      (  29 S)
 [ 1] count:    1,  width:   40 us [40;40]      (  40 S)
Gap width distribution:
 [ 0] count:    1,  width:   24 us [24;24]      (  24 S)
Pulse period distribution:
 [ 0] count:    1,  width:   53 us [53;53]      (  53 S)
Level estimates [high, low]:   1395,      9
RSSI: -10.7 dB SNR: 21.9 dB Noise: -32.6 dB
Frequency offsets [F1, F2]:    -378,      0     (-5.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_width: 29, long_width: 40, reset_limit: 25, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=29,l=40,r=25,g=0,t=4,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] { 2} 80        : 10

zuckschwerdt, 2020-09-22:
`rtl_433 -f 868.5M -X 'n=ventus,m=FSK_PCM,s=120,l=120,r=3000'` could work to get the data, but there seems to be noise (the very short pulses).
Otherwise use `rtl_433 -f 868.5M -S unknown` and drop the resulting files on https://triq.org/iqs/ to check if the spectrum is clean (just "two parallel lines" from your one sensor).
And use `rtl_433 -s1M -w GFILE.ook GIFLE.cu8` on the files and drop on https://triq.org/pdv/ to check the encoding.

RebeckaJohansson, 2020-09-22:
This is what i got so far from your description
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.500MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:23:45
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 205          data      : aaaaaaaaaa2dd4ba1c1910084d18ffffff0188129480fff0ce00
codes     : {205}aaaaaaaaaa2dd4ba1c1910084d18ffffff0188129480fff0ce00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:23:57
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 205          data      : aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600
codes     : {205}aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600
^[[6~_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
 _ _
time      : 2020-09-22 10:24:09
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 205          data      : aaaaaaaaaa2dd4ba1c1910084d18ffffff0188129480fff0ce00
codes     : {205}aaaaaaaaaa2dd4ba1c1910084d18ffffff0188129480fff0ce00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:24:21
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 205          data      : aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600
codes     : {205}aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:24:33
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 203          data      : aaaaaaaaaa2dd4ba1c1910084d18ffffff0188129480fff0ce0
codes     : {203}aaaaaaaaaa2dd4ba1c1910084d18ffffff0188129480fff0ce0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:24:45
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 205          data      : aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600
codes     : {205}aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:25:09
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 204          data      : 55555555545ba9d9403220109a31fffffe0311fffffffe03cc0
codes     : {204}55555555545ba9d9403220109a31fffffe0311fffffffe03cc0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:25:21
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 205          data      : aaaaaaaaaa2dd4d6d21910084d18ffffff0188130480fff05df8
codes     : {205}aaaaaaaaaa2dd4d6d21910084d18ffffff0188130480fff05df8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 10:25:33
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 204          data      : 55555555545ba9d9403220109a31fffffe0311fffffffe03cc0
codes     : {204}55555555545ba9d9403220109a31fffffe0311fffffffe03cc0

merbanan, 2020-09-22:
`codes : {205}aaaaaaaaaa2dd4eca01910084d18ffffff0188ffffffff01e600`

Here we can observe the very common 2dd4 sync word. If you add preamble=aaaa2dd4 to the command line you should get the aligned payload.

`rtl_433 -f 868.5M -X 'n=ventus,m=FSK_PCM,s=120,l=120,r=3000,preamble=aaaa2dd4'`

After this you observe the output depending on changed environment parameters. Start with temperature and try to keep all the other parameters the same. Record the payload output and a corresponding parameter reading from the receiver.

You can collect the data in bitbench to ease the identification of the parameter positions.

https://triq.net/bitbench

zuckschwerdt, 2020-09-22:
Looks like the newer Bresser-type data, 16-bit checksum in front and data could be BCD.
Was it by chance 12.9 °C?

RebeckaJohansson, 2020-09-22:
Im kinda new on this with decode im still trying understand the hex code haha. I got this output but still think how i got the temp, hum and more out of the code

 time      : 2020-09-22 12:32:22
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 149          data      : 32731910084d18ffffff0348ffffffff012400
codes     : {149}32731910084d18ffffff0348ffffffff012400
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 12:32:34
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 147          data      : 07d21910084d18ffffff0348182471fff085e
codes     : {147}07d21910084d18ffffff0348182471fff085e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 12:32:46
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 149          data      : 32731910084d18ffffff0348ffffffff012400
codes     : {149}32731910084d18ffffff0348ffffffff012400
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 12:32:58
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 150          data      : 07d21910084d18ffffff0348182471fff085fc
codes     : {150}07d21910084d18ffffff0348182471fff085fc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-09-22 12:33:10
model     : ventus       count     : 1             num_rows  : 1             rows      :
len       : 149          data      : 1dff1910084d18ffffff0378ffffffff01f400
codes     : {149}1dff1910084d18ffffff0378ffffffff01f400

And about bresser, yes its typical same as bresser 6in1 what i have been told. Ventus and Bresser 6in1 is near identical.

RebeckaJohansson, 2020-09-22:
> Looks like the newer Bresser-type data, 16-bit checksum in front and data could be BCD.
> Was it by chance 12.9 °C? (80 %H seems a lot but would be consistent)

The consol gives this via station screen
Out Temperature	18,3 °C
Dew Point 12,6 °C
Out Humidity	 70%

zuckschwerdt, 2020-09-22:
Humidity is usually the easiest to spot in the data. For 70%H it should be "70" (BCD) or "46" (hex).
Note that only every second code here has the data needed. This shows the fields:
`07d21910084d18ffffff0348` `182` `4` `71` `fff085e` : 18.2°C and 71%H.
It seems to work with the `feat-bresser6in1` branch. I guess this actually is a rebranded Bresser-6in1. We still need to figure out exactly what's in the data. Help out in #1214 if you can, we need to find the Wind speed and Rain fields.

closed 2020-09-22 by zuckschwerdt

RebeckaJohansson, 2020-09-27:
I think this mabey can help out from Bob Atchley Ws6in1

RebeckaJohansson, 2020-09-28:
> Humidity is usually the easiest to spot in the data. For 70%H it should be "70" (BCD) or "46" (hex).
> Note that only every second code here has the data needed. This shows the fields: [...]

I have find wind speed and wind gust in my code and also dubblecheck it when i moved the wind pointer so here its the bitbench code for it.