merbanan/rtl_433 #1500: High pressure TPMS sensors

mr-pl3r, 2020-09-24:
Hi @zuckschwerdt ,
I am working with these sensors -link below-, you helped me before with the Jansite sensors version TY-568-US which I was able to get Temperature and PSi for some brewing process I needed them for.

Anyway, I decided to move on with these new sensors - link below-, considering that I am trying to capture measurements close to 120 Psi which these sensors can take.
[Wireless External Sensors](https://www.aliexpress.com/item/32242371449.html?spm=a2g0s.9042311.0.0.fd624c4dz4d9ck)

I have done the same steps you mentioned to follow back then under  [the previous discussion about Jansite](https://github.com/merbanan/rtl_433/issues/1007#issuecomment-661674281) , but I can't really get any grasp of the same data or even close to it. Any advice will be appreciated, here is what I  have tried so far: 

- `rtl_433` command to confirm none of the already saved information will be able to retrieve any data. [confirmed]

- `rtl_433 G 4`  command to confirm none of the already saved information will be able to retrieve any data. [confirmed]

- `rtl_433 S all`  command to obtain the data, then run` rtl_433 -w YOURFILE.ook YOURFILE.cu8` and drop on https://triq.org/pdv/ to view, - ook files attached on this msg - .

None of these commands was able to guess the modulation used on these sensors, therefore after reading the pulses via https://triq.org/pdv , and based on the g002_433.92M_250k  I tried the next command `rtl_433 -R 0 -X 'n=TPMS,m=FSK_MC_ZEROBIT, s=100, l=100, r=300'` with no success. -readings attached- 
[rtl_433_tpms_120psi.zip](https://github.com/merbanan/rtl_433/files/5278954/rtl_433_tpms_120psi.zip)

Then I tried a different modulation. `rtl_433 -R 0 -X 'n=TPMS,m=FSK_MC_ZEROBIT, s=85, l=460, r=
[rtl_433_tpms_120psi.zip](https://github.com/merbanan/rtl_433/files/5278947/rtl_433_tpms_120psi.zip)
4350'` based on the rtl_433 analyzer with no success as well.  -readings attached-

Any help will be appreciated. 
Thank you in advance.

zuckschwerdt, 2020-09-25:
(for those with no intuitive grasp on, ahm freedom units, that's 827 kPa)

The OOK captures look consistent but the gaps are implausible short compared to the pulses. Also the gaps are roughly 40 and 80 µs while the pulses are 60 µs multiples. This couldn't be PCM. I suspect the demod failed here.

Do you have a .cu8 sample?

mr-pl3r, 2020-09-25:
Sure thing I do, please find them attached on this msg.
[g004_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/5284251/g004_433.92M_250k.zip)

zuckschwerdt, 2020-09-25:
I looks like GFSK maybe. It's too close to DC to visually make sense of though. Can you record again with `-s 1024k -f 433.98M`?
[image]

mr-pl3r, 2020-09-26:
Here you go! I just finished recording them...and after plugging them to the [SPECTROGRAM & PULSEDATA
]( https://triq.org/pdv/) it seems like there's data in there. Here is the zip file attached with the cu8 and ook files.

[image]
[433.98M_1024k.zip](https://github.com/merbanan/rtl_433/files/5287172/433.98M_1024k.zip)

zuckschwerdt, 2020-09-27:
That's a very clean OOK decode (not FSK after all, just DC interference). It's Manchester with a half-bit width of 52 µs -- somewhat "standard" for TPMS.
This will work to collect data: `-X 'n=TPMS,m=OOK_MC_ZEROBIT,s=52,r=1000'`
It should work even on default 250k sample rate. (you need `-s 1024k` when reading back non-default sample rate captures.)

mr-pl3r, 2020-09-29:
Okay, after getting some readings under Bitbench I was able to start analyzing and playing with the data coming from the sensors. This is the results after getting the information coming thru - BTW I needed to apply the "invert" to get the right information.
[Going up & Down ](https://triq.net/bitbench?c=407456f4fab4c&c=407456f4fab4c&c=337456f4edb4c&c=337456f4edb4c&c=277456f4e1b4c&c=277456f4e1b4c&c=197456f4d3b4c&c=197456f4d3b4c&c=107456f4cab4c&c=107456f4cab4c&c=087456f4c2b4c&c=087456f4c2b4c&c=7e7456f4b8b4c&c=7e7456f4b8b4c&c=&c=&c=2e7456f4c9b3e&c=2e7456f4c9b3e&c=377456f4d2b3e&c=3d7456f4d8b3e&c=3d7456f4d8b3e&c=457456f4e0b3e&c=457456f4e0b3e&c=4b7456f4e6b3e&c=4b7456f4e6b3e&c=507456f4ebb3e&c=507456f4ebb3e&c=557456f4f0b3e&c=557456f4f0b3e&c=587456f4f3b3e&c=5e7456f4f9b3e&c=5e7456f4f9b3e&c=607456f4fbb3e&c=607456f4fbb3e&c=617456f4fcb3e&c=617456f4fcb3e&c=637456f4feb3e&c=637456f4feb3e&c=647456f4ffb3e&c=647456f4ffb3e&c=647456f4ffb3e&c=647456f4ffb3e&c=647456f4ffb3e&c=647456f4ffb3e&c=647456f4ffb3e&c=647456f4ffb3e&f=4d%204d%20ID%3Ahhhhhh%2C%20PRESSURE%3A9d%20%2CTEMP%3A9d%2C%20FLAG%3A%20d&i=true)
The command that was able to gather the information is `rtl_433 -R 0 -X 'n=TPMS,m=OOK_MC_ZEROBIT, s=52, r=1000' -s 1024k -f 433.98M`  for some reason while trying to get it without the -s1024 was not good. 

* One thing I haven't figured out is how to get the battery status on these sensors, may be I am missing a bit or something.

Thank you again for  your help.

zuckschwerdt, 2020-09-29:
If you pad the message with a `f` nibble (`0` after invert) then the first byte is the sum of all other bytes, see [this BitBench](https://triq.net/bitbench?c=407456f4fab4cf&c=337456f4edb4cf&c=277456f4e1b4cf&c=197456f4d3b4cf&c=107456f4cab4cf&c=087456f4c2b4cf&c=7e7456f4b8b4cf&c=2e7456f4c9b3ef&c=377456f4d2b3ef&c=3d7456f4d8b3ef&c=457456f4e0b3ef&c=4b7456f4e6b3ef&c=507456f4ebb3ef&c=557456f4f0b3ef&c=587456f4f3b3ef&c=5e7456f4f9b3ef&c=607456f4fbb3ef&c=617456f4fcb3ef&c=637456f4feb3ef&c=647456f4ffb3ef&f=CHK%3A8h%20ID%3Ahhhhhh%20PRESSURE%3A8d%20TEMP%3A8d%20FLAG%3A%204h%20PAD%3A4h&i=true&cf=ADD&co=8&cl=48&cw=8) (note the checksum error in the 7th code).

I don't think that pressure and temp would be 9 bit. If you look at a field the lowest order bit should appear mostly random. If you take pressure as 9 bit it's always zero though.

mr-pl3r, 2020-09-30:
Excellent. That worked perfectly, I just needed to add to the code the next formula - this based on the fact that I am measuring these sensors against the previous ones you helped me with. 

Temperature is in Fahrenheit - without further calculations.
Pressure: (Bitbench # / 3 ) Psi .

Let me know if you want me to send you the code I am using to decode these sensors to include it as part of rtl_433.   Or I just close this topic "as is"?

Thank you again for all your help.

zuckschwerdt, 2020-09-30:
If the temperature is Fahrenheit then Psi would seem logical. Haven't seen a div 3 so far though. Mul 2 in kPa is somewhat close -- hard to verify I guess. Does the package mention a max pressure? 255/3 is only 85 (586 kPa). There could be a ninth bit somewhere.

Let's keep this open, I'll add a decoder shortly.

merbanan, 2021-03-22:
@zuckschwerdt have you had time to write a decoder?

gdt, 2023-09-29:
Status?