merbanan/rtl_433 #1501: error in rain indicator of inovalley-kw9015b.c

rtega, 2020-09-27:
My TFA 30.3161 rain sensor has it rain values trip over in the temperature indication:

```
2020-09-27 06:52:46,,,37,Inovalley-kw9015b,0,,,12.700,,255
2020-09-27 06:53:21,,,37,Inovalley-kw9015b,0,,,12.712,,0
2020-09-27 07:19:14,,,37,Inovalley-kw9015b,0,,,12.913,,0                                                                                                       
2020-09-27 07:29:54,,,37,Inovalley-kw9015b,0,,,13.113,,1
2020-09-27 08:59:17,,,37,Inovalley-kw9015b,0,,,13.712,,5

```
Does anybody see how this could be corrected?

zuckschwerdt, 2020-09-27:
Yes, the assumed data layout is

`AAAAAAAA BBBBBBBB BBBBBBBB CCCCCCCC DDDD`

- A : ID
- B : Temp
- C : Rain
- D : checksum

But that gives temp in 1/160 C which is unreasonable and 12 bit temp would be more likely. That would leave 12 bit for the rain then. Does that seem plausible? I.e.

`AAAAAAAA BBBBBBBB BBBBCCCC CCCCCCCC DDDD`

I'll just apply that change now, please keep an eye on the values and report if that works!

zuckschwerdt, 2020-09-27:
Closes with that fix, but please report if this works as intended.

rtega, 2020-09-27:
Hi thanks for the quick fix! Just wondering whether this is supposed to be normal: the rain count jumps from 264 (corrected from 9) to 40 now with the fixed code. Is this supposed to be normal with this fix or should it read 264?

rtega, 2020-09-27:
The temperature seems to be OK.

rtega, 2020-09-27:
Changing this line:
https://github.com/merbanan/rtl_433/blob/44acd9f15ed2edaf8c712c148ae7e20e66fb47b8/src/devices/inovalley-kw9015b.c#L47

to

`    rain     = ((reverse8(b[1]) & 0x0f) << 7) | reverse8(b[3]);
`

seems to solve the problem.

zuckschwerdt, 2020-09-27:
The bits are transmitted in reverse order and the field seem to be partially in reverse order also. Can you add
`fprintf(stderr, "Inovalley bits: %02x %02x %02x %02x\n", reverse8(b[0]), reverse8(b[1]), reverse8(b[2]), reverse8(b[3]));`
in https://github.com/merbanan/rtl_433/blob/master/src/devices/inovalley-kw9015b.c#L55 recompile and overserve the output?
We currently assume `IIIIIIII RRRRtttt TTTTTTTT rrrrrrrr CCCC` (lower-case = lower order bits). But that must be wrong.

zuckschwerdt, 2020-09-27:
Oh, sorry, yes it should be `<< 8` though.

rtega, 2020-09-27:
I'll put it at `<< 8` and see what happens when it trips over next.

rtega, 2020-10-29:
Just tripped over and we go from 767 to 1024. I guess that the bits RRRR are part of some checksum instead of part of the real counter. According to this site there are only 8 bits for the rain sensor:

`https://forum.fhem.de/index.php?topic=85746.0`

gdt, 2023-09-25:
What's the status of this?  @rtega are you still seeing problems with the latest git master?

rtega, 2023-09-26:
Nope, this should be OK as far as I can check. If I notice anything untoward in the future I'll reopen.

commit 44acd9f1 (2020-09-27): Fix Inovalley kw9015b temp/rain fields (closes #1501)
closed 2023-09-26 by gdt