merbanan/rtl_433 #1503: SCM+ duplicate devices

JonGilmore, 2020-10-01:
I'm running rtl_433 with the following: 
```
/usr/local/bin/rtl_433 -d 1 -f 915M -F mqtt://192.168.45.249 1883 user username pass password retain 1 events rtl_915[/model][/id]
```
this works well for an `ERT-SCM` device, but doesn't work well for my SCM+ water meter. The SDR picks up multiple water meters from around the neighborhood and values jump all over the place because it doesn't have an `id`. Can I filter this differently so this doesn't happen? I've shown a screenshot from MQTT Explorer below

[image]

zuckschwerdt, 2020-10-01:
Is the "EndpointID" field unique to the different meters? Perhaps we should map that to "id"? Also "PacketCRC" should likely be dropped, keys lowercased, and the "+"...

JonGilmore, 2020-10-01:
> Is the "EndpointID" field unique to the different meters? Perhaps we should map that to "id"? Also "PacketCRC" should likely be dropped, keys lowercased, and the "+"...

Yes, `EndpointID` is unique for the different meters near me. I think it would make sense to map to `id`, that would seem to follow standards of other devices that I have observed using this. I'm happy to help contribute with what is necessary to upstream these changes, but I'm no C developer and SDR's are brand new to me :/ 
Let me know what you need from me, I'm happy to help!

merbanan, 2020-10-01:
> s the "EndpointID" field unique to the different meters? Perhaps we should map that to "id"? Also "PacketCRC" should likely be dropped, keys lowercased, and the "+"...

The reason for the naming is to keep compatibility with rtlamr IIRC. Maybe add a duplicate id key?

JonGilmore, 2020-10-01:
> The reason for the naming is to keep compatibility with rtlamr IIRC. Maybe add a duplicate id key?

Ahh yeah, that would make sense. I think a duplicate key would solve my issue

zuckschwerdt, 2020-10-04:
`id` and some notes added, esp. "SCM+" is a bad MQTT topic name.

closed 2020-10-04 by zuckschwerdt

JonGilmore, 2020-10-04:
thanks @zuckschwerdt - I just confirmed this fixed my issues :D

commit 5ee8c8ce (2020-10-04): Add id key to scmplus (closes  #1503)