merbanan/rtl_433 #1504: Support for Vivint V-DW21R-345

kingbutter, 2020-10-02:
I've got some door sensors that came with an Vivint system that was in the house. I've submitted a pull request in rtl_433_tests with the captures: [https://github.com/merbanan/rtl_433_tests/pull/364]()

I see data when I run with "-X ‘2gigdw10-re208:OOK_MC_ZEROBIT:150:150:300’", which I found on a previous issue that had been posted. Unfortunately it appears that the vivint sensors are returning 96 bytes, where as the 2Gig are returning 64 bytes. 

Here's what -a -A gave me when I did a replay:

```
Test mode active. Reading samples from file: ./g001_344.975M_250k.cu8
Detected OOK package    @0.265964s
Analyzing pulses...
Total count:   79,  width: 25.41 ms             ( 6352 S)
Pulse width distribution:
 [ 0] count:   62,  width:  144 us [140;180]    (  36 S)
 [ 1] count:   17,  width:  276 us [276;280]    (  69 S)
Gap width distribution:
 [ 0] count:   62,  width:  120 us [96;128]     (  30 S)
 [ 1] count:   16,  width:  256 us [256;260]    (  64 S)
Pulse period distribution:
 [ 0] count:   52,  width:  264 us [264;276]    (  66 S)
 [ 1] count:   19,  width:  400 us [400;404]    ( 100 S)
 [ 2] count:    7,  width:  532 us [532;536]    ( 133 S)
Level estimates [high, low]:  15936,     39
RSSI: -0.1 dB SNR: 26.1 dB Noise: -26.2 dB
Frequency offsets [F1, F2]:    3502,      0     (+13.4 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 144, long_width: 0, reset_limit: 264, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=144,l=0,r=264'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {96} 00 01 85 ff bd ef fe c8 41 25 ba c3

Detected OOK package    @0.402300s
Analyzing pulses...
Total count:   79,  width: 25.41 ms             ( 6353 S)
Pulse width distribution:
 [ 0] count:   62,  width:  140 us [136;176]    (  35 S)
 [ 1] count:   17,  width:  268 us [268;276]    (  67 S)
Gap width distribution:
 [ 0] count:   62,  width:  124 us [96;132]     (  31 S)
 [ 1] count:   16,  width:  260 us [260;268]    (  65 S)
Pulse period distribution:
 [ 0] count:   52,  width:  264 us [264;272]    (  66 S)
 [ 1] count:   19,  width:  400 us [396;404]    ( 100 S)
 [ 2] count:    7,  width:  536 us [536;536]    ( 134 S)
Level estimates [high, low]:  14044,     16
RSSI: -0.7 dB SNR: 29.4 dB Noise: -30.1 dB
Frequency offsets [F1, F2]:    3845,      0     (+14.7 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 140, long_width: 0, reset_limit: 272, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=140,l=0,r=272'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {96} 00 01 85 ff bd ef fe c8 41 25 ba c3

Detected OOK package    @0.538620s
Analyzing pulses...
Total count:   79,  width: 25.41 ms             ( 6352 S)
Pulse width distribution:
 [ 0] count:   62,  width:  140 us [136;180]    (  35 S)
 [ 1] count:   17,  width:  272 us [272;276]    (  68 S)
Gap width distribution:
 [ 0] count:   62,  width:  124 us [92;132]     (  31 S)
 [ 1] count:   16,  width:  256 us [256;264]    (  64 S)
Pulse period distribution:
 [ 0] count:   52,  width:  264 us [264;272]    (  66 S)
 [ 1] count:   19,  width:  400 us [400;404]    ( 100 S)
 [ 2] count:    7,  width:  532 us [532;536]    ( 133 S)
Level estimates [high, low]:  16000,     36
RSSI: -0.1 dB SNR: 26.5 dB Noise: -26.6 dB
Frequency offsets [F1, F2]:    3370,      0     (+12.9 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 140, long_width: 0, reset_limit: 268, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=140,l=0,r=268'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {96} 00 01 85 ff bd ef fe c8 41 25 ba c3

Detected OOK package    @0.674964s
Analyzing pulses...
Total count:   79,  width: 25.41 ms             ( 6353 S)
Pulse width distribution:
 [ 0] count:   62,  width:  144 us [144;176]    (  36 S)
 [ 1] count:   17,  width:  280 us [280;284]    (  70 S)
Gap width distribution:
 [ 0] count:   62,  width:  120 us [100;124]    (  30 S)
 [ 1] count:   16,  width:  252 us [252;256]    (  63 S)
Pulse period distribution:
 [ 0] count:   52,  width:  264 us [264;276]    (  66 S)
 [ 1] count:   19,  width:  400 us [396;404]    ( 100 S)
 [ 2] count:    7,  width:  532 us [532;536]    ( 133 S)
Level estimates [high, low]:  15981,     25
RSSI: -0.1 dB SNR: 28.1 dB Noise: -28.2 dB
Frequency offsets [F1, F2]:    3852,      0     (+14.7 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 144, long_width: 0, reset_limit: 260, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=144,l=0,r=260'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {96} 00 01 85 ff bd ef fe c8 41 25 ba c3

Detected OOK package    @0.811248s
Analyzing pulses...
Total count:   79,  width: 25.41 ms             ( 6352 S)
Pulse width distribution:
 [ 0] count:   62,  width:  148 us [144;180]    (  37 S)
 [ 1] count:   17,  width:  280 us [280;284]    (  70 S)
Gap width distribution:
 [ 0] count:   62,  width:  116 us [96;124]     (  29 S)
 [ 1] count:   16,  width:  252 us [248;256]    (  63 S)
Pulse period distribution:
 [ 0] count:   52,  width:  264 us [264;276]    (  66 S)
 [ 1] count:   19,  width:  400 us [396;404]    ( 100 S)
 [ 2] count:    7,  width:  536 us [536;540]    ( 134 S)
Level estimates [high, low]:  15993,     24
RSSI: -0.1 dB SNR: 28.2 dB Noise: -28.3 dB
Frequency offsets [F1, F2]:    3033,      0     (+11.6 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 148, long_width: 0, reset_limit: 260, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=148,l=0,r=260'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {96} 00 01 85 ff bd ef fe c8 41 25 ba c3

Detected OOK package    @0.947628s
Analyzing pulses...
Total count:   79,  width: 25.41 ms             ( 6352 S)
Pulse width distribution:
 [ 0] count:   62,  width:  148 us [144;176]    (  37 S)
 [ 1] count:   17,  width:  280 us [280;288]    (  70 S)
Gap width distribution:
 [ 0] count:   62,  width:  116 us [100;124]    (  29 S)
 [ 1] count:   16,  width:  252 us [248;256]    (  63 S)
Pulse period distribution:
 [ 0] count:   52,  width:  264 us [264;276]    (  66 S)
 [ 1] count:   19,  width:  400 us [396;404]    ( 100 S)
 [ 2] count:    7,  width:  532 us [532;536]    ( 133 S)
Level estimates [high, low]:  15973,     28
RSSI: -0.1 dB SNR: 27.6 dB Noise: -27.7 dB
Frequency offsets [F1, F2]:    2892,      0     (+11.0 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 148, long_width: 0, reset_limit: 260, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=148,l=0,r=260'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {96} 00 01 85 ff bd ef fe c8 41 25 ba c3

*** signal_start = 56495, signal_end = 253260, signal_len = 196765, pulses_found = 474
Iteration 1. t: 53    min: 36 (372)    max: 70 (102)    delta 18
Iteration 2. t: 53    min: 36 (372)    max: 70 (102)    delta 0
Distance coding: Pulse length 53

Short distance: 30, long distance: 63, packet distance: 27733

p_limit: 53
bitbuffer:: Number of rows: 6
[00] {78} 00 01 10 04 44 00 a8 85 53 40
[01] {78} 00 01 10 04 44 00 a8 85 53 40
[02] {78} 00 01 10 04 44 00 a8 85 53 40
[03] {78} 00 01 10 04 44 00 a8 85 53 40
[04] {78} 00 01 10 04 44 00 a8 85 53 40
[05] {78} 00 01 10 04 44 00 a8 85 53 40
Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta 3567587328
Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
Distance coding: Pulse length 0

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 0
bitbuffer:: Number of rows: 0
```

Let me know what other information you need from me, or if there's another data grab I need to do.

Thanks!

zuckschwerdt, 2020-10-02:
This might be related to #1481 (but we don't have clean data there).

Yes, OOK_MC_ZEROBIT with 150 µs half-bit width looks like the Honeywell protocol.
Your data seems to always be `{96} 00 01 85 ff bd ef fe c8 41 25 ba c3`.

Per Honeywell protocol this data needs to be inverted (to `ff fe 7a 00 42 10 01 37 be da 45 3c`), we expect `PP PP C IIIII EE SS SS` with
- P: 16bit Preamble and sync bit (always ff fe)
- C: 4bit Channel
- I: 20bit Device serial number / or counter value
- E: 8bit Event, where 0x80 = Open/Close, 0x04 = Heartbeat / or id
- S: 16bit CRC

See [this BitBench](https://triq.net/bitbench?c=%7B96%7D%2000%2001%2085%20ff%20bd%20ef%20fe%20c8%2041%2025%20ba%20c3&f=HONEYWELL%3A%20PRE%3A16h%20CH%3A4h%20ID%3A20h%20EVENT%3A8h%20CRC%3A16h%20%3F8h8h8h8h&i=true&cw=4). But that doesn't fit (obviously). Can you gather different codes and annotate with the sensors action? That way we can locate the event and checksum fields.

kingbutter, 2020-10-02:
Here is what I get for closed and open (please let me know if this isn't what you are asking for)

with: `-X 'n=name,m=OOK_MC_ZEROBIT,s=150,l=150,r=300'`

Closed:
```
{"time" : "@0.123724s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe387fec84125e24a"}], "codes" : ["{96}000185ffe387fec84125e24a"]}
{"time" : "@0.288688s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe387fec84125e24a"}], "codes" : ["{96}000185ffe387fec84125e24a"]}
{"time" : "@0.453620s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe387fec84125e24a"}], "codes" : ["{96}000185ffe387fec84125e24a"]}
{"time" : "@0.618328s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe387fec84125e24a"}], "codes" : ["{96}000185ffe387fec84125e24a"]}
{"time" : "@0.782912s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe387fec84125e24a"}], "codes" : ["{96}000185ffe387fec84125e24a"]}
{"time" : "@0.947548s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe387fec84125e24a"}], "codes" : ["{96}000185ffe387fec84125e24a"]}
```

Open:

```
{"time" : "@0.121640s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe273fec841255190"}], "codes" : ["{96}000185ffe273fec841255190"]}
{"time" : "@0.286792s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe273fec841255190"}], "codes" : ["{96}000185ffe273fec841255190"]}
{"time" : "@0.451980s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe273fec841255190"}], "codes" : ["{96}000185ffe273fec841255190"]}
{"time" : "@0.617148s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe273fec841255190"}], "codes" : ["{96}000185ffe273fec841255190"]}
{"time" : "@0.782300s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe273fec841255190"}], "codes" : ["{96}000185ffe273fec841255190"]}
{"time" : "@0.947540s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe273fec841255190"}], "codes" : ["{96}000185ffe273fec841255190"]}
```

-a -A output:
Closed:
```
Detected OOK package    @0.123724s
Analyzing pulses...
Total count:   80,  width: 25.49 ms             ( 6373 S)
Pulse width distribution:
 [ 0] count:   63,  width:  140 us [132;172]    (  35 S)
 [ 1] count:   16,  width:  272 us [268;276]    (  68 S)
 [ 2] count:    1,  width:  100 us [100;100]    (  25 S)
Gap width distribution:
 [ 0] count:   63,  width:  124 us [100;136]    (  31 S)
 [ 1] count:   16,  width:  260 us [256;264]    (  65 S)
Pulse period distribution:
 [ 0] count:   57,  width:  264 us [264;272]    (  66 S)
 [ 1] count:   12,  width:  400 us [396;404]    ( 100 S)
 [ 2] count:   10,  width:  532 us [532;540]    ( 133 S)
Level estimates [high, low]:   8348,     39
RSSI: -2.9 dB SNR: 23.3 dB Noise: -26.2 dB
Frequency offsets [F1, F2]:    3995,      0     (+15.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 140, long_width: 272, reset_limit: 268, sync_width: 100
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=140,l=272,r=268,g=0,t=0,y=100'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {79} ff fd e7 ff df 7f db bd 4f a8
[01] { 0}                               :

Detected OOK package    @0.288688s
Analyzing pulses...
Total count:   80,  width: 25.49 ms             ( 6372 S)
Pulse width distribution:
 [ 0] count:   63,  width:  140 us [132;168]    (  35 S)
 [ 1] count:   16,  width:  272 us [268;276]    (  68 S)
 [ 2] count:    1,  width:   96 us [96;96]      (  24 S)
Gap width distribution:
 [ 0] count:   63,  width:  124 us [104;136]    (  31 S)
 [ 1] count:   16,  width:  260 us [256;268]    (  65 S)
Pulse period distribution:
 [ 0] count:   57,  width:  264 us [260;272]    (  66 S)
 [ 1] count:   12,  width:  400 us [396;408]    ( 100 S)
 [ 2] count:   10,  width:  532 us [528;540]    ( 133 S)
Level estimates [high, low]:   2552,     34
RSSI: -8.1 dB SNR: 18.8 dB Noise: -26.8 dB
Frequency offsets [F1, F2]:    3244,      0     (+12.4 kHz, +0.0 kHz)
[... 262 lines cut ...]
Detected OOK package    @0.947540s
Analyzing pulses...
Total count:   79,  width: 25.50 ms             ( 6374 S)
Pulse width distribution:
 [ 0] count:   61,  width:  140 us [132;164]    (  35 S)
 [ 1] count:   17,  width:  268 us [268;280]    (  67 S)
 [ 2] count:    1,  width:  104 us [104;104]    (  26 S)
Gap width distribution:
 [ 0] count:   61,  width:  124 us [104;132]    (  31 S)
 [ 1] count:   17,  width:  260 us [260;268]    (  65 S)
Pulse period distribution:
 [ 0] count:   55,  width:  264 us [260;272]    (  66 S)
 [ 1] count:   12,  width:  400 us [396;404]    ( 100 S)
 [ 2] count:   11,  width:  532 us [532;540]    ( 133 S)
Level estimates [high, low]:   2980,     42
RSSI: -7.4 dB SNR: 18.5 dB Noise: -25.9 dB
Frequency offsets [F1, F2]:    3252,      0     (+12.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 140, long_width: 268, reset_limit: 272, sync_width: 104
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=140,l=268,r=272,g=0,t=0,y=104'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {78} ff fd e7 ff d7 7f b7 7a 86 dc
[01] { 0}    
```

zuckschwerdt, 2020-10-02:
Yes, the codes like `{96}000185ffe273fec841255190` are good. We need a couple more to figure things out. Do you have more sensor to test? I.e. same state (e.g. open) but different sensor: we see where the id is. Toggling the state on different sensors: the state is where the bytes are the same across multiple sensors.

kingbutter, 2020-10-02:
Different sensor:

Closed:

```
{"time" : "@0.282500s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfd2bfec840a0ed4c"}], "codes" : ["{96}0001857bfd2bfec840a0ed4c"]}
{"time" : "@0.415460s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfd2bfec840a0ed4c"}], "codes" : ["{96}0001857bfd2bfec840a0ed4c"]}
{"time" : "@0.548464s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfd2bfec840a0ed4c"}], "codes" : ["{96}0001857bfd2bfec840a0ed4c"]}
{"time" : "@0.681456s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfd2bfec840a0ed4c"}], "codes" : ["{96}0001857bfd2bfec840a0ed4c"]}
{"time" : "@0.814428s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfd2bfec840a0ed4c"}], "codes" : ["{96}0001857bfd2bfec840a0ed4c"]}
{"time" : "@0.947456s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfd2bfec840a0ed4c"}], "codes" : ["{96}0001857bfd2bfec840a0ed4c"]}
```

Open:

```
{"time" : "@0.211476s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfc1bfec840a05826"}], "codes" : ["{96}0001857bfc1bfec840a05826"]}
{"time" : "@0.358652s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfc1bfec840a05826"}], "codes" : ["{96}0001857bfc1bfec840a05826"]}
{"time" : "@0.505856s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfc1bfec840a05826"}], "codes" : ["{96}0001857bfc1bfec840a05826"]}
{"time" : "@0.653056s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfc1bfec840a05826"}], "codes" : ["{96}0001857bfc1bfec840a05826"]}
{"time" : "@0.800188s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfc1bfec840a05826"}], "codes" : ["{96}0001857bfc1bfec840a05826"]}
{"time" : "@0.947460s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "0001857bfc1bfec840a05826"}], "codes" : ["{96}0001857bfc1bfec840a05826"]}
```

zuckschwerdt, 2020-10-03:
That gives (inverted):
```
0001 85ffe387fec84125 e24a [1 Closed]
0001 85ffe273fec84125 5190 [1 Open]
0001 857bfd2bfec840a0 ed4c [2 Closed]
0001 857bfc1bfec840a0 5826 [2 Open]
```
There are too much bits changing for a simple state change. The data might be encrypted.
The last two bytes are likely a checksum, it's not a plain CRC or Digest though.
The checksum might be over the unencrypted data then.
Do you have a tamper trigger on those devices? You might gain more codes by using that.
Try to record as many codes as you can (do they change over time, do they change with a battery reset?)
Then write all the codes out as above and try to find a pattern.

kingbutter, 2020-10-05:
Battery Insert Device ID 0019-050-7610:

```
{"time" : "@0.165656s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 2, "data" : "4"}], "codes" : ["{2}4"]}
{"time" : "@0.258760s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "00012fcbf0bffec841259bef"}], "codes" : ["{96}00012fcbf0bffec841259bef"]}
{"time" : "@0.393168s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "00012fcbf0bffec841259bef"}], "codes" : ["{96}00012fcbf0bffec841259bef"]}
{"time" : "@0.527520s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "00012fcbf0bffec841259bef"}], "codes" : ["{96}00012fcbf0bffec841259bef"]}
{"time" : "@0.661948s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "00012fcbf0bffec841259bef"}], "codes" : ["{96}00012fcbf0bffec841259bef"]}
{"time" : "@0.796384s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "00012fcbf0bffec841259bef"}], "codes" : ["{96}00012fcbf0bffec841259bef"]}
{"time" : "@0.930820s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "00012fcbf0bffec841259bef"}], "codes" : ["{96}00012fcbf0bffec841259bef"]}
{"time" : "@1.345828s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@1.475920s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@1.605952s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@1.736004s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@1.866064s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@1.996216s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.167688s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.297732s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.427780s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.557836s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.687892s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.817960s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe617fec841254539"}], "codes" : ["{96}000185ffe617fec841254539"]}
{"time" : "@0.986592s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@1.136036s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@1.285524s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@1.435020s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@1.584472s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@1.733976s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@0.200304s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@0.349756s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@0.499216s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@0.648668s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@0.798112s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
{"time" : "@0.947536s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe55ffec841252f72"}], "codes" : ["{96}000185ffe55ffec841252f72"]}
```

[... cut ...]
{"time" : "@0.649160s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe5b7fec840a08900"}], "codes" : ["{96}000185ffe5b7fec840a08900"]}
{"time" : "@0.798312s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe5b7fec840a08900"}], "codes" : ["{96}000185ffe5b7fec840a08900"]}
{"time" : "@0.947460s", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 96, "data" : "000185ffe5b7fec840a08900"}], "codes" : ["{96}000185ffe5b7fec840a08900"]}
```

zuckschwerdt, 2020-10-05:
The Device ID matches the last 16 bit (or likely 32 bit) field before the checksum, see [this BitBench](https://triq.net/bitbench?c=000185ffe387fec84125e24a%5BClosed%5D&c=000185ffe273fec841255190%5BOpen%5D&c=00012fcbf0bffec841259bef%5BBattery%20Insert%20Device%20ID%200019-050-7610%5D&c=000185ffe617fec841254539&c=000185ffe55ffec841252f72&c=0001857bfd2bfec840a0ed4c%5BClosed%5D&c=0001857bfc1bfec840a05826%5BOpen%5D&c=00012f00f0cffec840a0285f%5BBattery%20Insert%20Device%20ID%200019-050-7743%5D&c=000185ffe60bfec840a06e52&c=000185ffe5b7fec840a08900&f=VIVINT%3A%20PRE%3A16h%208h8h%20EVENT%3F8b%208b%2016d%2016d%20CRC%3F8h8h&i=true&cw=4).

48858 to 48991 is 133 and also ID 0019-050-7610 to ID 0019-050-7743 is 133.

The two battery inserts have `d034 0f40` and `d0ff 0f30` which is significantly different from the usual `7a00 ...` or `7a84` -- still not sure what's what in those fields.

kingbutter, 2020-10-05:
[User-manual-1674521.pdf](https://github.com/merbanan/rtl_433/files/5328306/User-manual-1674521.pdf)

I also found this user manual, not sure if it helps

zuckschwerdt, 2020-10-05:
All the collected data here in [this BitBench](https://triq.net/bitbench?c=00012fcbf0bffec841259bef%5BBattery%20Insert%20Device%20ID%200019-050-7610%5D&c=000185ffe387fec84125e24a%5BClosed%5D&c=000185ffe273fec841255190%5BOpen%5D&c=000185ffe617fec841254539&c=000185ffe55ffec841252f72&c=00012fdff0bffec84125e2ff%5Bbattery%20insert%5D&c=000185ffe417fec8412520ba%5Bclosed%5D&c=000185ffe307fec84125e979%5Bopen%5D&c=000185ffe413fec8412524a2%5BClosed%5D&c=000185ffe2f3fec841255aa3%5BClosed%5D&c=000185ffe1abfec84125abed%5BOpen%5D&c=00012f00f0cffec840a0285f%5BBattery%20Insert%20Device%20ID%200019-050-7743%5D&c=0001857bfd2bfec840a0ed4c%5BClosed%5D&c=0001857bfc1bfec840a05826%5BOpen%5D&c=000185ffe60bfec840a06e52&c=000185ffe5b7fec840a08900&c=00012ff1f0bffec840a0956f%5BBattery%20Insert%5D&c=000185ffe457fec840a0dce4%5Bclosed%5D&c=000185ffe33ffec840a09f25%5Bopen%5D&c=00012fe6f0bffec840a0467f%5BBattery%20Insert%5D&c=000185ffe33bfec840a09b3d%5BOpen%5D&c=000185ffe227fec840a0d890%5BClosed%5D&c=000185ffe18bfec840a0bca4%5BOpen%5D&f=VIVINT%3A%20PRE%3A16h%20EVENT%3A8h%20KEY%3F8h%20COUNTER%3F14h%202b%20ID%3A16d%2016d%20CRC%3F8h8h&i=true).

I'd say the first byte (7a / d0) is the event type, the next byte maybe a re-keying of sorts. The next 16 bits (last 2 bits always zero) looks somewhat like a counter -- which make sense for a security device, it protects against replay attacks.
That "counter" goes to a low value on battery insert and when the "key" is present. Otherwise it seems to go up.
I guess if you open and close over and over you see that counter going up and only go down when "key" is not `00`.
Still not sure how to extract the open/close value. Maybe you need to open/close one sensor 50 times or so and see what the "counter" does.

kingbutter, 2020-10-05:
Dropped these items (the ones with 96 byte length) into [this bitbench](https://triq.net/bitbench?c=00012fcbf0bffec841259bef%5BBattery%20Insert%20Device%20ID%200019-050-7610%5D&c=000185ffe387fec84125e24a%5BClosed%5D&c=000185ffe273fec841255190%5BOpen%5D&c=000185ffe617fec841254539&c=000185ffe55ffec841252f72&c=00012fdff0bffec84125e2ff%5Bbattery%20insert%5D&c=000185ffe417fec8412520ba%5Bclosed%5D&c=000185ffe307fec84125e979%5Bopen%5D&c=000185ffe413fec8412524a2%5BClosed%5D&c=000185ffe2f3fec841255aa3%5BClosed%5D&c=000185ffe1abfec84125abed%5BOpen%5D&c=00012f00f0cffec840a0285f%5BBattery%20Insert%20Device%20ID%200019-050-7743%5D&c=0001857bfd2bfec840a0ed4c%5BClosed%5D&c=0001857bfc1bfec840a05826%5BOpen%5D&c=000185ffe60bfec840a06e52&c=000185ffe5b7fec840a08900&c=00012ff1f0bffec840a0956f%5BBattery%20Insert%5D&c=000185ffe457fec840a0dce4%5Bclosed%5D&c=000185ffe33ffec840a09f25%5Bopen%5D&c=00012fe6f0bffec840a0467f%5BBattery%20Insert%5D&c=000185ffe33bfec840a09b3d%5BOpen%5D&c=000185ffe227fec840a0d890%5BClosed%5D&c=000185ffe18bfec840a0bca4%5BOpen%5D&c=000185ffe413fec8412524a2&c=000185ffe303fec84125ed61&c=000185ffe2f3fec841255aa3&c=000185ffe1affec84125aff5&c=000185ffe0c7fec841259852&c=000185ffe043fec841259779&c=000185ffdf2bfec84125fd4e&c=000185ffdfabfec84125f67d&c=000185ffdee3fec841253b35&c=000185ffde67fec84125341e&c=000185ffdddffec84125ae77&c=000185ffdd5ffec84125a544&c=000185ffdc9ffec84125a98f&c=000185ffdc1ffec84125a2bc&c=000185ffdb93fec841251d22&c=000185ffdb13fec841251611&c=000185ffdac3fec84125435f&c=000185ffda43fec84125486c&c=000185ffd9dffec8412555d1&c=000185ffd95ffec841255ee2&c=000185ffd8effec841256bb8&c=000185ffd86ffec84125608b&c=000185ffd733fec841253492&c=000185ffd68bfec84125aa38&c=000185ffd5dffec841258dbd&c=000185ffd55bfec841258296&c=000185ffd4fbfec84125f62a&c=000185ffd47ffec84125f901&c=000185ffd387fec8412553d5&c=000185ffd303fec841255cfe&c=000185ffd237fec841258ccc&c=000185ffd2b7fec8412587ff&c=000185ffd187fec84125ecb5&c=000185ffd103fec84125e39e&c=000185ffd003fec84125bc1d&c=000185ffd083fec84125b72e&c=000185ffcfdbfec841251b18&c=000186b0dd43fec5c3f3e91c&c=000185ffcf5bfec84125102b&c=000185ffce5bfec84125676b&c=000185ffcedbfec841256c58&c=000185ffcdcffec841252306&c=000185ffcd4ffec841252835&c=000185ffcca3fec8412571f9&c=000185ffcc23fec841257aca&c=000185ffcbd3fec8412589ce&c=000185ffcb57fec8412586e5&c=000185ffca3ffec84125d002&c=000185ffcabbfec84125df29&c=000185ffc96ffec84125505f&c=000185ffc9ebfec841255f74&c=000185ffc8d7fec841255d95&c=000185ffc857fec8412556a6&c=000185ffc7abfec841257860&c=000185ffc72bfec841257353&c=000185ffc65bfec841250d22&c=000185ffc6dffec841250209&c=000185ffc53ffec84125638e&c=000185ffc5bbfec841256ca5&c=000185ffc483fec84125137f&c=000185ffc407fec841251c54&c=000185ffc32ffec841258ced&c=000185ffc3abfec8412583c6&c=000185ffc23ffec841258aeb&c=000185ffc2bffec8412581d8&c=000185ffc153fec841254fb7&f=VIVINT%3A%20PRE%3A16h%20EVENT%3A8h%20KEY%3F8h%20COUNTER%3F14h%202b%20ID%3A16d%2016d%20CRC%3F8h8h&i=true&cw=4).

Outside of a couple weird ones, most of them are incrementing up it appears

```
{96}000185ffe413fec8412524a2
{96}000185ffe303fec84125ed61
{96}000185ffe2f3fec841255aa3
{96}000185ffe1affec84125aff5
{96}000185ffe0c7fec841259852
{96}000185ffe043fec841259779
{96}000185ffdf2bfec84125fd4e
{96}000185ffdfabfec84125f67d
{96}000185ffdee3fec841253b35
{96}000185ffde67fec84125341e
{96}000185ffdddffec84125ae77
{96}000185ffdd5ffec84125a544
{96}000185ffdc9ffec84125a98f
{96}000185ffdc1ffec84125a2bc
{96}000185ffdb93fec841251d22
{96}000185ffdb13fec841251611
{96}000185ffdac3fec84125435f
{96}000185ffda43fec84125486c
{96}000185ffd9dffec8412555d1
{96}000185ffd95ffec841255ee2
{96}000185ffd8effec841256bb8
{96}000185ffd86ffec84125608b
{96}000185ffd733fec841253492
{96}000185ffd68bfec84125aa38
{96}000185ffd5dffec841258dbd
{96}000185ffd55bfec841258296
{96}000185ffd4fbfec84125f62a
{96}000185ffd47ffec84125f901
{96}000185ffd387fec8412553d5
{96}000185ffd303fec841255cfe
{96}000185ffd237fec841258ccc
{96}000185ffd2b7fec8412587ff
{96}000185ffd187fec84125ecb5
{96}000185ffd103fec84125e39e
{96}000185ffd003fec84125bc1d
{96}000185ffd083fec84125b72e
{96}000185ffcfdbfec841251b18
{96}000186b0dd43fec5c3f3e91c
{138}000185ffcfdbfecf83ffd0ffb170fcfa470
{96}000185ffcf5bfec84125102b
{97}000186b0dd43fec5c3f3e91c8
{60}05bfec84125102b
{96}000185ffce5bfec84125676b
{96}000185ffcedbfec841256c58
{96}000185ffcdcffec841252306
{96}000185ffcd4ffec841252835
{96}000185ffcca3fec8412571f9
{96}000185ffcc23fec841257aca
{96}000185ffcbd3fec8412589ce
{96}000185ffcb57fec8412586e5
{96}000185ffca3ffec84125d002
{96}000185ffcabbfec84125df29
{96}000185ffc96ffec84125505f
{96}000185ffc9ebfec841255f74
{96}000185ffc8d7fec841255d95
{96}000185ffc857fec8412556a6
{96}000185ffc7abfec841257860
{96}000185ffc72bfec841257353
{96}000185ffc65bfec841250d22
{96}000185ffc6dffec841250209
{96}000185ffc53ffec84125638e
{96}000185ffc5bbfec841256ca5
{96}000185ffc483fec84125137f
{96}000185ffc407fec841251c54
{96}000185ffc32ffec841258ced
{96}000185ffc3abfec8412583c6
{96}000185ffc23ffec841258aeb
{96}000185ffc2bffec8412581d8
{96}000185ffc153fec841254fb7

```

zuckschwerdt, 2020-10-05:
Great! Now we roughly know what's going on. Maybe the difference in counter values tells what event it is. You might want to investigate in that direction -- can't see a "rhythm" at a quick glance though . Otherwise I'm out of ideas.

clsferguson, 2021-06-09:
is this related to #1261?

gdt, 2023-09-29:
@kingbutter Are you still working on this, heading to a PR?

NorthernMan54, 2026-08-19:
@n8henrie Range with the cc1101 will be a challenge.  I was doing testing with my CC1101, and range is really short.  I was able to get just a handful of meters of range.  So it works as a proof of concept, but is it practical?

If you have one and want to play around, PR to OpenMQTTGateway is here - https://github.com/1technophile/OpenMQTTGateway/pull/2358

n8henrie, 2026-08-19:
Sweet. Maybe a little 1/4 wavelength antenna?

EDIT: Here's a guy using a dipole: https://community.home-assistant.io/t/another-honeywell-vista-alarm-panel-external-component-with-hardware-uart-and-native-345-mhz-rf-via-cc1101/1006176?u=n8henrie

ilektron, 2026-08-19:
I was going to ask what antenna you are using? I made a 1/4 wave dipole for my SDR out of coax that seems to work really well. The 315MHz and 433MHz antennas are probably pretty crappy at 345MHz. I'll probably purchase a CC1101 to play around with.

NorthernMan54, 2026-08-19:
@ilektron My unit looks like this.

[image]

NorthernMan54, 2026-08-24:
@n8henrie I spent some time on tuning the settings on the cc1101, and it dramatically increased the range.  I went from a few meters to across the house now ( about 10 meters).  And it currently performs better than my rtl_sdr.

Top row is my cc1101 running OMG and bottom row is my rtl_sdr.  In top right I'm tracking loop1 output from OMG, and bottom right is rtl_433.  The other corners are reported rssi

[image]

This is my patented signal generator 2000

[image]

It's an ESP8266, running Tasmota, connected to a relay, connected to loop1 on a DW11.  And the relay opens every 30 seconds, then closes 30 seconds later.

To determine the tuning required, I had ChatGPT construct a knob turning setup, then left it running testing all the available settings.  After a few hours it came up with some recommended settings for the CC1101.

ilektron, 2026-09-01:
I'm taking way to long to wrap this up, but think I'm about ready to open a pull request back to the main repo. I grabbed some of the cc1101 modules to play with. I'm curious if a large 345mhz dipole will solve the range issue and will let you know. I'd also love to be able to simulate transmitting vivint events with it.

A quick info dump for posterity

I did order a VS-FLD001-345 and it surprisingly uses a 0x71 packet that isn't encrypted. The WPS/pair button on the flood sensor also sends a 0x74 packet oddly. I'm going to have to dump the firmware and see what's going on. The VS-SKEY2-345 uses a different packet format that's very similar but will be a different effort. It seems to use the same 0xd0 packet format on startup but uses a different CRC and possibly encryption. I'll eventually try dumping its firmware, but that will be a later effort. The DW11, DW21R, and PIR2 all can send a 0x71 packet when encryption isn't enabled, but  I have yet to see it send one. The SKEY2 has functionality that changes the packet format from Legacy, Legacy + garage door, to the new encrypted format if you hold a button down for 10 seconds. Once you enable the 'encrypted' mode, though, you can't downgrade the fob back to the legacy formats. The code on the dw sensors is very similar and makes me wonder if a brand new sensor has some carryover logic from the 2gig secure keyfobs which turned into the vivint fobs.

The 0x72 packet seems to only be sent once or very infrequently. Initially I thought it was a heartbeat, but it seems to be more likely to be just an informative dump of the battery voltage and the threshold that it is being compared to. The 0xd0 packet also includes the battery level and a unique byte that is constant for each different device. If received, you could know exactly what kind of device the sensor is. 

If anybody has a PIR3 or GB3, or even another flood sensor, I would love some captures of power-on and then their events to see if they are the same as the previous models. I also haven't been able to trigger a freeze event on the flood sensor,  which in the manual is Loop 3. I'm not sure which bit in the event is Loop 3. Yet another reason to dump the firmware. 

For now, my pr that I'll submit should at least work with most of the Vivint 345 sensors, minus the keyfobs.

NorthernMan54, 2026-09-03:
@ilektron I swapped the stock CC1101 ducky antenna on my test device for a straight 20.6 cm (8.11 inch) piece of solid wire, and the performance leap was night and day. Range expanded from just a few rooms to reliably reaching all the way out to my back shed.

Additionally, I can confirm the ESP32 successfully cracks the seed, making it incredibly easy to pair and onboard new DW11 devices natively. This setup is officially ready for deployment!

Going to add this here for others working with DW11's ( and possibly other similar devices ) in the future and needing some tips for integration.  Should this be in vivint.c ?

**Key Insights for Deploying DW11 ( and possibly other similar devices ) Sensors:**

**1 - Packet Loss & Placement Monitoring via Rolling Counter**

The DW11 increments an internal message counter with every single transmission. By tracking this integer in your code, any jump or gap in the sequence immediately signals a missed packet. This serves as an excellent diagnostic metric to flag localized interference or poor receiver placement.

**2 - Supervisory Heartbeat Timeout (70-Minute Window)**

The DW11 broadcasts a periodic supervisory heartbeat roughly every 70 minutes. Implementing a software timeout threshold at 80 minutes creates the perfect "dead man's switch." If this window passes without a check-in, you can reliably trigger a "Sensor Offline" or "Flaky Placement" alert.

ilektron, 2026-09-11:
I keep on digging deeper in this rabbit hole, but finally got a PR submitted. Since all these devices can also broadcast 

@NorthernMan54 My opinion is that I don't think that those items should be in the decoder. Those seem like automations that belong in something like Home Assistant to monitor packet loss and loss of communication. The recent releases of Home Assistant allegedly makes that type of automation trivial.

It also appears that my "new" DW21R and FLD001 aren't using the encrypted messages, and therefore also don't even send the counter. This seems to confirm my suspicion that there's some installation trick for new devices to determine if they should upgrade to encrypted comms. I'll need to poke through the firmware again to figure out why. Maybe the sensors that end up in the DIY market aren't configured for encrypted comms.

NorthernMan54, 2026-09-15:
@ilektron Sorry bad wording on my part, I was thinking as a comment and not in the code itself.  But in the issue would be good as well as others may see it and leverage.

n8henrie, 2026-10-01:
I renamed vivint-decode to "revivint" (<https://github.com/n8henrie/revivint-rs>), which I thought was clever given the intended purpose.I repackaged it into a workspace including a no_std core upon which an ESP32C3 (via CC1101) firmware is based, in addition to the CLI tool that is fairly similar to what I had previously published. Additionally, I have packaged both into a nix flake that should make the build environment and process simple and highly reproducible for nix users on macos and linux.

While not particularly relevant to rtl_433, so far I'm very pleased with the ESP32C3 version -- I get excellent reception from across my house (roughly 60 ft, several plaster walls, and 1 floor) with a cheap aliexpress CC1101 and its built-in antenna. I have it configured to relay the open / closed status to HomeAssistant via MQTT over WiFi, and it's practically instantaneous.

I think I'm finally ready to desolder all the debugging wires and reinstall the sensors for real-world use!