merbanan/rtl_433 #1509: [BUG] rtl_433 is sometime reporting 5 decimals in rain_mm

vistalba, 2020-10-05:
I use rtl_433 with mqtt to forward the messages to a mqtt broker on my homeassistant.

My weatherstaiont is a FineOffset WHx080. (`-R 32`)

Sometimes I now see that the value `rain_mm` has valus with 5 decimals.
Normally it only shows 1 decimals as the smallest amount of rain it can measure is 0.3mm.

So basicly only values from 0.0, 0.3, 0.6, 0.9, 1.2, 1.5, 1.8, 2.1, ... are possible.

Issue in mqtt explorer:
[image]

Without issue in mqtt explorer:
[image]

Command for rtl_433:
```ExecStart=/usr/local/bin/rtl_433 -f 868.319M -R 32 -d2 -p 28 -Y classic -s 250k -F mqtt://192.168.1.xx:1883,user=anyuser,pass=hiGhSeCurE,retain=1,devices=rtl_433[/id]"```

Any ideas why this happens?

zuckschwerdt, 2020-10-05:
We cut off all trailing zeros when printing values, but the default format is 5 decimals.
The real problem is that `rain = rain_raw * 0.3` here, which does not play nice with dual numbers.
I don't think we can do anything about that in the JSON or MQTT output.

vistalba, 2020-10-05:
Just round to 1 decimal after *0.3 should do the trick.
I do this now in Homeassistant as it is very annoying to get 99.60001 mm as value as it simply make no sense at all. ;)

zuckschwerdt, 2020-10-05:
That number is already rounded. The calculation is in float: 332 *0.3f = 99.6000061
Calculation in double then converting to float might round better: 99.59999847, but neither is great.
Maybe we should round depending on scale (i.e. small number -> lots of decimals vs. big number fewer decimals).

vistalba, 2020-10-05:
Sorry, I don‘t get the point. Why not just round the end result to 1 decimal?

99.6000061 = 99.6
99.59999847 = 99.6

Or why so much decimals should make sense?

E.g. 0.59999847 would be 599.99847 ml of rain.
Are this 0.00153 ml really relevent?

zuckschwerdt, 2020-10-05:
The number is a float value, not a string. What you see as "99.6" is 9.6000061 in the computer, even after rounding :)
There is only 99.5999984741 and then 99.6000061035, no number in between. (see e.g. https://www.h-schmidt.net/FloatConverter/IEEE754.html)
Any we can't round again in the ouput because some values need those decimals.

vistalba, 2020-10-05:
Ok, I undersrand. But why it works on console output? The output is rounded. Never got the wrong value there. So wgy not just rount the float to 1 decimal before publishing via mqtt?

zuckschwerdt, 2020-10-05:
In the mqtt output we don't really know how many decimals are needed, 1 or 2 is not always sufficient.
The console output uses a different formatting, given by the decoder. But there are other special characters added, like " C" or "%".
Perhaps I can add something to chop off "00001", I'll take a look.

vistalba, 2020-10-06:
Okay.. thanks.
For me I solved it with a workaroud in Homeassitant and just round the value I receive from rtl_433 to 1 decimal.

merbanan, 2021-03-16:
@zuckschwerdt floating point is evil. Have you looked at this any more? Or can we close the ticket? I don't really see any sane way to fix this.

closed 2021-03-17 by zuckschwerdt