merbanan/rtl_433 #1510: Home Assistant MQTT Auto Discovery example state_topic missing subtype (message type)

rct, 2020-10-05:
Summary: The `state_topic` that is auto generated by the hass mqtt example python script doesn't include `subtype` for rtl_433 devices that have a subtype, causing Home Assistant to subscribe to the wrong topic.

The details:

The rtl_433 MQTT device topic includes `subtype` for devices like Acurite 5n1 and Acurite Atlas that send different message types with different payloads and have `subtype` field.  The topic bases are

* rtl_433/{host}/devices/Acurite-5n1/56/{channel}/{id}
* rtl_433/{host}/devices/Acurite-5n1/49/{channel}/{id}

(56 and 49 are the different message type IDs), so full topics are:

* rtl_433/{host}/devices/Acurite-5n1/**56/**{channel}/{id}/temperature_F
* rtl_433/{host}/devices/Acurite-5n1/**49/**{channel}/{id}/wind_dir_deg

The home assistant MQTT auto discovery example script doesn't capture/use the subtype if it is part of the topic.  So the `state_topic` in the generated config for Home Assistant is incorrect:

* rtl_433/{host}/devices/Acurite-5n1/{channel}/{id}/temperature_F
* rtl_433/{host}/devices/Acurite-5n1/{channel}/{id}/wind_dir_deg

zuckschwerdt, 2020-10-05:
Yes, the default topic for MQTT is `devices[/type][/model][/subtype][/channel][/id]` a workaround would be to just request `devices[/type][/model][/channel][/id]` but your suggested change is way better (type might need to be there too). Do you want to PR this?

rct, 2020-10-05:
I want to mention a separate issue - This example script is a very nice starting point for Home Assistant users ONCE you figure out what the script does and it's purpose. There are almost no descriptions for many of the example scripts. Most are simpler than this one.

A basic description along the lines of  "rtl_433_mqtt_hass.py subscribes the rtl_433 event stream that is published to MQTT by rtl_433. The script publishes additional MQTT topics that can be used by Home Assistant to automatically discover and minimally configure new devices."

rct, 2020-10-05:
> 
> Yes, the default topic for MQTT is `devices[/type][/model][/subtype][/channel][/id]` a workaround would be to just request `devices[/type][/model][/channel][/id]` but your suggested change is way better (type might need to be there too). Do you want to PR this?

Thanks, that helps.

I will probably make an attempt. However, I don't believe I have (or have seen) devices that contain a type field. To be useful to me I've got to get subtype into the state topic.  So I'm not sure I'll correctly cover all the cases.  

I haven't looked at the mqtt data make code.  If the order is always [/type][/model][/subtype][/channel][/id] then this should be pretty straightforward.

(A separate matter is I think I've got to also implement some white and/or black listing because I've got too many false positives showing up that I don't want to clutter up Home Assistant with.)

rct, 2020-10-10:
Slightly different issue, I'm pretty new to MQTT - These configuration messages don't appear to be set to be retained, though they are republished every 10 minutes. So after restarting Home Assistant, it might be up to 10 minutes before all the sensors are back. 

It seems to me these config messages should probably be retained.  However the downside is false positive and transient signals will stay around for ever. 

What's the rational and what are people's experiences with this?

rct, 2020-10-11:
A special case comes up for devices that have multiple subtypes (message IDs) that have common elements across message types. For example the Acurite 5n1 and Atlas send wind speed in every message type since it changes so often (gusts). The Atlas with the lightning sensor also sends the lightning data in each message type. 

With the current scheme, you'd wind up only subscribed to the data element in one of the message type.  This could be avoided by wildcarding the subtype for that element.  So instead of subscribing to `rtl_433/wp1/devices/Acurite-Atlas/37/A/924/wind_avg_mi_h` subscribe to `rtl_433/wp1/devices/Acurite-Atlas/+/A/924/wind_avg_mi_h`.

On a separate note, the script is also lacking a translation for `wind_avg_mi_h`.  There are probably a bunch more that needed to be added.

rct, 2020-10-11:
One more subtype related wrinkle, The `device->identifiers` shouldn't include `subtype` so that a device with multiple message types gets grouped together as part of the parent device.

In other words, Home Assistant uses `identifiers` in the device object to group the separate sensors together under one device.   Using the Acurite 5n1 and Atlas example, there are 2 or 3 separate subtypes, but they are all part of the same device, and should be grouped together in Home Assistant. 

I can pull `subtype` out which will work for the devices I'm interested in.  Not sure if there are other users of subtype where this wouldn't be correct.

zuckschwerdt, 2020-10-12:
I'd say `type:model:subtype:id` should form a fixed "id". I.e. `subtype` is just more bits for `id`.
So `subtype` to me isn't "different function of the same device" but "different devices with other capabilities".

At a glance I think cotech, current_cost, dsc, elv, honeywell_wdb, interlogix, prologue, thermopro_tx2 assumes this but acurite, fineoffset_wh1080 might be in trouble there.

Maybe a breaking change to rename the fields there to e.g. `msgtype` is required.

rct, 2020-10-12:
> I'd say `type:model:subtype:id` should form a fixed "id". I.e. `subtype` is just more bits for `id`.
> So `subtype` to me isn't "different function of the same device" but "different devices with [...]

This makes sense to me.    Putting it another way `subtype` is useful for breaking out distinct sub-devices that might be aggregated behind some sort of gateway. 

In the case of the Acurite weather stations, `subtype` would make sense if it was breaking out the messages from the different sensors.  However, there are common fields across message types and it isn't breaking out different sensors, it's just using different message types to manage the frequency of different data items. Wind and lightning are sent in every message because they change frequently. Temperature, humidity, rain, light, uv are sent in every 3rd message in the Atlas case. 

I'm not familiar with the fineoffset_wh1080, but I'm guessing it might be similar?

Now that I see how `subtype` is used in MQTT for DSC it is clearly wrong and should be removed. It wouldn't be as bad if it was just the numeric first digit of the ESN, but it's still nearly arbitrary since they didn't really follow the guidelines.

rotragit, 2020-10-29:
@rct  can you post some Acurite-5n1 events (rtl_433/+/events from HA) to let me better understand the issue?
I think that the identifier should include the subtype and somehow the definition for HA should define the grouping.

rct, 2020-10-29:
@rotragit - See #1520.  With that merged there is no longer a subtype in the 5n1 or Atlas message.  So now the fields from the 2 or 3 different message types are all merged under a single base topic. So the naming problem is a moot point.  

So now instead of having data split between these two topics

* rtl_433/{host}/devices/Acurite-5n1/56/{channel}/{id}/{field}
* rtl_433/{host}/devices/Acurite-5n1/49/{channel}/{id}/{field}

They will all be under

* rtl_433/{host}/devices/Acurite-5n1/{channel}/{id}/

Regardless, I did implement a fix to construct the topic name per the form @Zuchschwerdt mentioned `rtl_433/{host}/devices[/type][/model][/subtype][/channel][/id]` but I haven't submitted a PR yet. There is still more clean up like getting definitions for all of the Acurite fields.  I'll try to get a PR done this weekend.

rotragit, 2020-10-30:
@rct @zuckschwerdt OK, the problem (and the solutions) seem to be different and not to conflict each other. With the Prologue-TH it's only a problem with the discovery script that doesn't take in account the subtype value even if it's used by rtl_433 to create the instance id. While the patch on rtl_433 is Acurite secific, the lines [(Prologue-TH missing subtype in instance definition in rtl_433_mqtt_hass.py #1540)](./1540)  added to the discovery script are more general.

rct, 2020-11-02:
@rotragit - You are correct, there are two solutions. They don't conflict and should be be implemented. 

If `subtype` is used, it needs to be part of the topic string. #1546 fixes that.  Can you take a look at that see if it resolves the issue with the Prologue TH?

The Prologue-TH, like the Acurite devices mentioned above should likely not be using `subtype`. Changing that will make the device topic consistent with the other devices.  @zuckschwerdt mentioned in #1540:

> For Prologue the first nibble really is a (hardware version?) type. We only ever see 9 and 5 and the value is fixed (i.e. "a Prologue9" or a "Prologue5" device), it could be seen as part of the ID though, https://github.com/merbanan/rtl_433/blob/master/src/devices/prologue.c#L19

rotragit, 2020-11-02:
@rct checked, it  works right. Great job! Minor issue: the button attribute is skipped. I think if a value is read from the sensor it should be always passed.

rct, 2021-01-26:
I'm going to go ahead and close this.

- #1546 got merged Nov. 2020.
- Devices that have subtypes in their topics should be handled correctly now. 
- The prologue TH subtype question is being discussed in another issue.

closed 2021-01-26 by rct