merbanan/rtl_433 #1512: hcs200 not decoding the KeyLoq key

majkrzak, 2020-10-07:
I've tried to check `hcs200` decoder with my KeyLoq key. It is not handling it by default either when forced to.

Running:
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=370,l=772,r=14000,g=4000,t=152,y=0,preamble={12}0xfff'`according to https://github.com/merbanan/rtl_433/blob/master/src/devices/hcs200.c results in more or less correct capture:

```
name: bitbuffer:: Number of rows: 2 
[00] { 0}                            : 
[01] {66} b5 52 57 3e 5f 6d 98 08 00 
```

merbanan, 2020-10-07:
Looks like we should loop over the rows. Can you attach a sample recording, that should help with the implementation?

majkrzak, 2020-10-07:
Looks like it is not about iterating the rows, but increasing the `gap` helps. Button 1 and 2 are detected correctly together with all buttons combination, but buttons 3 and 4 not.

How can I make proper recording for future testing? (guide on project wiki seems not be up to date)

merbanan, 2020-10-07:
Use rtl_433 -S unknown.

merbanan, 2021-03-07:
No samples attached. Reopen with samples and we can fix the code.

azriek, 2021-07-01:
Hi,
same issue here:
no detection with rtl_433 -f 868M -R 131
detection with rtl_433 -f 868M -R 0 -X 'n=name,m=OOK_PWM,s=370,l=772,r=14000,g=4000,t=152,y=0,preamble={12}0xfff'
```
time      : 2021-07-01 23:05:20
model     : name         count     : 1             num_rows  : 2             rows      : 
len       : 0            data      : , 
len       : 66           data      : a23bb443c9532a020
codes     : {0}, {66}a23bb443c9532a020
```

i attach a sample.
[g001_868M_1000k.zip](https://github.com/merbanan/rtl_433/files/6750845/g001_868M_1000k.zip)

zuckschwerdt, 2021-07-02:
Thanks! The decoder tries too hard to fit the preamble in the same row. Changing it to two roes now.

closed 2021-07-02 by zuckschwerdt

zuckschwerdt, 2021-07-02:
@azriek can you check if all buttons work? Especially 3 and 4? If not can you give the output of
`rtl_433 -f 868M -R 0 -X 'n=hcs200,m=OOK_PWM,s=370,l=772,r=9000,g=1500,t=152`
for each button.

azriek, 2021-07-02:
Great Work ! Device is detected now and button 1 and 2 are OK. (respectively Up and Stop in this rolling shutter remote)
```
time      : 2021-07-02 22:26:37
model     : Microchip-HCS200                       id        : 0607B250
id_rev    : 0A4DE060     encrypted : C7AB3A4F      encrypted_rev: F25CD5E3   button1   : ON            button2   : OFF           button3   : OFF           button4   : OFF
misc      :              Battery   : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-02 22:26:43
model     : Microchip-HCS200                       id        : 0607B250
id_rev    : 0A4DE060     encrypted : 49586907      encrypted_rev: E0961A92   button1   : OFF           button2   : ON            button3   : OFF           button4   : OFF
misc      :              Battery   : 1
```

as you guessed, down is not decoded (assume it will be button 3) and i don't have button 4 (but if needed i will disassemble the remote and try to activate the pin)
```
time      : 2021-07-02 22:26:46
model     : Microchip-HCS200                       id        : 0607B250
id_rev    : 0A4DE060     encrypted : 0F1B1A27      encrypted_rev: E458D8F0   button1   : OFF           button2   : OFF           button3   : OFF           button4   : OFF
misc      :              Battery   : 1
```

May be relevant or not, chip is HCS300 but non-encrypted part seems the same in the datasheets

Here is the log for Up, Stop and Down (assumed 1, 2 and 3 )

```
time      : 2021-07-02 22:03:59
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : f5f78715607b25040
codes     : {12}fff, {66}f5f78715607b25040
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:00
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : f5f78715607b25044
codes     : {12}fff, {66}f5f78715607b25044
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:00
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : f5f78715607b25044
codes     : {12}fff, {66}f5f78715607b25044
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:00
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : f5f78715607b25044
codes     : {12}fff, {66}f5f78715607b25044
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:00
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : f5f78715607b25044
codes     : {12}fff, {66}f5f78715607b25044
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:00
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 35           data      : f5f787156
codes     : {12}fff, {35}f5f787156

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:03
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : a4e2adc0607b25020
codes     : {12}fff, {66}a4e2adc0607b25020
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:03
[... 42 lines cut ...]
time      : 2021-07-02 22:04:07
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : 0a5459b3607b25014
codes     : {12}fff, {66}0a5459b3607b25014
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:07
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 66           data      : 0a5459b3607b25014
codes     : {12}fff, {66}0a5459b3607b25014
[... cut ...]
len       : 66           data      : 0a5459b3607b25014
codes     : {12}fff, {66}0a5459b3607b25014
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-07-02 22:04:07
model     : hcs200       count     : 2             num_rows  : 2             rows      : 
len       : 12           data      : fff, 
len       : 35           data      : 0a5459b36
codes     : {12}fff, {35}0a5459b36

```

zuckschwerdt, 2021-07-03:
Thanks! That confirms a suspicion where the original author had a remote with a quirk: the reported "button3" and "button4" where actually a combination of "1"+"4" and "1"+"2".

Also the button numbers are mixed (current output in quotes, protocol numbers plain): "1" is really 2, "2" is really 3, "3" is really 1+4, "4" is really 2+3.

I'm going to change that, and also remove the "ALL_BUTTONS" output. I get that it might be useful for some, but rtl_433 can't realistically cover special cases for generic protocols -- button-states in the protocol do no map to labels on (different) remotes easily (see the case above).

We might want to add some "flexible output remapping" someday perhaps.

Can you find out if there is some ID printed on the remote, the PCB, or packaging?
Is it 607B250 or A4DE060?

zuckschwerdt, 2021-07-03:
For reference, the original PR was #1081 from @667bdrm with test data in https://github.com/merbanan/rtl_433_tests/pull/287

Looking at the manual the button bit names are (in order MSB/first sent to LSB) S3, S0, S1, S2. And I was wrong about the button labels being "quirky":
Those bits are some encoding of the hardware pins (note there is no S3 pin), with all bits set a special case of "Seed Transmission" aka learn-mode.
The outputs currently in the decoder are the only valid states according to the HCS200 manual. It's different for the HCS300 though (and there is a S3 pin now). Also the last bit now is a "repeat" indicator.
- HCS200: http://ww1.microchip.com/downloads/en/devicedoc/40138c.pdf
- HCS300: http://ww1.microchip.com/downloads/en/devicedoc/21137g.pdf

With the HCS300 the buttons are simply a bit-field, we might want to output just that, or decode to a single button number 1-14 (0=Standby, 15=Learn). This would give numbers 1,2,3,12,13,14 for the HCS200 (note that there is an error in the HCS200 document: 13 is shown as 14).

It's clear however that he ID and encrypted part are LSB fist, thus only the current "_rev" outputs are true. (with the trailing "0" removed).

zuckschwerdt, 2021-07-03:
A proposed output would be
```
model     : Microchip-HCS200                       id        : 054CA93
Battery   : 1            Button    : 2
Button S3 : 0             Button S0 : 0             Button S1 : 1             Button S2 : 0
Learn mode: 0            Repeat    : 1             encrypted : 3864F950
```
With button number and the S-bits. No special decoding for the HCS200 as we can't known if it's a HCS200 or HCS300.

merbanan, 2021-07-03:
A lot of devices have implemented KeeLoq. We can not support all but the generic protocol should be. So the proposed output looks good.

azriek, 2021-07-11:
Hi

> Can you find out if there is some ID printed on the remote, the PCB, or packaging?
> Is it 607B250 or A4DE060?

i didn't find any id on this remote.

I made a test with the feat_hcs300 branch and result is Ok ( don't bother the function mapping as S bit are Ok and as said,  interpretation depends on device implementation) with this another remote (simple hcs300 to 868 transmitter), as i could directly activate S0, S1, S2 and S3 pin with the blue cable (3V) for the test
[image]

for reference, this is the result

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-11 09:36:04
model     : Microchip-HCS200                       id        : 0BEC235
Battery   : 1            Button    : 1             
Button S3 : 0             Button S0 : 1             Button S1 : 0           Button S2 : 0            
Learn mode: 0             Repeat    : 0             encrypted : 77F3E053
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-11 09:36:14
model     : Microchip-HCS200                       id        : 0BEC235
Battery   : 1            Button    : 2             
Button S3 : 0             Button S0 : 0             Button S1 : 1            Button S2 : 0            
Learn mode: 0             Repeat    : 0             encrypted : 8A1B5193
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-11 09:36:23
model     : Microchip-HCS200                       id        : 0BEC235
Battery   : 1            Button    : 4             
Button S3 : 0             Button S0 : 0             Button S1 : 0            Button S2 : 1            
Learn mode: 0             Repeat    : 0             encrypted : BC2C5507
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-07-11 09:37:04
model     : Microchip-HCS200                       id        : 0BEC235
Battery   : 1            Button    : 8             
Button S3 : 1             Button S0 : 0             Button S1 : 0            Button S2 : 0            
Learn mode: 0             Repeat    : 0             encrypted : 403411FA
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

```

zuckschwerdt, 2021-07-11:
Thanks for confirming the decoding in such detail! If only we could poke all the chips to make them reveal secrets this nicely :)

I'm still leaning towards removing the "Button S" outputs. One can't just look at just one of those (like with the HCS200 before) to determine the state and "Button" has all the state in one number.
If there is no objection I'll update and merge the changed decoder soon.

commit 7e60e4f4 (2021-07-02): Improve Microchip-HCS200 timing (closes #1512)
commit 2f42f154 (2021-07-02): Improve Microchip-HCS200 timing (closes #1512)