merbanan/rtl_433 #1517: Feature Request: Customize tag field name used in -K <tag>

rdica, 2020-10-13:
The field name used in JSON output is `tag` and its value is whatever you gave for -K, ie; -K rtl433 results in `{"tag": "rtl433"}`. Requesting the field name be customizable.
Thanks

zuckschwerdt, 2020-10-13:
Can you give an example where this is useful for you and what other software you use this with?

rdica, 2020-10-14:
I'm ingesting rtl_433 json output into ELK. I'm having to pipe to `jq --unbuffered -c '. + {"env": "rtl433"}'` to add a field called `env` so my logstash tcp listener can properly parse and route data based on `env` field.
The raw json that gets sent to the logstash tcp listener is formatted as such:

`{"time":"1602634188","protocol":82,"model":"Citroen","type":"TPMS","state":"df","id":"c14f4627","flags":0,"repeat":10,"pressure_PSI":15.827,"temperature_F":75.2,"maybe_battery":29,"mic":"CHECKSUM","mod":"FSK","freq1":433.959,"freq2":433.899,"rssi":-8.766,"snr":17.358,"noise":-26.124,"env":"rtl433"}`

This is the commandline I'm using:

`rtl_433 -F json -M protocol -M time:unix -M level -C customary|jq --unbuffered -c '. + {"env": "rtl433"}'|nc str002 5050`

ELK stack already has a field called `tags`, and this could be confused with a field called `tag`, which is what prompted me to not use `-K tagname` and use `jq` to add the field I needed instead.

merbanan, 2020-10-14:
I need this too, I have several devices that sends messages to a mqtt server, I would like to know what devices a message comes from. This feature would fix that.

zuckschwerdt, 2020-10-14:
@rdica but you would still need some pipe or bridge script to get the output to port 5050, right?

@merbanan you need all devices to report to a single shared topic?

zuckschwerdt, 2020-10-14:
@merbanan I get it now: you are using multiple devices for reliability, not on different bands. You want the values at the same topic but also know which devices had reception. `tag` is inconvenient? Which key are you thinking about?

We could just enhance to also support `-K foo=bar`, only a single "tag" though, or do you need multiple "tags"?

merbanan, 2020-10-14:
Correct, and I need a custom key,value pair. I will use the mac address and then do a lookup in the backend.

closed 2020-10-14 by zuckschwerdt

merbanan, 2020-10-14:
Jolly good.

commit dd8da88c (2020-10-14): Add named output tag option (closes #1517)