merbanan/rtl_433 #1521: Long listening time for the 433 button switches

AlexMKX, 2020-10-17:
I'm using the rtl_433 to operate the light using button switches.
When I'm pressing the button shortly, I get the 1-2 events in the rtl_433 log.
When I'm pressing the button for a long time, I get the bunch of events only after the button is released.

Such behaviour is problematic for the radio switches. User expect switched to be triggered after button press, not after the release.

Suggested : 
### Use the time-interval for the events : 
time-interval is 100ms. If button pressed for a 1 second, the first event should be fired no later than 100ms after first event is received. Later events should be fired in 100ms interval as they are received.

### Release event and release interval
When the button not transmitting anything during the release interval after the transmission, button is considered "released"
After such consideration the release event should be fired : same event as a push, but with field indicating how long the button has been pressed.

zuckschwerdt, 2020-10-17:
We are not collecting (coalescing) any events. What you are seeing are likely the last few events only and the continued transmission before that isn't picked up.
Record the whole transmission (e.g. `-w FILE.cu8 -T 60`) and then inspect it.

closed 2020-10-17 by zuckschwerdt

merbanan, 2020-10-17:
This is a hard problem as rtl_433 segments the signal after there is no rf-energy detected for a while and then decodes it. This suggestion is not really compatible with the code logic. As rtl_433 only relays what it can decode "on air". So any implicit events you have to create yourself in the consumer.

zuckschwerdt, 2020-10-17:
I though about the framing too, the one source with a continous signal I have are smoke detectors and framing isn't a problem there -- and those are occuping the band really hard. Perhaps we need to check and document in easy numbers what the framing requirements are.

merbanan, 2020-10-17:
I have a remote that transmits as long as I press the button. Those type of devices does not conform to the ISM band rules in some regions of the world. At least they are hard to handle but if you have a shift register receiver you get the redundancy you need. Anyway the rtl_433 design can not handle this well. It might be possible to release a signal segment after 100-200ms and try to decode it. But it opens up the problem of handling the rest of the signal. To me the only real solution is to create a matched receiver implementing all the logic one want with regards to pressevents and real time behaviour.