merbanan/rtl_433 #1523: Missing fields in csv output for Acurite 6045M

evoluder, 2020-10-18:
I am using this command to run rtl_433 :
$ rtl_433 -C customary -M newmodel -F csv:file.csv -M time:unix -M level

I am using version :

rtl_433 version 19.08-3-g04e818a branch master at 201909021329 inputs file rtl_tcp RTL-SDR

I'm trying to gather signals from several Acurite devices, one of these devices (6045M) has two fields "strike_count" and "storm_distance" which show up fine in kv mode. However, csv mode doesn't seem to have them. What am I doing wrong? Is this a bug?

zuckschwerdt, 2020-10-18:
Thanks, the fields are missing here: https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1337
I'll fix it soon.

zuckschwerdt, 2020-10-18:
Fixed with da0db3a.

closed 2020-10-18 by zuckschwerdt