merbanan/rtl_433 #1524: New F007TH sensors not being detected

abstractvector, 2020-10-19:
I've been using rtl-433 for a little while to very successfully intercept signals from my F007TH weather sensors. I have 3 sensors but just bought four more - I actually bought them in a bundle with a new display since it only cost $3 more to get the display.

I've just put batteries in the new sensor but rtl-433 isn't detecting its pulses. I can see the red LED on the front of the sensor flash (which it does when it transmits), but nothing appears in rtl-433.

I have tried configuring the DIP switches to make it transmit on channel 1 (same as one of my existing sensors) and channel 4 (a channel I'm not currently using) but no difference.

Any pointers on how to debug?

zuckschwerdt, 2020-10-19:
Make sure the signal is within the band, use CubicSDR, Gqrx or maybe try -A or record for some time (-T 60). Proceed to run -A to make sure something is detected.

abstractvector, 2020-10-19:
I'm no expert with CubicSDR, but when I insert the batteries I do see a signal at 433.92MHz. Actually, two signals - one long and then one short, corresponding to a long then short flash of the red LED on the sensor.

Running `rtl_433 -A`, I see the following output when the red LEDs flash to say it's transmitting:

```
Detected OOK package	2020-10-19 16:58:03
Analyzing pulses...
Total count:    1,  width: 2096.50 ms		(524126 S)
Pulse width distribution:
 [ 0] count:    1,  width: 2096504 us [2096504;2096504]	(524126 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   8120,     60
RSSI: -3.0 dB SNR: 21.3 dB Noise: -24.4 dB
Frequency offsets [F1, F2]:     -63,      0	(-0.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package	2020-10-19 16:58:05
Analyzing pulses...
Total count:    1,  width: 191.67 ms		(47917 S)
Pulse width distribution:
 [ 0] count:    1,  width: 191668 us [191668;191668]	(47917 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   6172,      3
RSSI: -4.2 dB SNR: 33.1 dB Noise: -37.4 dB
Frequency offsets [F1, F2]:     -61,      0	(-0.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
```

abstractvector, 2020-10-19:
I've just tried it again with `rtl_433 -A -T 60` and see the following output when I first insert the batteries:

```
% rtl_433 -A -T 60
rtl_433 version 20.02-165-g105342a branch master at 202009221012 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/matt/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 138 out of 167 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-167 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Detected OOK package	2020-10-19 17:07:40
Analyzing pulses...
Total count:    1,  width: 2095.94 ms		(523985 S)
Pulse width distribution:
 [ 0] count:    1,  width: 2095940 us [2095940;2095940]	(523985 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   7615,     42
RSSI: -3.3 dB SNR: 22.6 dB Noise: -25.9 dB
Frequency offsets [F1, F2]:     -61,      0	(-0.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package	2020-10-19 17:07:43
Analyzing pulses...
Total count:    6,  width: 191.11 ms		(47777 S)
Pulse width distribution:
 [ 0] count:    1,  width: 14464 us [14464;14464]	(3616 S)
 [ 1] count:    2,  width: 19308 us [19260;19360]	(4827 S)
 [ 2] count:    1,  width: 85744 us [85744;85744]	(21436 S)
 [ 3] count:    1,  width: 9672 us [9672;9672]	(2418 S)
 [ 4] count:    1,  width: 41656 us [41656;41656]	(10414 S)
Gap width distribution:
 [ 0] count:    2,  width:  252 us [228;276]	(  63 S)
 [ 1] count:    2,  width:  176 us [164;192]	(  44 S)
 [ 2] count:    1,  width:   92 us [92;92]	(  23 S)
Pulse period distribution:
 [ 0] count:    1,  width: 14692 us [14692;14692]	(3673 S)
 [ 1] count:    2,  width: 19528 us [19524;19536]	(4882 S)
 [ 2] count:    1,  width: 85936 us [85936;85936]	(21484 S)
 [ 3] count:    1,  width: 9764 us [9764;9764]	(2441 S)
Level estimates [high, low]:   7910,      1
RSSI: -3.2 dB SNR: 39.0 dB Noise: -42.1 dB
Frequency offsets [F1, F2]:     -62,      0	(-0.2 kHz, +0.0 kHz)
Guessing modulation: No clue...
```

zuckschwerdt, 2020-10-19:
There might be problems when you try to pick up a signal at exactly the center frequency. See if `-f 433.85M` works better?

abstractvector, 2020-10-19:
Yup, changing the frequency got it!

```
% rtl_433 -A -T 60 -f 433.85M
rtl_433 version 20.02-165-g105342a branch master at 202009221012 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/matt/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 138 out of 167 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-167 ]
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.850MHz.
Detected OOK package	2020-10-19 17:13:23
Analyzing pulses...
Total count:    1,  width: 2095.45 ms		(523862 S)
Pulse width distribution:
 [ 0] count:    1,  width: 2095448 us [2095448;2095448]	(523862 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:  15762,     47
RSSI: -0.2 dB SNR: 25.3 dB Noise: -25.4 dB
Frequency offsets [F1, F2]:   16022,      0	(+61.1 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package	2020-10-19 17:13:26
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-19 17:13:26
model     : Ambientweather-F007TH                  House Code: 164
Channel   : 4            Battery   : 1             Temperature: 82.4 F       Humidity  : 44 %          Integrity : CRC
Analyzing pulses...
Total count:  148,  width: 190.08 ms		(47520 S)
Pulse width distribution:
 [ 0] count:    1,  width:  112 us [112;112]	(  28 S)
 [ 1] count:  100,  width:  528 us [504;576]	( 132 S)
 [ 2] count:   47,  width: 1012 us [1004;1040]	( 253 S)
Gap width distribution:
 [ 0] count:   99,  width:  444 us [420;472]	( 111 S)
 [ 1] count:   48,  width:  936 us [912;952]	( 234 S)
Pulse period distribution:
 [ 0] count:    1,  width:  544 us [544;544]	( 136 S)
 [ 1] count:   81,  width:  976 us [968;988]	( 244 S)
 [ 2] count:   35,  width: 1464 us [1444;1480]	( 366 S)
 [ 3] count:   30,  width: 1952 us [1948;1960]	( 488 S)
Level estimates [high, low]:  15906,      2
RSSI: -0.1 dB SNR: 39.0 dB Noise: -39.1 dB
Frequency offsets [F1, F2]:   16404,      0	(+62.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 528, long_width: 1012, reset_limit: 956, sync_width: 112
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=528,l=1012,r=956,g=0,t=0,y=112'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {147} ff cc ae ad e7 67 7f e6 57 56 f3 b3 bf f3 2b ab 79 d9 e0 

Detected OOK package	2020-10-19 17:13:26
Analyzing pulses...
Total count:    1,  width: 20.09 ms		( 5023 S)
Pulse width distribution:
 [ 0] count:    1,  width: 20092 us [20092;20092]	(5023 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1000,      4
RSSI: -12.1 dB SNR: 24.0 dB Noise: -36.1 dB
Frequency offsets [F1, F2]:     -58,      0	(-0.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
```

abstractvector, 2020-10-19:
I've just run that through and it appears to be correctly identifying all my sensors, including the original ones that didn't need the frequency adjusting.

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-19 17:16:15
model     : Ambientweather-F007TH                  House Code: 87
Channel   : 1            Battery   : 1             Temperature: 74.7 F       Humidity  : 49 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-19 17:16:16
model     : Ambientweather-F007TH                  House Code: 217
Channel   : 2            Battery   : 1             Temperature: 60.2 F       Humidity  : 66 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-19 17:16:21
model     : Ambientweather-F007TH                  House Code: 189
Channel   : 3            Battery   : 1             Temperature: 56.6 F       Humidity  : 76 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-19 17:16:29
model     : Ambientweather-F007TH                  House Code: 164
Channel   : 4            Battery   : 1             Temperature: 81.9 F       Humidity  : 43 %          Integrity : CRC
```

Thank you for your prompt response! Out of curiosity, could you explain to me a little about why this adjustment was needed? I'm new to SDR but trying to learn.

zuckschwerdt, 2020-10-19:
At exactly the center frequency there is no "frequency" part any more (0 Hz), it's called DC (direct current). So the signal is not picked up any more. Some Receivers even block that "center" range.

closed 2020-10-19 by zuckschwerdt