merbanan/rtl_433 #1525: ECODHOME smart Socket

vannetta, 2020-10-20:
I've a EcpDhome Smart Socket which transmit at 433,92 Mhz. I don't know anything else about il.
I installed rtl4332mqtt e run it with -G 4 and -S all:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Starting RTL_433 with parameters:
MQTT Host = xxxx
MQTT User = xxxx
MQTT Password = xxxx
MQTT Topic = homeassistant/sensor/ecodhome
RTL_433 Protocol = 36
RTL_433 Frequency = 433920000
RTL_433 Gain = 60
RTL_433 Frequency Offset = 350
rtl_433 version 20.02-190-gf648297 branch master at 202010181332 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
	Use -G for testing only. Enable protocols with -R if you really need them.
Registered 162 out of 168 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-168 ]
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Frequency correction set to 350 ppm.
Tuned to 433.920MHz.
*** Saving signal to file g001_433.92M_250k.cu8 (51737 samples, 131072 bytes)
*** Saving signal to file g002_433.92M_250k.cu8 (35280 samples, 131072 bytes)
*** Saving signal to file g003_433.92M_250k.cu8 (35281 samples, 131072 bytes)
*** Saving signal to file g004_433.92M_250k.cu8 (35281 samples, 131072 bytes)
*** Saving signal to file g005_433.92M_250k.cu8 (67113 samples, 262144 bytes)
*** Saving signal to file g006_433.92M_250k.cu8 (35279 samples, 131072 bytes)
*** Saving signal to file g007_433.92M_250k.cu8 (131991 samples, 393216 bytes)
*** Saving signal to file g008_433.92M_250k.cu8 (131413 samples, 393216 bytes)
*** Saving signal to file g009_433.92M_250k.cu8 (51752 samples, 131072 bytes)
*** Saving signal to file g010_433.92M_250k.cu8 (80280 samples, 262144 bytes)
*** Saving signal to file g011_433.92M_250k.cu8 (35282 samples, 131072 bytes)
*** Saving signal to file g012_433.92M_250k.cu8 (385594 samples, 786432 bytes)
Async read stalled, exiting!
[cmd] /bin/ash exited 0
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

I can't find the .cu8 files  to understand if it finds something (I already knows that on protocol 12 there is a Oregon wheteher station of my neighbors.
Thanks a lot!

zuckschwerdt, 2020-10-20:
The [EcoDHOME Smart Socket](https://www.ecodhome.com/en/products/mcee-energy-monitors/mcee-solar/smart-socket.html) seems to only receive, not transmit. You need to capture samples (`-S all`) from something like the Smart Switch.
The `.cu8` will be in the directory where you started rtl_433. Drop them on https://triq.org/iqs/ to see if there is some signal. Then use `-A` on a good sample to inspect the protocol.

vannetta, 2020-10-20:
I installed the addon on Hassio, but I can't find the directory. Can you give me the path?
The socket trasmits to the [MCEE SOLAR MONITOR](https://www.ecodhome.com/en/products/mcee-energy-monitors/mcee-solar.html) that receives data from the socket and from two trasmitters (Photovoltaic system and power)

zuckschwerdt, 2020-10-20:
The `rtl4332mqtt` script is probably not the right way to use rtl_433. There is now a built-in MQTT output in rtl_433.
You need to open a terminal and run `rtl_433` with some parameters.
Your missing .cu8 files will likely be at the users home path running the script. Try `find / -name \*.cu8 -ls`

vannetta, 2020-10-20:
Ok, I'll try on a Raspberry Pi with Raspbian. I'll send you the results...

`find / -name \*.cu8 -ls`  doesn't find anything...

zuckschwerdt, 2020-10-20:
Where were you running this before? Did you run `find` on the same machine? Is this perhaps with Docker and your files are inside some container?

vannetta, 2020-10-20:
I was running it on Hassio on a dedicated Raspberry. I also entered inside Docker container, but I cannot find it.
Now I would like to install rtl_433 on Raspbian in another Raspberry

vannetta, 2020-10-20:
These are the cu8 files from the Raspberry (raspbian)
[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/5410310/g001_433.92M_250k.zip)
In the log appeared the two Oregon station of my neighbor

zuckschwerdt, 2020-10-20:
There is a transmission only in these:
- g004_433.92M_250k.cu8 : Oregon-THR228N
- g016_433.92M_250k.cu8 : OOK Manchester data
- g022_433.92M_250k.cu8 : Maybe FSK
- g028_433.92M_250k.cu8 : Oregon-THR228N
- g032_433.92M_250k.cu8 : Maybe OOK
- g034_433.92M_250k.cu8 : Maybe FSK

Look at the "shapes" in https://triq.org/iqs/ compare with what you get on button presses or such and isolate the EcoDHOME. Four different signals here don't help.
Otherwise remove the antenna from the RTL-SDR, put the sender at about 10 cm, then you receive only that sender.

vannetta, 2020-10-21:
I removed the antenna and put the Raspberry very near to the socket ( 5 cm).
All the files seem recorder the same; I upload three of them:

[g118_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/5414279/g118_433.92M_250k.zip)

 I could also try with the PV trasmitter, but it's a little complicated to put the antenna very near of it.

zuckschwerdt, 2020-10-21:
That looks good. All three files show a [pattern like this](https://triq.org/pdv/#AAB0360801000000FC01F402FC06F808F803FC007C82919191A1919391919191919191919191919391A1B19193919293A39191B3A19192B1B555+AAB0190801000000FC01F402FC06F808F803FC007C9191A291A192C555+AAB0160801000000FC01F402FC06F808F803FC007C9192A19655+AAB0140801000000FC01F402FC06F808F803FC007C929755).

You can use `-X 'n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=3000'` to get the data.
It's `{155}2ad455555516ea2918ae353b802b2d3f8029a12` consistently. You now need to find out what that data means :)

Perhaps catch that code from different Sockets, or reset the batteries or reset pairing mode. If you get different codes post them here and we can figure out if there is a checksum.

vannetta, 2020-10-21:
Here we are:

```
pi@raspberrypi:~/rtl_433/build $ rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=3000"                                                                                                                          rtl_433 version 20.02-190-gf648297 branch master at 202010181332 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 140 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:21:13
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f8029a12
codes     : {155}2ad455555516ea2918ae353b802b2d3f8029a12
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:21:51
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f8029a12
codes     : {155}2ad455555516ea2918ae353b802b2d3f8029a12
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:22:29
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f0029a0a
codes     : {155}2ad455555516ea2918ae353b802b2d3f0029a0a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:22:34
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5828029d22
codes     : {155}2ad455555516ea2918ae353b8020a5828029d22
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:23:07
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 156          data      : 2ad455555516ea2918ae353b802b2d3f0029a0a
codes     : {156}2ad455555516ea2918ae353b802b2d3f0029a0a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:23:45
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f0029a0a
codes     : {155}2ad455555516ea2918ae353b802b2d3f0029a0a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:24:22
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f0029a0a
codes     : {155}2ad455555516ea2918ae353b802b2d3f0029a0a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:25:00
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3e8029a02
codes     : {155}2ad455555516ea2918ae353b802b2d3e8029a02
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:25:38
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f0029a0a
codes     : {155}2ad455555516ea2918ae353b802b2d3f0029a0a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 11:26:15
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
[... cut ...]
(From the manual of the socket) To view the data of each sensor socket, please just press the socket’s corresponding numbered button on the controller under normal working condition. For example, by pressing button 1 on the controller, you will
see the data of socket 1. The first line is power of socket 1 under consumption. The second
line is the accumulated electricity consumption after the socket is paired with the controller. The 'k' means that is 10 times tha value.

So data should be: 5 W, 10.69 kWh. (Don't worry about the date and time string; it doesn't work)

zuckschwerdt, 2020-10-21:
This gives these unique codes:
```
{155}2ad455555516ea2918ae353b802 b2d3 f802 9a12
{155}2ad455555516ea2918ae353b802 b2d3 f002 9a0a
{155}2ad455555516ea2918ae353b802 0a58 2802 9d22
{155}2ad455555516ea2918ae353b802 b2d3 e802 9a02
```
You likely need to trim off 3 leading bits to align the data, see [this BitBench](https://triq.net/bitbench?c=%7B155%7D2ad455555516ea2918ae353b802%20b2d3%20f802%209a12&c=%7B155%7D2ad455555516ea2918ae353b802%20b2d3%20f002%209a0a&c=%7B155%7D2ad455555516ea2918ae353b802%200a58%202802%209d22&c=%7B155%7D2ad455555516ea2918ae353b802%20b2d3%20e802%209a02&f=8h%20&s=-3&cf=ADD&cl=144&cw=8).
The last byte could be a add (with carry) checksum. Simple adding seems to consistently be 3 bigger.
But that is just a guess much more codes are needed. Table about 20 different codes with the numbers from the display.

vannetta, 2020-10-21:
This is the record of 20 minutes. the monitor shows sometimes 4 W, sometimes 5 W. The total amount is always 10,69 kWh because the power is too low to change it.
There are almost 10 codes.

```pi@raspberrypi:~ $ rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=3000"
rtl_433 version 20.02-190-gf648297 branch master at 202010181332 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 140 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:28:25
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 156          data      : 2ad455555516ea2918ae353b8020a5820029d1a
codes     : {156}2ad455555516ea2918ae353b8020a5820029d1a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:28:44
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 134          data      : 2ad455555516ea2918ae353b802b2d4000
codes     : {134}2ad455555516ea2918ae353b802b2d4000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:28:44
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 17           data      : a6868
codes     : {17}a6868
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:28:46
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 156          data      : 2ad455555516ea2918ae353b8020a5820029d1a
codes     : {156}2ad455555516ea2918ae353b8020a5820029d1a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:29:22
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d408029a22
codes     : {155}2ad455555516ea2918ae353b802b2d408029a22
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:29:30
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 156          data      : 156a2aaaaa8b75148c571a9dc01052c10014e8d
codes     : {156}156a2aaaaa8b75148c571a9dc01052c10014e8d
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:29:44
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5828029d22
codes     : {155}2ad455555516ea2918ae353b8020a5828029d22
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:29:59
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 134          data      : 2ad455555516ea2918ae353b802b2d4000
codes     : {134}2ad455555516ea2918ae353b802b2d4000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:29:59
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 17           data      : a6868
codes     : {17}a6868
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 18:30:14
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5820029d1a
codes     : {155}2ad455555516ea2918ae353b8020a5820029d1a
[... cut ...]
time      : 2020-10-21 18:49:29
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d410029a2a
codes     : {155}2ad455555516ea2918ae353b802b2d410029a2a
^Z
[1]+  Fermato                 rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=3000"
```
I'm trying now to do also something different, I'm going to register with a more variable power and with also another wifi socket that register data to Home Assistant.

zuckschwerdt, 2020-10-21:
The shorter codes maybe bad reception but try to raise the `reset` value: `-X 'n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000'`
These are the unique codes, I'll try and see if I can spot the checksum.
```
{156}2ad455555516ea2918ae353b8020a5820029d1a
{155}2ad455555516ea2918ae353b802b2d408029a22
{155}2ad455555516ea2918ae353b8020a5828029d22
{155}2ad455555516ea2918ae353b8020a5820029d1a
{155}2ad455555516ea2918ae353b802b2d3f8029a12
{155}2ad455555516ea2918ae353b802b2d410029a2a
{156}2ad455555516ea2918ae353b802b2d408029a22
```
(the codes starting `156a` are good but duplicates.)

vannetta, 2020-10-22:
Hi,
I've tried again with your new code:

``````
pi@raspberrypi:~/rtl_433/build` $ rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000"
rtl_433 version 20.02-190-gf648297 branch master at 202010181332 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 140 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:04
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5830129d3a
codes     : {155}2ad455555516ea2918ae353b8020a5830129d3a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:13
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5880129d8a
codes     : {155}2ad455555516ea2918ae353b8020a5880129d8a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:20
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a58a8129db2
codes     : {155}2ad455555516ea2918ae353b8020a58a8129db2
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:32
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 156          data      : 156a2aaaaa8b75148c571a9dc015969fc014d09
codes     : {156}156a2aaaaa8b75148c571a9dc015969fc014d09
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:37
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a58a0129daa
codes     : {155}2ad455555516ea2918ae353b8020a58a0129daa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:43
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a58a0129daa
codes     : {155}2ad455555516ea2918ae353b8020a58a0129daa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:47:54
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a58a0129daa
codes     : {155}2ad455555516ea2918ae353b8020a58a0129daa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:48:10
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f0029a0a
codes     : {155}2ad455555516ea2918ae353b802b2d3f0029a0a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:48:18
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5880129d8a
codes     : {155}2ad455555516ea2918ae353b8020a5880129d8a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 16:48:28
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5888129d92
codes     : {155}2ad455555516ea2918ae353b8020a5888129d92
[... 420 lines cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 17:08:44
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b8020a5850029d4a
codes     : {155}2ad455555516ea2918ae353b8020a5850029d4a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 17:08:51
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 155          data      : 2ad455555516ea2918ae353b802b2d3f8029a12
codes     : {155}2ad455555516ea2918ae353b802b2d3f8029a12
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-22 17:08:53
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
[... cut ...] a      : 2ad455555516ea2918ae353b8020a5850029d4a
codes     : {155}2ad455555516ea2918ae353b8020a5850029d4a
^Z
[1]+  Fermato                 rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000"
pi@raspberrypi:~/rtl_433/build
``` `$
```
I'm attaching also a .xls file with data from the mcee monitor: there is the data (W) for every time (17:08:53 etc)
[mcee_monitor.xlsx](https://github.com/merbanan/rtl_433/files/5423771/mcee_monitor.xlsx)

Please let me know if I can do something else!
Thanks a lot!

zuckschwerdt, 2020-10-22:
I've collected the data to [this BitBench](https://triq.net/bitbench?c=%7B155%7D2ad455555516ea2918ae353b8020a5830129d3a%20%5B16%3A47%3A04%20528W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5880129d8a%20%5B16%3A47%3A13%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A47%3A20%20533W%5D&c=%7B156%7D156a2aaaaa8b75148c571a9dc015969fc014d09%20%5B16%3A47%3A32%20532W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a0129daa%20%5B16%3A47%3A37%20532W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a0129daa%20%5B16%3A47%3A43%20532W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a0129daa%20%5B16%3A47%3A54%20532W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f0029a0a%20%5B16%3A48%3A10%20528W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5880129d8a%20%5B16%3A48%3A18%20529W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5888129d92%20%5B16%3A48%3A28%20529W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5880129d8a%20%5B16%3A48%3A41%20528W%5D&c=%7B156%7D156a2aaaaa8b75148c571a9dc015969fc014d09%20%5B16%3A48%3A48%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A48%3A52%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5880129d8a%20%5B16%3A49%3A02%20528W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f0029a0a%20%5B16%3A49%3A25%20527W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5878129d82%20%5B16%3A49%3A30%20527W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A50%3A03%20527W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5888129d92%20%5B16%3A50%3A36%20529W%5D&c=%7B156%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A50%3A40%20531W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5898129da2%20%5B16%3A50%3A47%20531W%5D&c=%7B156%7D156a2aaaaa8b75148c571a9dc015969fc014d09%20%5B16%3A51%3A18%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A51%3A22%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A51%3A28%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58c8129dd2%20%5B16%3A51%3A40%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58c8129dd2%20%5B16%3A51%3A46%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A51%3A55%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58c8129dd2%20%5B16%3A51%3A58%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58c8129dd2%20%5B16%3A52%3A04%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A52%3A32%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58c8129dd2%20%5B16%3A52%3A42%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58c8129dd2%20%5B16%3A52%3A48%20537W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A53%3A10%20529W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5888129d92%20%5B16%3A53%3A18%20529W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A53%3A23%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5880129d8a%20%5B16%3A53%3A33%20528W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5878129d82%20%5B16%3A53%3A38%20527W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a0129daa%20%5B16%3A53%3A47%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A53%3A47%20532W%5D&c=%7B156%7D2ad455555516ea2918ae353b8020a58a0129daa%20%5B16%3A54%3A00%20532W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58b0129dba%20%5B16%3A54%3A12%20534W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A54%3A26%20535W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58b8129dc2%20%5B16%3A54%3A29%20535W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A54%3A46%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f0029a0a%20%5B16%3A55%3A04%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a58a8129db2%20%5B16%3A55%3A11%20533W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5800029cfa%20%5B16%3A55%3A28%200W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d400029a1a%20%5B16%3A55%3A43%200W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5800029cfa%20%5B16%3A56%3A06%200W%5D&c=%7B156%7D156a2aaaaa8b75148c571a9dc015969fc014d09%20%5B16%3A56%3A21%200W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d400029a1a%20%5B16%3A56%3A59%200W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5800029cfa%20%5B16%3A57%3A33%200W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A57%3A36%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5850029d4a%20%5B16%3A57%3A54%2010W%5D&c=%7B156%7D2ad455555516ea2918ae353b8020a5858029d52%20%5B16%3A58%3A10%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A58%3A13%2011W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5850029d4a%20%5B16%3A58%3A30%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A58%3A51%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5850029d4a%20%5B16%3A58%3A53%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5850029d4a%20%5B16%3A59%3A20%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B16%3A59%3A28%2010W%5D&c=%7B156%7D2ad455555516ea2918ae353b8020a5850029d4a%20%5B16%3A59%3A41%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B17%3A00%3A06%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B17%3A00%3A45%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B17%3A01%3A22%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B17%3A02%3A00%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B17%3A02%3A38%2011W%5D&c=%7B155%7D2ad455555516ea2918ae353b802b2d3f8029a12%20%5B17%3A03%3A16%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5858029d52%20%5B17%3A03%3A21%2010W%5D&c=%7B155%7D2ad455555516ea2918ae353b8020a5850029d4a%20%5B17%3A03%3A43%2010W%5D&c=%7B155%7D2ad455555516 [... cut ...]
On `414b` the two byte (little endian) power value follows. For the other type it should be the kWh probably.
Lastly there is a fixed `53` (status? stop?) and a checksum byte.

Interesting to note that `414b`, `565a`, and `53` are "AK", "VZ", and "S" which might not be a coincidence.

If you use `-X 'n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000,preamble=2dd4' you should be able to read the codes following the outlined format:)

The checksum works nicely: add all bytes after the sync word (mod 0xff).

We need to confirm if that's really a serial number, then I'll add a decoder.

vannetta, 2020-10-22:
This is the picture of the device:

Inviato da iPhone

zuckschwerdt, 2020-10-22:
The picture upload didn't work.

vannetta, 2020-10-22:
[image]

vannetta, 2020-10-23:
```
`pi@raspberrypi:~ $ rtl_433 -X "n=EcoDhome,m=FSK_PCM,s=250,l=250,r=6000,preamble=                                                                                                                               2dd4"
rtl_433 version 20.02-190-gf648297 branch master at 202010181332 inputs file rtl                                                                                                                               _tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 140 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23                                                                                                                                25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-1                                                                                                                               61 163-168 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:21:38
model     : EcoDhome     count     : 1             num_rows  : 1
rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:22:16
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:22:54
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:23:31
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:24:09
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:24:46
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:25:24
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:26:02
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:26:40
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:27:18
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 07:32:21
model     : EcoDhome     count     : 1             num_rows  : 1             rows      :
len       : 98           data      : 52315c6a7700565a7f0053424
codes     : {98}52315c6a7700565a7f0053424
^Z
[1]+  Fermato                 rtl_433 -X "n=EcoDhome,m=FSK_PCM,s=250,l=250,r=6000,preamble=2dd4"
`
```
This is the record with -X 'n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000,preamble=2dd4'
The power is the same than before (almost 530 and 0 at the end)

vannetta, 2020-10-23:
```

pi@raspberrypi:~/rtl_433/build $ rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=                         6000"
rtl_433 version 20.02-190-gf648297 branch master at 202010181332 inputs file rtl                         _tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 140 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23                          25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-1                         61 163-168 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:22:16
model     : EcoDHOME     count     : 1             num_rows  : 1
rows      :
len       : 144          data      : aaaaaa2dd48c74d4b93eb322384451550000
codes     : {144}aaaaaa2dd48c74d4b93eb322384451550000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:22:25
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : d5555516ea463b6a5cb8d9c419b1832a80000
codes     : {145}d5555516ea463b6a5cb8d9c419b1832a80000
13^[[C22_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:22:34
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 144          data      : aaaaaa2dd48c74d4b93eb335394465550000
codes     : {144}aaaaaa2dd48c74d4b93eb335394465550000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:22:43
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 144          data      : aaaaaa2dd48c74d4b93eb335394465550000
codes     : {144}aaaaaa2dd48c74d4b93eb335394465550000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:22:52
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 55555516ea463a6a5c9f599a9ca232aa80000
codes     : {145}55555516ea463a6a5c9f599a9ca232aa80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:23:01
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 55555516ea463a6a5c9f599a9ca232aa80000
codes     : {145}55555516ea463a6a5c9f599a9ca232aa80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:23:10
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : d5555516ea463a6a5c9f599a9ca232aa80000
codes     : {145}d5555516ea463a6a5c9f599a9ca232aa80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:23:18
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 55555516ea463a6a5c9f599a9ca232aa80000
codes     : {145}55555516ea463a6a5c9f599a9ca232aa80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:23:27
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : d5555516ea463a6a5c9f598e9c22262a80000
codes     : {145}d5555516ea463a6a5c9f598e9c22262a80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:23:36
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 55555516ea463b6a5cb8d9c419b1832a80000
codes     : {145}55555516ea463b6a5cb8d9c419b1832a80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[... 335 lines cut ...]
time      : 2020-10-23 13:33:39
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 143          data      : aaaaaa2dd48c74d4b93eb328384457550000
codes     : {143}aaaaaa2dd48c74d4b93eb328384457550000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:33:48
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 145          data      : 55555516ea463a6a5c9f59941c222baa80000
codes     : {145}55555516ea463a6a5c9f59941c222baa80000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-23 13:33:57
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
[... cut ...]
model     : EcoDHOME     count     : 1             num_rows  : 1             rows      :
len       : 144          data      : 55555516ea463a6a5c9f59941c222baa8000
codes     : {144}55555516ea463a6a5c9f59941c222baa8000
^Z
[1]+  Fermato                 rtl_433 -X "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000"
```
This is the PV trasmitter (P/N 01333-5847-00); I've done with the code: "n=EcoDHOME,m=FSK_PCM,s=250,l=250,r=6000"
There is the file attached with the data from monitor
[mcee_monitor.xlsx](https://github.com/merbanan/rtl_433/files/5429185/mcee_monitor.xlsx)

zuckschwerdt, 2020-10-23:
Mostly works out like in [this BitBench](https://triq.net/bitbench?c=%7B144%7Daaaaaa2dd48c74d4b93eb322384451550000%5B13%3A22%3A16%201519%20W%5D&c=%7B145%7Dd5555516ea463b6a5cb8d9c419b1832a80000%5B13%3A22%3A25%201538%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb335394465550000%5B13%3A22%3A34%201538%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb335394465550000%5B13%3A22%3A43%201538%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599a9ca232aa80000%5B13%3A22%3A52%201538%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599a9ca232aa80000%5B13%3A23%3A01%201538%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f599a9ca232aa80000%5B13%3A23%3A10%201538%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599a9ca232aa80000%5B13%3A23%3A18%5D&c=%7B145%7Dd5555516ea463a6a5c9f598e9c22262a80000%5B13%3A23%3A27%201514%20W%5D&c=%7B145%7D55555516ea463b6a5cb8d9c419b1832a80000%5B13%3A23%3A36%201514%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598e9c22262a80000%5B13%3A23%3A45%201514%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598e9c22262a80000%5B13%3A23%3A54%201514%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb31d38444c550000%5B13%3A24%3A03%201514%20W%5D&c=%7B144%7D55555516ea463a6a5c9f598e9c22262a8000%5B13%3A24%3A12%201514%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f59821c2219aa80000%5B13%3A24%3A20%201489%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59821c2219aa80000%5B13%3A24%3A29%201489%20W%5D&c=%7B143%7Daaaaaa2dd48c74d4b93eb3a33844d2550000%5B13%3A24%3A38%201392%20W%5D&c=%7B145%7D55555516ea463b6a5cb8d9c419b1832a80000%5B13%3A24%3A47%201430%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb3c93844f8550000%5B13%3A24%3A56%201430%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb311384440550000%5B13%3A25%3A05%201502%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59889c22202a80000%5B13%3A25%3A14%201502%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb31d38444c550000%5B13%3A25%3A23%201514%20W%5D&c=%7B143%7Daaaaaa2dd48c74d4b93eb33e39446e550000%5B13%3A25%3A31%201547%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599f1ca2372a80000%5B13%3A25%3A40%201547%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599f1ca2372a80000%5B13%3A25%3A49%201547%20W%5D&c=%7B145%7D55555516ea463b6a5cb8d9c419b1832a80000%5B13%3A25%3A58%201547%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599f1ca2372a80000%5B13%3A26%3A07%201547%20W%5D&c=%7B144%7D55555516ea463a6a5c9f599f1ca2372a8000%5B13%3A26%3A16%201547%20W%5D&c=%7B143%7D2aaaaa2dd48c74d4b93eb33e39446e550000%5B13%3A26%3A25%201547%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb33e39446e550000%5B13%3A26%3A33%201547%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599f1ca2372a80000%5B13%3A26%3A42%201547%20W%5D&c=%7B145%7D55555516ea463a6a5c9f599f1ca2372a80000%5B13%3A26%3A51%201547%20W%5D&c=%7B144%7D2aaaaa2dd48c74d4b93eb34c39447c550000%5B13%3A27%3A00%201561%20W%5D&c=%7B145%7D55555516ea463b6a5cb8d9c419b1832a80000%5B13%3A27%3A09%201561%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f59a61ca23e2a80000%5B13%3A27%3A18%201561%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f59a61ca23e2a80000%5B13%3A27%3A27%201561%20W%5D&c=%7B144%7D55555516ea463a6a5c9f59879c221f2a8000%5B13%3A27%3A36%201500%20W%5D&c=%7B144%7D2aaaaa2dd48c74d4b93eb314384443550000%5B13%3A27%3A44%201505%20W%5D&c=%7B143%7Daaaaaa2dd48c74d4b93eb31d38444c550000%5B13%3A27%3A53%201514%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598e9c22262a80000%5B13%3A28%3A02%201514%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598e9c22262a80000%5B13%3A28%3A11%201514%20W%5D&c=%7B144%7Daaaaaa2dd48c76d4b971b388336306550000%5B13%3A28%3A20%201514%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb31d38444c550000%5B13%3A28%3A29%201514%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598e9c22262a80000%5B13%3A28%3A38%201514%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb3ed38441c550000%5B13%3A28%3A46%201466%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59f69c220e2a80000%5B13%3A28%3A55%201466%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59dc9c22742a80000%5B13%3A29%3A04%201414%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59f69c220e2a80000%5B13%3A29%3A13%201466%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f59f69c220e2a80000%5B13%3A29%3A22%201466%20W%5D&c=%7B145%7Dd5555516ea463b6a5cb8d9c419b1832a80000%5B13%3A29%3A31%201466%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59f69c220e2a80000%5B13%3A29%3A40%201466%20W%5D&c=%7B144%7D55555516ea463a6a5c9f59f81c220faa8000%5B13%3A29%3A48%201491%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb306384435550000%5B13%3A29%3A57%201491%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59831c221aaa80000%5B13%3A30%3A06%201491%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59e39c227b2a80000%5B13%3A30%3A15%201428%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f59f69c220e2a80000%5B13%3A30%3A24%201466%20W%5D&c=%7B144%7D55555516ea463a6a5c9f59849c221c2a8000%5B13%3A30%3A33%201494%20W%5D&c=%7B145%7Dd5555516ea463b6a5cb8d9c419b1832a80000%5B13%3A30%3A42%201494%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb314384443550000%5B13%3A30%3A51%201505%20W%5D&c=%7B145%7Dd5555516ea463a6a5c9f598a1c2221aa80000%5B13%3A30%3A59%201505%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb314384443550000%5B13%3A31%3A08%201505%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598a1c2221aa80000%5B13%3A31%3A17%201505%20W%5D&c=%7B144%7D55555516ea463a6a5c9f598a1c2221aa8000%5B13%3A31%3A26%201505%20W%5D&c=%7B144%7D55555516ea463a6a5c9f598a1c2221aa8000%5B13%3A31%3A35%201505%20W%5D&c=%7B145%7D55555516ea463a6a5c9f598a1c2221aa80000%5B13%3A31%3A44%201505%20W%5D&c=%7B145%7D55555516ea463b6a5cb8d9c419b1832a80000%5B13%3A31%3A53%201505%20W%5D&c=%7B143%7D2aaaaa2dd48c74d4b93eb304384433550000%5B13%3A32%3A01%201489%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59821c2219aa80000%5B13%3A32%3A10%201489%20W%5D&c=%7B144%7Daaaaaa2dd48c74d4b93eb304384433550000%5B13%3A32%3A19%201489%20W%5D&c=%7B144%7D55555516ea463a6a5c9f59821c2219aa8000%5B13%3A32%3A28%201489%20W%5D&c=%7B145%7D555 [... cut ...] 5516ea463a6a5c9f59941c222baa80000%5B13%3A33%3A48%201525%20W%5D&c=%7B145%7D55555516ea463a6a5c9f59941c222baa80000%5B13%3A33%3A57%201525%20W%5D&c=%7B144%7D55555516ea463a6a5c9f59941c222baa8000%5B13%3A34%3A06%5D&f=ID%3F8h8h8h8h%20TYPE%3F8h8h%20%20PWR%3F8d%208h%20%3F8h%20CHK%3F8h%20TRAILER%3F8h8h8h&a=Preamble&m=2dd4&cf=ADD&cl=72&cw=8).
But the checksum is different and I don't see where the high-order bits (upper byte) of the value is.
That is the vlaue range of the PV transmitter? E.g. 0 W to 11,5kW?

vannetta, 2020-10-23:
0 W when the PV doesn't work (before sunrise, after sunset); the maximum suring summer is almost 3000 W Now for istance (15:27) is 522W.

vannetta, 2020-10-26:
Can I do some other test?

zuckschwerdt, 2020-10-26:
Ideally we need some different power ranges to figure out if there are upper bits to e.g. form a 16 bit (0-65kW) value or if there is an exponent (typical seen with power meters). If you can measure dusk or dawn so we get a nice ramp of values.

vannetta, 2020-10-29:
These are data at dusk:
https://pastebin.com/nxvyYc2m
the monitor shows values between 80 W and 41 W. Consider that, as today is a little cloudy, values go up and down. 
Please let me know if data are ok!

zuckschwerdt, 2020-10-29:
This gives us:
```
44 33 092  41-80 W?
44 33 094  41-80 W?
44 33 097  41-80 W?
44 33 103  41-80 W?
44 33 106  41-80 W?
44 33 108  41-80 W?
44 33 111  41-80 W?
44 33 117  41-80 W?
44 33 119  41-80 W?
44 33 125  41-80 W?
44 33 128  41-80 W?
44 33 131  41-80 W?
44 38 004   1489 W (1489-004 = 1485)
44 38 006   1491 W (1491-006 = 1485)
44 38 009   1494 W (1494-009 = 1485)
44 38 015   1500 W (1500-015 = 1485)
44 38 015   1525 W (1525-015 = 1510)
44 38 017   1502 W (1502-017 = 1485)
44 38 020   1505 W (1505-020 = 1485)
44 38 029   1514 W (1514-029 = 1485)
44 38 034   1519 W (1519-034 = 1485)
44 38 040   1525 W (1525-040 = 1485)
44 38 163   1392 W (1392-163 = 1229)
44 38 182   1505 W (1505-182 = 1323)
44 38 185   1414 W (1414-185 = 1229)
44 38 199   1428 W (1428-199 = 1229)
44 38 201   1430 W (1430-201 = 1229)
44 38 221   1450 W (1450-221 = 1229)
44 38 237   1466 W (1466-237 = 1229)
44 38 240   1491 W (1491-240 = 1251)
44 39 053   1538 W (1538-053 = 1485)
44 39 062   1547 W (1547-062 = 1485)
44 39 076   1561 W (1561-076 = 1485)
```
Which show `33` for those low values, `38` for high values, and `39` starting between 1491 to 1538.
But the difference there is still 1485 which is strange. Also there is sometime 1229 which is plain weird.

To see how these values map we need to see a clean progression over a wide range.
Please collect the values for multiple hours at dusk or dawn so we see a range from low to high clearly.

Paste a list with just the values, each line like `{145}55555516ea463a6a5c9f59b519a24a2a80000 [x W]` (x can be a guess, just so we know what to expect)

vannetta, 2020-10-31:
This is a progression of a whole day, between 9 am and 6 pm (with the maximum production until the PV stops - 0W)
The MCEE monitor shows data in Windows xp, every 10 minutes. So I put together the output from rtl_433 with the output of MCEE from Win XP. 
As I said I have data from MCEE monitor only every 10 minutes, but the values should be quite right; today was a sunny day, only some clouds, so data should be quite similar within the 10 minutes of missing data.
Please let me know if this is ok, or if I can do something different.
[trasmitter ECODHOME.xlsx](https://github.com/merbanan/rtl_433/files/5469593/trasmitter.ECODHOME.xlsx)

zuckschwerdt, 2020-10-31:
This gives [this BitBench for just the power field](https://triq.net/bitbench?c=bd33%5B121%5D&c=dc33%5B138%5D&c=b733%5B135%5D&c=ce33%5B144%5D&c=d337%5B1067%5D&c=8c38%5B1317%5D&c=0638%5B1502%5D&c=4139%5B1519%5D&c=6239%5B1550%5D&c=6d39%5B1569%5D&c=9939%5B1616%5D&c=af39%5B1669%5D&c=b539%5B1669%5D&c=ba39%5B1658%5D&c=b839%5B1671%5D&c=9939%5B1605%5D&c=8939%5B1591%5D&c=7b39%5B1605%5D&c=5f39%5B1561%5D&c=4939%5B1505%5D&c=1f38%5B1516%5D&c=0c38%5B1505%5D&c=b938%5B1447%5D&c=c938%5B1411%5D&c=9d38%5B1405%5D&c=6e38%5B1331%5D&c=3738%5B1300%5D&c=0237%5B1189%5D&c=5e37%5B1080%5D&c=af36%5B942%5D&c=2835%5B793%5D&c=fb35%5B734%5D&c=b635%5B638%5D&c=6b35%5B582%5D&c=e734%5B458%5D&c=a134%5B400%5D&c=3d34%5B258%5D&c=1633%5B258%5D&c=c533%5B135%5D&c=a133%5B124%5D&c=8033%5B80%5D&c=7533%5B60%5D&c=6c33%5B80%5D&c=6c33%5B74%5D&c=6733%5B88%5D&c=3333%5B0%5D&f=8h%20&cl=72&cw=8).

The ramp looks about right, the MSB is offset by 0x33 and the 0 W value is transmitted as 0x3333 which is weird -- we can substract 0x33 from the LSB, it would sometimes go negative then. But I guess the values with this calculation would be ballpark-right.

vannetta, 2020-11-01:
That looks great!
Do you want me to do some other test? Do you need values from the consumption trnsmitter? It has a bigger range of values (from 300 W to 4500 W)

zuckschwerdt, 2020-11-01:
We know now the trailer (`aaaa`) and sync word (`2dd4`), the checksum (add with initial value of 0xcb or 0xcc if you add the sync word too), and we have a guess what the `ID? 8c74d4b9 TYPE? 3eb3` messages contain.

We still need to figure the `ID? 8c76d4b9 TYPE? 71b3` messages out that arrive every 71 seconds, but with all the codes you sent there is only ever this `8c76d4b9 71b3 863363 04 550000` which might be some kind of status then and not a reading.
I'll soon add a draft decoder to test

zuckschwerdt, 2020-11-01:
Can you give a overview: what components do you have?
- PV Transmitter?
- consumption transmitter?
- Smart Sockets?

You can now test #1544 -- please watch the output carefully if there is e.g. "checksum fail". Also review if the values are what you expect. You should see
- PV Transmitter type 3eb3 messages
- PV Transmitter type 71b3 messages (is the raw always the same, 863363?)
- Smart Socket type 7700 subtype 414b
- Smart Socket type 7700 subtype 565a

Are all the id numbers okay? Which ids do you get?

vannetta, 2020-11-01:
I have two identical transmitters, one for PV and one for consumption, and two smart sockets.
I'm going to test #1544.

vannetta, 2020-11-02:
This is the output; all the values look right!

Inviato da iPhone

zuckschwerdt, 2020-11-02:
Not sure if the PV Transmitter type 3eb3 messages are really the right value. Seems close, but needs careful checking as the calculatio is rather strange.

For the PV Transmitter type 71b3 messages the raw is always the same `863363`?

Both Smart Socket type 7700 subtype 414b and subtype 565a are the right values? Remind me what is the difference, is it W and kWh?

vannetta, 2020-11-03:
I need to test it at dawn/dusk; at the moment I've done only one test near noon. 
In this test I only used one sockets, the other one is not connected at the moment. 
Data are in Watt. The only value in kWh is the total consumption, that is only shown for the sockets, not for PV or consumption of the general panel box.

zuckschwerdt, 2020-11-03:
Then what is meaning of the Smart Socket type 7700 subtype 414b and subtype 565a values?
Is the subtype 565a value anything useful? Is it even "power W" or is it something else?

vannetta, 2020-11-08:
Sorry for these past days, I didn't feel good.
The socket send a value for consumption (W) and total consumptions since the moment that was connected (kWh).
Moreover from the monitor is possibile to switch on and off the socket. So it should be something that identify on and off status.
The trasmitter only send a Watt value.
Tomorrow I can connect the other socket I have and send a complete test.

vannetta, 2020-11-10:
I've made a complete record of one day, with two transmitters and one socket, from late dawn to dusk.
This is what I noticed:
1. The consumption values (id 8c7412d6) seem always almost right (maybe because is always more than 500 W)
2. Low values of production (id 8c74d4b9) are wrong (maybe under 200 W); when the production is 0, the value is 51W. Hight values seem right.
3. The socket was almost all the day without charge (0W); I don't know what is the message: 565a Power:127W The only data when the socket is not charging is the total consumption that is 10,69 kWh (always the same)
4. Sometimes there is a value for the socket, messge:414b      Power:   x W; this value is right.

I'm attaching two files: one is the registration with rtl_433, the other one is the record from MCEE monitor.
[ecodhome1.xlsx](https://github.com/merbanan/rtl_433/files/5516474/ecodhome1.xlsx)
[History_power2020-11-09 19-49-43.xlsx](https://github.com/merbanan/rtl_433/files/5516475/History_power2020-11-09.19-49-43.xlsx)

zuckschwerdt, 2020-11-10:
I'll take a look.
There is a new message in there we have not seen before (but it looks empty):
```
2020-11-09 07:57:33  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
2020-11-09 09:58:50  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
2020-11-09 10:08:22  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
2020-11-09 10:10:46  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
2020-11-09 10:18:53  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
2020-11-09 10:19:23  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
2020-11-09 10:43:37  EcoDHOME-SmartSocket  46315c6a  Type: 7700  Subtype: 5053  Power: 0.0 W    Raw: 000000
```

zuckschwerdt, 2020-11-10:
I just noticed that we always see these:
```
EcoDHOME-Transmitter  8c74 d4b9  Type: 3eb3  Power: 111.0 W  Raw: 6f3344
EcoDHOME-Transmitter  8c74 12d6  Type: 3eb3  Power: 700.0 W  Raw: bc3544
EcoDHOME-Transmitter  8c76 12d6  Type: 71b3  Raw: 333363
EcoDHOME-Transmitter  8c76 d4b9  Type: 71b3  Raw: 863363
```

I.e. EcoDHOME-Transmitter Type: 3eb3 is from 8c7**4** d4b9 and 8c7**4** 12d6
while EcoDHOME-Transmitter Type: 71b3 is from 8c7**6** 12d6 and 8c7**6** d4b9

zuckschwerdt, 2020-11-10:
Ok, the mystery of the Transmitter 3eb3 message payload is uncovered: both power bytes start at 0x33 (=0) and wrap up to 0x32 (=255). The reading then fits your MCEE values neatly. I'll merge this. You have to observe all other message carefully and try to see where there is switch on/off or such events.

closed 2020-11-10 by zuckschwerdt

vannetta, 2020-11-10:
I'm testing now: 
the 71b3 message is always 333363 for the consumption; as the sun is below horizon, there is only consumption: 8c74 12d6 (W) and 8c76 d4b9 (message 333363); so it could be a status.
 8c74 d4b9 and 8c76 d4b9 don't appear at all.

The smart socket always shows 565a message; values are 00800, 007f00, 007e00, 008100, 008200. So it can't be a status because nothing has changed.
Another thing is that the 414b (power) message appears very few times, even though something is charging.

Tomorrow I'm going to test with the PV switched on.