merbanan/rtl_433 #1527: Suspicious values for sensor Abarth 124 Spider

samueltardieu, 2020-10-21:
I happen to receive messages sent from "Abarth 124 Spider" TPMS sensors as I work close to some garages. The temperature outside is around 13°C and the reported temperature and pressure are suspicious even though the consistency between measures of the same sensor and the number of sensors (4 wheels) make it plausible that they are indeed TPMS sensor:

```
time      : 2020-10-21 07:56:58
model     : Abarth 124 Spider                      type      : TPMS          id        : f217b8c0
flags     : b5           Pressure  : 86 kPa        Temperature: -12 C        status    : 177
mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:56:59
model     : Abarth 124 Spider                      type      : TPMS          id        : f217b8c0
flags     : b5           Pressure  : 86 kPa        Temperature: -12 C        status    : 177
mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:56:59
model     : Abarth 124 Spider                      type      : TPMS          id        : f217b8c0
flags     : b5           Pressure  : 86 kPa        Temperature: -12 C        status    : 177
mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:56:59
model     : Abarth 124 Spider                      type      : TPMS          id        : f40374c0
flags     : ad           Pressure  : 87 kPa        Temperature: -14 C        status    : 117
mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:56:59
model     : Abarth 124 Spider                      type      : TPMS          id        : f40374c0
flags     : ad           Pressure  : 87 kPa        Temperature: -14 C        status    : 117
mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:57:00
model     : Abarth 124 Spider                      type      : TPMS          id        : f3fcecc0
flags     : b4           Pressure  : 87 kPa        Temperature: -18 C        status    : 9             mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:57:00
model     : Abarth 124 Spider                      type      : TPMS          id        : f3fcecc0
flags     : b4           Pressure  : 87 kPa        Temperature: -18 C        status    : 9             mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-21 07:57:03
model     : Abarth 124 Spider                      type      : TPMS          id        : f217b8c0
flags     : b5           Pressure  : 86 kPa        Temperature: -25 C        status    : 141
mic       : CHECKSUM
```

zuckschwerdt, 2020-10-21:
Seems to be a different sensor with same protocol indeed. Can you compare the `unknown (seems static)` value, see https://github.com/merbanan/rtl_433/blob/master/src/devices/tpms_abarth124.c#L29

merbanan, 2020-10-29:
@samueltardieu can we get some sample signals and approximativ readings?

samueltardieu, 2020-10-30:
Right now I couldn't capture some more. As I wrote I got them by chance, I'll see if I encounter them again.

samueltardieu, 2020-11-07:
This is strange. I have since switched RTL-SDR cards (I now have one from rtl-sdr.com, before I used nameless clone in a plastic case), and I was able to read consistent values without being able to make any guarantee that they correspond to the same device as before:

```
time      : @0.277912s
model     : Abarth 124 Spider                      type      : TPMS          id        : c8a673d0
flags     : 07           Pressure  : 184 kPa       Temperature: 19 C         status    : 4             mic       : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.448680s
model     : Abarth 124 Spider                      type      : TPMS          id        : c8a673d0
flags     : 07           Pressure  : 184 kPa       Temperature: 19 C         status    : 4             mic       : CHECKSUM
```

If I can find again and can record signals with the bogus interpretation I'll reopen this issue.

closed 2020-11-07 by samueltardieu

SphtKr, 2021-12-29:
I just bought a set of TPMS sensors from Amazon that are reporting as Abarth 124 Spider… and the values are all wrong… a “Qiilu” branded system: Qiilu Tire Pressure Monitor... https://www.amazon.com/dp/B08KVXCDX5 … it’s probably more closely related to the Jansite sensors but showing up as the Abarth.

 It shows values about 300 kPa where the included receiver shows 6.8 bar (680 kPa)… though 0 appears to be 0, so maybe it’s just some weird fraction.

 If someone wants to point me to exactly what to look at I can give it a shot, though I’ve only got about 3 days’ exposure to this SDR stuff :-D

zuckschwerdt, 2021-12-29:
Put a `bitrow_printf(b, 72, "%s: ", __func__);` here https://github.com/merbanan/rtl_433/blob/master/src/devices/tpms_abarth124.c#L74 to see the full codes.

Or use `rtl_433 -X "n=tpms=FSK_PCM,s=52,l=52,r=150,invert,preamble=aaaaa9"` to get raw data.