merbanan/rtl_433 #1530: HackRF and Windows problem ?

schmiernippel, 2020-10-22:
Is it possible to use my hackrf with rtl_433 on windows with this soapy drivers ? I dont get it howto...?

When its possible, can someone please write a quick tutorial ?

greets

zuckschwerdt, 2020-10-22:
Yes, you need to compile or install SoapySDR and drivers, then build rtl_433 with `-DENABLE_SOAPYSDR=ON `.
There are some hints in https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md

schmiernippel, 2020-10-22:
I have installed soapySDR(photosSDR)but my knowledge is to less to build rtl_433 by myself... but thanks for your answer !

zuckschwerdt, 2020-10-22:
It is somewhat involved to get the drivers included. Basically some trial and error to get all paths setup. Do you have any experience with either MinGW or Visual Studio?
We currently don't provide binaries with SoapySDR support as getting all dependencies setup is tricky. I might take a look at that, would you be interested in testing some builds? It could be some back and forth until we get it right though.

schmiernippel, 2020-10-22:
I have no experience in MinGW or visual studio. But i would like to test some builds, this would be nice !

zuckschwerdt, 2020-10-23:
Can you try rtl_433-20.02-195-gc92751e-win-x64.zip from https://bintray.com/beta/#/chzu/dist/rtl_433/20.02-195-gc92751e?tab=files
Then not-static version should include support for SoapySDR.

schmiernippel, 2020-10-23:
I have installed PhotosSDR with the SoapySDR drivers and have run rtl_433.exe. No supportet device found....
CubicSDR, GQRX SDR, SDR# are working nice.
Any workaround what should i do ?

zuckschwerdt, 2020-10-23:
That is the first line in the output of `rtl_433.exe`?

schmiernippel, 2020-10-23:
Microsoft Windows [Version 10.0.19041.508]
(c) 2020 Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\schmiernippel>C:\Daten\Downloads\SDR\rtl_433\Soapy_support\rtl_433.exe
rtl_433 version 20.02-195-gc92751e branch  at 202010231657 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Daten\Downloads\SDR\rtl_433\Soapy_support\rtl_433.conf"...
Trying conf file at "C:\Users\schmiernippel\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 139 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]
No supported devices found.

C:\Users\schmiernippel>

Thats the output!

zuckschwerdt, 2020-10-23:
Great! That looks fine.
`rtl_433 version 20.02-195-gc92751e branch at 202010231657 inputs file rtl_tcp RTL-SDR SoapySDR` <-
And selecting your HackRF doesn't work? I.e. `rtl_433 -d driver=hackrf`

schmiernippel, 2020-10-24:
Microsoft Windows [Version 10.0.19041.572]
(c) 2020 Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\schmiernippel>C:\Daten\Downloads\SDR\rtl_433\Soapy_support\rtl_433.exe -d driver=hackrf
rtl_433 version 20.02-195-gc92751e branch  at 202010231657 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Daten\Downloads\SDR\rtl_433\Soapy_support\rtl_433.conf"...
Trying conf file at "C:\Users\schmiernippel\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 139 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]

C:\Users\schmiernippel>

The same output, without "No supportet devices found."

zuckschwerdt, 2020-10-24:
The build includes the SoapySDR.dll from the https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2020.01.26-vc14-x64.exe version found on https://downloads.myriadrf.org/builds/PothosSDR/
Is that the same version you have installed?
Perhaps just copy `rtl_433.exe` to the directory where PothosSDR installed all the .exe and .dll, do not copy `SoapySDR.dll` so your version is used --  it should already be in that directory along with `libusb-1.0.dll`.
Running `SoapySDRUtil.exe --info` does list the HackRF module, right? And `--probe` or `--make` show the HackRF?

schmiernippel, 2020-10-24:
"libusb-1.0.dll" was not in the folder "C:\Program Files\PothosSDR\bin". So i copied it.

C:\Users\schmiernippel>"C:\Program Files\PothosSDR\bin\rtl_433.exe" -d driver=hackrf
rtl_433 version 20.02-195-gc92751e branch  at 202010231657 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Program Files\PothosSDR\bin\rtl_433.conf"...
Trying conf file at "C:\Users\schmiernippel\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 139 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]
[INFO] Opening HackRF One #0 17c467dc2154bbc3...

C:\Users\schmiernippel>

And here is the SoapySDRUtil.exe Output:

C:\Users\schmiernippel>"C:\Program Files\PothosSDR\bin\SoapySDRUtil.exe" --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-PothosSDR-2020.01.26-vc14-x64
API Version: v0.8.0
ABI Version: v0.8
Install root: C:\Program Files\PothosSDR
Search path:  C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspyhfSupport.dll (0.1.1-81ca737)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/airspySupport.dll   (0.1.2-10d697b)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/audioSupport.dll    (0.1.1-2ae84e3)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/bladeRFSupport.dll  (0.4.1-1c1e8aa)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/HackRFSupport.dll   (0.3.3-3c514ce)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/IrisSupport.dll     (2019.07.0.1-423a351)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/LMS7Support.dll     (19.04.1-538e186b)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/miriSupport.dll     (0.2.6-5a9674c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/netSDRSupport.dll   (0.1.0-dc4e579)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/osmosdrSupport.dll  (0.2.6-5a9674c)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/PlutoSDRSupport.dll (0.2.0-e28e4f5)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/RedPitaya.dll       (0.1.1-3d576f8)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/remoteSupport.dll   (0.5.2-6d9bd82)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/rtlsdrSupport.dll   (0.3.1-8ba18f1)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/sdrPlaySupport.dll  (0.3.0-14ec39e)
Module found: C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/uhdSupport.dll      (0.3.6-47972ba)
Available factories... airspy, airspyhf, audio, bladerf, hackrf, iris, lime, miri, netsdr, osmosdr, plutosdr, redpitaya, remote, rtlsdr, sdrplay, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

C:\Users\schmiernippel>"C:\Program Files\PothosSDR\bin\SoapySDRUtil.exe" --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=Audio
  hardware=Audio
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Stereo Sample Offset - Offset stereo samples for off-by-one audio inputs.
       [key=sample_offset, default=0, type=string, options=(-2, -1, 0, 1, 2)]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS16 [full-scale=65536]
  Stream args:
     * Channel Setup - Input channel configuration.
       [key=chan, default=mono_l, type=string, options=(mono_l, mono_r, stereo_iq, stereo_qi)]
  Antennas: RX
  Full gain range: [0, 0] dB
  Full freq range: [0, 6000] MHz
    RF freq range: [0, 6000] MHz
  Sample rates: 0.004, 0.005512, 0.008, 0.0096, 0.011025, 0.048, 0.0882, 0.096, 0.1764, 0.192 MSps

C:\Users\schmiernippel>"C:\Program Files\PothosSDR\bin\SoapySDRUtil.exe" --make
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.15.0.0-0-gaea0e2de
  driver=Audio
  hardware=Audio
  device_id=0
  origin=https://github.com/pothosware/SoapyAudio

C:\Users\schmiernippel>

I see no hackrf ? But in windows device manager its listed under "USB-devices" as "HackRF One"

zuckschwerdt, 2020-10-24:
At least it does say ` [INFO] Opening HackRF One #0 17c467dc2154bbc3...` now. But immediatly stops? This could be the hackrf driver looking for a device but not finding it.
Does adding `-vv` get more useful info?
I suspect that the device is not available, perhaps blocked/in-use by another program? Some other user reported something about outdated firmware he had to update.

schmiernippel, 2020-10-24:
This is with -vv 

C:\Users\schmiernippel>"C:\Program Files\PothosSDR\bin\rtl_433.exe" -d driver=hackrf -vv
rtl_433 version 20.02-195-gc92751e branch  at 202010231657 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Program Files\PothosSDR\bin\rtl_433.conf"...
Trying conf file at "C:\Users\schmiernippel\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registering protocol [1] "Silvercrest Remote Control"
Registering protocol [2] "Rubicson Temperature Sensor"
Registering protocol [3] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [4] "Waveman Switch Transmitter"
Registering protocol [8] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [11] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [12] "Oregon Scientific Weather Sensor"
Registering protocol [15] "KlikAanKlikUit Wireless Switch"
Registering protocol [16] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [17] "Cardin S466-TX2"
Registering protocol [18] "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor"
Registering protocol [19] "Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor"
Registering protocol [20] "Ambient Weather Temperature Sensor"
Registering protocol [21] "Calibeur RF-104 Sensor"
Registering protocol [23] "DSC Security Contact"
Registering protocol [25] "Globaltronics GT-WT-02 Sensor"
Registering protocol [26] "Danfoss CFR Thermostat"
Registering protocol [29] "Chuango Security Technology"
Registering protocol [30] "Generic Remote SC226x EV1527"
Registering protocol [31] "TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999"
Registering protocol [32] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [33] "WT450, WT260H, WT405H"
Registering protocol [34] "LaCrosse WS-2310 / WS-3600 Weather Station"
Registering protocol [35] "Esperanza EWS"
Registering protocol [36] "Efergy e2 classic"
Registering protocol [38] "Generic temperature sensor 1"
Registering protocol [39] "WG-PB12V1 Temperature Sensor"
Registering protocol [40] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 3N1, Atlas"
Registering protocol [41] "Acurite 986 Refrigerator / Freezer Thermometer"
Registering protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor"
Registering protocol [43] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"
Registering protocol [44] "CurrentCost Current Sensor"
Registering protocol [45] "emonTx OpenEnergyMonitor"
Registering protocol [46] "HT680 Remote control"
Registering protocol [47] "Conrad S3318P, FreeTec NC-5849-913 temperature humidity sensor"
Registering protocol [48] "Akhan 100F14 remote keyless entry"
Registering protocol [49] "Quhwa"
Registering protocol [50] "OSv1 Temperature Sensor"
Registering protocol [51] "Proove / Nexa / KlikAanKlikUit Wireless Switch"
Registering protocol [52] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [53] "Springfield Temperature and Soil Moisture"
Registering protocol [54] "Oregon Scientific SL109H Remote Thermal Hygro Sensor"
Registering protocol [55] "Acurite 606TX Temperature Sensor"
Registering protocol [56] "TFA pool temperature sensor"
Registering protocol [57] "Kedsum Temperature & Humidity Sensor, Pearl NC-7415"
Registering protocol [58] "Blyss DC5-UK-WH"
Registering protocol [59] "Steelmate TPMS"
Registering protocol [60] "Schrader TPMS"
Registering protocol [63] "Efergy Optical"
Registering protocol [67] "Radiohead ASK"
Registering protocol [68] "Kerui PIR / Contact Sensor"
Registering protocol [69] "Fine Offset WH1050 Weather Station"
Registering protocol [70] "Honeywell Door/Window Sensor, 2Gig DW10/DW11, RE208 repeater"
Registering protocol [71] "Maverick ET-732/733 BBQ Sensor"
Registering protocol [73] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor"
Registering protocol [74] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [75] "LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor"
Registering protocol [76] "LaCrosse TX29IT Temperature sensor"
Registering protocol [77] "Vaillant calorMatic VRT340f Central Heating Control"
Registering protocol [78] "Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor"
Registering protocol [79] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor"
Registering protocol [80] "IBIS beacon"
[... 59 lines cut ...]
Registering protocol [149] "ERT"
Registering protocol [151] "Visonic powercode"
Registering protocol [152] "Eurochron EFTH-800 temperature and humidity sensor"
Registering protocol [153] "Cotech 36-7959 wireless weather station with USB"
Registering protocol [154] "Standard Consumption Message Plus (SCMplus)"
Registering protocol [155] "Fine Offset Electronics WH1080/WH3080 Weather Station (FSK)"
Registering protocol [156] "Abarth 124 Spider TPMS"
Registering protocol [157] "Missil ML0757 weather station"
Registering protocol [158] "Sharp SPC775 weather station"
Registering protocol [159] "Insteon"
Registering protocol [160] "Interval Data Message (IDM)"
Registering protocol [161] "Interval Data Message (IDM) for Net Meters"
Registering protocol [163] "Acurite 590TX Temperature with optional Humidity"
[... cut ...]
Registering protocol [165] "TFA Dostmann 30.3221.02 T/H Outdoor Sensor"
Registering protocol [166] "LaCrosse Technology View LTV-WSDTH01 Breeze Pro Wind Sensor"
Registering protocol [167] "Somfy RTS"
Registering protocol [168] "Schrader TPMS SMD3MA4 (Subaru)"
Registered 139 out of 168 device decoding protocols
[INFO] Opening HackRF One #0 17c467dc2154bbc3...
Using device HackRF One:
C:\Users\schmiernippel>

Is it possible that i must ad a line with settings for the hackrf ? Like gain and so on ??? 
Firmware is the newest for HackRF.

zuckschwerdt, 2020-10-24:
Looks like rtl_433 is able to acquire a handle to the driver but the device isn't initializing or working.
You need to try things until `SoapySDRUtil.exe --make` works, then continue with rtl_433.

schmiernippel, 2020-10-24:
Ok, thanks.I will google some stuff :-)

merbanan, 2021-03-08:
No update in a long time, closing issue.

closed 2021-03-08 by merbanan