merbanan/rtl_433 #1539: Buffer overflow

merbanan, 2020-10-28:
```
Detected OOK package	2020-10-28 21:44:34
Analyzing pulses...
Total count: 1200,  width: 174.58 ms		(174583 S)
Pulse width distribution:
 [ 0] count:    1,  width:   19 us [19;19]	(  19 S)
 [ 1] count:    1,  width:  190 us [190;190]	( 190 S)
 [ 2] count:  722,  width:   61 us [57;70]	(  61 S)
 [ 3] count:  476,  width:  112 us [110;118]	( 112 S)
Gap width distribution:
 [ 0] count:    2,  width:   16 us [15;18]	(  16 S)
 [ 1] count:  721,  width:   43 us [36;47]	(  43 S)
 [ 2] count:  476,  width:   95 us [90;99]	(  95 S)
Pulse period distribution:
 [ 0] count:    1,  width:   37 us [37;37]	(  37 S)
 [ 1] count:  351,  width:  208 us [205;211]	( 208 S)
 [ 2] count:  595,  width:  104 us [102;107]	( 104 S)
 [ 3] count:  252,  width:  156 us [154;158]	( 156 S)
Pulse timing distribution:
 [ 0] count:    3,  width:   17 us [15;19]	(  17 S)
 [ 1] count:    1,  width:  190 us [190;190]	( 190 S)
 [ 2] count:  722,  width:   61 us [57;70]	(  61 S)
 [ 3] count:  952,  width:  104 us [90;118]	( 104 S)
 [ 4] count:  722,  width:   43 us [36;47]	(  43 S)
Level estimates [high, low]:   8851,     70
RSSI: -2.7 dB SNR: 21.0 dB Noise: -23.7 dB
Frequency offsets [F1, F2]:  -14542,      0	(-221.9 kHz, +0.0 kHz)
Guessing modulation: No clue...
=================================================================
==1291151==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7fffffff02a2 at pc 0x5555558c7994 bp 0x7ffffffe4ec0 sp 0x7ffffffe4eb0
WRITE of size 1 at 0x7fffffff02a2 thread T0
    #0 0x5555558c7993 in pulse_analyzer /home/benjamin/project/rtl_433/src/pulse_analyzer.c:395
    #1 0x55555581033a in sdr_callback /home/benjamin/project/rtl_433/src/rtl_433.c:417
    #2 0x7ffff7561bf0  (/lib/x86_64-linux-gnu/librtlsdr.so.0+0x5bf0)
    #3 0x7ffff754ab65  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xab65)
    #4 0x7ffff754f49b  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xf49b)
    #5 0x7ffff7550d97  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0x10d97)
    #6 0x7ffff754a5a8  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xa5a8)
    #7 0x7ffff754b65b in libusb_handle_events_timeout_completed (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xb65b)
    #8 0x7ffff75618dc in rtlsdr_read_async (/lib/x86_64-linux-gnu/librtlsdr.so.0+0x58dc)
    #9 0x555555905a83 in sdr_start /home/benjamin/project/rtl_433/src/sdr.c:1199
    #10 0x555555839e34 in main /home/benjamin/project/rtl_433/src/rtl_433.c:1636
    #11 0x7ffff68b90b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)
    #12 0x55555580c7ad in _start (/home/benjamin/project/rtl_433/build/src/rtl_433+0x2b87ad)

Address 0x7fffffff02a2 is located in stack of thread T0 at offset 45858 in frame
    #0 0x5555558c1c24 in pulse_analyzer /home/benjamin/project/rtl_433/src/pulse_analyzer.c:204

  This frame has 8 object(s):
    [48, 200) 'device' (line 256)
    [272, 596) 'hist_pulses' (line 217)
    [672, 996) 'hist_gaps' (line 218)
    [1072, 1396) 'hist_periods' (line 219)
    [1472, 1796) 'hist_timings' (line 220)
    [1872, 2900) 'hexstr' (line 342)
    [3040, 12704) 'pulse_periods' (line 209)
    [12960, 45856) 'hexstrs' (line 368) <== Memory access at offset 45858 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /home/benjamin/project/rtl_433/src/pulse_analyzer.c:395 in pulse_analyzer
Shadow bytes around the buggy address:
  0x10007fff6000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff6010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff6020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff6030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff6040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x10007fff6050: 00 00 00 00[f3]f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3
  0x10007fff6060: f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3
  0x10007fff6070: f3 f3 f3 f3 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff6080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff6090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff60a0: 00 00 00 00 00 00 00 00 f1 f1 f1 f1 00 f2 f2 f2
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc

```

Changing to hexstr_t hexstrs[256] = {{{0}}}; fixes it.

zuckschwerdt, 2020-10-28:
I was lazy there, probably should check for a defined maximum string number.

Mindavi, 2020-11-02:
I have also found this issue, reproduced using this file: 
[crash.zip](https://github.com/merbanan/rtl_433/files/5477733/crash.zip)

<details><summary>Valgrind crash log</summary>

```
[nix-shell:~/hobby/rtl/recordings]$ ../work/build/src/rtl_433 -A -r crash.cu8 
rtl_433 version 20.02-224-g61da659 branch enable-more-warnings-and-fix-them at 202011011830 inputs file rtl_tcp
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/rick/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 143 out of 173 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-173 ]
Test mode active. Reading samples from file: crash.cu8
Detected OOK package    @0.271052s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.271052s
model     : Nexus-TH     House Code: 130
Channel   : 1            Battery   : 1             Temperature: 27.60 C      Humidity  : 41 %
Analyzing pulses...
Total count:  880,  width: 1750.52 ms           (437630 S)
Pulse width distribution:
 [ 0] count:  866,  width:  500 us [420;520]    ( 125 S)
 [ 1] count:    3,  width:  980 us [972;1004]   ( 245 S)
 [ 2] count:   10,  width:  728 us [700;788]    ( 182 S)
 [ 3] count:    1,  width:  260 us [260;260]    (  65 S)
Gap width distribution:
 [ 0] count:   23,  width: 3892 us [3872;3904]  ( 973 S)
 [ 1] count:  406,  width: 1936 us [1736;1968]  ( 484 S)
 [ 2] count:  435,  width:  960 us [940;1168]   ( 240 S)
 [ 3] count:   13,  width:  716 us [672;760]    ( 179 S)
 [ 4] count:    2,  width:  268 us [268;272]    (  67 S)
Pulse period distribution:
 [ 0] count:   23,  width: 4388 us [4316;4412]  (1097 S)
 [ 1] count:  406,  width: 2444 us [2424;2940]  ( 611 S)
 [ 2] count:  447,  width: 1460 us [1184;1688]  ( 365 S)
 [ 3] count:    1,  width: 1948 us [1948;1948]  ( 487 S)
 [ 4] count:    2,  width:  768 us [768;772]    ( 192 S)
Pulse timing distribution:
 [ 0] count:  866,  width:  500 us [420;520]    ( 125 S)
 [ 1] count:  438,  width:  960 us [940;1168]   ( 240 S)
 [ 2] count:   23,  width:  720 us [672;788]    ( 180 S)
 [ 3] count:    3,  width:  264 us [260;272]    (  66 S)
 [ 4] count:   23,  width: 3892 us [3872;3904]  ( 973 S)
 [ 5] count:  406,  width: 1936 us [1736;1968]  ( 484 S)
 [ 6] count:    1,  width: 10044 us [10044;10044]       (2511 S)
Level estimates [high, low]:  15936,    106
RSSI: -0.1 dB SNR: 21.8 dB Noise: -21.9 dB
Frequency offsets [F1, F2]:  -18830,      0     (-71.8 kHz, +0.0 kHz)
Guessing modulation: No clue...
../src/pulse_analyzer.c:372:44: runtime error: index 33 out of bounds for type 'hexstr_t [32]'
../src/pulse_analyzer.c:396:46: runtime error: index 32 out of bounds for type 'hexstr_t [32]'
../src/pulse_analyzer.c:396:62: runtime error: load of address 0x72811509e4f0 with insufficient space for an object of type 'unsigned int'
0x72811509e4f0: note: pointer points here
 00 00 00 00  15 00 00 00 aa b0 11 07  01 01 f4 03 c0 02 d0 01  08 0f 34 07 90 27 3c 85  55 00 00 00
              ^ 
../src/pulse_analyzer.c:397:44: runtime error: index 32 out of bounds for type 'hexstr_t [32]'
../src/pulse_analyzer.c:399:50: runtime error: index 32 out of bounds for type 'hexstr_t [32]'
../src/pulse_analyzer.c:399:50: runtime error: load of address 0x72811509e0f4 with insufficient space for an object of type 'uint8_t'
0x72811509e0f4: note: pointer points here
  aa b0 11 07 01 01 f4 03  c0 02 d0 01 08 0f 34 07  90 27 3c 85 55 00 00 00  00 00 02 00 00 00 00 00
              ^ 
../src/pulse_analyzer.c:399:50: runtime error: index 32 out of bounds for type 'hexstr_t [32]'
../src/pulse_analyzer.c:399:50: runtime error: store to address 0x72811509e0f4 with insufficient space for an object of type 'uint8_t'
0x72811509e0f4: note: pointer points here
  aa b0 11 07 01 01 f4 03  c0 02 d0 01 08 0f 34 07  90 27 3c 85 55 00 00 00  00 00 02 00 00 00 00 00
              ^ 
Segmentation fault (core dumped)
```

</details>

zuckschwerdt, 2020-11-02:
Thanks for the nice test case!

commit c72b5618 (2020-11-02): Fix rfraw builder overflow (closes #1539)
closed 2020-11-02 by zuckschwerdt