merbanan/rtl_433 #1540: Prologue-TH missing subtype in instance definition

rotragit, 2020-10-29:
Prologue-TH sensors have a subtype attribute:
``{
    "time": "2020-10-29 04:32:58",
    "model": "Prologue-TH",
    "subtype": 5,
    "id": 37,
    "channel": 3,
    "battery_ok": 1,
    "temperature_C": 19.8,
    "humidity": 66,
    "button": 0
}``

so the states are published in `rtl_433/[hostname]/devices/Prologue-TH/[subtype]/[channel]/[id]/humidity`

Adding the following lines between 394 and 395 solves the problem:

``
   if "subtype" in data:
        subtype = str(data["subtype"])
        instance = subtype + "/" + instance
``

zuckschwerdt, 2020-10-29:
@rct is working on this in #1510 with some deeper analysis, please coordinate changes there.

rct, 2020-10-31:
@rotragit - What does subtype: 5 represent for Prologue sensors?  Is that a message type?  Does the device only send a single value for that field?  If so the field name should probably be changed to `message_type` - see #1520 

@zuckschwerdt mentioned in @1510 `subtype` is intended to be used when there are somewhat distinct sub-devices that it makes sense to break out. This is why I believe it 

I'll also publish the fixes I have so far for the MQTT discovery script.

zuckschwerdt, 2020-10-31:
For Prologue the first nibble really is a (hardware version?) type. We only ever see 9 and 5 and the value is fixed (i.e. "a Prologue9" or a "Prologue5" device), it could be seen as part of the ID though, https://github.com/merbanan/rtl_433/blob/master/src/devices/prologue.c#L19

rct, 2020-11-01:
That does seem more like part of device identification rather than identifying a discrete sub-device.

rotragit, 2020-11-02:
@zuckschwerdt I own only 3 Prologue sensors and all of them have number 5 as subtype. As I see there exists a "prologue IW007 weather station", I think the answer should be related to that: if the protocol is the same and the data are marked as subtypes then - probably - the subtype sholud be part of the id, if not than the subtype is some kind of hardware model and Prologue9, Prologue5 make sense.
I don't own that station, of course :-)
If you have no experience with that weather station, and because there is not a significant difference in the case of Prologue-TH sensors, I suggest to pass the subtype ias part of the id in the discovery script. And because the suggested version is general for any device that has a subtype, it probably will solve other problems with other sensors where the difference is not so important :-)

zuckschwerdt, 2020-11-02:
When the Prologue decoder was written den first nibble was fixed to `9` and just seen/used as a check. Then later `5` was also seen.
So maybe we should have added the subtype to the ID back then. It is a breaking change and will invalidate ID for existing users, so we need to consider carefully if we want to break this :/

rotragit, 2020-11-02:
> So maybe we should have added the subtype to the ID back then. It is a breaking change and will invalidate ID for existing users, so we need to consider carefully if we want to break this :/

I agree

merbanan, 2020-11-02:
Well the id is randomly generated anyway so people will have to reconfigure anyway on battery replacement. If people use this sensor I think they will only care about the model. I'd say do the change after the new release.

zuckschwerdt, 2020-11-02:
Good point, let's do that.

rotragit, 2020-11-02:
I never have seen the id changed after a battery replacement on the Prologue-TH. I have also four Nexus-TH and yes, they have id changed after battery replacement (and that's very annoying...)

merbanan, 2020-11-02:
@rotragit get wmbus sensors. Unique ids and 10+ years of battery.

zuckschwerdt, 2020-11-02:
Uhh, nice. Can we get a list of tested and well behaving hardware?

rotragit, 2020-11-02:
Never seen wmbus devices "affordable" :-)
Prologue-TH are not so bad for me, considering that I've got them for less than 10€ each....

gdt, 2023-09-29:
Where are we on this?   Has the bugfix  been made?  If not, is @rotragit able to make a PR?

gdt, 2024-06-04:
abandoned

closed 2024-06-04 by gdt