merbanan/rtl_433 #1541: Home Assistant Autodiscovery setup

sotatech, 2020-10-29:
Hi,
I posted this question on Google Groups but didn't get an answer. As several people here seem to have it working, I'm hoping for a little assistance!

I have a number of 433MHz devices that work with rtl_433 and I want to integrate them with Home Assistant using the python examples provided. However, there seems to be little or no documentation.
I have run the rtl_433_mqtt_hass.py which seems to expect input via MQTT from rtl_433_mqtt_relay.py which it then republishes as a HA discovery topic.
However, how does the mqtt_relay connect to rtl_433? I can see that it's using MQTT and syslog, but trying both of those output formats with rtl_433 produces nothing. I'm monitoring the topics with MQTT explorer, so I can see that rtl_433 is generating traffic.
Any help would be appreciated.

zuckschwerdt, 2020-10-29:
You would use the `rtl_433_mqtt_relay.py` only if you want to filter or remap the MQTT topics (it reads from the Syslog format and publishes custom topics). Usually you should use just `-F mqtt`.
The `rtl_433_mqtt_hass.py` example can then pick up those default topics and generate discovery information.

sotatech, 2020-10-29:
Ah, thank you @zuckschwerdt! I was over-thinking this, I can see the HA topics appearing now.

zuckschwerdt, 2020-10-29:
No worries, the examples are "things some people use" but don't explain the use-case. I'll try to add some preface there when time allows.

sotatech, 2020-10-29:
That would be great, even a couple of lines like above would have saved me many hours :). One last thing, it doesn't recognise my CurrentCost power meter (R 44) - `{"time":"2020-10-29 13:43:46","model":"CurrentCost-TX","id":77,"power0_W":1513,"power1_W":0,"power2_W":0}`, so I added this:

`    "power0_W": {
        "device_type": "sensor",
        "object_suffix": "watts",
        "config": {
            "device_class": "power",
            "name": "Power",
            "unit_of_measurement": "W",
            "value_template": "{{ value|float }}"
        }
    },`

Finally, I want to run this as a background process, do I just uncomment the lines at the end?

Thanks again.

zuckschwerdt, 2020-10-29:
The `daemon` stuff are just some hints what needs to be configured. If you are not sure perhaps use `supervisord` to run rtl_433 and that script as background services.

sotatech, 2020-10-29:
Ah, ok understood. I'll run with supervisord for the moment but I'll use the daemon as a project in my attempt to learn Python.

rct, 2020-10-29:
@sotatech - First, I fumbled trying to understand the role and topology for this script. I'm hoping to submit a PR with some changes to the script soon including at least a comment that say something like:

    "rtl_433_mqtt_hass.py connects to MQTT and subscribes the rtl_433 event stream that is published to MQTT by rtl_433. The script publishes additional MQTT topics that can be used by Home Assistant to automatically discover and minimally configure new devices."

Some other notes:

* If you are using the Acurite 5n1 or Atlas weather station, make sure you're running the most recent version of rtl_433 with #1520 merged.  See #1510 for more info.

* With regards to running this script as a daemon. The set of rtl_433 devices I'm using changes very infrequently. So I've changed this script to publish the home assistant configurations to MQTT as retained messages. So I only need to run this script periodically when something changes. When Home Assistant restarts up and connects to MQTT it gets all of the configuration messages right away, so I don't have to wait for up to 10 minutes for the messages to be republished.

`    mqttc.publish(path, json.dumps(config), retain=Retain)`

Set `Retain` to True.

sotatech, 2020-10-29:
Thanks for the heads up @rct - I hope that might solve my problem with my Watchman oil tank guage, I'm only seeing the temperature and not the depth_cm. I'll do a fresh build in the morning. I don't have a weather station at the moment, but it's on my Chrismas list.
I was under the impression that the script had to be kept running in the background, but I'll try your tip regarding the retained messages.

sotatech, 2020-10-30:
No change with building the latest release. The script is seeing the depth reading and then returning a sensor with device_class: depth. But I don't think this is a valid class for HA sensors according to the docs - [https://www.home-assistant.io/integrations/sensor/](url)

This is the error in the HA logs:
2020-10-30 08:37:04 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'device_class': 'depth', 'name': 'Oil-SonicSmart-137574874-D', 'unit_of_measurement': 'cm', 'value_template': '{{ value|float }}', 'state_topic': 'rtl_433/ubuntu/devices/Oil-SonicSmart/137574874/depth_cm', 'unique_id': 'Oil-SonicSmart-137574874-D', 'device': {'identifiers': 'Oil-SonicSmart-137574874', 'name': 'Oil-SonicSmart-137574874', 'manufacturer': 'rtl_433'}, 'platform': 'mqtt'},)
voluptuous.error.MultipleInvalid: value is not allowed for dictionary value @ data['device_class']

sotatech, 2020-10-30:
As a test, I changed the device_class in the script to "pressure" and Home Assistant is now seeing the sensor. I've made a feature request to have the device_class depth added to HA.

rct, 2020-11-02:
@sotatech - I just started trying to figure out the rules for Home Assistant device classes, units of measurements. `device_class` can be omitted and then it's a generic sensor. Setting it to `pressure` indicates that the value is in hPa or mbar.  so that might not be what you want.  See https://www.home-assistant.io/integrations/sensor

I think a number of the entries in the mapping table are WIP.

sotatech, 2020-11-02:
Thanks, @rct I just wanted to document what was needed to get the script working for my setup. I totally understand that it's a WIP, but I was interested in trying to get it to work as an exercise for myself.
Changing device_id to pressure what just to prove I had found the issue, however it's been working fine since so I'll wait and see if the HA develpers take up my feature request to add depth to the device_class.

scyto, 2021-01-24:
sorry for hijacking, so am i right that the first stage is to setup rtl_433 to listen on whatever frequency / service type i want (in my case DSC alarm sensors) - collect all the JSON that produces and then  modify the scripts to work with those json objects?

zuckschwerdt, 2021-01-25:
@scyto the HA autodiscovery script should work without changes. Use (and modify) an example script if you want to parse rtl_433 output to custom destinations or actions.

rct, 2021-01-25:
@scyto - In theory, the auto-discovery script should just work. The reality is right now is it is a WIP. There are some un-merged and conflicting changes which correct some problems in field definitions. There are some more definitions that need to be added.

Note: all of the definitions right now are global for all fields of that name. Works great for things like temperature. Not so much for less well defined fields that might be interpreted differently across devices.

IIRC the current code doesn't match the DSC sensors. This is what I added to make DSC "closed" work:

```
352,364d351
<     # rct added 2020/12/13,
<     # temp for DSC sensors, might not work well as a general attribute
<     "closed": {
<         "device_type": "binary_sensor",
<         "object_suffix": "closed",
<         "config": {
<             "device_class": "door",
<             "force_update": "true",
<             "payload_on": "1",
<             "payload_off": "0"
<         }
<     },
<
```

Currently the DSC fields that are being matched: `closed, tamper, battery_ok, rssi, snr, noise`
Unmatched:  `event, xactivity, xtamper1, xtamper2, esn, status, status_hex`

The important one that needs to be added is `event`.  I suspect that a single global definition for event might be problematic for some devices but if you are only concerned about the limited set of devices you actually own, just modify the closed example above.

----

A note about device_class. This can be set to indicate what type of device it is (door, window, smoke, gas (CO), etc.  (See Hass docs for device_class, also alarm panel, and envisalink.)

The discovery script will *never* have enough information to get that right. Only you know what each sensor is for.

The approach I've taken is to run the auto-discovery script, let it publish to a different topic, copy and edit the entries, then publish to the homeassistant topic when I'm happy.  I publish them as retained.  I also save my generated entries in a text file for republishing if ever necessary.

Hope this helps.

rct, 2021-01-26:
I think this issue could be closed now.  The original set of issues @sotatech mentioned I believe have been addressed. There is still some data mapping clean up that needs to be done (and will probably be on-going). Ideally we'll get the mappings in to an external file like maxlock's PR which currently has some unresolved conflicts.

closed 2021-01-26 by zuckschwerdt

scyto, 2021-01-26:
> Hope this helps.

yes, very much, thanks!