merbanan/rtl_433 #1545: Add Dooya remote protocoll

vampywiz17, 2020-11-01:
Hello there,

I use A Dooya Curtain Remote (DC1602)

https://www.aliexpress.com/i/32954197821.html

It is a 15 channel remote. all channel able to 3 function. Open/Close/Stop

I do some sample:

1 channel 

Open
```

Detected OOK package    2020-11-01 13:11:38
Analyzing pulses...
Total count:  369,  width: 539.47 ms            (134867 S)
Pulse width distribution:
 [ 0] count:    9,  width: 4916 us [4912;4936]  (1229 S)
 [ 1] count:  249,  width:  344 us [340;352]    (  86 S)
 [ 2] count:  111,  width:  716 us [708;724]    ( 179 S)
Gap width distribution:
 [ 0] count:    9,  width: 1588 us [1588;1596]  ( 397 S)
 [ 1] count:  243,  width:  784 us [780;804]    ( 196 S)
 [ 2] count:  108,  width:  416 us [408;432]    ( 104 S)
 [ 3] count:    7,  width: 9016 us [8848;9252]  (2254 S)
 [ 4] count:    1,  width: 15548 us [15548;15548]       (3887 S)
Pulse period distribution:
 [ 0] count:    9,  width: 6508 us [6504;6524]  (1627 S)
 [ 1] count:  351,  width: 1132 us [1124;1156]  ( 283 S)
 [ 2] count:    7,  width: 9468 us [9364;9592]  (2367 S)
 [ 3] count:    1,  width: 16264 us [16264;16264]       (4066 S)
Pulse timing distribution:
 [ 0] count:    9,  width: 4916 us [4912;4936]  (1229 S)
 [ 1] count:  357,  width:  364 us [340;432]    (  91 S)
 [ 2] count:  354,  width:  764 us [708;804]    ( 191 S)
 [ 3] count:    9,  width: 1588 us [1588;1596]  ( 397 S)
 [ 4] count:    7,  width: 9016 us [8848;9252]  (2254 S)
 [ 5] count:    1,  width: 15548 us [15548;15548]       (3887 S)
 [ 6] count:    1,  width: 49364 us [49364;49364]       (12341 S)
Level estimates [high, low]:  10026,      5
RSSI: -2.1 dB SNR: 33.0 dB Noise: -35.2 dB
Frequency offsets [F1, F2]:    3181,      0     (+12.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB03907021334016C02FC063423383CBCC0D48392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A1929292A1929292A455+AAB03907011334016C02FC063423383CBCC0D48392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A1929292A1929292A555+AAB03907051334016C02FC063423383CBCC0D48392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A1929292A1A1A1A19455+AAB03907011334016C02FC063423383CBCC0D48392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A1929292A1A1A1A19655
Attempting demodulation... short_width: 344, long_width: 716, reset_limit: 15552, sync_width: 4916
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=344,l=716,r=15552,g=0,t=0,y=4916'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 9
[00] {40} a9 c7 fe fe ee : 10101001 11000111 11111110 11111110 11101110
[01] {40} a9 c7 fe fe ee : 10101001 11000111 11111110 11111110 11101110
[02] {40} a9 c7 fe fe ee : 10101001 11000111 11111110 11111110 11101110
[03] {40} a9 c7 fe fe e1 : 10101001 11000111 11111110 11111110 11100001
[04] {40} a9 c7 fe fe e1 : 10101001 11000111 11111110 11111110 11100001
[05] {40} a9 c7 fe fe e1 : 10101001 11000111 11111110 11111110 11100001
[06] {40} a9 c7 fe fe e1 : 10101001 11000111 11111110 11111110 11100001
[07] {40} a9 c7 fe fe e1 : 10101001 11000111 11111110 11111110 11100001
[08] {40} a9 c7 fe fe e1 : 10101001 11000111 11111110 11111110 11100001

```

Close

```
Detected OOK package    2020-11-01 13:12:26
Analyzing pulses...
Total count:  369,  width: 532.16 ms            (133040 S)
Pulse width distribution:
 [ 0] count:    9,  width: 4928 us [4924;4944]  (1232 S)
 [ 1] count:  243,  width:  356 us [348;368]    (  89 S)
 [ 2] count:  117,  width:  728 us [720;736]    ( 182 S)
Gap width distribution:
 [ 0] count:    9,  width: 1576 us [1576;1580]  ( 394 S)
 [ 1] count:  237,  width:  772 us [764;796]    ( 193 S)
 [ 2] count:  114,  width:  404 us [396;424]    ( 101 S)
 [ 3] count:    8,  width: 8912 us [8564;9040]  (2228 S)
Pulse period distribution:
 [ 0] count:    9,  width: 6504 us [6500;6524]  (1626 S)
 [ 1] count:  351,  width: 1132 us [1120;1156]  ( 283 S)
 [ 2] count:    8,  width: 9408 us [9292;9552]  (2352 S)
Pulse timing distribution:
 [ 0] count:    9,  width: 4928 us [4924;4944]  (1232 S)
 [ 1] count:  357,  width:  372 us [348;424]    (  93 S)
 [ 2] count:  354,  width:  760 us [720;796]    ( 190 S)
 [ 3] count:    9,  width: 1576 us [1576;1580]  ( 394 S)
 [ 4] count:    8,  width: 8912 us [8564;9040]  (2228 S)
 [ 5] count:    1,  width: 49444 us [49444;49444]       (12361 S)
Level estimates [high, low]:  15939,      4
RSSI: -0.1 dB SNR: 36.0 dB Noise: -36.1 dB
Frequency offsets [F1, F2]:    3616,      0     (+13.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB03706031340017402F8062822D0C1248392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A19292A1A19292A1A455+AAB03706051340017402F8062822D0C1248392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A19292A1A1A1A1929455+AAB03706011340017402F8062822D0C1248392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A19292A1A1A1A1929555
Attempting demodulation... short_width: 356, long_width: 728, reset_limit: 9044, sync_width: 4928
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=356,l=728,r=9044,g=0,t=0,y=4928'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 9
[00] {40} a9 c7 fe fe cc : 10101001 11000111 11111110 11111110 11001100
[01] {40} a9 c7 fe fe cc : 10101001 11000111 11111110 11111110 11001100
[02] {40} a9 c7 fe fe cc : 10101001 11000111 11111110 11111110 11001100
[03] {40} a9 c7 fe fe c3 : 10101001 11000111 11111110 11111110 11000011
[04] {40} a9 c7 fe fe c3 : 10101001 11000111 11111110 11111110 11000011
[05] {40} a9 c7 fe fe c3 : 10101001 11000111 11111110 11111110 11000011
[06] {40} a9 c7 fe fe c3 : 10101001 11000111 11111110 11111110 11000011
[07] {40} a9 c7 fe fe c3 : 10101001 11000111 11111110 11111110 11000011
[08] {40} a9 c7 fe fe c3 : 10101001 11000111 11111110 11111110 11000011
```

Stop

```
Detected OOK package    2020-11-01 13:13:27
Analyzing pulses...
Total count:  123,  width: 172.09 ms            (43022 S)
Pulse width distribution:
 [ 0] count:    3,  width: 4932 us [4928;4944]  (1233 S)
 [ 1] count:   81,  width:  352 us [348;360]    (  88 S)
 [ 2] count:   39,  width:  724 us [720;728]    ( 181 S)
Gap width distribution:
 [ 0] count:    3,  width: 1580 us [1580;1584]  ( 395 S)
 [ 1] count:   81,  width:  776 us [772;796]    ( 194 S)
 [ 2] count:   36,  width:  408 us [404;424]    ( 102 S)
 [ 3] count:    2,  width: 8840 us [8836;8844]  (2210 S)
Pulse period distribution:
 [ 0] count:    3,  width: 6512 us [6508;6524]  (1628 S)
 [ 1] count:  117,  width: 1132 us [1124;1152]  ( 283 S)
 [ 2] count:    2,  width: 9568 us [9564;9572]  (2392 S)
Pulse timing distribution:
 [ 0] count:    3,  width: 4932 us [4928;4944]  (1233 S)
 [ 1] count:  117,  width:  368 us [348;424]    (  92 S)
 [ 2] count:  120,  width:  760 us [720;796]    ( 190 S)
 [ 3] count:    3,  width: 1580 us [1580;1584]  ( 395 S)
 [ 4] count:    2,  width: 8840 us [8836;8844]  (2210 S)
 [ 5] count:    1,  width: 49444 us [49444;49444]       (12361 S)
Level estimates [high, low]:  15957,      4
RSSI: -0.1 dB SNR: 36.0 dB Noise: -36.1 dB
Frequency offsets [F1, F2]:    2726,      0     (+10.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB03706021344017002F8062C2288C1248392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A192A192A192A192A455+AAB03706011344017002F8062C2288C1248392A192A192A1A1929292A1A1A192929292929292929292A192929292929292A192A192A192A192A555
Attempting demodulation... short_width: 352, long_width: 724, reset_limit: 8848, sync_width: 4932
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=352,l=724,r=8848,g=0,t=0,y=4932'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 3
[00] {40} a9 c7 fe fe aa : 10101001 11000111 11111110 11111110 10101010
[01] {40} a9 c7 fe fe aa : 10101001 11000111 11111110 11111110 10101010
[02] {40} a9 c7 fe fe aa : 10101001 11000111 11111110 11111110 10101010
```

Save some files (attach it)

A record channel 0,1,2 and 15. If need more, please ask and record more.
[channel_sample.zip](https://github.com/merbanan/rtl_433/files/5470805/channel_sample.zip)

zuckschwerdt, 2020-11-01:
The PWM suggestion looks good. You can check in the given https://triq.org/pdv/ ... url.
Write a conf file (copy a conf/ examples) with a flex decoder of `-X 'n=Dooya-Curtain,m=OOK_PWM,s=350,l=750,y=4900,r=9900'

Seems like the first 4 bytes give an ID (`a9 c7 fe fe`) and the last byte is a command
- Open: {40} a9 c7 fe fe ee / {40} a9 c7 fe fe e1
- Close: {40} a9 c7 fe fe cc / {40} a9 c7 fe fe c3
- Stop: {40} a9 c7 fe fe aa

Something like this will work:
```
    get=@0:{32}:id,
    get=@32:{8}:button,
```

Press buttons on all channels and see where that maps.

zuckschwerdt, 2020-11-01:
Get back with a summary of results for more help or just PR the documented .conf file

vampywiz17, 2020-11-01:
[all_buttons.zip](https://github.com/merbanan/rtl_433/files/5470856/all_buttons.zip)

Here all buttons.

the order is open/close/stop 0-15 (sometime it recognise a different device the close buttons, so i push it again)

zuckschwerdt, 2020-11-01:
Looks like this:
```
    invert
    get=@0:{24}:id,
    get=@28:{4}:channel
    get=@32:{4}:button:[1:open 3:close 5:stop],
,
```
The last nibble is the button number plus possibly some repeat indicator.

closed 2020-11-01 by zuckschwerdt

vampywiz17, 2020-11-01:
@zuckschwerdt 

It seems to be good,  thanks. one problem, that it recognise some noise...

```
time      : 2020-11-01 15:07:47
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 121           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:48
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:48
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:48
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:49
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:49
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:49
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:50
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:50
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:50
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:51
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:51
model     : Dooya-Curtain count     : 3            num_rows  : 3             len       : 40            data      : 5638010555    id        : 5650433
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-01 15:07:51
model     : Dooya-Curtain count     : 1            num_rows  : 1             len       : 122           data      : 0000000000000000000000000000000         id        : 0
channel   : 0            button    : 0             check     : 0
```

It possible to fix it?

zuckschwerdt, 2020-11-01:
Add `match=563801,` to only decode your fixed ID.

martin072, 2022-07-25:
Hi all, 

I have a Dooya DC2702 15-channel remote, do you know if this works as well? I briefly tried starting up rtl_433, but it did not detect the remote when pushing the buttons. Would there anything I need to configure or load? Or do I need to do an analyse for this remote?

zuckschwerdt, 2022-07-25:
You need to load [dooya_curtain.conf](https://github.com/merbanan/rtl_433/blob/master/conf/dooya_curtain.conf) with `-c`

martin072, 2022-07-25:
Thanks for the quick answer, will give it a try when I am home. 
I assume, rtl_433 als keeps on monitoring for other signals as well then? I would like to use the remote + a 433mhz weather station

martin072, 2022-07-30:
> 

All works perfect, but I seem to get a lot of "ghost" triggers. Anyone knows what this is, or how to filter these out? (see attached picture)
[image]

zuckschwerdt, 2022-07-30:
That's some other signal that vaguely passes as that protocol. You can further narrow the protocol by e.g. adding `bits<=41,` or be really strict and change the existing length check to `bits=40,`
Lastly you can also filter on the id, if that's the only sender you want to receive: `match=fa3807`

martin072, 2022-07-30:
Much appreciated!

commit afaefc6d (2020-11-01): Add Dooya Curtain Remote conf (closes #1545)