merbanan/rtl_433 #1550: Infineon SP40 Sensor

CaelanK, 2020-11-03:
Hello, big user of this project, have used it for endless applications, so I am fairly familiar with the process and features.... but forgive me if I get anything wrong.

I expect this to be similar to #1534 & #1007.

I have some Infineon SP40 sensors which I am trying to interpret. I believe the Jansite ones use the previous SP37 chips.

Right now I am not getting much useful information from the sensors when following all of the previous recommendations on the issues I mentioned before.

merbanan, 2020-11-03:
I suspect one needs to reverse engineer the receiver to get hold of the protocol for these type of devices

doomedtofail, 2020-11-03:
I do actually have access to the on-air protocol for these sensors, so I can sense check the data. This is how I know the current info is awry!

doomedtofail, 2020-11-03:
```
rtl_433 version 20.02-223-gc72b561 branch master at 202011022144 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 0 out of 173 device decoding protocols [ ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Detected FSK package	2020-11-03 15:38:11
Analyzing pulses...
Total count:  139,  width: 16.17 ms		( 4042 S)
Pulse width distribution:
 [ 0] count:  139,  width:   48 us [40;52]	(  12 S)
Gap width distribution:
 [ 0] count:   17,  width:  152 us [152;156]	(  38 S)
 [ 1] count:  121,  width:   52 us [52;64]	(  13 S)
Pulse period distribution:
 [ 0] count:   17,  width:  204 us [204;208]	(  51 S)
 [ 1] count:  121,  width:  104 us [92;116]	(  26 S)
Pulse timing distribution:
 [ 0] count:  255,  width:   48 us [40;56]	(  12 S)
 [ 1] count:   18,  width:  156 us [152;172]	(  39 S)
 [ 2] count:    5,  width:   60 us [60;64]	(  15 S)
Level estimates [high, low]:  15401,    675
RSSI: -0.3 dB SNR: 13.6 dB Noise: -13.9 dB
Frequency offsets [F1, F2]:   25755,  -3358	(+98.3 kHz, -12.8 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
view at https://triq.org/pdv/#AAB1030030009C003C8180828080808080818080808080808081808080808080808180808080808080818080808080808081808080808080808180808080808080828081808080808080808180808080808280818080808080808081808080808080808180808080808080818080808080808081808080808080808180808080808080818082808080808081808280808080808155
Attempting demodulation... short_width: 52, long_width: 152, reset_limit: 160, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PPM,s=52,l=152,g=160,r=160'
pulse_demod_ppm(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {138} 80 80 80 80 80 80 80 20 20 20 20 20 20 20 20 20 20 00 

Detected OOK package	2020-11-03 15:38:11
Analyzing pulses...
Total count:   10,  width: 7.10 ms		( 1775 S)
Pulse width distribution:
 [ 0] count:    1,  width:  144 us [144;144]	(  36 S)
 [ 1] count:    2,  width:  588 us [532;648]	( 147 S)
 [ 2] count:    2,  width:  228 us [224;232]	(  57 S)
 [ 3] count:    4,  width: 1056 us [956;1156]	( 264 S)
 [ 4] count:    1,  width:  336 us [336;336]	(  84 S)
Gap width distribution:
 [ 0] count:    1,  width:   64 us [64;64]	(  16 S)
 [ 1] count:    8,  width:   84 us [84;92]	(  21 S)
Pulse period distribution:
 [ 0] count:    1,  width:  208 us [208;208]	(  52 S)
 [ 1] count:    2,  width:  676 us [616;736]	( 169 S)
 [ 2] count:    2,  width:  312 us [308;320]	(  78 S)
 [ 3] count:    3,  width: 1176 us [1144;1244]	( 294 S)
 [ 4] count:    1,  width:  420 us [420;420]	( 105 S)
Pulse timing distribution:
 [ 0] count:    1,  width:  144 us [144;144]	(  36 S)
 [ 1] count:    2,  width:  588 us [532;648]	( 147 S)
 [ 2] count:    2,  width:  228 us [224;232]	(  57 S)
 [ 3] count:    4,  width: 1056 us [956;1156]	( 264 S)
 [ 4] count:    1,  width:  336 us [336;336]	(  84 S)
 [ 5] count:    1,  width:   64 us [64;64]	(  16 S)
 [ 6] count:    8,  width:   84 us [84;92]	(  21 S)
 [ 7] count:    1,  width: 11564 us [11564;11564]	(2891 S)
Level estimates [high, low]:  15915,    687
RSSI: -0.1 dB SNR: 13.6 dB Noise: -13.8 dB
Frequency offsets [F1, F2]:   16004,      0	(+61.1 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB1080090024C00E404200150004000542D2C8596A6A6B6B696C6B6B755

```

zuckschwerdt, 2020-11-03:
PCM, bit-width 84 µs should decode that. E.g. `-X 'n=name,m=OOK_PCM,s=84,l=84,r=1700'`

doomedtofail, 2020-11-03:
Massive thanks for the support both!

Output of: `-X 'n=name,m=OOK_PCM,s=84,l=84,r=1700'`

[image]

```
time      : 2020-11-03 15:54:15
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 139          data      : ffffffffd75df77ddfffbf77fff77e00000
codes     : {139}ffffffffd75df77ddfffbf77fff77e00000
```

The data I would have expected to see within this frame above is:
80 64 3B 7C 83 83

Will try 50, 42 right away.

zuckschwerdt, 2020-11-03:
To really pin down the timing parameters record (`-S unknown`) a good sample (try https://triq.org/iqs) and study it (maybe drop as a zip here). I suspect the modulation isn't picked up cleanly.

doomedtofail, 2020-11-03:
Ok sorry, I power-cycled my receiver and got some interesting data with 42u

I shall also give the `-S unknown` a whirl.

The data bears a striking resemblance to this:
[https://github.com/merbanan/rtl_433/issues/1007#issuecomment-472160379](https://github.com/merbanan/rtl_433/issues/1007#issuecomment-472160379)

[image]

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-03 16:08:45
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 491          data      : aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa996aa69a555556555aa96a9a95aaaa5555e0000000000
codes     : {491}aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa996aa69a555556555aa96a9a95aaaa5555e0000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-03 16:09:45
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 491          data      : aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa996aa69a555556555aa96a9a6aaaaa5a99e0000000000
codes     : {491}aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa996aa69a555556555aa96a9a6aaaaa5a99e0000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-11-03 16:10:14
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 491          data      : aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa99a6596a695a59555aa96a9a6aaaaa9a56e0000000000
codes     : {491}aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa99a6596a695a59555aa96a9a6aaaaa9a56e0000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-11-03 16:10:44
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 468          data      : aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab555555555555555555554cb5534d2aaaab2aad54b54d4ad5552aaaf0000000000
codes     : {468}aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaab555555555555555555554cb5534d2aaaab2aad54b54d4ad5552aaaf0000000000
```

merbanan, 2020-11-03:
> I do actually have access to the on-air protocol for these sensors, so I can sense check the data. This is how I know the current info is awry!

Can you share the protocol? Or details of it?

zuckschwerdt, 2020-11-03:
Fast switches of bits, like `aa`or `55` will work with a wide range of timings. The problem is with longer runs of 0's or 1's -- we need an exact timing there to known how many bits that was. E.g. 200 µs pulse is detected as 5 bits at 42 µs bit-width, but 4 bits at 50 µs…

zuckschwerdt, 2020-11-03:
Ah, wait a second. If you see lots of a, 5, 6, 9s then that indicates Manchester code which will be much easier than plain PCM.
Just pop a data code into https://triq.net/bitbench with a format of just `v` to visualize the bit runs.

doomedtofail, 2020-11-03:
.......After a full day of poking at this, Manchester decoding has fully solved this for me....

Last piece of the neatness puzzle would be being able to remove preamble from the data within rtl_433 or is that more of a post-processing control?

So:
`aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa996aa69a555556555aa96a9a6aaaaa5a99e0000000000`

Becomes:
`996aa69a555556555aa96a9a6aaaaa5a99e0000000000`

zuckschwerdt, 2020-11-03:
A preamble match of `55 54` would be best suited here: on -X add `,preamble=5554`
Also try to use `OOK_MC_ZEROBIT` instead of `OOK_PCM` to decode the MC.

merbanan, 2021-03-07:
I just added a decoder for a model of Jansite Solar TPMS sensors. Maybe it can handle this also. If not reopen with some sample signals.

closed 2021-03-07 by merbanan