merbanan/rtl_433 #1551: High cpu on raspberry PI 2

jamaisx, 2020-11-03:
I'm experiences fixed 50% cpu when using this command:

rtl_433 -f 868M -R119 -F json -M utc

-vvv doesn't give me any hint

lsusb reports:

Bus 001 Device 021: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

The Bresser 5in1 data is decode successfully, but the cpu work seems to high for me, any ideas?

zuckschwerdt, 2020-11-03:
This is expected (50% of one core, i.e. full load is 400% on the 4-core Raspi). FSK decoding is the main task here.

merbanan, 2020-11-03:
Do you know the code path that takes most of the time? Is it the dsp stuff?

zuckschwerdt, 2020-11-03:
I did some profiling way back, but on an i7 not ARM-v7. 50% is what I'm used to see on my Raspis (100% on the RPi 1).
I remember @evilpete did some extensive profiling a short time ago?

merbanan, 2020-11-03:
Is 100% with -O3 enabled? I would think that inlining would give quite a speedup.

zuckschwerdt, 2020-11-03:
The 100% was on a RPi 1: 700MHz single core ARMv6 :)

merbanan, 2020-11-03:
Well, I'm running on mips cores not much faster frequency wise (Mediatek 7621) that reports a load of ca 15%.

zuckschwerdt, 2020-11-03:
I don't have the RPi 1s anymore, but I'll profile on a RPi 3 then.

rct, 2020-11-03:
Are these numbers with FSK decoding?

FWIW, I guess I don't have anything doing FSK,  just pulled the current code and built on an RPI 1. Still seems to be around 10-12% of the single core without enabling/disabling any specific protocols.

jamaisx, 2020-11-04:
some tests (Raspberyy PI model B rev 2):

---------
rtl_433  (without any parameter)

Registered 139 out of 168 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers

CPU 14 %

-----------
Enabling only Bresser5in1 protocol (1 mb/s sample rate, FSK-PCM encoding)
rtl_433 -f 868M -R119 -F json -M utc

Registered 1 out of 168 device decoding protocols [ 119 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.000MHz.
Allocating 15 zero-copy buffers

CPU 53 %

--------------
Enabling only "ambientweather_tx8300" (112)  protocol

rtl_433 -R 112 -F json -M utc

rtl_433 version 20.02-185-g46041c5 branch master at 202010101031 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 168 device decoding protocols [ 112 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers

CPU 8 %

zuckschwerdt, 2020-11-04:
Ah, thanks! There is the clue hidden I was missing.
- I run at a sample rate of **1M** too: **50% with FSK**.
- Very roughly, quarter of that rate (the 433.92M default of **250k**) is a quarter of cpu: **14%**
- And bare bones **250k only OOK**: **8%**

jamaisx, 2020-11-04:
So the device works in streaming mode? I supposed something event driven, sorry I'm a noob here...

zuckschwerdt, 2020-11-04:
Event driven would be with something like an CC1101, a Sonoff bridge, a CUL, … delivering demodulated data packets. rtl_433 streams the raw I/Q rf signal and demodulating that is the load you see. Actual decoding (as events) isn't a noticeable load. Hardware or off-loaded solutions need fixed demodulation (format and timings) or even decoding parameters, rtl_433 can try to demodulate and match everything.

rct, 2020-11-06:
> 
>  *  I run at a sample rate of 1M too: 50% with FSK.
>  *  Very roughly, quarter of that rate (the 433.92M default of 250k) is a quarter of cpu: 14%
>  * And bare bones 250k only OOK: 8%

So with no parameters is rtl_433 doing FSK at 250ksps?   That's the 14% cpu case that @jamaisx listed first. 

Is the difference between 1 decoder at 8% and 139 decoders at 14% due to FSK or just the overhead of attempting to match for the 139 decoders?

merbanan, 2020-11-06:
FSK demod, the decoder matching code is insignificant compared to the dsp code that does things per sample.

merbanan, 2021-03-22:
Everything works like it was designed. There is room for optimisation as always. Closing issue.

closed 2021-03-22 by merbanan