merbanan/rtl_433 #1554: Cardin s500 433Mhz

jochimsergoyne, 2020-11-05:
I see that RTL supports a Cardin S466-TX2. I know it's a different protocol than the Cardin s500. I try to read the signal so I can send it out and integrate it with MQTT, but I do not get any signal on rtl_433. Any help from someone or suggestions?

zuckschwerdt, 2020-11-05:
Try `-A`. Otherwise `-S unknown` then https://triq.org/iqs/ and `-w FILENAME.ook FILENAME.cu8` then https://triq.org/pdv/
Look at https://github.com/merbanan/rtl_433/tree/master/conf and create something similar.

jochimsergoyne, 2020-11-05:
Thanks for your input!! That gave me a lot of information. I am still trying to read through the documentation on how to interpret it correctly. It seems to be a rolling code at first sight...

zuckschwerdt, 2020-11-05:
There is some information on a "Cardin TXQ504C2" which mentions rolling codes. There might still be fixed bits for the buttons and the rolling code just for authentication. Try to build a `-X` line (like perhaps suggested in the `-A` output) and grab many codes -- there might be a pattern.

jochimsergoyne, 2020-11-06:
This is my first time trying to work with RF codes and integrate it with home assistant. I try to understand and see a pattern but got nowhere yet. Sorry to abuse this forum to try to get forward, but you are very friendly and helpful :-) Thanks for that!

Click #1

Detected FSK package    2020-11-06 08:35:49
Analyzing pulses...
Total count:  360,  width: 85.86 ms             (21466 S)
Pulse width distribution:
 [ 0] count:  325,  width:  104 us [96;128]     (  26 S)
 [ 1] count:   34,  width:  208 us [200;224]    (  52 S)
 [ 2] count:    1,  width:  520 us [520;520]    ( 130 S)
Gap width distribution:
 [ 0] count:    1,  width:  544 us [544;544]    ( 136 S)
 [ 1] count:  324,  width:  104 us [88;124]     (  26 S)
 [ 2] count:   33,  width:  212 us [208;224]    (  53 S)
 [ 3] count:    1,  width: 2004 us [2004;2004]  ( 501 S)
Pulse period distribution:
 [ 0] count:    1,  width:  668 us [668;668]    ( 167 S)
 [ 1] count:  307,  width:  208 us [200;224]    (  52 S)
 [ 2] count:   33,  width:  316 us [304;340]    (  79 S)
 [ 3] count:   17,  width:  420 us [412;440]    ( 105 S)
 [ 4] count:    1,  width: 2108 us [2108;2108]  ( 527 S)
Pulse timing distribution:
 [ 0] count:  649,  width:  104 us [88;128]     (  26 S)
 [ 1] count:   67,  width:  208 us [200;224]    (  52 S)
 [ 2] count:    2,  width:  532 us [520;544]    ( 133 S)
 [ 3] count:    1,  width: 2004 us [2004;2004]  ( 501 S)
 [ 4] count:    1,  width:    8 us [8;8]        (   2 S)
Level estimates [high, low]:  15876,   4770
RSSI: -0.1 dB SNR: 5.2 dB Noise: -5.4 dB
Frequency offsets [F1, F2]:    3209,  -5100     (+12.2 kHz, -19.5 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB0000501006800D0021407D40008828080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080908180908080819180908180808080908081808080808091809180919081808080809180809081918090808080818090818080919080818080808091809180908081919081809081908080808180808091808090808180918090818091919180918090808081809190808355+AAB00D0501006800D0021407D40008A455
Attempting demodulation... short_width: 104, long_width: 208, reset_limit: 2008, sync_width: 520
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=104,l=208,r=2008,g=0,t=0,y=520'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {359} ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff f6 eb ef ea 7d ab ee 7e ac d7 f6 eb 15 e6
[01] { 0}                                                                                                                                        :

Detected FSK package    2020-11-06 08:35:50
Analyzing pulses...
Total count:  725,  width: 175.13 ms            (43782 S)
Pulse width distribution:
 [ 0] count:    1,  width:  628 us [628;628]    ( 157 S)
 [ 1] count:  651,  width:  104 us [92;128]     (  26 S)
 [ 2] count:   68,  width:  208 us [200;240]    (  52 S)
 [ 3] count:    3,  width:  132 us [132;136]    (  33 S)
 [ 4] count:    1,  width: 1256 us [1256;1256]  ( 314 S)
 [ 5] count:    1,  width: 2424 us [2424;2424]  ( 606 S)
Gap width distribution:
 [ 0] count:  655,  width:  100 us [76;124]     (  25 S)
 [ 1] count:   66,  width:  212 us [204;224]    (  53 S)
 [ 2] count:    2,  width: 2020 us [2020;2020]  ( 505 S)
 [ 3] count:    1,  width:   76 us [76;76]      (  19 S)
Pulse period distribution:
 [ 0] count:    1,  width:  704 us [704;704]    ( 176 S)
 [ 1] count:  620,  width:  208 us [200;228]    (  52 S)
 [ 2] count:   66,  width:  316 us [304;336]    (  79 S)
 [ 3] count:   34,  width:  420 us [412;440]    ( 105 S)
 [ 4] count:    2,  width: 2120 us [2116;2128]  ( 530 S)
 [ 5] count:    1,  width: 1332 us [1332;1332]  ( 333 S)
Pulse timing distribution:
 [ 0] count:    1,  width:  628 us [628;628]    ( 157 S)
 [ 1] count: 1305,  width:  104 us [88;128]     (  26 S)
 [ 2] count:  134,  width:  208 us [200;240]    (  52 S)
 [ 3] count:    3,  width:  132 us [132;136]    (  33 S)
 [ 4] count:    1,  width: 1256 us [1256;1256]  ( 314 S)
 [ 5] count:    3,  width: 2152 us [2020;2424]  ( 538 S)
 [ 6] count:    2,  width:   76 us [76;76]      (  19 S)
[... 307 lines cut ...]
 [ 1] count:  111,  width:  208 us [200;228]    (  52 S)
 [ 2] count:    4,  width: 2112 us [1828;2220]  ( 528 S)
 [ 3] count:    1,  width:  132 us [132;132]    (  33 S)
Gap width distribution:
 [ 0] count:    1,  width:  324 us [324;324]    (  81 S)
 [ 1] count:  968,  width:  104 us [88;124]     (  26 S)
 [ 2] count:  111,  width:  212 us [204;224]    (  53 S)
 [ 3] count:    1,  width:   80 us [80;80]      (  20 S)
 [ 4] count:    1,  width:  140 us [140;140]    (  35 S)
Pulse period distribution:
 [ 0] count:   58,  width:  420 us [408;456]    ( 105 S)
 [ 1] count:  913,  width:  208 us [200;228]    (  52 S)
 [ 2] count:  108,  width:  316 us [304;336]    (  79 S)
 [ 3] count:    3,  width: 2324 us [2308;2360]  ( 581 S)
Pulse timing distribution:
 [ 0] count: 1936,  width:  104 us [80;132]     (  26 S)
 [ 1] count:  222,  width:  208 us [200;228]    (  52 S)
 [ 2] count:    4,  width: 2112 us [1828;2220]  ( 528 S)
 [ 3] count:    2,  width:  136 us [132;140]    (  34 S)
 [ 4] count:    1,  width:  324 us [324;324]    (  81 S)
 [ 5] count:    1,  width:    0 us [0;0]        (   0 S)
Level estimates [high, low]:  14052,   4458
RSSI: -0.7 dB SNR: 5.0 dB Noise: -5.7 dB
Frequency offsets [F1, F2]:     443,  -8051     (+1.7 kHz, -30.7 kHz)
Guessing modulation: No clue...

jochimsergoyne, 2020-11-06:
rtl_433 -X 'n=name,m=OOK_PWM,s=104,l=208,r=2008,g=0,t=0,y=520'
gives me the following, but only seems to respond to some clicks, not all of them (I always press the same button, but only sometimes something shows up)

time      : 2020-11-06 08:42:58
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-06 08:43:06
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-06 08:43:09
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 0            data      :
codes     : {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-06 08:43:13
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 2            data      : c
codes     : {2}c
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-11-06 08:43:24
model     : name         count     : 2             num_rows  : 2             rows      :
len       : 2            data      : c,
len       : 0            data      :
codes     : {2}c, {0}

jochimsergoyne, 2020-11-06:
I thought there was a pattern as seen below with the -X params:
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2008,g=0,t=0,y=520'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2040,g=0,t=0,y=436'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2008,g=0,t=0,y=520'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2040,g=0,t=0,y=436'

But when I ran it a couple of minutes later and just randomly pressed the buttons, I took the following one.
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2016,g=0,t=0,y=132'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2032,g=0,t=0,y=260'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2008,g=0,t=0,y=132'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2020,g=0,t=0,y=360'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2020,g=0,t=0,y=412'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2024,g=0,t=0,y=384'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2032,g=0,t=0,y=144'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2024,g=0,t=0,y=268'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2016,g=0,t=0,y=52'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2016,g=0,t=0,y=388'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2020,g=0,t=0,y=404'
-X 'n=name,m=OOK_PWM,s=104,l=208,r=2028,g=0,t=0,y=356'

Basically what I'm trying to do is send the same signal as that one button would send.

zuckschwerdt, 2020-11-06:
The `-X 'n=name,m=OOK_PWM,s=104,l=208,r=300'` looks suitable to grab the codes.
This will however not help you to reproduce this signal, there are more parameters in the RF signal.
Either analyze the raw .cu8 sample to get those parameters or just replay the .cu8 -- what is your TX hardware?

jochimsergoyne, 2020-11-06:
I have the RTL-SDR usb as well as a Sonoff RF bridge, flashed with Tasmota and Portisch

zuckschwerdt, 2020-11-06:
With Portisch you might be interested in the RfRaw codes given by `-A` specifically the `AAB0` commands: https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB0
But you need to analyze and shorten the given rfraw into a single code (note the `+` separating multiple commands).

jochimsergoyne, 2020-11-16:
I have tried many things in the mean time but no success. I read through the complete manual and got no wiser. One thing I tried to do is to duplicate the remote (following the manual, you press a button (like a reset button with a small pin on a router for example). If you do this for both (current receiver and new one), you can then copy the signal to the new remote. I do not have a new remote, but I caught the signal from the current remote with RTL 433 and got the following:
Detected OOK package    2020-11-16 13:22:11
Analyzing pulses...
Total count:   57,  width: 31.37 ms             ( 7842 S)
Pulse width distribution:
 [ 0] count:   26,  width:  248 us [224;296]    (  62 S)
 [ 1] count:   23,  width:  408 us [316;464]    ( 102 S)
 [ 2] count:    1,  width: 1460 us [1460;1460]  ( 365 S)
 [ 3] count:    1,  width:   44 us [44;44]      (  11 S)
 [ 4] count:    4,  width:  812 us [768;848]    ( 203 S)
 [ 5] count:    1,  width:  560 us [560;560]    ( 140 S)
 [ 6] count:    1,  width:  168 us [168;168]    (  42 S)
Gap width distribution:
 [ 0] count:   27,  width:  256 us [208;280]    (  64 S)
 [ 1] count:    5,  width:  176 us [156;196]    (  44 S)
 [ 2] count:   14,  width:   92 us [80;112]     (  23 S)
 [ 3] count:    7,  width:   68 us [60;76]      (  17 S)
 [ 4] count:    1,  width:  124 us [124;124]    (  31 S)
 [ 5] count:    2,  width:   48 us [44;52]      (  12 S)
Pulse period distribution:
 [ 0] count:   48,  width:  508 us [504;512]    ( 127 S)
 [ 1] count:    1,  width: 1520 us [1520;1520]  ( 380 S)
 [ 2] count:    1,  width:  380 us [380;380]    (  95 S)
 [ 3] count:    1,  width:  132 us [132;132]    (  33 S)
 [ 4] count:    4,  width: 1016 us [1016;1020]  ( 254 S)
 [ 5] count:    1,  width:  684 us [684;684]    ( 171 S)
Pulse timing distribution:
 [ 0] count:   53,  width:  252 us [208;296]    (  63 S)
 [ 1] count:   23,  width:  408 us [316;464]    ( 102 S)
 [ 2] count:    1,  width: 1460 us [1460;1460]  ( 365 S)
 [ 3] count:    3,  width:   44 us [44;52]      (  11 S)
 [ 4] count:    4,  width:  812 us [768;848]    ( 203 S)
 [ 5] count:    1,  width:  560 us [560;560]    ( 140 S)
 [ 6] count:    6,  width:  172 us [156;196]    (  43 S)
 [ 7] count:   14,  width:   92 us [80;112]     (  23 S)
 [ 8] count:    7,  width:   68 us [60;76]      (  17 S)
 [ 9] count:    1,  width:  124 us [124;124]    (  31 S)
 [10] count:    1,  width: 14604 us [14604;14604]       (3651 S)
Level estimates [high, low]:  15744,   3535
RSSI: -0.2 dB SNR: 6.5 dB Noise: -6.7 dB
Frequency offsets [F1, F2]:   -3516,      0     (-13.4 kHz, +0.0 kHz)
Guessing modulation: No clue...

and a second time to compare:
Detected OOK package    2020-11-16 13:22:30
Analyzing pulses...
Total count:   93,  width: 50.60 ms             (12649 S)
Pulse width distribution:
 [ 0] count:   53,  width:  408 us [328;460]    ( 102 S)
 [ 1] count:    4,  width:  840 us [828;856]    ( 210 S)
 [ 2] count:   33,  width:  260 us [228;296]    (  65 S)
 [ 3] count:    2,  width:  592 us [588;600]    ( 148 S)
 [ 4] count:    1,  width: 1312 us [1312;1312]  ( 328 S)
Gap width distribution:
 [ 0] count:   33,  width:  100 us [88;124]     (  25 S)
 [ 1] count:   37,  width:  240 us [180;280]    (  60 S)
 [ 2] count:   12,  width:   68 us [56;76]      (  17 S)
 [ 3] count:    3,  width:   44 us [44;44]      (  11 S)
 [ 4] count:    7,  width:  156 us [140;176]    (  39 S)
Pulse period distribution:
 [ 0] count:   85,  width:  508 us [500;520]    ( 127 S)
 [ 1] count:    4,  width: 1016 us [1016;1020]  ( 254 S)
 [ 2] count:    2,  width:  676 us [676;680]    ( 169 S)
 [ 3] count:    1,  width: 1524 us [1524;1524]  ( 381 S)
Pulse timing distribution:
 [ 0] count:   53,  width:  408 us [328;460]    ( 102 S)
 [ 1] count:    4,  width:  840 us [828;856]    ( 210 S)
 [ 2] count:   67,  width:  252 us [212;296]    (  63 S)
 [ 3] count:    2,  width:  592 us [588;600]    ( 148 S)
 [ 4] count:    1,  width: 1312 us [1312;1312]  ( 328 S)
 [ 5] count:   33,  width:  100 us [88;124]     (  25 S)
 [ 6] count:   10,  width:  164 us [140;188]    (  41 S)
 [ 7] count:   12,  width:   68 us [56;76]      (  17 S)
 [ 8] count:    3,  width:   44 us [44;44]      (  11 S)
 [ 9] count:    1,  width: 13124 us [13124;13124]       (3281 S)
Level estimates [high, low]:  15750,   3382
RSSI: -0.2 dB SNR: 6.7 dB Noise: -6.9 dB
Frequency offsets [F1, F2]:   -1131,      0     (-4.3 kHz, +0.0 kHz)
Guessing modulation: No clue...

Is this something to check or not worth it?

zuckschwerdt, 2020-11-16:
Looks like a valid PWM signal. But it's now OOK, it was FSK before? For FSK you would need:
 `-X 'n=name,m=FSK_PWM,s=104,l=208,r=300'`
Generally those SNR / Noise levels look bad. Usually you'd want them way bigger, like 20 dB. Our default minimum SNR is 9 dB, not sure this is working at all.

jochimsergoyne, 2020-11-16:
This signal was generated to replicate the signal from an existing remote to a new remote. Or is this not the direction I should be looking towards? All looks OOK to me?

zuckschwerdt, 2020-11-16:
The original was FSK with maybe a 30k-50k deviation, wasn't it? My initial hints were likely bad, I didn't spot he `Detected FSK package` there.
And you had 100/200µs pulses/gaps which could perhaps be Manchester. Your new signal is 260/400 µs OOK.

jochimsergoyne, 2020-11-16:
.\rtl_433.exe -X 'n=name,m=FSK_PWM,s=104,l=208,r=2008,g=0,t=0,y=520'

**Pushing it once:**
time      : 2020-11-16 15:40:13
model     : name         count     : 4             num_rows  : 4             rows      :
len       : 359          data      : fffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb779e6de7d875cff9c9f5dff5ca,
len       : 365          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedde79b79f61d73fe727d77fd728,
len       : 365          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedde79b79f61d73fe727d77fd728,
len       : 0            data      :
codes     : {359}fffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb779e6de7d875cff9c9f5dff5ca, {365}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedde79b79f61d73fe727d77fd728, {365}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedde79b79f61d73fe727d77fd728, {0}

**pushed it a second time**
time      : 2020-11-16 15:40:35
model     : name         count     : 4             num_rows  : 4             rows      :
len       : 358          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4,
len       : 366          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4,
len       : 366          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4,
len       : 0            data      :
codes     : {358}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4, {366}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4, {366}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4, {0}

**pushed it a third time**
time      : 2020-11-16 15:41:02
model     : name         count     : 4             num_rows  : 4             rows      :
len       : 346          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff6ed756ab57c5d7beda2a0dc14,
len       : 357          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffeddaead56af8baf7db4541b828,
len       : 357          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffeddaead56af8baf7db4541b828,
len       : 0            data      :
codes     : {346}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff6ed756ab57c5d7beda2a0dc14, {357}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffeddaead56af8baf7db4541b828, {357}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffeddaead56af8baf7db4541b828, {0}

**pushed a fourth time**
time      : 2020-11-16 15:41:36
model     : name         count     : 4             num_rows  : 4             rows      :
len       : 355          data      : fffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb7630f62dfa3c569d99efaaffe,
len       : 361          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedd8c3d8b7e8f15a7667beabff8,
len       : 361          data      : ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedd8c3d8b7e8f15a7667beabff8,
len       : 0            data      :
codes     : {355}fffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb7630f62dfa3c569d99efaaffe, {361}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedd8c3d8b7e8f15a7667beabff8, {361}ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedd8c3d8b7e8f15a7667beabff8, {0}

zuckschwerdt, 2020-11-16:
That looks good, see [this BitBench](https://triq.net/bitbench?c=%7B359%7Dfffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb779e6de7d875cff9c9f5dff5ca&c=%7B365%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedde79b79f61d73fe727d77fd728&c=%7B365%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedde79b79f61d73fe727d77fd728&c=%7B358%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4&c=%7B366%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4&c=%7B366%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedddfb9dbd9eabbce85fbf76adf4&c=%7B346%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff6ed756ab57c5d7beda2a0dc14&c=%7B357%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffeddaead56af8baf7db4541b828&c=%7B357%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffeddaead56af8baf7db4541b828&c=%7B355%7Dfffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffb7630f62dfa3c569d99efaaffe&c=%7B361%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedd8c3d8b7e8f15a7667beabff8&c=%7B361%7Dffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffedd8c3d8b7e8f15a7667beabff8&f=8h%20&a=Sync&m=ffb7&i=true).
But the payload could be encoded with a rolling code (observe the first variable byte counting up).

gdt, 2023-09-29:
Status and plan for this issue?  Is someone working on a PR?