merbanan/rtl_433 #1558: Noise that i can not recognise it.

vampywiz17, 2020-11-09:
Hello there,

sometimes for hours i catch this noise https://triq.org/pdv/#AAB1050050003C002C00BC27148383838383838383838393838383938383838383838393838383938383838383839383A38383838383838383938383838383838383838383838383838383838383838383838383838383839383838383839383838383838383838383838383839383838383838383838383838383838383838383838383838455

I catch it every second,.. It really hard to find out the "usefull" signal. I try to use ferit ring USB cable, try to use USB hub, Try a other PC, but it not help at all...

The hardware is a Nooelec NESDR SMArt. 

Sometimes it not see this noise about hours and i use the device well.

Any idea what cause it?

Mindavi, 2020-11-09:
Walking around with an antenna could help pinpoint the general direction of the signal (signal will get stronger the closer you come to it). Using a tool like gqrx or sdr# or something like that to visualise the signal 'live' will help.

This could be anything. I'm not too well versed with the rf space per se, but I know things like solar inverters or 'homeplugs' (ethernet over power cables) can generate quite a lot of noise.

The 1 second time between transmissions does point to something different than a 'constant' polluter though. I really wouldn't know.

You could also check if there's any pattern to the data, does it happen in the day? Does it happen in the night? Does it happen around dinner time?

There's not much more I can do for you, hope you can find the issue and stop the noise from polluting your measurements.

Mindavi, 2020-11-11:
Closing this because I don't think we can provide any more help than my tips.

closed 2020-11-11 by Mindavi