merbanan/rtl_433 #1560: Digoo DG-R8H

Ruderbaker, 2020-11-12:
I have 3 Digoo DG-R8H sensors but i cannot see data from any of them. I also have 4 Acurite and 1 Fine Offset sensors that are working fine.

Do I need to enable a protocol for them to work?

zuckschwerdt, 2020-11-12:
I don't think we have support specifically for a "Digoo DG-R8H". It might be a rebranded device or use another known protocol.
Make sure you are receiving the transmission. Use CubicSDR, Gqrx, or SDRangel to see if the transmissions are coming in and are near 433.92M. Otherwise run `rtl_433 -A` and see if they show up. Then record samples with `rtl_433 -S unknown` and inspect those. Use https://triq.org/iqs/ for the .cu8 files and `rtl_433 -w FILENAME.ook FILENAME.cu8` to get .ook files and use them on https://triq.org/pdv/

lslomczynski, 2020-11-14:
Hello!

I have a Digoo DG-R8H from AliExpress with declared frequency on the sticker: 433.92 and I am using it with rtl_433 in Node-RED.
msg.payload.model says: "Nexus-TH" and gives both temperature and relative humidity readings.

Have you placed it close to receiver? From my experience the signal emitted by the unit is kind of weak and I had to put it behind the window of the room where the receiver is to get the readings, while my other sensor from a different manufacturer running on 2xAA batteries works from across my flat without any issues (where Digoo couldn't).

merbanan, 2021-03-22:
No sample signals where ever posted. Closing issue.

closed 2021-03-22 by merbanan