merbanan/rtl_433 #1561: Telldus FT007TH / 313169 misidentification (aka Renkforce, Ambient Weather, Frogit,)

justinphelps, 2020-11-14:
I have been using rtl_433 ( https://www.domoticz.com/forum/viewtopic.php?t=27310 ) with the built in mqtt client. 
rtl_433 detects them as  Fineoffset-TelldusProove (probably WH58)
I had 3 devices, each device had a unique ID according to rtl_433 and decided to add more. 
The FT007TH actually uses dip switches to assign 1-8 channels, not IDs, but it worked fine with 3 devices and the unique IDs.
I got more FT007TH hygrothermometers and assigned unique channel to one of them, but the the new device was assigned the same ID as another existing device, despite having different channel settings via the dip switches. 
RTL_433 is correctly decoding the Temperature, Humidity and CRC. from all the devices despite the ID collision.
I've tried changing the dip switches, but ended up with other ID collisions. 

It appears that some other projects have possibly decoded the protocol specific to the FT007TH. 
Here: https://github.com/Xorok/FT007TH-Arduino-Basestation/blob/master/WebServerSensorDemo.ino
or here: https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/14_SD_WS.pm

Here is the summary of one device having ID 135 (g009_433.92M_250k.ook)
time      : 2020-11-14 01:19:21
model     : Fineoffset-TelldusProove               ID        : 135
Temperature: 4.6 C       Humidity  : 65 %          Integrity : CRC
*** Saving signal to file g009_433.92M_250k.cu8 (59794 samples, 131072 bytes)

Then another: (g011_433.92M_250k.ook)
time      : 2020-11-14 01:20:57
model     : Fineoffset-TelldusProove               ID        : 135
Temperature: 4.4 C       Humidity  : 65 %          Integrity : CRC
*** Saving signal to file g011_433.92M_250k.cu8 (59897

[ft007th_ooks.zip](https://github.com/merbanan/rtl_433/files/5540005/ft007th_ooks.zip)
 samples, 131072 bytes)

justinphelps, 2020-11-14:
via the dip switches, the samples were set to 
g009  channel 3 (up up down) 
g011  channel 7 (down up down)

merbanan, 2020-11-14:
rc/rtl_433 -r ~/temp/g011_433.92M_250k.ook
[00] {49} ff a4 38 16 20 db 80 : 11111111 10100100 00111000 00010110 00100000 11011011 1
model     : Fineoffset-TelldusProove               ID        : 135
Temperature: 4.4 C       Humidity  : 65 %          Integrity : CRC

src/rtl_433 -r ~/temp/g009_433.92M_250k.ook
[00] {49} ff a4 38 17 20 b7 00 : 11111111 10100100 00111000 00010111 00100000 10110111 0
model     : Fineoffset-TelldusProove               ID        : 135
Temperature: 4.6 C       Humidity  : 65 %          Integrity : CRC

From the documentation:
The data is grouped in 6 bytes / 12 nibbles.

    [pre] [pre]    [type] [id]     [id] [temp]     [temp] [temp]      [humi] [humi]     [crc] [crc]

I would think that the type nibble holds the battery state also, but it is the same in both payloads.  Anyway one can clearly see that the only difference is in the temperature bit and crc. And the values decoded properly so the crc value is fine. Thus there are no bits unaccounted for that can represent the dip switch state you posted.

So someone is lying, it is either me, you, rtl_433 or the sensor. My bet is either the sensor or the test setup. Maybe you recorded the wrong sensor? Hopefully your sensor is just not supported and rtl_433 picks up one from some where else.

merbanan, 2020-11-14:
So remove the antenna and lower the gain value (-g). Then place the sensors nearby and try again with -A and -S known/unknown.

justinphelps, 2020-11-14:
I think it's me. I realize now that I've sent captures from the same device. Will re-capture and send new ones

justinphelps, 2020-11-14:
This time I put 3 sensors in very different climates so they'd be easy to differentiate. 
I'm my new attempt to scan and capture it appears the new devices that I bought this week have different firmware despite having identical model numbers. 
The original devices that I bought over a year ago is being identified as  Fineoffset-TelldusProove, the new devices are  Fineoffset-WH2. 
 I guess it's less of an issue for you unless you want to represent ID as the Telldus # , so  I'm attaching my captures 

time      : 2020-11-14 13:41:30
model     : Fineoffset-TelldusProove               ID        : 167

Temperature: 10.4 C      Humidity  : 50 %          Integrity : CRC
*** Saving signal to file g011_433.92M_250k.cu8 (60311 samples, 131072 bytes)

time      : 2020-11-14 13:42:27
model     : Fineoffset-WH2                         ID        : 135
Temperature: 22.7 C      Humidity  : 47 %          Integrity : CRC
*** Saving signal to file g014_433.92M_250k.cu8 (59610 samples, 131072 bytes)
*** Saving signal to file g015_433.92M_250k.cu8 (59081 samples, 131072 bytes)
*** Saving signal to file g016_433.92M_250k.cu8 (36300 samples, 131072 bytes)
*** Saving signal to file g017_433.92M_250k.cu8 (59416 samples, 131072 bytes)
*** Saving signal to file g018_433.92M_250k.cu8 (59564 samples, 131072 bytes)
*** Saving signal to file g019_433.92M_250k.cu8 (59599 samples, 131072 bytes)

time      : 2020-11-14 13:44:27
model     : Fineoffset-TelldusProove               ID        : 231
Temperature: 36.8 C      Humidity  : 27 %          Integrity : CRC
*** Saving signal to file g020_433.92M_250k.cu8 (59578 samples, 131072 bytes)
*** Saving signal to file g021_433.92M_250k.cu8 (35707 samples, 131072 bytes)

[Telldus captures.zip](https://github.com/merbanan/rtl_433/files/5540723/Telldus.captures.zip)

merbanan, 2020-11-14:
So is there now still an issue? It looks good to me.

merbanan, 2020-11-14:
The channel should be part of the id somehow. The low or high 3-4 bits should be static.

merbanan, 2021-03-07:
No update in a long time. Closing.

closed 2021-03-07 by merbanan