merbanan/rtl_433 #1564: Demo mode

Mindavi, 2020-11-16:
In #1562 (which is in turn based on #1263) I started on a demo mode.

My main reason for implementing it is because I want to improve the automated testing capabilities for rtl-433, and I thought that I could combine that nicely. However, I'm not sure if the 'demo mode' is the right place to do so, because I don't think it would be logical to add the expected output there as well. So I think we should do that in another way (possibly duplicating the patterns between the two places?).

I've also started wondering who will be the target audience of the demo mode and how it could be useful for them.

- Will probably be first-time users
- Useful for finding out how the outputs of rtl_433 look
- Useful for finding out which data from the device is exposed

Could you fill me in further on the background of the feature and possible uses @zuckschwerdt?

We could use this to write documentation and maybe help first-time users on their way by pointing to the feature.

jedi7, 2021-01-27:
Isn't easier to just provide real captured data of added devices? It will be also good for testing.

zuckschwerdt, 2021-01-27:
We already collect sample data in rtl_433_tests. This is meant to embed just the raw or processed codes for quick testing and demo of the decoders. Originally bitbuffers seemed easiest, but RfRaw has benefits at roughly the same ease of use.
We still need to figure out where to put the codes: config data in the r_device? Annotation comments? Current idea is the later to group this with "Data layout" info and allow free text notes.

gdt, 2023-09-29:
@Mindavi Seems like this has run its course and should be closed.  What do you think?

Mindavi, 2023-09-30:
I still think something like this is valuable, but I am not sure if I'll pick it up. So let's close it.

I envision a directory with (textual) patterns that can be used as a demo/example, but also for regression testing.

Probably generated from samples we already have

closed 2023-09-30 by Mindavi