merbanan/rtl_433 #1565: Door / window contact naming convention not uniform

ai-xyz, 2020-11-18:
Hi,

i thought about adding to the `rtl_433_mqtt_hass.py` the ability to send auto discovery also for door / window contacts. As far as i can see the naming of the state if the contact is open or closed is different for all the sensors:

`honeywell.c` -> contact_open
`dsc.c`-> closed
`kerui.c` -> opened
`simplisafe.c` -> state
`interlogix.c` -> switch1 (i guess)

Is this somehow intended because some sensors are sending additional data or could we add there a naming convention in order to homgenize this type of devices and add the possibility for auto discovery?

zuckschwerdt, 2020-11-18:
It's just that so far nobody came forward with a general scheme. There are some quirks to consider with each different device. I guess most users integrating these are interested in a specific sensor only.

gdt, 2023-09-29:
I'm calling this is a bug; it seems obvious that naming should be regularized.