merbanan/rtl_433 #1566: Acurite/Chaney 985 wireless refrigerator & freezer thermometer support

jfbauer432, 2020-11-19:
I see rtl_433 has support for Acurite 986, but not obviously the 985.  It looks nearly identical, but I am not seeing anything from it when running rtl_433 (even with -G), so I suspect the wireless part is different.

Anyone know if getting the 985 working is doable?

I am using a [NooElec NESDR Mini 2 SDR & DVB-T USB Stick (RTL2832 + R820T2)](https://www.amazon.com/gp/product/B00P2UOU72/)
with a Raspberry Pi 3b+, running Raspbian 10.  Using local build of rtl_433 from branch master.

manual:  https://www.acurite.com/media/manuals/00985-INST.pdf

rct, 2020-11-19:
I don't get any relevant bits for 985 from Google. 

From the instructions that looks to be the 986 sensors + display. Maybe that's designation for a non-433 mhz version for countries that doesn't use 433 mhz for ISM devices? 

What frequency are you expecting the sensors to transmit on, 433.92?   Are you picking anything else up? 

Are you trying with the sensors outside of the fridge/freezer somewhat close to the RTL-SDR?

The signal from the 986 sensors is somewhat weak and directional. Orientation of the sensor (vertical/horizontal) makes a big difference. IIRC the antenna on the sensor is near the bottom which seems like a bad choice.

The metal box of the freezer/freezer tends to attenuate the signals somewhat as does food/beverage items around the sensor. Think about the straight path (line of sight) between the sensor and receiver.

jfbauer432, 2020-11-19:
I am receiving an outdoor wireless thermometer (Acurite-606TX) and sometimes something from a neighbor.  But nothing from the 985 sensors.  I did have the antenna just outside the fridge near the 985 receiver that came with the sensors (and works).  But now I tried moving the temp sensor to be right next to the antenna: still nothing.

I really don't know what frequency it is using.  Is there a way I can tell?

rct, 2020-11-20:
> I really don't know what frequency it is using. Is there a way I can tell?

The country that you are in determines what the ISM bands are. For example in the US, many devices are 433 mhz though some are 900-915 mhz.  The 986 and most of the Acurite devices are all 433 mhz in the US. 

Is the 606TX that you are receiving yours or a neighbors?   Receiving the 606TX is a good sign that things are right on the receiving side.

Try running with the analyzer `-A` and saving unknown signals `-S unknown`.  On the 986 sensors there is an LED that flashes when it transmits so you should be able to see some output when your sensor transmits.

jfbauer432, 2020-11-20:
Running gqrx I see a small blip around 733.70 MHz when the light on the sensor flashes.  I tried capturing it with _rtl_433 -f 433700000 -S unknown_

Most of the files didn't seem to have anything useful, but there was one possiblity

```
$ rtl_433 -A -r g004_433.7M_250k.cu8
rtl_433 version 20.11 branch master at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g004_433.7M_250k.cu8
Detected OOK package    @0.117616s
Analyzing pulses...
Total count:  168,  width: 311.17 ms            (77793 S)
Pulse width distribution:
 [ 0] count:   87,  width: 1092 us [1080;1112]  ( 273 S)
 [ 1] count:    3,  width: 2984 us [2976;2996]  ( 746 S)
 [ 2] count:   78,  width:  544 us [536;564]    ( 136 S)
Gap width distribution:
 [ 0] count:   39,  width:  916 us [908;932]    ( 229 S)
 [ 1] count:    3,  width: 2988 us [2984;2996]  ( 747 S)
 [ 2] count:   75,  width:  624 us [488;684]    ( 156 S)
 [ 3] count:   48,  width: 1192 us [1148;1232]  ( 298 S)
 [ 4] count:    2,  width: 7636 us [7632;7644]  (1909 S)
Pulse period distribution:
 [ 0] count:   87,  width: 2164 us [2000;2328]  ( 541 S)
 [ 1] count:    3,  width: 5976 us [5968;5984]  (1494 S)
 [ 2] count:   75,  width: 1168 us [1036;1232]  ( 292 S)
 [ 3] count:    2,  width: 8184 us [8176;8196]  (2046 S)
Pulse timing distribution:
 [ 0] count:  174,  width: 1080 us [908;1232]   ( 270 S)
 [ 1] count:    6,  width: 2988 us [2976;2996]  ( 747 S)
 [ 2] count:  153,  width:  584 us [488;684]    ( 146 S)
 [ 3] count:    2,  width: 7636 us [7632;7644]  (1909 S)
 [ 4] count:    1,  width: 29964 us [29964;29964]       (7491 S)
Level estimates [high, low]:  15857,   2238
RSSI: -0.1 dB SNR: 8.5 dB Noise: -8.6 dB
Frequency offsets [F1, F2]:   -4962,      0     (-18.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB018050104380BAC02481DD4750C80808080808080808080809155+AAB038050104380BAC02481DD4750C80A280A280A2A2808080A2A2A2A280A28080A2A28080A28080A28080A2A2A2A2808080A2A2A2A2A2A2A2A2A355+AAB018050104380BAC02481DD4750C80808080808080808080809155+AAB038050104380BAC02481DD4750C80A280A280A2A2808080A2A2A2A280A28080A2A28080A28080A28080A2A2A2A2808080A2A2A2A2A2A2A2A2A355+AAB018050104380BAC02481DD4750C80808080808080808080809155+AAB038050104380BAC02481DD4750C80A280A280A2A2808080A2A2A2A280A28080A2A28080A28080A28080A2A2A2A2808080A2A2A2A2A2A2A2A2A455
Attempting demodulation... short_width: 544, long_width: 1092, reset_limit: 7648, sync_width: 2984
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=544,l=1092,r=7648,g=0,t=0,y=2984'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 4 
[00] {11} 00 00                : 00000000 000
[01] {55} 56 3d 32 4f 1f f0 00 
[02] {55} 56 3d 32 4f 1f f0 00 
[03] {44} 56 3d 32 4f 1f f0    : 01010110 00111101 00110010 01001111 00011111 1111

$ rtl_433 -r g004_433.7M_250k.cu8 -X 'n=name,m=OOK_PWM,s=544,l=1092,r=7648,g=0,t=0,y=2984'rtl_433 version 20.11 branch master at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 146 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g004_433.7M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.117616s
model     : name         count     : 4             num_rows  : 4             rows      : 
len       : 11           data      : 000, 
len       : 55           data      : 563d324f1ff000, 
len       : 55           data      : 563d324f1ff000, 
len       : 44           data      : 563d324f1ff
codes     : {11}000, {55}563d324f1ff000, {55}563d324f1ff000, {44}563d324f1ff

```

rct, 2020-11-20:
That sample looks promising.  It would be helpful to capture more so that you know you are receiving it reliably and we're looking at the right signal.  The 986 transmits once a minute or so.  I believe there are samples of my 986s in the tests repo. 

Regarding the frequency - If you are seeing the sensor at 433.770,  change the frequency to something like 433.820 or 433.720.  There is effectively a dead spot at the center frequency where the receiver is tuned.  (There is a DC spike at the center.)

jfbauer432, 2020-11-20:
Forgot to answer the question about the 606TX.  Yes that is mine.  I have seen a few other devices from neighbors: Some lacross device, a soil monitor, and a few tire pressure sensors.

Got a few more samples of the likely 985 sensor.   These corresponded to the light flashing and are about 61-64 seconds apart.   The first few are at 433.70MHz, then I got one at 433.82MHz and one at 433.72MHz.

```
$ rtl_433 -A -r g006_433.7M_250k.cu8 
rtl_433 version 20.11 branch master at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g006_433.7M_250k.cu8
Detected OOK package    @0.089768s
Analyzing pulses...
Total count:  168,  width: 339.02 ms            (84755 S)
Pulse width distribution:
 [ 0] count:  111,  width: 1096 us [1092;1112]  ( 274 S)
 [ 1] count:    3,  width: 2992 us [2988;2996]  ( 748 S)
 [ 2] count:   54,  width:  548 us [540;556]    ( 137 S)
Gap width distribution:
 [ 0] count:   39,  width:  912 us [908;920]    ( 228 S)
 [ 1] count:    3,  width: 2992 us [2992;2992]  ( 748 S)
 [ 2] count:   49,  width:  632 us [492;680]    ( 158 S)
 [ 3] count:   72,  width: 1200 us [1152;1228]  ( 300 S)
 [ 4] count:    2,  width: 7632 us [7628;7636]  (1908 S)
 [ 5] count:    2,  width:  488 us [488;488]    ( 122 S)
Pulse period distribution:
 [ 0] count:  111,  width: 2196 us [2008;2328]  ( 549 S)
 [ 1] count:    3,  width: 5984 us [5980;5988]  (1496 S)
 [ 2] count:   51,  width: 1176 us [1036;1228]  ( 294 S)
 [ 3] count:    2,  width: 8180 us [8180;8184]  (2045 S)
Pulse timing distribution:
 [ 0] count:  222,  width: 1096 us [908;1228]   ( 274 S)
 [ 1] count:    6,  width: 2992 us [2988;2996]  ( 748 S)
 [ 2] count:  105,  width:  584 us [488;680]    ( 146 S)
 [ 3] count:    2,  width: 7632 us [7628;7636]  (1908 S)
 [ 4] count:    1,  width: 29964 us [29964;29964]       (7491 S)
Level estimates [high, low]:  15912,    162
RSSI: -0.1 dB SNR: 19.9 dB Noise: -20.0 dB
Frequency offsets [F1, F2]:   -5482,      0     (-20.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB018050104480BB002481DD0750C80808080808080808080809155+AAB011050104480BB002481DD0750C80A280A28055+AAB00D050304480BB002481DD0750C8055+AAB00E050104480BB002481DD0750CA28055+AAB010050104480BB002481DD0750CA2A2A28055+AAB00D050404480BB002481DD0750C8055+AAB00E050104480BB002481DD0750CA28055+AAB00D050204480BB002481DD0750C8055+AAB00F050104480BB002481DD0750CA2A28055+AAB00D050104480BB002481DD0750C8055+AAB012050104480BB002481DD0750CA2A2A2A2A28055+AAB00D050304480BB002481DD0750C8055+AAB00E050104480BB002481DD0750CA28055+AAB00D050204480BB002481DD0750C8055+AAB00F050104480BB002481DD0750CA2A28055+AAB00D050104480BB002481DD0750C8055+AAB00D050104480BB002481DD0750CA355+AAB018050104480BB002481DD0750C80808080808080808080809155+AAB011050104480BB002481DD0750C80A280A28055+AAB00D050304480BB002481DD0750C8055+AAB00E050104480BB002481DD0750CA28055+AAB010050104480BB002481DD0750CA2A2A28055+AAB00D050404480BB002481DD0750C8055+AAB00E050104480BB002481DD0750CA28055+AAB00D050204480BB002481DD0750C8055+AAB00F050104480BB002481DD0750CA2A28055+AAB00D050104480BB002481DD0750C8055+AAB012050104480BB002481DD0750CA2A2A2A2A28055+AAB00D050304480BB002481DD0750C8055+AAB00E050104480BB002481DD0750CA28055+AAB00D050204480BB002481DD0750C8055+AAB00F050104480BB002481DD0750CA2A28055
Too many pulse groups (58 pulses missed in rfraw)
Attempting demodulation... short_width: 548, long_width: 1096, reset_limit: 7640, sync_width: 2992
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=548,l=1096,r=7640,g=0,t=0,y=2992'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 4 
[00] {11} 00 00                : 00000000 000
[01] {55} 50 b8 23 3e 11 90 00 
[02] {55} 50 b8 23 3e 11 90 00 
[03] {44} 50 b8 23 3e 11 90    : 01010000 10111000 00100011 00111110 00010001 1001

```

```
$ rtl_433 -A -r g007_433.7M_250k.cu8 
rtl_433 version 20.11 branch master at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g007_433.7M_250k.cu8
Detected OOK package    @0.103728s
Analyzing pulses...
Total count:  168,  width: 325.10 ms            (81275 S)
Pulse width distribution:
[... 209 lines cut ...]
 [ 1] count:    3,  width: 5984 us [5980;5996]  (1496 S)
 [ 2] count:   78,  width: 1176 us [1036;1228]  ( 294 S)
 [ 3] count:    2,  width: 9280 us [9280;9280]  (2320 S)
Pulse timing distribution:
 [ 0] count:  171,  width: 1076 us [924;1240]   ( 269 S)
 [ 1] count:    6,  width: 2992 us [2980;3016]  ( 748 S)
 [ 2] count:  156,  width:  588 us [500;692]    ( 147 S)
[... cut ...] 6C8355
Too many pulse groups (56 pulses missed in rfraw)
Attempting demodulation... short_width: 536, long_width: 1084, reset_limit: 8196, sync_width: 2980
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=536,l=1084,r=8196,g=0,t=0,y=2980'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 4 
[00] {11} 00 00                : 00000000 000
[01] {55} 5d d6 76 5f 1a c0 00 
[02] {55} 5d d6 76 5f 1a c0 00 
[03] {44} 5d d6 76 5f 1a c0    : 01011101 11010110 01110110 01011111 00011010 1100

```

jfbauer432, 2020-11-20:
And here is an screen capture of gqrx showing the blip at about 433.67 and a smaller one at 432.76.  They both show up with the led flash on the 985 sensor rather consistently.

[image]

zuckschwerdt, 2020-11-21:
That could be a harmonic. The stronger, upper one looks true.

rct, 2020-11-21:
Have you found a set up where you can reliable receive the 985's and the 606?  

You should be able to tune above 433.76, assuming that's the real signal, maybe around 433.810 to 433.860?

If the PWM signal you've captured above is in fact the 985, a new decoder will be needed. the 986 is PPM modulation not PWM.   It doesn't look like the first byte has the temperature like the 986 unless the 5 is part of the preamble and maybe there is a parity bit in the 2nd nybble. 

You can see the 986 details in acurite.c:
https://github.com/merbanan/rtl_433/blob/0bb4672b1afb716d107b3060afc2d80a2159dbef/src/devices/acurite.c#L912

rct, 2020-11-21:
Just checking, the sensors you have look like the 986, and not the newer 515's?

Are there any IDs marked on the back of the sensors  (FCC, ICC, Acurite part No.'s)?

What country are you located in?

jfbauer432, 2020-11-21:
I am in the US.  The back of each sensor has: "FCC ID: RNE00985TX" and "MADE IN CHINA".

I just looked up that [FCC ID](https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=YPLqFB20JDMDmWZYjJT5sw%3D%3D&fcc_id=RNE00985TX) (should have done that a while back), there is a lot of potentially useful information there.

Looks like I can pick up both the 606TX and the 985s at 433.810MHz.  For the 985 that was sensor #1 (fridge).

For the first time I tried sensor #2 (freezer) and it appears to be at a different frequency.
[image]  But I have been unable to capture anything from it.

merbanan, 2020-11-21:
If you are in the US check around 315MHz and 915MHz. 433MHz is used for other things in the US.

merbanan, 2020-11-21:
Hmm looks like it actually uses 433.975MHz.

merbanan, 2020-11-21:
You could try with -s 1000k and see if it works better.

jfbauer432, 2020-11-21:
I think I picked up sensor 2 (but not sensor 1) at 433.975

```
$ rtl_433 -A -r g007_433.975M_250k.cu8
rtl_433 version 20.11 branch master at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g007_433.975M_250k.cu8
Detected OOK package    @0.110828s
Analyzing pulses...
Total count:  168,  width: 318.12 ms            (79530 S)
Pulse width distribution:
 [ 0] count:   93,  width: 1084 us [1080;1096]  ( 271 S)
 [ 1] count:    3,  width: 2980 us [2980;2980]  ( 745 S)
 [ 2] count:   72,  width:  536 us [536;540]    ( 134 S)
Gap width distribution:
 [ 0] count:   39,  width:  924 us [924;928]    ( 231 S)
 [ 1] count:    3,  width: 3004 us [3004;3004]  ( 751 S)
 [ 2] count:    3,  width:  496 us [496;500]    ( 124 S)
 [ 3] count:   66,  width:  648 us [620;688]    ( 162 S)
 [ 4] count:   54,  width: 1192 us [1168;1236]  ( 298 S)
 [ 5] count:    2,  width: 7640 us [7640;7640]  (1910 S)
Pulse period distribution:
 [ 0] count:   93,  width: 2168 us [2012;2324]  ( 542 S)
 [ 1] count:    3,  width: 5984 us [5984;5984]  (1496 S)
 [ 2] count:   69,  width: 1180 us [1036;1224]  ( 295 S)
 [ 3] count:    2,  width: 8180 us [8180;8180]  (2045 S)
Pulse timing distribution:
 [ 0] count:  186,  width: 1084 us [924;1236]   ( 271 S)
 [ 1] count:    6,  width: 2992 us [2980;3004]  ( 748 S)
 [ 2] count:  141,  width:  588 us [496;688]    ( 147 S)
 [ 3] count:    2,  width: 7640 us [7640;7640]  (1910 S)
 [ 4] count:    1,  width: 29804 us [29804;29804]       (7451 S)
Level estimates [high, low]:  15938,    359
RSSI: -0.1 dB SNR: 16.5 dB Noise: -16.6 dB
Frequency offsets [F1, F2]:    3658,      0     (+14.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB0180501043C0BB0024C1DD8746C80808080808080808080809155+AAB0110501043C0BB0024C1DD8746C80A280A28055+AAB00D0501043C0BB0024C1DD8746C8055+AAB00E0501043C0BB0024C1DD8746CA28055+AAB00F0502043C0BB0024C1DD8746CA2A28055+AAB00D0501043C0BB0024C1DD8746C8055+AAB00E0501043C0BB0024C1DD8746CA28055+AAB0110501043C0BB0024C1DD8746CA2A2A2A28055+AAB00E0501043C0BB0024C1DD8746CA28055+AAB00D0502043C0BB0024C1DD8746C8055+AAB00F0501043C0BB0024C1DD8746CA2A28055+AAB0100501043C0BB0024C1DD8746CA2A2A28055+AAB00D0502043C0BB0024C1DD8746C8055+AAB0100501043C0BB0024C1DD8746CA2A2A28055+AAB00F0501043C0BB0024C1DD8746CA2A28055+AAB00D0501043C0BB0024C1DD8746C8055+AAB00D0501043C0BB0024C1DD8746CA355+AAB0180501043C0BB0024C1DD8746C80808080808080808080809155+AAB0110501043C0BB0024C1DD8746C80A280A28055+AAB00D0501043C0BB0024C1DD8746C8055+AAB00E0501043C0BB0024C1DD8746CA28055+AAB00F0502043C0BB0024C1DD8746CA2A28055+AAB00D0501043C0BB0024C1DD8746C8055+AAB00E0501043C0BB0024C1DD8746CA28055+AAB0110501043C0BB0024C1DD8746CA2A2A2A28055+AAB00E0501043C0BB0024C1DD8746CA28055+AAB00D0502043C0BB0024C1DD8746C8055+AAB00F0501043C0BB0024C1DD8746CA2A28055+AAB0100501043C0BB0024C1DD8746CA2A2A28055+AAB00D0502043C0BB0024C1DD8746C8055+AAB0100501043C0BB0024C1DD8746CA2A2A28055+AAB00F0501043C0BB0024C1DD8746CA2A28055
Too many pulse groups (58 pulses missed in rfraw)
Attempting demodulation... short_width: 536, long_width: 1084, reset_limit: 7644, sync_width: 2980
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=536,l=1084,r=7644,g=0,t=0,y=2980'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 4 
[00] {11} 00 00                : 00000000 000
[01] {55} 52 d9 7a 37 1d 90 00 
[02] {55} 52 d9 7a 37 1d 90 00 
[03] {44} 52 d9 7a 37 1d 90    : 01010010 11011001 01111010 00110111 00011101 1001

```

jfbauer432, 2020-11-21:
Tried _rtl_433 -f 433975000 -S unknown -s 1000k_.  It generated lots of files, but nothing looked good.

jfbauer432, 2020-11-21:
Upper left near 433.7 is sensor 1 and the lower near 434.0 is sensor 2.

[image]

merbanan, 2020-11-21:
Try tuning to the middle of them.

jfbauer432, 2020-11-21:
433.7MHz - just sensor 1
433.8MHz - neither
433.9MHz - just sensor 2

merbanan, 2020-11-21:
-s 300 or -s 900 can work. If not then you need to cycle the frequency.

jfbauer432, 2020-11-21:
With _rtl_433 -f 433800000 -S unknown -s 300_ (and -s 900), I got a lot of _sample rate too low for protocol nnn_ messages, bit it did seem to pick up sensor 1, although not sensor 2.

With _rtl_433 -f 433850000 -S unknown -s 300_ (and -s 900), got sensor 2 but not 1.

merbanan, 2020-11-21:
-s 300k -s 900k, my bad.

jfbauer432, 2020-11-21:
rtl_433 -f 433830000 -S unknown -s 900k seems to pick up both sensors

rct, 2020-11-23:
Should have asked earlier, are you using an antenna that is somewhat reasonably tuned (right length) for 433 mhz?

(Might not help since 200 Khz seems like a fairly big spread between sensors.)

Also are the sensors both inside the freezer/refrigerator?  Their frequency may drift a bit as their temperature changes.

jfbauer432, 2020-11-23:
The antenna I am using came with the [NooElec NESDR Mini 2 SDR & DVB-T USB Stick (RTL2832 + R820T2)](https://www.amazon.com/gp/product/B00P2UOU72/) I am using and is about 12 inches long.  That is fully extended.

During must of the testing the sensors were next to me on a desk approximately 5 feet (sometimes closer) from the antenna.  The antenna is vertical with the base at desk height.

zuckschwerdt, 2020-11-23:
That's not the right antenna. The other ones (fixed, hand length) that usually come with NooElec or off-brand receviers are suitable.

jfbauer432, 2020-11-23:
Can you provide a link please?

zuckschwerdt, 2020-11-23:
Just a generic 433 MHz antenna that looks like this: https://www.amazon.de/dp/B01E564FPU

rct, 2020-11-23:
If the antenna is collapsible, see if you can get it down to about 6.5-6.8 inches, so that you've got a 1/4 wave "monopole" antenna.  The base should be sitting on something metal to from a ground plane that makes up the other half of the antenna.

Standing upright the antenna is "vertically polarized" and will receive signals from vertically oriented antennas the best. Your sensors usually use a coiled spring like antenna for size and to reduce the effects of polarization.  

If your RTL-SDR has an SMA connector on it, you could put a 433mhz "rubber duck" antenna that's made for handheld radios.

jfbauer432, 2020-11-23:
It collapses to 4.5" from the bottom of the base.  Base is currently sitting on a 2-door metal filing cabinet.  I now have it at about 6.6" above the bottom of the base.

jfbauer432, 2020-11-23:
_rtl_433 -f 433830000 -S unknown -s 900k_ isn't giving me much today (the sensors are colder today: just came out of freezer/fridge)

But _rtl_433 -f 433700000 -S unknown -s 900k_ works for sensor 1
and _rtl_433 -f 433900000 -S unknown -s 900k_ works for sensor 2.

jfbauer432, 2020-12-14:
I have been trying to decode the signals and have finally made some progress.  It was very similar to the 986.

I am not totally certain on the values for r_device struct.  Most came from one of the `rtl_433 -A` outputs, so they might be reasonable.  The gap_limit, however, was rather trail and error.  Values 3000 and below had a lot of decode failures, but not all the time.  4000 to (I think 6000) worked a lot better.

I was confused for a long time with the output of `rtl_433 -A`.  Specifically lines like
`[01] {55} 56 db 2f 05 11 20 00`
I could not figure any way for the temperature to be encoded in that.  And I did try flipping the bit order and a bunch of other things.  But when I created a new decoded -- modelled on the Acurite 986 -- I discovered that it was seeing a different set of bytes and the temperature showed up rather clearly.  So I guess that line in the output wasn't what I thought it was (unless it is a bug).

I am getting two random bytes preceding the actual data which I am guessing may be part of the sync pulse.  Is there a way to eliminate that or is just skipping them good enough?

This is what I came up with for decoding the 985.
https://github.com/jfbauer432/rtl_433/commit/d9bbea4ae3454231f9a57fab7ba1b83c502721d4
https://github.com/jfbauer432/rtl_433_tests/commit/cf46408cc69c1fd85496131b8a15157bc417768e

I have not gone back to try and capture both sensors at the same time (same frequency).

zuckschwerdt, 2020-12-14:
Quick note: you know you're expecting 7 bytes, if I read that right. Don't use any `browlen` then, fail if it's less than 7 bytes, otherwise just process those 7 bytes.

jfbauer432, 2020-12-21:
BTW, I made that little change and there is now a pull request pending.

I have been running this for a while now and it is looking pretty good.  I never did find a way to capture both sensors from a single frequency (-f) argument.  But this seems to be working reasonable enough.

rtl_433 -R 176 -f 433.92M -f 433.7M -H 480

I chose a hop value of 480 (8 minutes) as after the sensors are running for a while (20 min), sensors 1 and 2 send at intervals of 7.5 and 6 minutes respectively. I could probably shrink the hop to be closer to 7.5 minutes, but I have not tried that.

As a bonus (after adding -R 55), I can also pick up my outdoor thermometer.

closed 2020-12-21 by jfbauer432