merbanan/rtl_433 #1572: Error during make "undefined reference to `libusb_strerror'", "libusb_dev_mem_free", "libusb_dev_mem_alloc"

dwanderer, 2020-12-01:
I tried installing RTL_433 in Raspberry PI3. Find the error log below. Requesting your support to resolve the issue. Thanks.

pi@PI3A:~/code/rtl_433 $ cmake .//
-- Build type not specified: defaulting to release.
-- Found Git version: refs/heads/master commit 84114f21771f3ee3e43b72f63a3125322f3bd30f from 2020-11-30T12:55:54+0100
-- Using Git version tag: 20.11-1-g84114f2 on master at 202011301255
-- RTL-SDR device input will be compiled.
-- SoapySDR device input will be compiled.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/code/rtl_433
pi@PI3A:~/code/rtl_433 $ sudo make install
[ 89%] Built target r_433
[ 89%] Linking C executable rtl_433
/usr/bin/ld: libr_433.a(sdr.c.o): in function `sdr_start':
sdr.c:(.text+0x2444): undefined reference to `libusb_strerror'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/librtlsdr.so: undefined reference to `libusb_dev_mem_free'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/librtlsdr.so: undefined reference to `libusb_dev_mem_alloc'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/rtl_433.dir/build.make:88: src/rtl_433] Error 1
make[1]: *** [CMakeFiles/Makefile2:1041: src/CMakeFiles/rtl_433.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
pi@PI3A:~/code/rtl_433 $ cmake .//
-- Build type not specified: defaulting to release.
-- Found Git version: refs/heads/master commit 84114f21771f3ee3e43b72f63a3125322f3bd30f from 2020-11-30T12:55:54+0100
-- Using Git version tag: 20.11-1-g84114f2 on master at 202011301255
-- RTL-SDR device input will be compiled.
-- SoapySDR device input will be compiled.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/code/rtl_433
pi@PI3A:~/code/rtl_433 $ sudo make install
[ 89%] Built target r_433
[ 89%] Linking C executable rtl_433
/usr/bin/ld: libr_433.a(sdr.c.o): in function `sdr_start':
sdr.c:(.text+0x2444): undefined reference to `libusb_strerror'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/librtlsdr.so: undefined reference to `libusb_dev_mem_free'
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/librtlsdr.so: undefined reference to `libusb_dev_mem_alloc'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/rtl_433.dir/build.make:88: src/rtl_433] Error 1
make[1]: *** [CMakeFiles/Makefile2:1041: src/CMakeFiles/rtl_433.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

merbanan, 2020-12-01:
Do you have libusb-dev installed?

vu3vyd, 2020-12-01:
Yes, It is installed.

Mindavi, 2020-12-01:
Maybe it should be libusb1 instead?

zuckschwerdt, 2020-12-01:
Some related: #1459 #1235
IIRC libusb-1.0-0-dev is needed, https://packages.debian.org/de/buster/libusb-1.0-0-dev
libusb-dev is version 0: https://packages.debian.org/de/buster/libusb-dev

vu3vyd, 2020-12-02:
I have re-installed the library - libusb-1.0-0-dev. Then built the code again from the beginning. Attaching the log to this comment. No luck yet!
[RTL_433_BuildError_CompleteLog.txt](https://github.com/merbanan/rtl_433/files/5626705/RTL_433_BuildError_CompleteLog.txt)

Mindavi, 2020-12-02:
FYI, I have these dependencies installed on my raspberry pi. I'd assume one of them will work / a combination of them. Note that I'm still on an older version of Raspbian, so YMMV.

```
pi@homeserver:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.11 (stretch)
Release:        9.11
Codename:       stretch
pi@homeserver:~$ uname -a
Linux homeserver 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux
pi@homeserver:~$
pi@homeserver:~$ apt list --installed | grep "usb\|rtl"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

librtlsdr-dev/stable,now 0.5.3-11+rpt1 armhf [installed]
librtlsdr0/stable,now 0.5.3-11+rpt1 armhf [installed]
libusb-0.1-4/oldstable,now 2:0.1.12-30 armhf [installed]
libusb-1.0-0/oldstable,now 2:1.0.21-1 armhf [installed,automatic]
libusb-1.0-0-dev/oldstable,now 2:1.0.21-1 armhf [installed,automatic]
libusb-1.0-doc/oldstable,now 2:1.0.21-1 all [installed,automatic]
rtl-sdr/stable,now 0.5.3-11+rpt1 armhf [installed]
usb-modeswitch/oldstable,now 2.5.0+repack0-1 armhf [installed]
usb-modeswitch-data/oldstable,now 20170120-1 all [installed,automatic]
usbutils/oldstable,now 1:007-4+deb9u1 armhf [installed]
```

zuckschwerdt, 2020-12-02:
There are likely other libusb header and lib files around, both rtlsdr and rtl_433 pick the wrong ones. To diagnose try:
`mkdir build && cd build && cmake -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON .. && make VERBOSE=1`
You should see e.g. only `-I/home/pi/rtl_433/include`and `-I/usr/include/libusb-1.0` and not any `/usr/local` or other paths.

closed 2021-01-10 by zuckschwerdt