merbanan/rtl_433 #1574: rtl_433 on Raspberry Pi Zero

Chemical-Nils, 2020-12-05:
Hi
I am trying to run rtl_433 on a Raspberry Pi Zero. My problem is that protocol 79 (Fineoffset) is not working on the Pi Zero whereas it works perfectly on a Pi 3. Other protocols as e.g. 19 (Nexus TH) works fine on both platforms. Is it a performance problem with the Pi Zero or ??
I have attached debug output for Pi Zero (both protocols) and for Pi 3  (both protocols).
Any clues to what can be done ?
[Debug Fineoffset.txt](https://github.com/merbanan/rtl_433/files/5647424/Debug.Fineoffset.txt)

zuckschwerdt, 2020-12-05:
Can you measure the load (`uptime` or `top`)?
Also make sure to use an extension cable, don't plug the RTL-SDR directly into the Pi -- there will be noticeable interference.

Chemical-Nils, 2020-12-06:
Hi
Thanks. Good suggestions but unfortunalely:
- Load seems not to be an issue. Attached load muasurements on both pi's
- USB cable on the Rasberry Pi Zero was 15 cm. I added a ferrite and a 3m expension cable. No improvements.
Another observation: Exchanging the two RTL-SDR's (Fitipower FC0012) makes the Pi Zero not receive anything on the two monitored protocols whereas the Pi 3 receives only the Nexus-TH protocol ???  Changing back results in the previously described situatiation. 
I am really confused. Is it the problem RTL_433, the Fineoffset plugin, the RTL-SDR's or is it the Raspberries?
[rtl_433 load.txt](https://github.com/merbanan/rtl_433/files/5648994/rtl_433.load.txt)

zuckschwerdt, 2020-12-06:
We are not aware of any such problems. No stress tests have been done for the case where data is slower to process than real-time -- but that's not the issue here anyway.
You can try to grab frames (`-S all`) and then compare, try to process on each device and examine.
Fineoffset-WH0530 is OOK_PULSE_PWM, 504µs / 1480µs and Nexus-TH is OOK_PULSE_PPM, 1000µs / 2000µs, they are roughly pretty much the same.

merbanan, 2020-12-06:
Fitipower tuners needs a fixed gain. Try -g [0-20] to see if it helps.

Chemical-Nils, 2020-12-07:
Hi

Thanks a lot to both of you!!

The hint with the fixed gain led me to move the WH0530 5m or more away from the receiver. That worked.

Again thank you for takin the time to help !

Best

Nils

closed 2020-12-07 by Mindavi