merbanan/rtl_433 #1576: Protocol 19 (model Nexus-T) - no data reported when temperature is 0 degrees

badrpc, 2020-12-06:
Hello,

I use rtl_433 to decode data from a couple of sensors and noticed that data from one of the sensors disappear when temperature crosses value of 0 degrees Celsius (see graph below). Temperature station is Auriol HG00073B and internal module will keep showing correct temperature even when it reaches 0. The model reported by rtl_433 is Nexus-T and the protocol number is 19. What can I do to help debug this problem further?

[image]

zuckschwerdt, 2020-12-06:
Thanks for analyzing this so spot on. There was a bug with false positive detection at 0 degrees. Fixed!

closed 2020-12-06 by zuckschwerdt

badrpc, 2020-12-07:
Wow! This was a very fast response with a fix. Thank you! I can confirm it works: 
[image]

commit cb372a57 (2020-12-06): Fix false positive bug in Nexus (closes #1576)