merbanan/rtl_433 #1577: No signal recognized with Airspy / SoapySDR

Akito01, 2020-12-11:
I'm trying to run RTL_433 with an Airspy. I've created the SoapySDR driver and I can successfully start RTL_433:
`rtl_433 -C si
rtl_433 version 20.11-5-g67ee1d69 branch master at 202012081410 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Reading conf from "/usr/local/etc/rtl_433/rtl_433.conf".
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
[INFO] Using format CS16.
Sample rate set to 250000 S/s.
Tuner set to manual gain.
Setting gain element "lgain: 5.000000 dB
Setting gain element vgain: 9.000000 dB
Setting gain element mgain: 2.000000 dB
Tuned to 433.920MHz.
OOOOOOOOOOOOOOOOOOOOOOOO`

However, no matter how long I let it run nothing is being detected.
With an RTL-SDR dongle the display keeps scrolling with signals, so.... what am I doing wrong?

zuckschwerdt, 2020-12-11:
- The `O` means overflow of the stream coming in. It should not appear regularly.
- Could be the 250k sample rate isn't great for the Airspy, try `-s 1M`.
- Is the gain well selected? Try to record with `-w FILE.cs16 -T 120` and look at the spectrum (https://triq.org/iqs/)

closed 2021-01-10 by zuckschwerdt

dwarning, 2022-03-16:
Sorry for posting in this closed issue.
But I have same problem as the TO.
`dietmar@Inspiron:~/Projects/sdr$ rtl_433 -f 433.94M -s 1M -g "LNA=14 MIX=7 VGA=5" -d driver=airspy
rtl_433 version 21.12-75-g45e8fc54 branch master at 202203091346 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dietmar/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 184 out of 215 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 ]
[INFO] Using format CS16.
Sample rate set to 1000000 S/s.
Tuner set to manual gain.
Setting gain element LNA: 14.000000 dB
Tuned to 433.940MHz.
baseband_demod_FM_cs16: low pass filter for 1000000 Hz at cutoff 100000 Hz, 10.0 us
`
I used different settings for gain and sample rate for the airspy mini. Also my SoapySDR seems to recognize the airspy:
`dietmar@Inspiron:~/Projects/sdr$ SoapySDRUtil --probe "driver=airspy"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=Airspy
  hardware=Airspy
  serial=a74068c8305d2693

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * Bias tee - Enable the 4.5v DC Bias tee to power SpyVerter / LNA / etc. via antenna connection.
       [key=biastee, default=false, type=bool]
     * Bit Pack - Enable packing 4 12-bit samples into 3 16-bit words for 25% less USB trafic.
       [key=bitpack, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Full gain range: [0, 45] dB
    LNA gain range: [0, 15] dB
    MIX gain range: [0, 15] dB
    VGA gain range: [0, 15] dB
  Full freq range: [24, 1800] MHz
    RF freq range: [24, 1800] MHz
  Sample rates: 6, 3 MSps
`
rtl_sdr stick show signals from different sensors arround. Also the airspy with gqrx.
I checked the spectrum with your proposal (-w FILE.cs16 -T 120) and see similar waterfall as with gqrx.
How can I debug this case?

zuckschwerdt, 2022-03-16:
Try with `-Y autolevel`, quality SDR don't use the full dB range and rtl_433 needs to adjust the levels down to accept quieter transmissions.

dwarning, 2022-03-16:
Thanks for answering!
Made a trial with autolevel:
`dietmar@Inspiron:~/Projects/sdr$ rtl_433 -f 433.94M -s 3M -g "LNA=15 MIX=10 VGA=7" -d driver=airspy -Y autolevel
rtl_433 version 21.12-75-g45e8fc54 branch master at 202203091346 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dietmar/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 184 out of 215 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 ]
[INFO] Using format CS16.
Sample rate set to 3000000 S/s.
Tuner set to manual gain.
Setting gain element LNA: 15.000000 dB
Tuned to 433.940MHz.
Estimated noise level is -22.2 dB, adjusting minimum detection level to -19.2 dB
baseband_demod_FM_cs16: low pass filter for 3000000 Hz at cutoff 300000 Hz, 3.3 us
Estimated noise level is -28.5 dB, adjusting minimum detection level to -25.5 dB
Estimated noise level is -34.0 dB, adjusting minimum detection level to -31.0 dB
Estimated noise level is -38.8 dB, adjusting minimum detection level to -35.8 dB
Estimated noise level is -43.0 dB, adjusting minimum detection level to -40.0 dB
Estimated noise level is -46.7 dB, adjusting minimum detection level to -43.7 dB
Estimated noise level is -49.9 dB, adjusting minimum detection level to -46.9 dB
Estimated noise level is -52.7 dB, adjusting minimum detection level to -49.7 dB
Estimated noise level is -55.2 dB, adjusting minimum detection level to -52.2 dB
Estimated noise level is -57.4 dB, adjusting minimum detection level to -54.4 dB
Estimated noise level is -59.3 dB, adjusting minimum detection level to -56.3 dB
Estimated noise level is -60.9 dB, adjusting minimum detection level to -57.9 dB
Estimated noise level is -62.4 dB, adjusting minimum detection level to -59.4 dB
Estimated noise level is -63.7 dB, adjusting minimum detection level to -60.7 dB
Estimated noise level is -64.8 dB, adjusting minimum detection level to -61.8 dB
Estimated noise level is -66.6 dB, adjusting minimum detection level to -63.6 dB
Estimated noise level is -68.0 dB, adjusting minimum detection level to -65.0 dB
Estimated noise level is -69.0 dB, adjusting minimum detection level to -66.0 dB
Estimated noise level is -70.2 dB, adjusting minimum detection level to -67.2 dB
Estimated noise level is -71.3 dB, adjusting minimum detection level to -68.3 dB
Estimated noise level is -72.3 dB, adjusting minimum detection level to -69.3 dB
`
-M level show nothing different.

I checked rtlsdr stick over SoapySDR interface and it works same as with direct interface.
I agree that in my experience the airspy delivers lower level than the rtlsdr stick. But the usual noise level is about -100 dB and so 20 dB lower then rtlsdr stick.
The catched file .cs16 is a bit large to attache. Should I attache a screendump from (https://triq.org/iqs/)?

zuckschwerdt, 2022-03-16:
> usual noise level is about -100 dB and so 20 dB lower then rtlsdr stick.

That does not sound right. The bottom level of the RTL-SDR (8bit sample depth) is -42 dBFS,
for a 12bit SDR it would be -66 dBFS, or with CS16 not properly transmitted (not aligned to MSB) it would be -90 dBFS

> The catched file .cs16 is a bit large to attache. Should I attache a screendump from (https://triq.org/iqs/)?

Sure, best to use https://triq.org/pdv/

dwarning, 2022-03-16:
Attached are the both cases with rtlsdr (above) and airspy (below).

Can't interpret the difference. In both diagrams are the events, but 
only the rtlsdr shows decoded signals with rtl_433.

zuckschwerdt, 2022-03-16:
I guess file upload doesn't work by mail. Anyway, if the spectrograms look mostly the same, then perhaps try the same lower sample rate on the AirSpy (`-s 1000k`).
You can also use the rtl_tcp pass-through (`-F rtl_tcp`) to watch live data from e.g. Gqrx or any other SoapySDR-enabled UI using this driver https://github.com/pothosware/SoapyRTLTCP

dwarning, 2022-03-16:
OK, didn't know:
[image]

Can you comment please.

Airspy mini only support 3 and 6Ms. Perhaps decimation would help.

zuckschwerdt, 2022-03-16:
There is too much range cut of in the screenshots. You need to play with the gain settings on the spectroplot. Basically you want a proper dB distribution, the violet curve at the right, it's just a lump at the bottom in those pictures ;)

CtrlC-Root, 2022-07-22:
My attempt with an Airspy R1:

[image]

I was also unable to decode any data using it. I tried to get a live view using the Soapy RTL TCP driver above but for some reason qgrx wouldn't work correctly; it would connect, issue a few commands, rtl_433 would print out `rtl_tcp send not ready for write?`, then it would disconnect, and when I tried to run it I would see garbage data until it would crash shortly after.

starvald, 2022-07-22:
i have noticed this too with my Airspy R2, sdrplay rsp2 i receive lots of incoming signals, the R2 nothing.

zuckschwerdt, 2022-07-23:
The error likely means that the buffer is overflowing. Did you check that both programs are configured to the same sample rate?

CtrlC-Root, 2022-07-25:
> The error likely means that the buffer is overflowing. Did you check that both programs are configured to the same sample rate?

Yes, I verified that was the case.

samysousa, 2022-08-09:
Seeing same results the outputed cs16 file has nothing. Exactly like the graphs above. 
Seen this with other programs where the cs16output of SoapySDR does not work well. 
For example; in SDR++, i have to set to cs8 for the audio to not be choppy. This only happens trough SoapySDR, not with any native driver. 

If only someone would add native airspy support to rtl_433 :-)

CtrlC-Root, 2024-01-08:
I'm still running into this issue. Does anyone have any advice on how I could troubleshoot it further?

```
Seen this with other programs where the cs16output of SoapySDR does not work well.
```

Interesting. Any links you can share for me to follow this up?

zuckschwerdt, 2024-01-08:
What's the used sample rate? Can you upload a short .cs16 from rtl_433 (`rtl_433 OTHER-NEEDED-ARGS -w airspy_433.92M_1000k.cs16 -T 5` -- Replace the actual center freq and sample rate in the file name)? Perhaps try do a few recordings, check with https://triq.org/pdv/ and upload an interesting one.

CtrlC-Root, 2024-01-08:
```bash
rtl_433 -d 'driver=airspy' -Y autolevel -w airpsy_2.cs16 -T 5
7z a airspy_2.zip airpsy_2.cs16
```

I clicked the TX button on a sensor I have on my desk a few times and indeed it was captured in the signal.

[airspy_2.zip](https://github.com/merbanan/rtl_433/files/13865419/airspy_2.zip)
[image]

CtrlC-Root, 2024-01-08:
I also tried without `-Y autolevel` and while the signal is still visible it's a lot weaker. I've messed around with various gain settings too but honestly gqrx works fine with the automatic hardware AGC and right now I just want to confirm it's *possible* to receive a signal. If it turns out to be a gain issue I will search the parameter space to find good values.

zuckschwerdt, 2024-01-08:
You can swap bytes with e.g. `dd conv=swab <airpsy_433.92M_2048k.cs16 >airpsy-swap_433.92M_2048k.cs16` -- if you want a look yourself. rtl_433 does decode something with that file then.

CtrlC-Root, 2024-01-09:
Thank you for this, I'll dig into this and report back here if I figure out anything useful. It seems to me it may be an issue with Soapy and not the hardware or rtl_433, without digging into it further.

zuckschwerdt, 2024-01-10:
It could be an issue with SoapyAirspy. But I suspect it's not as simple as swapped endianess as it would only be 8-bit deep data then (the "other" 8-bit never change /no noise). More likely it's really low gain, then the upper 8-bit are okay to always be "0".
Can you make similar cs16 recordings with some Airspy-native software? Also a cf32 recording would be interesting, rx_tools should be able to do that. Those two are the supported formats in SoapyAirspy.