merbanan/rtl_433 #1578: Acurite/Chaney 515 wireless refrigerator & freezer thermometer support

osenft, 2020-12-12:
I just got a new pair of Acurite 00515 fridge / freezer thermometers.

`rtl_433 -A` shows some data. I recorded the received packets and what the display shows and ended up with the following list (displayed temperature in F is appended at each line):

Freezer:
```
[00] {50} 39 64 36 f3 d2 9c c0 : 00111001 01100100 00110110 11110011 11010010 10011100 11 | 10.1
[00] {50} 39 64 36 f3 44 0e c0 : 00111001 01100100 00110110 11110011 01000100 00001110 11 | 11.5
[00] {50} 39 64 36 f3 3f 09 c0 : 00111001 01100100 00110110 11110011 00111111 00001001 11 | 12.0
[00] {50} 39 64 36 f3 3a 04 c0 : 00111001 01100100 00110110 11110011 00111010 00000100 11 | 12.5
[00] {50} 39 64 36 f3 b7 81 c0 : 00111001 01100100 00110110 11110011 10110111 10000001 11 | 12.8
[00] {50} 39 64 36 f3 39 03 c0 : 00111001 01100100 00110110 11110011 00111001 00000011 11 | 12.6
[00] {50} 39 64 36 f3 44 0e c0 : 00111001 01100100 00110110 11110011 01000100 00001110 11 | 11.5
[00] {50} 39 64 36 f3 d2 9c c0 : 00111001 01100100 00110110 11110011 11010010 10011100 11 | 10.1
[00] {50} 39 64 36 f3 e1 ab c0 : 00111001 01100100 00110110 11110011 11100001 10101011 11 |  8.6
[00] {50} 39 64 36 f3 ee b8 c0 : 00111001 01100100 00110110 11110011 11101110 10111000 11 |  7.3
[00] {50} 39 64 36 f3 7d 47 c0 : 00111001 01100100 00110110 11110011 01111101 01000111 11 |  5.8
[00] {50} 39 64 36 74 09 54 c0 : 00111001 01100100 00110110 01110100 00001001 01010100 11 |  4.6
[00] {50} 39 64 36 74 96 e1 c0 : 00111001 01100100 00110110 01110100 10010110 11100001 11 |  3.3
[00] {50} 39 64 36 74 1e 69 c0 : 00111001 01100100 00110110 01110100 00011110 01101001 11 |  2.5
[00] {50} 39 64 36 74 a5 f0 c0 : 00111001 01100100 00110110 01110100 10100101 11110000 11 |  1.8
[00] {50} 39 64 36 74 a5 f0 c0 : 00111001 01100100 00110110 01110100 10100101 11110000 11 |  1.8
[00] {50} 39 64 36 74 a3 ee c0 : 00111001 01100100 00110110 01110100 10100011 11101110 11 |  2.0
[00] {50} 39 64 36 74 a0 eb c0 : 00111001 01100100 00110110 01110100 10100000 11101011 11 |  2.3
[00] {50} 39 64 36 74 1b 66 c0 : 00111001 01100100 00110110 01110100 00011011 01100110 11 |  2.8
[00] {50} 39 64 36 74 18 63 c0 : 00111001 01100100 00110110 01110100 00011000 01100011 11 |  3.1
[00] {50} 39 64 36 74 93 de c0 : 00111001 01100100 00110110 01110100 10010011 11011110 11 |  3.6
[00] {50} 39 64 36 74 8d d8 c0 : 00111001 01100100 00110110 01110100 10001101 11011000 11 |  4.2
[00] {50} 39 64 36 74 87 d2 c0 : 00111001 01100100 00110110 01110100 10000111 11010010 11 |  4.8
[00] {50} 39 64 36 74 82 cd c0 : 00111001 01100100 00110110 01110100 10000010 11001101 11 |  5.3
[00] {50} 39 64 36 f3 7d 47 c0 : 00111001 01100100 00110110 11110011 01111101 01000111 11 |  5.8
```

Fridge
```
[00] {50} 3d 77 b7 71 4e 2e c0 : 00111101 01110111 10110111 01110001 01001110 00101110 11 | 36.1
[00] {50} 3d 77 b7 71 4d 2d c0 : 00111101 01110111 10110111 01110001 01001101 00101101 11 | 36.2
[00] {50} 3d 77 b7 71 4b 2b c0 : 00111101 01110111 10110111 01110001 01001011 00101011 11 | 36.4
[00] {50} 3d 77 b7 71 4b 2b c0 : 00111101 01110111 10110111 01110001 01001011 00101011 11 | 36.4
[00] {50} 3d 77 b7 71 4b 2b c0 : 00111101 01110111 10110111 01110001 01001011 00101011 11 | 36.4
[00] {50} 3d 77 b7 71 ca aa c0 : 00111101 01110111 10110111 01110001 11001010 10101010 11 | 36.5
[00] {50} 3d 77 b7 71 ca aa c0 : 00111101 01110111 10110111 01110001 11001010 10101010 11 | 36.5
[00] {50} 3d 77 b7 71 ca aa c0 : 00111101 01110111 10110111 01110001 11001010 10101010 11 | 36.5
[00] {50} 3d 77 b7 71 ca aa c0 : 00111101 01110111 10110111 01110001 11001010 10101010 11 | 36.5
[00] {50} 3d 77 b7 71 4b 2b c0 : 00111101 01110111 10110111 01110001 01001011 00101011 11 | 36.4
[00] {50} 3d 77 b7 71 4b 2b c0 : 00111101 01110111 10110111 01110001 01001011 00101011 11 | 36.4
[00] {50} 3d 77 b7 71 4d 2d c0 : 00111101 01110111 10110111 01110001 01001101 00101101 11 | 36.2
[00] {50} 3d 77 b7 71 4e 2e c0 : 00111101 01110111 10110111 01110001 01001110 00101110 11 | 36.1
[00] {50} 3d 77 b7 71 cf af c0 : 00111101 01110111 10110111 01110001 11001111 10101111 11 | 36.0
[00] {50} 3d 77 b7 71 d1 b1 c0 : 00111101 01110111 10110111 01110001 11010001 10110001 11 | 35.8
[00] {50} 3d 77 b7 71 53 33 c0 : 00111101 01110111 10110111 01110001 01010011 00110011 11 | 35.6
[00] {50} 3d 77 b7 71 d4 b4 c0 : 00111101 01110111 10110111 01110001 11010100 10110100 11 | 35.5
[00] {50} 3d 77 b7 71 d4 b4 c0 : 00111101 01110111 10110111 01110001 11010100 10110100 11 | 35.5
[00] {50} 3d 77 b7 71 d4 b4 c0 : 00111101 01110111 10110111 01110001 11010100 10110100 11 | 35.5
[00] {50} 3d 77 b7 71 d4 b4 c0 : 00111101 01110111 10110111 01110001 11010100 10110100 11 | 35.5
[00] {50} 3d 77 b7 71 d4 b4 c0 : 00111101 01110111 10110111 01110001 11010100 10110100 11 | 35.5
[00] {50} 3d 77 b7 71 d4 b4 c0 : 00111101 01110111 10110111 01110001 11010100 10110100 11 | 35.5
[00] {50} 3d 77 b7 71 53 33 c0 : 00111101 01110111 10110111 01110001 01010011 00110011 11 | 35.6 
[00] {50} 3d 77 b7 71 53 33 c0 : 00111101 01110111 10110111 01110001 01010011 00110011 11 | 35.6 
[00] {50} 3d 77 b7 71 53 33 c0 : 00111101 01110111 10110111 01110001 01010011 00110011 11 | 35.6
[00] {50} 3d 77 b7 71 d1 b1 c0 : 00111101 01110111 10110111 01110001 11010001 10110001 11 | 35.8
```

FCC ID is RNE00515TX. It seems to be operating on the same frequency as the 00592TX so assume that the packets are modulated similarly, which leads me to assume that the demodulation that I got is "right".

I've been trying to make sense of the data and found that the lower 7 bits of the 5th byte contain the temperature in F with some offset. 

For the freezer I ended up with: `byte = 183 - (temp * 10)`. E.g. `0x7d = 183 - 5.8 * 10`.

For the fridge this would be: `byte = 439 - (temp * 10)`. E.g. `0x53 = 439 - 35.6 * 10`.

Note that 439-183 = 256 = 0x100.

I'm pretty sure that there's a simpler way and that the offsets for fridge and freezer and somehow be calculated from a little less crazy numbers.

Has anyone looked at these sensors yet? I'd be happy to provide further captures ...

zuckschwerdt, 2020-12-12:
Using `-A` isn't great as the timings might be wrong. Likely you need to invert the bits.
Here is [a BitBench with these values](https://triq.net/bitbench?c=%7B50%7D39643674a5f0c0%5B%201.8%5D&c=%7B50%7D39643674a3eec0%5B%202.0%5D&c=%7B50%7D39643674a0ebc0%5B%202.3%5D&c=%7B50%7D396436741e69c0%5B%202.5%5D&c=%7B50%7D396436741b66c0%5B%202.8%5D&c=%7B50%7D396436741863c0%5B%203.1%5D&c=%7B50%7D3964367496e1c0%5B%203.3%5D&c=%7B50%7D3964367493dec0%5B%203.6%5D&c=%7B50%7D396436748dd8c0%5B%204.2%5D&c=%7B50%7D396436740954c0%5B%204.6%5D&c=%7B50%7D3964367487d2c0%5B%204.8%5D&c=%7B50%7D3964367482cdc0%5B%205.3%5D&c=%7B50%7D396436f37d47c0%5B%205.8%5D&c=%7B50%7D396436f3eeb8c0%5B%207.3%5D&c=%7B50%7D396436f3e1abc0%5B%208.6%5D&c=%7B50%7D396436f3d29cc0%5B10.1%5D&c=%7B50%7D396436f3440ec0%5B11.5%5D&c=%7B50%7D396436f33f09c0%5B12.0%5D&c=%7B50%7D396436f33a04c0%5B12.5%5D&c=%7B50%7D396436f33903c0%5B12.6%5D&c=%7B50%7D396436f3b781c0%5B12.8%5D&c=&c=%7B50%7D3d77b771d4b4c0%5B35.5%5D&c=%7B50%7D3d77b7715333c0%5B35.6%5D&c=%7B50%7D3d77b771d1b1c0%5B35.8%5D&c=%7B50%7D3d77b771cfafc0%5B36.0%5D&c=%7B50%7D3d77b7714e2ec0%5B36.1%5D&c=%7B50%7D3d77b7714d2dc0%5B36.2%5D&c=%7B50%7D3d77b7714b2bc0%5B36.4%5D&c=%7B50%7D3d77b771caaac0%5B36.5%5D&f=CH%3A~2h%20ID6h8h%20P1b%207h%20P1b%20TH7d%20P1b%20TL7d%208h%202h&i=true) -- and it exactly matches.
I.e. Calculate: TH*128+TL - 1480 and you get deg F x10

osenft, 2020-12-12:
Wow, that is awesome, thank you! How did you figure out which bits to ignore, i.e. what the "P" bits are? I'll have a look on how to add this to the existing Acurite decoder.

zuckschwerdt, 2020-12-12:
The usual Acurite protocol has a parity bit as MSB on the data bytes.
You would really just need to document the timings, the rest seems like the protocol we already have.

zuckschwerdt, 2020-12-12:
Something similar to https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L1103
What's the timing?

osenft, 2020-12-12:
We don't have humidity, from what I can tell, and the temperature is in F rather than C. I also need to figure out message_type 8 (fridge sensor) vs message_type 9 (freezer sensor).

But here the rough PoC draft:
```diff
diff --git a/src/devices/acurite.c b/src/devices/acurite.c
index 4974cb9e..930ad046 100644
--- a/src/devices/acurite.c
+++ b/src/devices/acurite.c
@@ -29,6 +29,7 @@ Acurite weather stations and temperature / humidity sensors.
 
 // ** Acurite 5n1 functions **
 
+#define ACURITE_515_BITLEN        50
 #define ACURITE_TXR_BITLEN        56
 #define ACURITE_5N1_BITLEN        64
 #define ACURITE_6045_BITLEN       72
@@ -673,7 +674,8 @@ static int acurite_txr_decode(r_device *decoder, bitbuffer_t *bitbuffer)
         if ((bitbuffer->bits_per_row[brow] < ACURITE_TXR_BITLEN ||
                 bitbuffer->bits_per_row[brow] > ACURITE_5N1_BITLEN + 1)
                 && bitbuffer->bits_per_row[brow] != ACURITE_6045_BITLEN
-                && bitbuffer->bits_per_row[brow] != ACURITE_ATLAS_BITLEN) {
+                && bitbuffer->bits_per_row[brow] != ACURITE_ATLAS_BITLEN
+               && bitbuffer->bits_per_row[brow] != ACURITE_515_BITLEN) {
             if (decoder->verbose > 1 && bitbuffer->bits_per_row[brow] > 16)
                 fprintf(stderr, "%s: skipping wrong len\n", __func__);
             continue; // DECODE_ABORT_LENGTH
@@ -745,6 +747,37 @@ static int acurite_txr_decode(r_device *decoder, bitbuffer_t *bitbuffer)
             valid++;
         }
 
+        // 00515 sensor messages are 6 bytes.
+        if (browlen == ACURITE_515_BITLEN / 8) {
+            channel = acurite_getChannel(bb[0]);
+            // Tower sensor ID is the last 14 bits of byte 0 and 1
+            // CCII IIII | IIII IIII
+            sensor_id = ((bb[0] & 0x3f) << 8) | bb[1];
+            // temperature encoding used by 515 sensors
+            // 14 bits available after removing both parity bits.
+            int temp_raw = ((bb[3] & 0x7F) << 7) | (bb[4] & 0x7F);
+            tempf = (temp_raw - 1480) * 0.1f;
+            sprintf(channel_str, "%c", channel);
+            // Battery status is the 7th bit 0x40. 1 = normal, 0 = low
+            battery_low = (bb[2] & 0x40) == 0;
+
+            /* clang-format off */
+            data = data_make(
+                    "model",                "",             DATA_STRING, _X("Acurite 00515","Acurite 00515 sensor"),
+                   "message_type", NULL,   DATA_INT,       message_type,
+                    "id",                   "",             DATA_INT,    sensor_id,
+                    "channel",              NULL,           DATA_STRING, &channel_str,
+                    _X("battery_ok","battery_low"), "",     DATA_INT,    _X(!battery_low,battery_low),
+                    "temperature_C",        "Temperature",  DATA_FORMAT, "%.1f F", DATA_DOUBLE, tempf,
+                    "mic",                  "Integrity",    DATA_STRING, "CHECKSUM",
+                    NULL);
+            /* clang-format on */
+
+            decoder_output_data(decoder, data);
+            valid++;
+        }
+
+
         // The 5-n-1 weather sensor messages are 8 bytes.
         else if (message_type == ACURITE_MSGTYPE_5N1_WINDSPEED_WINDDIR_RAINFALL ||
                  message_type == ACURITE_MSGTYPE_5N1_WINDSPEED_TEMP_HUMIDITY ||

```

zuckschwerdt, 2020-12-12:
Looks good already.
Are you sure there is a message type? Aren't both the same hardware, just different IDs?
If there is no humidity what is the first data byte (48 / 49) in your examples?

rct, 2020-12-12:
Haven't looked at the inverted data for this new device yet, however from the 986 there is a hard coded part of the message to indicate sensor 1 or 2 (freezer vs. refrigerator).  

The 986 uses a different modulation, where the 985 seems to follow the rest of "TXR" family. Were different timings required?

Regarding the possible ID, would be good to do some power cycles and see if the ID changes or not.

osenft, 2020-12-12:
> Are you sure there is a message type? Aren't both the same hardware, just different IDs?
I think you're right. I'll change it so that it reports fridge and freezer for message_type 8 and 9 respectively instead.

> If there is no humidity what is the first data byte (48 / 49) in your examples?
I think the 8 and 9 indicate the sensor "type" fridge vs. freezer. I was just using the existing code for that:
```C
        // acurite sensors with a common format appear to have a message type
        // in the lower 6 bits of the 3rd byte.
        // Format: PBMMMMMM
        // P = Parity
        // B = Battery Normal
        // M = Message type
        message_type = bb[2] & 0x3f;
```

> Haven't looked at the inverted data for this new device yet, however from the 986 there is a hard coded part of the message to indicate sensor 1 or 2 (freezer vs. refrigerator).
Yes, I think you're right. I'll follow that same method.

> The 986 uses a different modulation, where the 985 seems to follow the rest of "TXR" family. Were different timings required?
No, I don't think so. It just worked for the 00515 the same as it does for the existing 00592.

> Regarding the possible ID, would be good to do some power cycles and see if the ID changes or not.
It does. Power cycle assigns a new ID to the sensor. What is the ID usually being used for?

osenft, 2020-12-12:
So, with a few changes, this is how it looks now:

```
time      : 2020-12-12 14:05:33
model     : Acurite-515  id        : 1691
channel   : AF           battery_ok: 1             Temperature: 2.3 F        Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:05:33
model     : Acurite-515  id        : 1691
channel   : AF           battery_ok: 1             Temperature: 2.3 F        Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:05:33
model     : Acurite-515  id        : 1691
channel   : AF           battery_ok: 1             Temperature: 2.3 F        Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:05:36
model     : Acurite-515  id        : 648
channel   : AR           battery_ok: 1             Temperature: 35.5 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:05:36
model     : Acurite-515  id        : 648
channel   : AR           battery_ok: 1             Temperature: 35.5 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:05:36
model     : Acurite-515  id        : 648
channel   : AR           battery_ok: 1             Temperature: 35.5 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:06:17
model     : Acurite-515  id        : 8633
channel   : BF           battery_ok: 1             Temperature: -3.5 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:06:17
model     : Acurite-515  id        : 8633
channel   : BF           battery_ok: 1             Temperature: -3.5 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:06:17
model     : Acurite-515  id        : 8633
channel   : BF           battery_ok: 1             Temperature: -3.5 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:06:19
model     : Acurite-515  id        : 6055
channel   : BR           battery_ok: 1             Temperature: 33.6 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:06:19
model     : Acurite-515  id        : 6055
channel   : BR           battery_ok: 1             Temperature: 33.6 F       Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-12 14:06:19
model     : Acurite-515  id        : 6055
channel   : BR           battery_ok: 1             Temperature: 33.6 F       Integrity : CHECKSUM
```

Questions:
* I'm following the 00986 code with appending R (refrigerator) or F (freezer) to the channel ID. Is that ok?
* I noticed that the existing `acurite_txr_decode` code doesn't actually calculate and compare the checksum. Should it?
* The sensors send the same data 3 times in a row. Does that need to be filtered or is it ok to just process it and decode it individually?

osenft, 2020-12-12:
Alright, here the PRs:
* https://github.com/merbanan/rtl_433_tests/pull/378
* https://github.com/merbanan/rtl_433/pull/1579

I'll send a second PR for the rtl_433_tests with the JSON files once the code is checked in. Is that the right order?

osenft, 2020-12-12:
Oh, I noticed that the temperature is reported to MQTT as `temperature_C` - that's obviously not right. I'll dig into that.

rct, 2020-12-12:
For the devices that follow the "TXR" encoding, the third byte is essentially used for differentiating message type and battery status.  So in this implementation if they've essentially allocated message type 8 and 9 to this sensor, I think there should be #defines for it along with the others, I think it will make things easier to understand.

Also I recommend checking the message type == 8 or 9, rather than just one bit.  I think we'll get less false positives that way.  It's not being checked for some of the devices that were decoded earlier. I intend to clean that up one of these days.

If possible can you test with either some low batteries or an adjustable power supply to verify the battery status bit is the same for this device?   Many of these devices signal low battery somewhere around 2.2-2.4Vdc.  They stop functioning below about 2.2Vdc.

> I'm following the 00986 code with appending R (refrigerator) or F (freezer) to the channel ID. Is that ok?

Does this sensor actually have an A-B-C switch?    If not, how are you getting different values for channel ID?

>  noticed that the existing acurite_txr_decode code doesn't actually calculate and compare the checksum. Should it?

The checksum is verified in common code above. Parity checking isn't currently done, but should be. 

> The sensors send the same data 3 times in a row. Does that need to be filtered or is it ok to just process it and decode it individually?

Yes, that's fine. That's what the other sensors do.  Consumers can remove what they consider duplicates based on their application specific requirements.

> >    Regarding the possible ID, would be good to do some power cycles and see if the ID changes or not.

> It does. Power cycle assigns a new ID to the sensor. What is the ID usually being used for?

For devices with volatile IDs and no channel switches, the ID is usually used to "pair" the sensor and display, which then needs to be done whenever the batteries are changed. 

Please make a note in the comments that describe the message type that the ID for this device is volatile. I think most/all of the other TXR type devices have a factory fixed ID.

rct, 2020-12-12:
Ok, I was briefly confused. I thought this was the implementation for the 985's that was mentioned in the last week or two.  I didn't catch that this implementation is for the 515s, the new sensors. 

Now I found the manual and answered my own question about the A-B-C switch.

I actually just got a set of these sensors that I haven't unboxed yet.  Thanks for doing the work for me!   I should be able to test this next week.  I'm interested to see if the Acurite Atlas will pick these up or not.  I suspect they haven't added support yet. Though if they did it would be helpful for confirming the decoding.

WRT to the channel.  I've got some reservations about concatenating channel with the fixed sensor type. On the one hand, there is a discrete set of values for channel that match the physical switch on these and other devices. So I feel like the F or R shouldn't be part of the channel.  On the other hand if the device doesn't have a fixed ID, I'm not sure where the device type should really go, particularly when MQTT and other downstream consumers are factored in.  The `subtype` field might actually be useful for this. 

FWIW, Acurite's "online" ecosystem (the access publishing to their cloud service and weather underground) completely ignores Channel and only used ID, since all of the other devices they support have fixed IDs.

BTW, I've come to loath devices with volatile IDs, If you are trying to maintain history in a database, they can be a real nuisance. If the battery voltage drops low enough and then comes back up, you wind up with a new devices with no warning.

zuckschwerdt, 2020-12-13:
As it is likely the exact same hardware for both it would be interesting to see if there is a hidden switch (R/F) -- maybe a cut trace on the board.

osenft, 2020-12-13:
> WRT to the channel. I've got some reservations about concatenating channel with the fixed sensor type. On the one hand, there is a discrete set of values for channel that match the physical switch [...]

I think logically it does make sense for the `R` and `F` to be part of the channel - at least as far as the user is concerned. The user can select A-B-C via switch and even with both fridge and freezer sensors configured to the same channel, they see different values, so it's logically a different channel. I agree that it's weird since it's implemented with different message types. But isn't that just an implementation detail.

I got my inspiration from the 986 sensor decoding code, which names the channels `1R` for the fridge and `2F` for the freezer.

I experimented with MQTT and Home Assistant and identifying the sensors by their type and channel (incl. combined fridge / freezer) works well.

osenft, 2020-12-14:
I addressed the review comments on PR #1579. Is there anything else left to do? Also see https://github.com/merbanan/rtl_433_tests/pull/378.

closed 2020-12-15 by zuckschwerdt

rct, 2020-12-16:
@osenft - Implementation looks good! I got my new 515s out of the box. Battery status is working, I put low batteries in one to test. 

Thanks for working on this.